libusb/libusb #1000: Coverity scan report -- libusb/os/linux_usbfs.c

mcuee, 2021-09-24:
This is the Defect reported by Coverity Scan for libusb/libusb

```
** CID 338869:    (TAINTED_SCALAR)
/libusb/os/linux_usbfs.c: 690 in parse_config_descriptors()
/libusb/os/linux_usbfs.c: 697 in parse_config_descriptors()

________________________________________________________________________________________________________
*** CID 338869:    (TAINTED_SCALAR)
/libusb/os/linux_usbfs.c: 734 in parse_config_descriptors()
728                              * with an invalid bLength removed.
729                              */
730                             uint16_t sysfs_config_len;
731                             int offset;
732     
733                             if (num_configs > 1 && idx < num_configs - 1) {
>>>     CID 338869:    (TAINTED_SCALAR)
>>>     Passing tainted expression "remaining" to "seek_to_next_config", which uses it as a loop boundary.
734                                     offset = seek_to_next_config(ctx, buffer, remaining);
735                                     if (offset < 0)
736                                             return offset;
737                                     sysfs_config_len = (uint16_t)offset;
738                             } else {
739                                     sysfs_config_len = (uint16_t)remaining;
/libusb/os/linux_usbfs.c: 690 in parse_config_descriptors()
684             device_desc = priv->descriptors;
685             num_configs = device_desc->bNumConfigurations;
686     
687             if (num_configs == 0)
688                     return 0;       /* no configurations? */
689     
>>>     CID 338869:    (TAINTED_SCALAR)
>>>     Passing tainted expression "num_configs * 16UL" to "malloc", which uses it as an allocation size.
690             priv->config_descriptors = malloc(num_configs * sizeof(priv->config_descriptors[0]));
691             if (!priv->config_descriptors)
692                     return LIBUSB_ERROR_NO_MEM;
693     
694             buffer = (uint8_t *)priv->descriptors + LIBUSB_DT_DEVICE_SIZE;
695             remaining = priv->descriptors_len - LIBUSB_DT_DEVICE_SIZE;
/libusb/os/linux_usbfs.c: 697 in parse_config_descriptors()
691             if (!priv->config_descriptors)
692                     return LIBUSB_ERROR_NO_MEM;
693     
694             buffer = (uint8_t *)priv->descriptors + LIBUSB_DT_DEVICE_SIZE;
695             remaining = priv->descriptors_len - LIBUSB_DT_DEVICE_SIZE;
696     
>>>     CID 338869:    (TAINTED_SCALAR)
>>>     Using tainted variable "num_configs" as a loop boundary.
697             for (idx = 0; idx < num_configs; idx++) {
698                     struct usbi_configuration_descriptor *config_desc;
699                     uint16_t config_len;
700     
701                     if (remaining < LIBUSB_DT_CONFIG_SIZE) {
702                             usbi_err(ctx, "short descriptor read %zu/%d",
/libusb/os/linux_usbfs.c: 734 in parse_config_descriptors()
728                              * with an invalid bLength removed.
729                              */
730                             uint16_t sysfs_config_len;
731                             int offset;
732     
733                             if (num_configs > 1 && idx < num_configs - 1) {
>>>     CID 338869:    (TAINTED_SCALAR)
>>>     Passing tainted expression "remaining" to "seek_to_next_config", which uses it as a loop boundary.
734                                     offset = seek_to_next_config(ctx, buffer, remaining);
735                                     if (offset < 0)
736                                             return offset;
737                                     sysfs_config_len = (uint16_t)offset;
738                             } else {
739                                     sysfs_config_len = (uint16_t)remaining;

```

mcuee, 2021-09-24:
Trace.

````
675static int parse_config_descriptors(struct libusb_device *dev)
676{
677        struct libusb_context *ctx = DEVICE_CTX(dev);
678        struct linux_device_priv *priv = usbi_get_device_priv(dev);
679        struct usbi_device_descriptor *device_desc;
680        uint8_t idx, num_configs;
681        uint8_t *buffer;
682        size_t remaining;
683
   	1. tainted_data_downcast: Downcasting priv->descriptors from void * to struct usbi_device_descriptor implies that the data that this pointer points to is tainted.
684        device_desc = priv->descriptors;
685        num_configs = device_desc->bNumConfigurations;
686
   	2. Condition num_configs == 0, taking false branch.
687        if (num_configs == 0)
688                return 0;       /* no configurations? */
689
     	CID 338869 (#2 of 4): Untrusted loop bound (TAINTED_SCALAR) [select issue]
690        priv->config_descriptors = malloc(num_configs * sizeof(priv->config_descriptors[0]));
   	3. Condition !priv->config_descriptors, taking false branch.
691        if (!priv->config_descriptors)
692                return LIBUSB_ERROR_NO_MEM;
693
   	4. var_assign_var: Assigning: buffer = (uint8_t *)priv->descriptors + 18. Both are now tainted.
694        buffer = (uint8_t *)priv->descriptors + LIBUSB_DT_DEVICE_SIZE;
695        remaining = priv->descriptors_len - LIBUSB_DT_DEVICE_SIZE;
696
     	CID 338869 (#3 of 4): Untrusted loop bound (TAINTED_SCALAR) [select issue]
   	5. Condition idx < num_configs, taking true branch.
   	28. Condition idx < num_configs, taking true branch.
   	47. Condition idx < num_configs, taking true branch.
697        for (idx = 0; idx < num_configs; idx++) {
698                struct usbi_configuration_descriptor *config_desc;
699                uint16_t config_len;
700
   	6. Condition remaining < 9, taking false branch.
   	29. Condition remaining < 9, taking false branch.
   	30. lower_bounds: Checking lower bounds of unsigned scalar remaining by taking the false branch of remaining < 9UL.
   	48. Condition remaining < 9, taking false branch.
   	49. lower_bounds: Checking lower bounds of unsigned scalar remaining by taking the false branch of remaining < 9UL.
701                if (remaining < LIBUSB_DT_CONFIG_SIZE) {
702                        usbi_err(ctx, "short descriptor read %zu/%d",
703                                 remaining, LIBUSB_DT_CONFIG_SIZE);
704                        return LIBUSB_ERROR_IO;
705                }
706
707                config_desc = (struct usbi_configuration_descriptor *)buffer;
   	7. Condition config_desc->bDescriptorType != LIBUSB_DT_CONFIG, taking false branch.
   	31. Condition config_desc->bDescriptorType != LIBUSB_DT_CONFIG, taking false branch.
   	50. Condition config_desc->bDescriptorType != LIBUSB_DT_CONFIG, taking false branch.
708                if (config_desc->bDescriptorType != LIBUSB_DT_CONFIG) {
709                        usbi_err(ctx, "descriptor is not a config desc (type 0x%02x)",
710                                 config_desc->bDescriptorType);
711                        return LIBUSB_ERROR_IO;
   	8. Condition config_desc->bLength < 9, taking false branch.
   	32. Condition config_desc->bLength < 9, taking false branch.
   	51. Condition config_desc->bLength < 9, taking false branch.
712                } else if (config_desc->bLength < LIBUSB_DT_CONFIG_SIZE) {
713                        usbi_err(ctx, "invalid descriptor bLength %u",
714                                 config_desc->bLength);
715                        return LIBUSB_ERROR_IO;
716                }
717
718                config_len = libusb_le16_to_cpu(config_desc->wTotalLength);
   	9. Condition config_len < 9, taking false branch.
   	33. Condition config_len < 9, taking false branch.
   	52. Condition config_len < 9, taking false branch.
[... 66 lines cut ...]
755                        }
756                }
757
   	24. Condition config_desc->bConfigurationValue == 0, taking true branch.
   	43. Condition config_desc->bConfigurationValue == 0, taking true branch.
758                if (config_desc->bConfigurationValue == 0)
759                        usbi_warn(ctx, "device has configuration 0");
760
761                priv->config_descriptors[idx].desc = config_desc;
762                priv->config_descriptors[idx].actual_len = config_len;
763
   	25. var_assign_var: Compound assignment involving tainted variable config_len to variable buffer taints buffer.
   	44. var_assign_var: Compound assignment involving tainted variable config_len to variable buffer taints buffer.
764                buffer += config_len;
   	26. var_assign_var: Compound assignment involving tainted variable config_len to variable remaining taints remaining.
   	45. var_assign_var: Compound assignment involving tainted variable config_len to variable remaining taints remaining.
765                remaining -= config_len;
   	27. Jumping back to the beginning of the loop.
   	46. Jumping back to the beginning of the loop.
766        }
767
768        return LIBUSB_SUCCESS;
769}

```

photron, 2021-09-24:
I don't understand the problem here. I seems that Coverity complains about casting `priv->descriptors` from `void *` to a different type and then accessing that casted data. All information that is derived from that casted void pointer is assumed to be tainted/untrusted. But I don't get why.

Does Coverity give a more detailed explanation about this issue?

mcuee, 2021-09-24:
No, that is all it gives. We can triage it as not an issue if we think therer is no issue or false positive.

windfine, 2022-06-20:
I think therer  is no issue. The reasons are as follows：

>>>     CID 338869:    (TAINTED_SCALAR)
>>>     Passing tainted expression "remaining" to "seek_to_next_config", which uses it as a loop boundary.
The value of remaining has been checked befor calling seek_to_next_config.
		if (remaining < LIBUSB_DT_CONFIG_SIZE) {
			usbi_err(ctx, "short descriptor read %zu/%d",
				 remaining, LIBUSB_DT_CONFIG_SIZE);
			return LIBUSB_ERROR_IO;
		}

>>>     CID 338869:    (TAINTED_SCALAR)
>>>     Passing tainted expression "num_configs * 16UL" to "malloc", which uses it as an allocation size.
num_configs is 8UL，num_configs * 16UL would not exceed size_t（the input parameters units of malloc.）

>>>     CID 338869:    (TAINTED_SCALAR)
>>>     Using tainted variable "num_configs" as a loop boundary.
num_configs is 8UL，is no risk as the exit condition of the loop。

mcuee, 2026-03-12:
We may need to re-rub Coverity later. Close this for now.

closed 2026-03-12 by mcuee