libusb/libusb #1001: Unable to open Android phone on Windows.

AlynxZhou, 2021-09-24:
I am implementing [HID over AOAv2 for scrcpy](https://github.com/Genymobile/scrcpy/pull/2632), while libusb works quite well on Linux, it failed to open my Android phone on Windows.

I've tried `xusb.exe`, and it says `Failed to open device` (or something like that, I am under Linux so no screenshot).

I tried to use Zadig to install driver, however there is no device without driver, and I can see multiply devices for my phone if check `List all devices`, one is MTP, and another is `Samsung_ADB_Interface` (or something like it), I tried to force install WinUSB driver for them, but install failed.

Could you please tell me what's wrong with it? Or what should I do?

mcuee, 2021-09-24:
Read here on how to use Zadig. You may want to use WinUSB driver to replace the composite parent driver.
https://github.com/pbatard/libwdi/wiki/Zadig

closed 2021-09-24 by mcuee

mcuee, 2021-09-24:
BTW, if you have issues with Zadig, please create an issue in Zadig. It is a separate project.

AlynxZhou, 2021-09-24:
> Read here on how to use Zadig. You may want to use WinUSB driver to replace the composite parent driver.
> https://github.com/pbatard/libwdi/wiki/Zadig

How two decide which one is the composite parent driver? I've tried every one related and failed...

mcuee, 2021-09-24:
> > Read here on how to use Zadig. You may want to use WinUSB driver to replace the composite parent driver.
> > https://github.com/pbatard/libwdi/wiki/Zadig
> 
> How two decide which one is the composite parent driver? I've tried every one related and failed...

The one with the USB VID/PID of your phone and without saying Interface 0 or interface 1.

If you have issues using Zadig, you can use libusbk-inf-wizard which some users like more.
https://github.com/mcuee/libusbk/releases (install libusbK-3.1.0.0-setup.exe)

AlynxZhou, 2021-09-24:
> > > Read here on how to use Zadig. You may want to use WinUSB driver to replace the composite parent driver.
> > > https://github.com/pbatard/libwdi/wiki/Zadig
> > 
> > 
> > How two decide which one is the composite parent driver? I've tried every one related and failed...
> 
> You need to set the options so that the parent device will be listed.
> The one with the USB VID/PID of your phone and without saying Interface 0 or interface 1.
> 
> If you have issues using Zadig, you can use libusbk-inf-wizard which some users like more.
> https://github.com/mcuee/libusbk/releases (install libusbK-3.1.0.0-setup.exe)

Thanks, will try it tomorrow.