libusb/libusb #999: Coverity scan report -- core.c

mcuee, 2021-09-24:
This is the Defect reported by Coverity Scan for libusb/libusb

```

** CID 338870:  Program hangs  (LOCK)
/libusb/core.c: 2381 in libusb_init()

________________________________________________________________________________________________________
*** CID 338870:  Program hangs  (LOCK)
/libusb/core.c: 2381 in libusb_init()
2375     err_free_ctx:
2376            usbi_mutex_destroy(&_ctx->open_devs_lock);
2377            usbi_mutex_destroy(&_ctx->usb_devs_lock);
2378     
2379            free(_ctx);
2380     
>>>     CID 338870:  Program hangs  (LOCK)
>>>     Returning without unlocking "default_context_lock".
2381            return r;
2382     }
2383     
2384     /** \ingroup libusb_lib
2385      * Deinitialize libusb. Should be called after closing all open devices and
2386      * before your application terminates.

```

photron, 2021-09-24:
Well you didn't show the full trace, but I guess this is fixed by PR #995.

mcuee, 2021-09-24:
I guess the following is the full trace. I am new to this Coverity stuff as I am not a programmer myself, rather a hardware engineer.

```
2261/** \ingroup libusb_lib
2262 * Initialize libusb. This function must be called before calling any other
2263 * libusb function.
2264 *
2265 * If you do not provide an output location for a context pointer, a default
2266 * context will be created. If there was already a default context, it will
2267 * be reused (and nothing will be initialized/reinitialized).
2268 *
2269 * \param ctx Optional output location for context pointer.
2270 * Only valid on return code 0.
2271 * \returns 0 on success, or a LIBUSB_ERROR code on failure
2272 * \see libusb_contexts
2273 */
2274int API_EXPORTED libusb_init(libusb_context **ctx)
2275{
2276        size_t priv_size = usbi_backend.context_priv_size;
2277        struct libusb_context *_ctx;
2278        int r;
2279
    	1. lock: usbi_mutex_static_lock locks default_context_lock. [show details]
2280        usbi_mutex_static_lock(&default_context_lock);
2281
    	2. Condition !ctx, taking true branch.
    	3. Condition usbi_default_context, taking false branch.
2282        if (!ctx && usbi_default_context) {
2283                usbi_dbg(usbi_default_context, "reusing default context");
2284                default_context_refcnt++;
2285                usbi_mutex_static_unlock(&default_context_lock);
2286                return 0;
2287        }
2288
2289        /* check for first init */
    	4. Condition !active_contexts_list.next, taking true branch.
2290        if (!active_contexts_list.next) {
2291                list_init(&active_contexts_list);
2292                usbi_get_monotonic_time(&timestamp_origin);
2293        }
2294
2295        _ctx = calloc(1, PTR_ALIGN(sizeof(*_ctx)) + priv_size);
    	5. Condition !_ctx, taking false branch.
2296        if (!_ctx) {
2297                usbi_mutex_static_unlock(&default_context_lock);
2298                return LIBUSB_ERROR_NO_MEM;
2299        }
2300
2301#if defined(ENABLE_LOGGING) && !defined(ENABLE_DEBUG_LOGGING)
    	6. Condition NULL == ctx, taking true branch.
    	7. Condition default_context_options[LIBUSB_OPTION_LOG_LEVEL].is_set, taking true branch.
2302        if (NULL == ctx && default_context_options[LIBUSB_OPTION_LOG_LEVEL].is_set) {
2303                _ctx->debug = default_context_options[LIBUSB_OPTION_LOG_LEVEL].arg.ival;
    	8. Falling through to end of if statement.
2304        } else {
2305                _ctx->debug = get_env_debug_level();
2306        }
    	9. Condition _ctx->debug != LIBUSB_LOG_LEVEL_NONE, taking true branch.
2307        if (_ctx->debug != LIBUSB_LOG_LEVEL_NONE)
2308                _ctx->debug_fixed = 1;
2309#endif
2310
2311        usbi_mutex_init(&_ctx->usb_devs_lock);
2312        usbi_mutex_init(&_ctx->open_devs_lock);
2313        list_init(&_ctx->usb_devs);
2314        list_init(&_ctx->open_devs);
2315
2316        /* default context should be initialized before calling usbi_dbg */
    	10. Condition !ctx, taking true branch.
2317        if (!ctx) {
[... 62 lines cut ...]
2368        }
2369
2370        usbi_mutex_static_unlock(&default_context_lock);
2371
2372        usbi_hotplug_exit(_ctx);
2373        usbi_io_exit(_ctx);
2374
2375err_free_ctx:
2376        usbi_mutex_destroy(&_ctx->open_devs_lock);
2377        usbi_mutex_destroy(&_ctx->usb_devs_lock);
2378
2379        free(_ctx);
2380
    	
CID 338870 (#1 of 1): Missing unlock (LOCK)
23. missing_unlock: Returning without unlocking default_context_lock.
2381        return r;
2382}
2383
2384/** \ingroup libusb_lib
2385 * Deinitialize libusb. Should be called after closing all open devices and
2386 * before your application terminates.
2387 * \param ctx the context to deinitialize, or NULL for the default context
2388 */
```

photron, 2021-09-24:
With this full trace one can see the control flow that Coverity found that results in the reported problem. From this trace I can tell that this is fixed by PR #995.

mcuee, 2022-04-01:
Close this for now.

closed 2022-04-01 by mcuee