libusb/libusb #1002: Any suggestion on how to sign libusb driver for Windows 10?

dataq-instruments, 2021-09-28:
I use libusb for our products, and we create a .INF file to use our PID and VID and sign it vie DigiCert, everything works fine until Microsoft drops most certificates (https://docs.microsoft.com/en-us/windows-hardware/drivers/install/deprecation-of-software-publisher-certificates-and-commercial-release-certificates)

I am using the binary image of libusb, does anyone have recommendation on how to sign it now?

photron, 2021-09-28:
The best solution is to not have to ship a driver at all.

Do you control the firmware of your device? Then you can turn it into a WCID device that reports compatibility with WinUSB using "Microsoft OS String Descriptors". This way Windows autodetects you device as WinUSB compatible and automatically load the WinUSB driver for your device. Plug and Play how it should be.

There is a good article about what you need to change on your device to make that work:

https://github.com/pbatard/libwdi/wiki/WCID-Devices

We're doing this for our products for years now. Works flawlessly.

Anyway, the issue tracker is not meant for general support questions. The libusb mailing list is better suited for such things:

libusb-devel@lists.sourceforge.net

mcuee, 2021-09-29:
> I use libusb for our products, and we create a .INF file to use our PID and VID and sign it vie DigiCert, everything works fine until Microsoft drops most certificates [...]

Which driver are you using?  It is not recommended to use libusb0.sys with libusb Windows backend. 

If you use WinUSB driver and you have control over the firmware, then indeed the best is to use Microsoft OS descriptors.

But I understand you may not have control over the devices in the field, in that case, there are two methods.
1) attestation signing -- you need a EV certificate, this is the recommended method for most  driver developers
https://docs.microsoft.com/en-us/windows-hardware/drivers/dashboard/attestation-signing-a-kernel-driver-for-public-release
2) WHQL, not applicable for some drivers which do not have corresponding HLK tests.
https://docs.microsoft.com/en-us/windows-hardware/drivers/install/whql-release-signature

closed 2021-09-30 by mcuee