libusb/libusb #1003: no_enumeration should be per context, not global

tormodvolden, 2021-10-02:
This was brought up in a discussion somewhere, so I open this ticket so it is not forgotten. Likely related to #942.

tormodvolden, 2021-11-01:
Carried over from #942:
Now, if you want to have one context where you want to skip enumeration (because you use libusb_wrap_sys() anyway) and another context with enumeration, you will have to initialize the latter first, then set LIBUSB_OPTION_NO_DEVICE_DISCOVERY and initialize the former. If these are in independent plug-ins (like the example in the Contexts documentation) this is not ideal. I guess we could make libusb_set_option(NULL, LIBUSB_OPTION_NO_DEVICE_DISCOVERY, 0) work to turn it off, or we need something more sophisticated to manipulate the default options in general. If we were to start from scratch, I think there should have been a flag option argument to libusb_init(), and the options being bit masks...

Also, libusb_set_option(NULL, ...) does not affect an existing, default context. Existing applications may depend on this, if this was the behavior earlier(?), consistent to the use of NULL ctx in all other functions.

Maybe we should add a new libusb_set_default_options() instead?

xloem, 2021-11-04:
additional possible approaches:
- passing options in a list structure to libusb_init
- separating allocation from initialisation so early options can be set with data
- refactoring linux_usbfs to initialise on first use

tormodvolden, 2021-11-04:
By "initialisation" you mean also the device discovery?

tormodvolden, 2021-11-04:
To remain backwards compatible, maybe we can add something like a new libusb_create_context(&ctx, options, ...) and if you call libusb(ctx) with an already created context it can skip allocation and apply these settings for "initialisation" including whether to do device discovery. Well, I guess that's what you had in mind for your second approach above.

xloem, 2021-11-04:
yeah.  and I guess by initialisation I did mean initial device discovery and likely hotplug setup I suppose; I don't remember the internal design well.

tormodvolden, 2021-11-20:
@xloem did you do some work on this as part of some other merge request? There will be 1.0.25 RC soon, so last chance to get something in.

xloem, 2021-11-20:
58ac16bd1e0f5f5f01a26c40720501d5f2974a30 is part of #874 .  I have trouble thinking about the PR, and am unlikely to resolve that before the release.  There may be other changes in the branch that are relevant: or that commit may simply be mergeable.  Needs testing or a glance through the code at least.  The change is pretty simple.

hjelmn, 2021-11-20:
I think the best approach might be to fork the libusb_init function and create one that takes options. The whole mess with setting options on the default context before it is created is just a constant issue. I don't have a interface in mind yet but maybe something like:

```
struct libusb_options {
  enum libusb_option option;
  union {
    int64_t ival;
  } value;
};
int libusb_init_with_options(libusb_context **context, struct libusb_options options[], int num_options);
```

Then immediately deprecate ```libusb_set_option()```. I know it is a newer interface but this madness must end :)

hjelmn, 2021-11-20:
Threw together a change. Can open a PR for discussion.

hjelmn, 2021-11-20:
See #1026 for the proposal. If there is agreement that this is the way we should go I will go ahead and finish the documentation and make sure everything is ready.

tormodvolden, 2021-11-20:
Can we safely assume that no settings need to be changed after libusb_init*? I can only think of the debug level that can have a need for that. But we already have libusb_set_debug() which we can de-deprecate and we would be covered.

I agree that the current concept of a set of default options for new contexts (not exactly "options on default context") will just make more and more problems, and the sooner we change it the better.

tormodvolden, 2021-11-20:
Just want to reference #987 and #990 which had some related discussion.

hjelmn, 2021-11-21:
Yeah. Hard to say. It could be we leave set_option and error if trying to change an option that can only be set at init time. Leaving the interface as flexible as possible while disallowing the problematic cases seems like the best approach.

tormodvolden, 2023-01-16:
It seems that this is included in #1164. We should pick those changes separately, and 1164 needs rebasing anyway.

closed 2023-01-22 by tormodvolden

tormodvolden, 2023-01-22:
> [58ac16b](https://github.com/libusb/libusb/commit/58ac16bd1e0f5f5f01a26c40720501d5f2974a30) is part of #874 . I have trouble thinking about the PR, and am unlikely to resolve that before the release. There may be other changes in the branch that are relevant: or that commit may simply be mergeable. Needs testing or a glance through the code at least. The change is pretty simple.

@xloem, a reworked version of this has now been merged to master. Thanks for your work on this.

commit a92e21ba (2023-01-22): linux: Keep the no_device_discovery option per context

Fixes #1003
Closes #1231

[Tormod: Adapted to current master]