libusb/libusb #1008: problem with libusb_open_device_with_vid_pid()!

NoumanNawaz51, 2021-10-11:
I was having trouble when I got to know about this. Well actually it's not a big deal but still its very strange that I didn't find much about this(maybe my bad). this function open device with vid pid is not enough.

Problem Statement: 
There are 5 device with VID = 0x0789 PID = 0x6545 connected. and since all have same ids they will be mounted with unique instances in udev(matched against keys).

Now when you will call get_device_list you will get all 5 devices in the list you can open them one by one and all good. but now when you do open_device_with_vid_pid you will only able to get the instance of first device in the list since in the function
` libusb_open_device_with_vid_pid(libusb_context *ctx, uint16_t vendor_id, uint16_t product_id);`
device vid and pid will be match at the first index and it will return from the found case. 

There should be an overloaded method which has one or more different parameters like serialno for example that way we can search more efficiently. if I get the time i will apply the fix but I wanted to know what you guys have to say about this.

a1ien, 2021-10-11:
This is stated in the documentation `libusb_open_device_with_vid_pid`
>Convenience function for finding a device with a particular idVendor/idProduct combination. This function is intended for those scenarios where you are using libusb to knock up a quick test application - it allows you to avoid calling `libusb_get_device_list() `and worrying about traversing/freeing the list.

This function has limitations and is hence not intended for use in real applications: if multiple devices have the same IDs it will only give you the first one, etc.

mcuee, 2021-10-11:
I will close this issue since it is documented.

@NoumanNawaz51 If you want to contribute to the improvement, please go here and continue the unfinished work: https://github.com/libusb/libusb/pull/875

mcuee, 2021-10-11:
@NoumanNawaz51 Or you can propose new enhancements if you think #875 is not enough.

closed 2021-10-11 by mcuee