libusb/libusb #1017: Windows: synchronous control transfer errors are not translated to libusb errors

tormodvolden, 2021-11-02:
This is an issue that is older than 1.0.23, when libusb0 control transfers were otherwise working.

For example, as io.c says about an unsupported control transfer:
>  Even though control endpoints do not halt, a completed control transfer
>  may have a LIBUSB_TRANSFER_STALL status code. This indicates the control
>  request was not supported.

And sync.c says this about libusb_control_transfer():
> returns LIBUSB_ERROR_PIPE if the control request was not supported by the
> device

The expected behavior is what we see on the other backends and drivers, which are all asynchronous.
Example of WinUSB driver from https://github.com/libusb/libusb/issues/94#issuecomment-955999279:
```
libusb: debug [windows_handle_transfer_completion] handling transfer 000001BF963E7398 completion with errcode 31, length 0
libusb: debug [windows_handle_transfer_completion] detected endpoint stall
```
However with libusb0.sys driver we see:
```
libusb: warning [winusbx_submit_control_transfer] ControlTransfer failed: [31] A device attached to the system is not functioning.
libusb: error [libusb_get_bos_descriptor] failed to read BOS (-1)

```
We recognize the errcode 31 "A device attached to the system is not functioning." which is ERROR_GEN_FAILURE.

In the case of asynchronous transfers, the DLL ControlTransfer always returns ERROR_IO_PENDING, handled in winusbx_submit_control_transfer(). When the transfer at a later point is finished or cancelled, windows_handle_transfer_completion() maps for instance ERROR_GEN_FAILURE (31) to LIBUSB_TRANSFER_STALL, which
again is translated into LIBUSB_ERROR_PIPE by libusb_control_transfer().

In the case of synchronous control transfers (libusb0.sys), all errors (other than ERROR_IO_PENDING) are returned as LIBUSB_ERROR_IO.

tormodvolden, 2021-11-02:
We might need to copy the Windows error translation from windows_handle_transfer_completion, and write the transfer status into the transfer structure, then call windows_force_sync_completion() like we do for a successful transfer.

There is a usbd_status_to_libusb_transfer_status() function available but I don't know if a USBD status is available to us in this case.

tormodvolden, 2021-11-02:
Please give #1018 a try with libusb0.sys and report with debug log.

mcuee, 2021-11-03:
Here is for libusb0.sys 1.2.6.0, libusbk.dll 3.1.0.0 and a FTDI USB to Serial device.

```
C:\work\libusb\libusb\x64\Release\examples [master ↑5 +4 ~0 -0 !]> .\xusb.exe 0403:6001
Using libusb v1.0.24.11670

Opening device 0403:6001...

Reading device descriptor:
            length: 18
      device class: 0
               S/N: 3
           VID:PID: 0403:6001
         bcdDevice: 0600
   iMan:iProd:iSer: 1:2:3
          nb confs: 1

Reading BOS descriptor: libusb: error [libusb_get_bos_descriptor] short BOS read 0/5
no descriptor

Reading first configuration descriptor:
              total length: 32
         descriptor length: 9
             nb interfaces: 1
              interface[0]: id = 0
interface[0].altsetting[0]: num endpoints = 2
   Class.SubClass.Protocol: FF.FF.FF
       endpoint[0].address: 81
           max packet size: 0040
          polling interval: 00
       endpoint[1].address: 02
           max packet size: 0040
          polling interval: 00

Kernel driver attached for interface 0: -12

Claiming interface 0...

Reading string descriptors:
   String (0x01): "FTDI"
   String (0x02): "FT232R USB UART"
   String (0x03): "A8007Ub5"

Reading OS string descriptor: no descriptor

Releasing interface 0...
Closing device...
C:\work\libusb\libusb\x64\Release\examples [master ↑5 +4 ~0 -0 !]> $Env:LIBUSB_DEBUG=4
C:\work\libusb\libusb\x64\Release\examples [master ↑5 +4 ~0 -0 !]> .\xusb.exe 0403:6001
Using libusb v1.0.24.11670

[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.000258] [00006760] libusb: debug [libusb_init] created default context
[ 0.000409] [00006760] libusb: debug [libusb_init] libusb v1.0.24.11670
[ 0.000543] [00006760] libusb: debug [usbi_add_event_source] add HANDLE 0000000000000088 events 0
[ 0.000693] [00006760] libusb: debug [usbi_io_init] using timer for timeouts
[ 0.000777] [00006760] libusb: debug [usbi_add_event_source] add HANDLE 000000000000008C events 0
[ 0.000946] [00006760] libusb: debug [get_windows_version] Windows 8 64-bit
[ 0.001071] [00006760] libusb: debug [htab_create] using 1021 entries hash table
[ 0.007441] [00006760] libusb: info [winusbx_init] WinUSB DLL available (with isoch support)
[ 0.013853] [00006760] libusb: debug [winusbx_init] libusbK DLL found, version: 3.1.0.0
[ 0.020340] [00006760] libusb: error [load_usbdk_helper_dll] Failed to load UsbDkHelper.dll: [126] The specified module could not be found.
[ 0.020489] [00006760] libusb: info [windows_init] UsbDk backend is not available
Opening device 0403:6001...
[ 0.020908] [00006760] libusb: debug [libusb_get_device_list]
[ 0.021862] [00003474] libusb: debug [windows_iocp_thread] I/O completion thread started
[ 0.045840] [00006760] libusb: debug [winusb_get_device_list] allocating new device for session [A1]
[ 0.046099] [00006760] libusb: debug [winusb_get_device_list] allocating new device for session [A2]
[ 0.046675] [00006760] libusb: debug [get_api_type] driver(s): BTHUSB
[... 265 lines cut ...]
[ 0.090894] [00006760] libusb: debug [windows_force_sync_completion] transfer 0000025B28D00778, length 0
[ 0.090973] [00006760] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[ 0.090978] [00003474] libusb: debug [windows_iocp_thread] transfer 0000025B28D00778 completed, length 0
[ 0.091063] [00006760] libusb: debug [usbi_wait_for_events] WaitForMultipleObjects() for 2 HANDLEs with timeout in 60000ms
[ 0.091213] [00006760] libusb: debug [usbi_wait_for_events] WaitForMultipleObjects() returned 0
[ 0.091269] [00006760] libusb: debug [handle_event_trigger] event triggered
[ 0.091353] [00006760] libusb: debug [windows_handle_transfer_completion] handling transfer 0000025B28D00778 completion with errcode 0, length 0
[ 0.091432] [00006760] libusb: debug [arm_timer_for_next_timeout] no timeouts, disarming timer
[ 0.091500] [00006760] libusb: debug [usbi_handle_transfer_completion] transfer 0000025B28D00778 has callback 00007FF7D1058860
[ 0.091559] [00006760] libusb: debug [sync_transfer_cb] actual_length=0
[ 0.091613] [00006760] libusb: debug [libusb_free_transfer] transfer 0000025B28D00778
 no descriptor

Releasing interface 0...
[ 0.091878] [00006760] libusb: debug [libusb_release_interface] interface 0
Closing device...
[ 0.091976] [00006760] libusb: debug [libusb_close]
[ 0.092033] [00006760] libusb: debug [libusb_unref_device] destroy device 1.11
[ 0.092082] [00006760] libusb: debug [libusb_unref_device] destroy device 1.0
[ 0.092133] [00006760] libusb: debug [libusb_exit] destroying default context
[ 0.092182] [00003474] libusb: debug [windows_iocp_thread] I/O completion thread exiting
[ 0.094093] [00006760] libusb: debug [usbi_remove_event_source] remove HANDLE 000000000000008C
[ 0.094335] [00006760] libusb: debug [usbi_remove_event_source] remove HANDLE 0000000000000088

```

mcuee, 2021-11-05:
@tormodvolden 
It is very strange that I can not use your repo, and I can not use the git master and then add #1016 and #1018.

```
C:\work\libusb\libusb_test [master ≡]> git pull origin pull/1016/head
remote: Enumerating objects: 9, done.
remote: Counting objects: 100% (9/9), done.
remote: Compressing objects: 100% (1/1), done.
remote: Total 5 (delta 4), reused 5 (delta 4), pack-reused 0
Unpacking objects: 100% (5/5), 735 bytes | 13.00 KiB/s, done.
From https://github.com/libusb/libusb
 * branch            refs/pull/1016/head -> FETCH_HEAD
Merge made by the 'recursive' strategy.
 libusb/os/windows_winusb.c | 6 ++++--
 1 file changed, 4 insertions(+), 2 deletions(-)
C:\work\libusb\libusb_test [master ↑2]> git pull origin pull/1018/head
remote: Enumerating objects: 20, done.
remote: Counting objects: 100% (20/20), done.
remote: Compressing objects: 100% (2/2), done.
remote: Total 13 (delta 11), reused 13 (delta 11), pack-reused 0
Unpacking objects: 100% (13/13), 1.64 KiB | 16.00 KiB/s, done.
From https://github.com/libusb/libusb
 * branch            refs/pull/1018/head -> FETCH_HEAD
Auto-merging libusb/os/windows_winusb.c
CONFLICT (content): Merge conflict in libusb/os/windows_winusb.c
Automatic merge failed; fix conflicts and then commit the result.
```

mcuee, 2021-11-05:
So I will use your branch instead.
```
C:\work\libusb> git clone https://github.com/tormodvolden/libusb.git libusb_tormod
Cloning into 'libusb_tormod'...
remote: Enumerating objects: 16068, done.
remote: Counting objects: 100% (459/459), done.
remote: Compressing objects: 100% (198/198), done.
remote: Total 16068 (delta 270), reused 338 (delta 200), pack-reused 15609
Receiving objects: 100% (16068/16068), 5.15 MiB | 2.17 MiB/s, done.
Resolving deltas: 100% (11613/11613), done.
C:\work\libusb> cd .\libusb_tormod\
C:\work\libusb\libusb_tormod [master ≡]> git branch -a
* master
  remotes/origin/HEAD -> origin/master
  remotes/origin/api_bump
  remotes/origin/ghost-fix_zero_length_p_678
  remotes/origin/libusb0ctrlerror
  remotes/origin/libusb0ctrlxfer
  remotes/origin/master
  remotes/origin/multDeviceInterfaceGuids
  remotes/origin/nano_on_master
  remotes/origin/win_endpoints0-timeout
C:\work\libusb\libusb_tormod [master ≡]> git checkout libusb0ctrlerror
```

mcuee, 2021-11-05:
The following test is done with libusb0.sys 1.2.6.0 release and libusbk.dll 3.1.0.0 with https://github.com/tormodvolden/libusb/tree/libusb0ctrlerror

Run log:
```
C:\work\libusb\libusb_tormod\x64\Release\examples [libusb0ctrlerror ≡]> .\xusb.exe 0925:7001
Using libusb v1.0.24.11670

Opening device 0925:7001...

Reading device descriptor:
            length: 18
      device class: 0
               S/N: 0
           VID:PID: 0925:7001
         bcdDevice: 0001
   iMan:iProd:iSer: 1:2:0
          nb confs: 1

Reading BOS descriptor: libusb: error [libusb_get_bos_descriptor] short BOS read 0/5
no descriptor

Reading first configuration descriptor:
              total length: 41
         descriptor length: 9
             nb interfaces: 1
              interface[0]: id = 0
interface[0].altsetting[0]: num endpoints = 2
   Class.SubClass.Protocol: 03.00.00
       endpoint[0].address: 81
           max packet size: 0040
          polling interval: 01
       endpoint[1].address: 01
           max packet size: 0040
          polling interval: 01

Kernel driver attached for interface 0: -12

Claiming interface 0...

Reading string descriptors:
   String (0x01): "Lakeview Research"
   String (0x02): "Generic HID"

Reading OS string descriptor: no descriptor

Releasing interface 0...
Closing device...
```

Debug log.
<details>
<summary> click for the full debug log </summary>

```
C:\work\libusb\libusb_tormod\x64\Release\examples [libusb0ctrlerror ≡]> $Env:LIBUSB_DEBUG=4
C:\work\libusb\libusb_tormod\x64\Release\examples [libusb0ctrlerror ≡]> .\xusb.exe 0925:7001
Using libusb v1.0.24.11670

[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.000293] [0000510c] libusb: debug [libusb_init] created default context
[ 0.000459] [0000510c] libusb: debug [libusb_init] libusb v1.0.24.11670
[ 0.000615] [0000510c] libusb: debug [usbi_add_event_source] add HANDLE 0000000000000088 events 0
[ 0.000787] [0000510c] libusb: debug [usbi_io_init] using timer for timeouts
[ 0.000932] [0000510c] libusb: debug [usbi_add_event_source] add HANDLE 000000000000008C events 0
[ 0.001109] [0000510c] libusb: debug [get_windows_version] Windows 8 64-bit
[ 0.001252] [0000510c] libusb: debug [htab_create] using 1021 entries hash table
[ 0.007417] [0000510c] libusb: info [winusbx_init] WinUSB DLL available (with isoch support)
[ 0.012870] [0000510c] libusb: debug [winusbx_init] libusbK DLL found, version: 3.1.0.0
[ 0.018674] [0000510c] libusb: error [load_usbdk_helper_dll] Failed to load UsbDkHelper.dll: [126] The specified module could not be found.
[... 242 lines cut ...]
[ 0.068670] [0000510c] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[ 0.068716] [0000510c] libusb: debug [usbi_wait_for_events] WaitForMultipleObjects() for 2 HANDLEs with timeout in 60000ms
[ 0.068673] [00005978] libusb: debug [windows_iocp_thread] transfer 0000022B840A2A68 completed, length 0
[ 0.068830] [0000510c] libusb: debug [usbi_wait_for_events] WaitForMultipleObjects() returned 0
[ 0.068876] [0000510c] libusb: debug [handle_event_trigger] event triggered
[ 0.068920] [0000510c] libusb: debug [windows_handle_transfer_completion] handling transfer 0000022B840A2A68 completion with errcode 0, length 0
[ 0.068983] [0000510c] libusb: debug [arm_timer_for_next_timeout] no timeouts, disarming timer
[ 0.069029] [0000510c] libusb: debug [usbi_handle_transfer_completion] transfer 0000022B840A2A68 has callback 00007FF78FFB8860
[ 0.069091] [0000510c] libusb: debug [sync_transfer_cb] actual_length=0
[ 0.069136] [0000510c] libusb: debug [libusb_free_transfer] transfer 0000022B840A2A68
 no descriptor

Releasing interface 0...
[ 0.069295] [0000510c] libusb: debug [libusb_release_interface] interface 0
Closing device...
[ 0.069380] [0000510c] libusb: debug [libusb_close]
[ 0.069429] [0000510c] libusb: debug [libusb_unref_device] destroy device 1.15
[ 0.069474] [0000510c] libusb: debug [libusb_unref_device] destroy device 1.0
[ 0.069520] [0000510c] libusb: debug [libusb_exit] destroying default context
[ 0.069569] [00005978] libusb: debug [windows_iocp_thread] I/O completion thread exiting
[ 0.071237] [0000510c] libusb: debug [usbi_remove_event_source] remove HANDLE 000000000000008C
[ 0.071296] [0000510c] libusb: debug [usbi_remove_event_source] remove HANDLE 0000000000000088
```

</details>

mcuee, 2021-11-05:
Comparison for libusbk.sys 3.0.7.0 and libusbk.dll 3.1.0.0.

```
C:\work\libusb\libusb_tormod\x64\Release\examples [libusb0ctrlerror ≡]> .\xusb.exe 0925:7001
Using libusb v1.0.24.11670

[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.000247] [00002b04] libusb: debug [libusb_init] created default context
[ 0.000372] [00002b04] libusb: debug [libusb_init] libusb v1.0.24.11670
[ 0.000486] [00002b04] libusb: debug [usbi_add_event_source] add HANDLE 0000000000000088 events 0
[ 0.000607] [00002b04] libusb: debug [usbi_io_init] using timer for timeouts
[ 0.000715] [00002b04] libusb: debug [usbi_add_event_source] add HANDLE 000000000000008C events 0
[ 0.000848] [00002b04] libusb: debug [get_windows_version] Windows 8 64-bit
[ 0.000966] [00002b04] libusb: debug [htab_create] using 1021 entries hash table
[ 0.009669] [00002b04] libusb: info [winusbx_init] WinUSB DLL available (with isoch support)
[ 0.015692] [00002b04] libusb: debug [winusbx_init] libusbK DLL found, version: 3.1.0.0
[ 0.021027] [00002b04] libusb: error [load_usbdk_helper_dll] Failed to load UsbDkHelper.dll: [126] The specified module could not be found.
[ 0.021148] [00002b04] libusb: info [windows_init] UsbDk backend is not available
Opening device 0925:7001...
[ 0.021507] [00002b04] libusb: debug [libusb_get_device_list]
[ 0.022588] [00005ae8] libusb: debug [windows_iocp_thread] I/O completion thread started
[ 0.045676] [00002b04] libusb: debug [winusb_get_device_list] allocating new device for session [A0]
[ 0.045853] [00002b04] libusb: debug [winusb_get_device_list] allocating new device for session [A1]
[ 0.046234] [00002b04] libusb: debug [get_api_type] driver(s): BTHUSB
[ 0.046368] [00002b04] libusb: debug [get_api_type] lower filter driver(s): ibtusb
[ 0.046439] [00002b04] libusb: debug [winusb_get_device_list] allocating new device for session [16]
[ 0.046940] [00002b04] libusb: debug [get_api_type] driver(s): libusbK
[ 0.047042] [00002b04] libusb: debug [get_api_type] matched driver name against libusbK
[ 0.047110] [00002b04] libusb: debug [winusb_get_device_list] allocating new device for session [66]
[ 0.047401] [00002b04] libusb: debug [get_api_type] driver(s): usbccgp
[ 0.047563] [00002b04] libusb: debug [get_api_type] matched driver name against Composite API
[ 0.047724] [00002b04] libusb: debug [winusb_get_device_list] allocating new device for session [7E]
[ 0.048043] [00002b04] libusb: debug [get_api_type] driver(s): usbccgp
[ 0.048173] [00002b04] libusb: debug [get_api_type] matched driver name against Composite API
[ 0.048271] [00002b04] libusb: debug [winusb_get_device_list] allocating new device for session [7F]
[ 0.048532] [00002b04] libusb: debug [get_api_type] driver(s): usbccgp
[ 0.048663] [00002b04] libusb: debug [get_api_type] matched driver name against Composite API
[ 0.048765] [00002b04] libusb: debug [winusb_get_device_list] allocating new device for session [80]
[ 0.049011] [00002b04] libusb: debug [get_api_type] driver(s): usbccgp
[ 0.049138] [00002b04] libusb: debug [get_api_type] matched driver name against Composite API
[ 0.049235] [00002b04] libusb: debug [winusb_get_device_list] allocating new device for session [81]
[ 0.049536] [00002b04] libusb: debug [enumerate_hcd_root_hub] assigning HCD 'PCI\VEN_8086&DEV_A36D&SUBSYS_091A1028&REV_10\3&11583659&0&A0' bus number 1
[ 0.049666] [00002b04] libusb: debug [enumerate_hcd_root_hub] assigning HCD 'PCI\VEN_8086&DEV_15DB&SUBSYS_091A1028&REV_02\71D9F51854B3020000' bus number 2
[ 0.051089] [00002b04] libusb: debug [winusb_get_device_list] extra GUID: {865C4DDE-10B6-6373-FAEE-1506B321A619}
[ 0.051194] [00002b04] libusb: debug [winusb_get_device_list] found existing device for session [81]
[ 0.051425] [00002b04] libusb: debug [init_device] found 1 configurations (current config: 1)
[ 0.051522] [00002b04] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 59 bytes)
[ 0.051714] [00002b04] libusb: debug [init_device] (bus: 1, addr: 2, depth: 1, port: 1): 'USB\VID_046D&PID_C534\5&E9F3E45&0&1'
[ 0.051959] [00002b04] libusb: debug [winusb_get_device_list] found existing device for session [16]
[ 0.052123] [00002b04] libusb: debug [init_device] found 1 configurations (current config: 1)
[ 0.052207] [00002b04] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 200 bytes)
[ 0.052320] [00002b04] libusb: debug [init_device] (bus: 1, addr: 5, depth: 1, port: 14): 'USB\VID_8087&PID_0AAA\5&E9F3E45&0&14'
[ 0.052827] [00002b04] libusb: debug [winusb_get_device_list] found existing device for session [80]
[ 0.053001] [00002b04] libusb: debug [init_device] found 1 configurations (current config: 1)
[ 0.053059] [00002b04] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 288 bytes)
[ 0.053132] [00002b04] libusb: debug [init_device] (bus: 1, addr: 13, depth: 1, port: 2): 'USB\VID_047F&PID_C056\D1CEC32927974D5F9BD6B2AEBF2EA8E3'
[ 0.053572] [00002b04] libusb: debug [init_root_hub] root hub 'USB\ROOT_HUB30\4&36020D6F&0&0' reports 26 ports
[ 0.053696] [00002b04] libusb: debug [init_device] (bus: 1, addr: 0, depth: 0, port: 0): 'USB\ROOT_HUB30\4&36020D6F&0&0'
[ 0.053979] [00002b04] libusb: debug [winusb_get_device_list] found existing device for session [7F]
[ 0.054113] [00002b04] libusb: debug [init_device] found 1 configurations (current config: 1)
[... cut ...]
[ 0.075477] [00002b04] libusb: debug [libusb_unref_device] destroy device 1.16
[ 0.075521] [00002b04] libusb: debug [libusb_unref_device] destroy device 1.0
[ 0.075569] [00002b04] libusb: debug [libusb_exit] destroying default context
[ 0.075652] [00005ae8] libusb: debug [windows_iocp_thread] I/O completion thread exiting
[ 0.077488] [00002b04] libusb: debug [usbi_remove_event_source] remove HANDLE 000000000000008C
[ 0.077595] [00002b04] libusb: debug [usbi_remove_event_source] remove HANDLE 0000000000000088
```

mcuee, 2021-11-05:
libusb0.sys 1.2.7.2 seems to be fine (with libusbk.dll 3.1.0.0)

Run log
```
C:\work\libusb\libusb_tormod\x64\Release\examples [libusb0ctrlerror ≡]> .\xusb.exe 0925:7001
Using libusb v1.0.24.11670

Opening device 0925:7001...

Reading device descriptor:
            length: 18
      device class: 0
               S/N: 0
           VID:PID: 0925:7001
         bcdDevice: 0001
   iMan:iProd:iSer: 1:2:0
          nb confs: 1

Reading BOS descriptor: no descriptor

Reading first configuration descriptor:
              total length: 41
         descriptor length: 9
             nb interfaces: 1
              interface[0]: id = 0
interface[0].altsetting[0]: num endpoints = 2
   Class.SubClass.Protocol: 03.00.00
       endpoint[0].address: 81
           max packet size: 0040
          polling interval: 01
       endpoint[1].address: 01
           max packet size: 0040
          polling interval: 01

Kernel driver attached for interface 0: -12

Claiming interface 0...

Reading string descriptors:
   String (0x01): "Lakeview Research"
   String (0x02): "Generic HID"

Reading OS string descriptor: no descriptor

Releasing interface 0...
Closing device...
```

tormodvolden, 2021-11-05:
Thanks again!
> It is very strange that I can not use the git master and then add #1016 and #1018.

Sorry for the confusion, this branch includes the commit from 1016 (to make it easier to test with one pull) and previously you could pull 1016 and then 1018 which would then fast-forward merge. However, the new version of 1018 is rebased and then you get some auto-merging (not sure why it fails though). The simple solution is to check out git master and only pull in 1018.

> The error message from libusb seems to be better now. But there is still an error message libusb: error [libusb_get_bos_descriptor] short BOS read 0/5.

It still doesn't do what I wanted. The Windows API error 31 should have ended up (via LIBUSB_TRANSFER_STALL) as LIBUSB_ERROR_PIPE, which is recognized by libusb_get_bos_descriptor() so it wouldn't print an error.

This shows the error code didn't make it through:
`[windows_handle_transfer_completion] handling transfer 0000022B840A2A68 completion with errcode 0, length 0`

tormodvolden, 2021-11-07:
I added commit 235d88c to #1018 which could solve this, but there is a chance of breaking the real asynchronous drivers (WinUSB or libusbK). If this doesn't work we might have to add a new member to the window_transfer_priv struct for passing the status or for signalling that the transfer was forced to complete synchronously.

mcuee, 2021-11-08:
> I added commit [235d88c](https://github.com/libusb/libusb/commit/235d88cc5a5090b3ea18e7ec0f6fc2b180611fb7) to #1018 which could solve this, but there is a chance of breaking the real asynchronous [...]

This seems to perform better for libusb0.sys.

Debug log:

```
C:\work\libusb\libusb_tormod\x64\Release\examples [libusb0ctrlerror ≡]> $Env:LIBUSB_DEBUG=4
C:\work\libusb\libusb_tormod\x64\Release\examples [libusb0ctrlerror ≡]> .\xusb.exe 0925:7001
Using libusb v1.0.24.11670

[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.000231] [00004164] libusb: debug [libusb_init] created default context
[ 0.000366] [00004164] libusb: debug [libusb_init] libusb v1.0.24.11670
[ 0.000486] [00004164] libusb: debug [usbi_add_event_source] add HANDLE 0000000000000088 events 0
[ 0.000622] [00004164] libusb: debug [usbi_io_init] using timer for timeouts
[ 0.000697] [00004164] libusb: debug [usbi_add_event_source] add HANDLE 000000000000008C events 0
[ 0.000798] [00004164] libusb: debug [get_windows_version] Windows 8 64-bit
[ 0.000908] [00004164] libusb: debug [htab_create] using 1021 entries hash table
[ 0.008318] [00004164] libusb: info [winusbx_init] WinUSB DLL available (with isoch support)
[ 0.013927] [00004164] libusb: debug [winusbx_init] libusbK DLL found, version: 3.1.0.0
[ 0.019366] [00004164] libusb: error [load_usbdk_helper_dll] Failed to load UsbDkHelper.dll: [126] The specified module could not be found.
[ 0.019550] [00004164] libusb: info [windows_init] UsbDk backend is not available
Opening device 0925:7001...
[ 0.020172] [00004164] libusb: debug [libusb_get_device_list]
[ 0.021789] [000024e8] libusb: debug [windows_iocp_thread] I/O completion thread started
[ 0.045312] [00004164] libusb: debug [winusb_get_device_list] allocating new device for session [A1]
[ 0.045511] [00004164] libusb: debug [winusb_get_device_list] allocating new device for session [A2]
[ 0.045936] [00004164] libusb: debug [get_api_type] driver(s): BTHUSB
[ 0.046136] [00004164] libusb: debug [get_api_type] lower filter driver(s): ibtusb
[ 0.046217] [00004164] libusb: debug [winusb_get_device_list] allocating new device for session [15]
[ 0.046770] [00004164] libusb: debug [get_api_type] driver(s): libusb0
[ 0.046928] [00004164] libusb: debug [get_api_type] matched driver name against libusb0
[ 0.047039] [00004164] libusb: debug [winusb_get_device_list] allocating new device for session [6E]
[ 0.047402] [00004164] libusb: debug [get_api_type] driver(s): usbccgp
[ 0.047642] [00004164] libusb: debug [get_api_type] matched driver name against Composite API
[ 0.047776] [00004164] libusb: debug [winusb_get_device_list] allocating new device for session [7E]
[ 0.048136] [00004164] libusb: debug [get_api_type] driver(s): usbccgp
[ 0.048325] [00004164] libusb: debug [get_api_type] matched driver name against Composite API
[ 0.048476] [00004164] libusb: debug [winusb_get_device_list] allocating new device for session [7F]
[ 0.048840] [00004164] libusb: debug [get_api_type] driver(s): usbccgp
[ 0.049034] [00004164] libusb: debug [get_api_type] matched driver name against Composite API
[ 0.049181] [00004164] libusb: debug [winusb_get_device_list] allocating new device for session [80]
[ 0.049547] [00004164] libusb: debug [get_api_type] driver(s): usbccgp
[ 0.049732] [00004164] libusb: debug [get_api_type] matched driver name against Composite API
[ 0.049866] [00004164] libusb: debug [winusb_get_device_list] allocating new device for session [81]
[ 0.050249] [00004164] libusb: debug [enumerate_hcd_root_hub] assigning HCD 'PCI\VEN_8086&DEV_A36D&SUBSYS_091A1028&REV_10\3&11583659&0&A0' bus number 1
[ 0.050429] [00004164] libusb: debug [enumerate_hcd_root_hub] assigning HCD 'PCI\VEN_8086&DEV_15DB&SUBSYS_091A1028&REV_02\71D9F51854B3020000' bus number 2
[ 0.051833] [00004164] libusb: debug [winusb_get_device_list] extra GUID: {865C4DDE-10B6-6373-FAEE-1506B321A619}
[ 0.051998] [00004164] libusb: debug [winusb_get_device_list] found existing device for session [81]
[ 0.052248] [00004164] libusb: debug [init_device] found 1 configurations (current config: 1)
[ 0.052388] [00004164] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 59 bytes)
[ 0.052532] [00004164] libusb: debug [init_device] (bus: 1, addr: 1, depth: 1, port: 1): 'USB\VID_046D&PID_C534\5&E9F3E45&0&1'
[ 0.052836] [00004164] libusb: debug [winusb_get_device_list] found existing device for session [15]
[ 0.053240] [00004164] libusb: debug [init_device] found 1 configurations (current config: 1)
[ 0.053395] [00004164] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 200 bytes)
[ 0.053556] [00004164] libusb: debug [init_device] (bus: 1, addr: 6, depth: 1, port: 14): 'USB\VID_8087&PID_0AAA\5&E9F3E45&0&14'
[ 0.054200] [00004164] libusb: debug [winusb_get_device_list] found existing device for session [80]
[ 0.054423] [00004164] libusb: debug [init_device] found 1 configurations (current config: 1)
[ 0.054545] [00004164] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 288 bytes)
[ 0.054671] [00004164] libusb: debug [init_device] (bus: 1, addr: 2, depth: 1, port: 2): 'USB\VID_047F&PID_C056\D1CEC32927974D5F9BD6B2AEBF2EA8E3'
[ 0.055219] [00004164] libusb: debug [init_root_hub] root hub 'USB\ROOT_HUB30\4&36020D6F&0&0' reports 26 ports
[... cut ...]
[ 0.079135] [00004164] libusb: debug [libusb_unref_device] destroy device 1.7
[ 0.079178] [00004164] libusb: debug [libusb_unref_device] destroy device 1.0
[ 0.079227] [00004164] libusb: debug [libusb_exit] destroying default context
[ 0.079275] [000024e8] libusb: debug [windows_iocp_thread] I/O completion thread exiting
[ 0.080944] [00004164] libusb: debug [usbi_remove_event_source] remove HANDLE 000000000000008C
[ 0.081002] [00004164] libusb: debug [usbi_remove_event_source] remove HANDLE 0000000000000088
```

tormodvolden, 2021-11-08:
Thanks a lot! It now does exactly what it should for libusb0. I was unsure if the overlapped->Internal gets reset by the other kernel drivers on successful transfer and it sounds like it does. It would be good to have this part reviewed by someone familiar with Windows internals, and maybe this should wait til 1.0.26. But #1016 is probably safe for 1.0.25.

mcuee, 2021-11-08:
I agree that #1016 is safe to be merged. We can always delay #1018 if the review can not be done on time.

mcuee, 2022-06-18:
Hopefully someone can review #1018 and get it merged.

@tormodvolden 
BTW, other than the filter driver mode, right now libusb0.sys works pretty well based on my testing of avrdude under Windows.

mcuee, 2023-02-15:
New issues found from avrdude testing. But it may not be related to this issue.
* https://github.com/libusb/libusb/issues/1226

mcuee, 2023-02-15:
I just tried the latest git plus PR #1018, it seems to work fine if I use simple test program like xusb.

Interface 0 is using libusb0.sys driver.
Interface 1 is using Windows HID driver.

```
C:\work\libusb\libusb_pr1018 [master ≡]> git pull origin pull/1018/head
From https://github.com/libusb/libusb
 * branch              refs/pull/1018/head -> FETCH_HEAD
Auto-merging libusb/os/windows_common.c
Auto-merging libusb/os/windows_common.h
Auto-merging libusb/os/windows_usbdk.c
Auto-merging libusb/os/windows_winusb.c
Merge made by the 'ort' strategy.
 libusb/os/windows_common.c |  7 ++++---
 libusb/os/windows_common.h |  2 +-
 libusb/os/windows_usbdk.c  |  6 +++---
 libusb/os/windows_winusb.c | 10 +++++-----
 4 files changed, 13 insertions(+), 12 deletions(-)

C:\work\libusb\libusb_pr1018 [master ↑3]> .\examples\xusb.exe 16c0:05dc
Using libusb v1.0.26.11784

Opening device 16C0:05DC...

Reading device descriptor:
            length: 18
      device class: 0
               S/N: 3
           VID:PID: 16C0:05DC
         bcdDevice: 0110
   iMan:iProd:iSer: 1:2:3
          nb confs: 1

Reading BOS descriptor: 2 caps
    Container ID:
      {b9d357ad-6611-f843-8790-ebe14ddc7594}
    Platform descriptor:
      bLength                : 28
      PlatformCapabilityUUID : {df60ddd8-8945-c74c-9cd2-659d9e648a9f}

  00000000  00 00 03 06 36 01 5d 00                          ....6.].

Reading first configuration descriptor:
              total length: 50
         descriptor length: 9
             nb interfaces: 2
              interface[0]: id = 0
interface[0].altsetting[0]: num endpoints = 0
   Class.SubClass.Protocol: FF.00.00
              interface[1]: id = 1
interface[1].altsetting[0]: num endpoints = 2
   Class.SubClass.Protocol: 03.00.00
       endpoint[0].address: 81
           max packet size: 0008
          polling interval: 0A
       endpoint[1].address: 01
           max packet size: 0008
          polling interval: 0A

Kernel driver attached for interface 0: (not supported)

Claiming interface 0...

Kernel driver attached for interface 1: (not supported)

Claiming interface 1...

Reading string descriptors:
   String (0x01): "www.fischl.de"
   String (0x02): "USBasp"
   String (0x03): "0001"

Reading OS string descriptor: no descriptor

Reading interface association descriptors (IADs) for first configuration:
    nb IADs: 0

Releasing interface 0...
Releasing interface 1...
Closing device...
```

mcuee, 2023-02-15:
Another test with CH340 USB to Serial Converter and libusb0.sys 1.2.6.0 version in filter driver mode. The results are good as well with latest libusb git with or without this PR.
* https://github.com/libusb/libusb/issues/1019#issuecomment-1430783650