libusb/libusb #1019: Windows: libusb0.sys filter driver support improvement

mcuee, 2021-11-03:
Ref: https://github.com/libusb/libusb/issues/94 has been more or less resolved by #1016.

The next thing is to see whether we can improve the libusb-win32 (libusb0.sys) filter driver mode support. Take note the filter driver itself may not work for some devices.

mcuee, 2021-11-03:
Test with #1016 and #1018 on top of libusb git head https://github.com/libusb/libusb/commit/f33c9562a1dd7a0cb333fcd9a670ba14fbbfbc0e

mcuee, 2021-11-03:
It is the same for 1.0.23.

```
C:\work\libusb\libusb-1.0.23_bin\examples\bin64> $Env:LIBUSB_DEBUG=0
C:\work\libusb\libusb-1.0.23_bin\examples\bin64> .\xusb.exe 0403:6001
Using libusb v1.0.23.11397

Opening device 0403:6001...

Reading device descriptor:
            length: 18
      device class: 0
               S/N: 3
           VID:PID: 0403:6001
         bcdDevice: 0600
   iMan:iProd:iSer: 1:2:3
          nb confs: 1

Reading BOS descriptor: no descriptor

Reading first configuration descriptor:
             nb interfaces: 1
              interface[0]: id = 0
interface[0].altsetting[0]: num endpoints = 2
   Class.SubClass.Protocol: FF.FF.FF
       endpoint[0].address: 81
           max packet size: 0040
          polling interval: 00
       endpoint[1].address: 02
           max packet size: 0040
          polling interval: 00

Claiming interface 0...

Reading string descriptors:
   String (0x01): "FTDI"
   String (0x02): "FT232R USB UART"
   String (0x03): "A8007Ub5"

Releasing interface 0...
Closing device...
C:\work\libusb\libusb-1.0.23_bin\examples\bin64> .\xusb.exe 0403:6001
Using libusb v1.0.23.11397

Opening device 0403:6001...
  Failed.
```

mcuee, 2021-11-03:
Even if I take out the BOS descriptor fetching, the result is the same.

```
C:\work\libusb\libusb\x64\Release\examples [master ↑5 +4 ~1 -0 !]> .\xusb_nobos.exe 0403:6001
Using libusb v1.0.24.11670

Opening device 0403:6001...

Reading device descriptor:
            length: 18
      device class: 0
               S/N: 3
           VID:PID: 0403:6001
         bcdDevice: 0600
   iMan:iProd:iSer: 1:2:3
          nb confs: 1

Reading first configuration descriptor:
              total length: 32
         descriptor length: 9
             nb interfaces: 1
              interface[0]: id = 0
interface[0].altsetting[0]: num endpoints = 2
   Class.SubClass.Protocol: FF.FF.FF
       endpoint[0].address: 81
           max packet size: 0040
          polling interval: 00
       endpoint[1].address: 02
           max packet size: 0040
          polling interval: 00

Kernel driver attached for interface 0: -12

Claiming interface 0...

Reading string descriptors:
   String (0x01): "FTDI"
   String (0x02): "FT232R USB UART"
   String (0x03): "A8007Ub5"

Reading OS string descriptor: no descriptor

Releasing interface 0...
Closing device...
C:\work\libusb\libusb\x64\Release\examples [master ↑5 +4 ~1 -0 !]> .\xusb_nobos.exe 0403:6001
Using libusb v1.0.24.11670

Opening device 0403:6001...
  Failed.
```

mcuee, 2021-11-03:
So I take out BOS and MSOS descriptor fetching and the result is the same.

```
C:\work\libusb\libusb\x64\Release\examples [master ↑5 +4 ~1 -0 !]> .\xusb_nobos_nomsos.exe 0403:6001
Using libusb v1.0.24.11670

Opening device 0403:6001...

Reading device descriptor:
            length: 18
      device class: 0
               S/N: 3
           VID:PID: 0403:6001
         bcdDevice: 0600
   iMan:iProd:iSer: 1:2:3
          nb confs: 1

Reading first configuration descriptor:
              total length: 32
         descriptor length: 9
             nb interfaces: 1
              interface[0]: id = 0
interface[0].altsetting[0]: num endpoints = 2
   Class.SubClass.Protocol: FF.FF.FF
       endpoint[0].address: 81
           max packet size: 0040
          polling interval: 00
       endpoint[1].address: 02
           max packet size: 0040
          polling interval: 00

Kernel driver attached for interface 0: -12

Claiming interface 0...

Reading string descriptors:
   String (0x01): "FTDI"
   String (0x02): "FT232R USB UART"
   String (0x03): "A8007Ub5"

Releasing interface 0...
Closing device...
C:\work\libusb\libusb\x64\Release\examples [master ↑5 +4 ~1 -0 !]> .\xusb_nobos_nomsos.exe 0403:6001
Using libusb v1.0.24.11670

Opening device 0403:6001...
  Failed.

```

mcuee, 2021-11-03:
This may not be a libusb problem though. It could be a problem with the filter driver.

mcuee, 2021-11-03:
I build libftdi under Windows since 2010.
https://sourceforge.net/projects/picusb/files/
https://github.com/mcuee/libftdi/releases

libftdi-0.20git was the last version I built back in 2012 and it uses libusb-win32 API. After that, libftdi switched to use libusb-1.0 API.

Interestingly libftdi-0.20git works with libusb0.sys filter mode, but not libftdi-1.x, for the basic program find_all.exe.

Source code of find_all (from my mirror)
https://github.com/mcuee/libftdi/blob/master/examples/find_all.c

Run log with libftdi-0.20git
```
C:\work\libusb\libftdi_bin\libftdi_0.20git_win32\libftdi_0.20git_win32\bin> .\find_all.exe
Number of FTDI devices found: 1
Checking device: 0
Manufacturer: FTDI, Description: FT232R USB UART

C:\work\libusb\libftdi_bin\libftdi_0.20git_win32\libftdi_0.20git_win32\bin> .\simple.exe
ftdi_read_chipid: 0
FTDI chipid: 30B7EF1
```

Run log with a recent snapshot.
```
C:\work\libusb\libftdi_bin\libftdi_snapshot_bin\bin> .\find_all.exe
Number of FTDI devices found: 1
Checking device: 0
ftdi_usb_get_strings failed: -4 (libusb_open() failed)
C:\work\libusb\libftdi_bin\libftdi_snapshot_bin\bin> .\find_all.exe
Number of FTDI devices found: 1
Checking device: 0
ftdi_usb_get_strings failed: -4 (libusb_open() failed)

C:\work\libusb\libftdi_bin\libftdi_snapshot_bin\bin> .\simple.exe
Initialized libftdi 1.5 (major: 1, minor: 5, micro: 0, snapshot ver: v1.5-12-g0684c1b)
ftdi_read_chipid: 0
FTDI chipid: 30B7EF1
```

mcuee, 2021-11-03:
No issues with the device driver mode, using libusb0.sys 1.2.6.0 and libusbk.dll 3.1.0.0 and libusb-1.0.dll patched version with #1016 and #1018.
```
C:\work\libusb\libftdi_bin\libftdi_snapshot_bin\bin> .\find_all.exe
Number of FTDI devices found: 1
Checking device: 0
Manufacturer: FTDI, Description: FT232R USB UART

C:\work\libusb\libftdi_bin\libftdi_snapshot_bin\bin> .\find_all.exe
Number of FTDI devices found: 1
Checking device: 0
Manufacturer: FTDI, Description: FT232R USB UART
```

mcuee, 2023-02-15:
The following is a test with CH340 USB to Serial Converter using libusb0.sys filter driver. It seems to work well.

```
C:\work\libusb\libusb\mingw64\bin [master ≡ +2 ~0 -0 !]> .\xusb.exe 1a86:7523
Using libusb v1.0.26.11748

Opening device 1A86:7523...

Reading device descriptor:
            length: 18
      device class: 255
               S/N: 0
           VID:PID: 1A86:7523
         bcdDevice: 0264
   iMan:iProd:iSer: 0:2:0
          nb confs: 1

Reading BOS descriptor: no descriptor

Reading first configuration descriptor:
              total length: 39
         descriptor length: 9
             nb interfaces: 1
              interface[0]: id = 0
interface[0].altsetting[0]: num endpoints = 3
   Class.SubClass.Protocol: FF.01.02
       endpoint[0].address: 82
           max packet size: 0020
          polling interval: 00
       endpoint[1].address: 02
           max packet size: 0020
          polling interval: 00
       endpoint[2].address: 81
           max packet size: 0008
          polling interval: 01

Kernel driver attached for interface 0: -12

Claiming interface 0...

Reading string descriptors:
   String (0x02): "USB Serial"

Reading OS string descriptor: no descriptor

Releasing interface 0...
Closing device...

```

mcuee, 2023-02-15:
However, the issue with FT232 USB to Serial Converter (using libusb0.sys in filter driver mode) seems to still exist.

First run is okay, second run fails.

The following is with git HEAD plus PR #1018.

```
PS C:\work\libusb\libusb_pr1018> .\examples\xusb.exe 0403:6001
Using libusb v1.0.26.11784

Opening device 0403:6001...

Reading device descriptor:
            length: 18
      device class: 0
               S/N: 3
           VID:PID: 0403:6001
         bcdDevice: 0600
   iMan:iProd:iSer: 1:2:3
          nb confs: 1

Reading BOS descriptor: no descriptor

Reading first configuration descriptor:
              total length: 32
         descriptor length: 9
             nb interfaces: 1
              interface[0]: id = 0
interface[0].altsetting[0]: num endpoints = 2
   Class.SubClass.Protocol: FF.FF.FF
       endpoint[0].address: 81
           max packet size: 0040
          polling interval: 00
       endpoint[1].address: 02
           max packet size: 0040
          polling interval: 00

Kernel driver attached for interface 0: (not supported)

Claiming interface 0...

Reading string descriptors:
   String (0x01): "FTDI"
   String (0x02): "US232R"
   String (0x03): "FTDCRPIE"

Reading OS string descriptor: no descriptor

Reading interface association descriptors (IADs) for first configuration:
    nb IADs: 0

Releasing interface 0...
Closing device...

PS C:\work\libusb\libusb_pr1018> .\examples\xusb.exe 0403:6001
Using libusb v1.0.26.11784

Opening device 0403:6001...
libusb: error [winusbx_open] could not open device \\?\USB#VID_0403&PID_6001#FTDCRPIE#{A5DCBF10-6530-11D2-901F-00C04FB951ED} (interface 0): [87] The parameter is incorrect.
  Failed.

```

tormodvolden, 2023-02-15:
I think this must be a bug in libusb0.sys . Is it the same if the device is just opened and closed again, without any transfers?

mcuee, 2023-02-16:
> I think this must be a bug in libusb0.sys . Is it the same if the device is just opened and closed again, without any transfers?

It could be. However, please take not there is no issue if I use device driver mode. So it could still be issues with libusb backend. Or there is a hidden issue with the FTDI device which is somehow triggered by running xusb.

tormodvolden, 2023-02-16:
FWIW, the debug log is identical in the two runs, until the winusbx_open(), which itself doesn't involve device communication. Would it be possible to get the kernel USB debug messages? IIRC, this requires installing the debug build of libusb0.sys.

mcuee, 2023-02-16:
I will try to see if I can get the debug log working over the weekend.

I think this is a low priotity. The device filter is not guarateed to work anyway. It is not really use much based on the feedbacks from libusb-win32 mailing list. 

The replacement for the libusb-win32 filter is supposed to be the usbdk backend in the context of libusb project. However, I can see usbdk backend is also not popular either, maybe due to the need to use `libusb_set_option()`.

tormodvolden, 2023-02-16:
Yes, sure, this doesn't seem very important. What are exactly the use cases for the filter driver?

mcuee, 2023-02-16:
> Yes, sure, this doesn't seem very important. What are exactly the use cases for the filter driver?

The filter driver is supposed to be quite useful -- you do not to replace the original vendor device driver and just install the filter driver. Then you can use the original vendor application, and you can use libusb based application.

This sounds very good. However, it is not easy to get the filter driver work. There is actually very difficult. libusb0.sys filter driver mode works in some cases (worked pretty well back in the Windows 98SE days) but not in other cases. 

usbdk seems to be more advanced and it is supposed to be working for more devices.

mcuee, 2023-02-16:
For the FT232R USB to Serial adapter, the issue happens on both libusb0.sys filter driver 1.2.6.0 and 1.2.7.3 version.

mcuee, 2023-02-17:
It is indeed very difficult to get generic USB filter working. usbdk is more agressive than libusb0.sys device filter (only filter single USB device) and a bit similar to the depreacted libusb0.sys class filter (filter multiple USB devices). Then it may also be more likely to cause problems like BSODs.

Comments about usbdk backend here.
* https://github.com/libusb/libusb/issues/250#issuecomment-1434568980

> Dolphin has been using 1.0.26 since April 2022 but I think the majority of users are not using the usbdk backend. Sadly we haven't been able to recommend it for a while because of general instability issues and even reports of blue screens iirc.

mcuee, 2023-02-19:
The libusb0.sys device filter driver now works reasonably well for some use cases. The following is using the FT232R with avrdude.

Without filter driver, avrdude using libftdi-1.0 and libusb-1.0 will fail when using the FTDI vendor driver.
```
PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude_git  -c arduino-ft232r -p t13a -U flash:r:attiny13a_test.hex:i
avrdude_git error: cannot open ftdi device: usb_open() failed
avrdude_git error: unable to open programmer arduino-ft232r on port usb

avrdude_git done.  Thank you.
```

After installation of the filter driver, it works fine without switching out FTDI vendor driver.
```
PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude_git  -c arduino-ft232r -p t13a -U flash:r:attiny13a_test.hex:i

avrdude_git: AVR device initialized and ready to accept instructions
avrdude_git: device signature = 0x1e9007 (probably t13a)
avrdude_git: reading flash memory ...

Reading | ################################################## | 100% 0.64 s

avrdude_git: writing output file attiny13a_test.hex

avrdude_git done.  Thank you.

```

mcuee, 2023-02-19:
@tormodvolden 
BTW, pr #1018 does help with the libusb-1.0.27.3 filter driver usage.

libusb git master branch
```
PS C:\work\libusb\libusb_pr1018> .\examples\xusb.exe 0403:6001
Using libusb v1.0.26.11784

Opening device 0403:6001...
libusb: error [winusbx_open] could not open device \\?\USB#VID_0403&PID_6001#A50285BI#{A5DCBF10-6530-11D2-901F-00C04FB951ED} (interface 0): [87] The parameter is incorrect.
  Failed.
PS C:\work\libusb\libusb_pr1018> cd ..
PS C:\work\libusb> cd libusb
PS C:\work\libusb\libusb> .\examples\xusb.exe 0403:6001
Using libusb v1.0.26.11784

Opening device 0403:6001...

Reading device descriptor:
            length: 18
      device class: 0
               S/N: 3
           VID:PID: 0403:6001
         bcdDevice: 0600
   iMan:iProd:iSer: 1:2:3
          nb confs: 1

Reading BOS descriptor: libusb: warning [winusbx_submit_control_transfer] ControlTransfer failed: [31] A device attached to the system is not functioning.
libusb: error [libusb_get_bos_descriptor] failed to read BOS (-1)
no descriptor

Reading first configuration descriptor:
              total length: 32
         descriptor length: 9
             nb interfaces: 1
              interface[0]: id = 0
interface[0].altsetting[0]: num endpoints = 2
   Class.SubClass.Protocol: FF.FF.FF
       endpoint[0].address: 81
           max packet size: 0040
          polling interval: 00
       endpoint[1].address: 02
           max packet size: 0040
          polling interval: 00

Kernel driver attached for interface 0: (not supported)

Claiming interface 0...

Reading string descriptors:
   String (0x01): "FTDI"
   String (0x02): "FT232R USB UART"
   String (0x03): "A50285BI"

Reading OS string descriptor:libusb: warning [winusbx_submit_control_transfer] ControlTransfer failed: [31] A device attached to the system is not functioning.
 no descriptor

Reading interface association descriptors (IADs) for first configuration:
    nb IADs: 0

Releasing interface 0...
Closing device...
```

No more warning messages with PR #1018 integrated into git master.
```
PS C:\work\libusb\libusb_pr1018> .\examples\xusb.exe 0403:6001
Using libusb v1.0.26.11784

Opening device 0403:6001...

Reading device descriptor:
            length: 18
      device class: 0
               S/N: 3
           VID:PID: 0403:6001
         bcdDevice: 0600
   iMan:iProd:iSer: 1:2:3
          nb confs: 1

Reading BOS descriptor: no descriptor

Reading first configuration descriptor:
              total length: 32
         descriptor length: 9
             nb interfaces: 1
              interface[0]: id = 0
interface[0].altsetting[0]: num endpoints = 2
   Class.SubClass.Protocol: FF.FF.FF
       endpoint[0].address: 81
           max packet size: 0040
          polling interval: 00
       endpoint[1].address: 02
           max packet size: 0040
          polling interval: 00

Kernel driver attached for interface 0: (not supported)

Claiming interface 0...

Reading string descriptors:
   String (0x01): "FTDI"
   String (0x02): "FT232R USB UART"
   String (0x03): "A50285BI"

Reading OS string descriptor: no descriptor

Reading interface association descriptors (IADs) for first configuration:
    nb IADs: 0

Releasing interface 0...
Closing device...
```

mcuee, 2023-02-19:
@tormodvolden 

Running libusb-win32 1.2.6.0 debug version does not help much to see why it failed. It seems to me the FT232R USB to Serial Converter goes into a strange state and then it can no longer be used by libusb.

```
[8024] 
[8024] AllK required contiguous memory = 534200 (64bit)
[8024]   8 HotK Handles: HandleSize 2112 PoolSize 16912 (bytes)
[8024]   64 LstK Handles: HandleSize 64 PoolSize 4112 (bytes)
[8024]   1024 LstInfoK Handles: HandleSize 64 PoolSize 65552 (bytes)
[8024]   64 UsbK Handles: HandleSize 96 PoolSize 6160 (bytes)
[8024]   32 DevK Handles: HandleSize 112 PoolSize 3600 (bytes)
[8024]   4096 OvlK Handles: HandleSize 104 PoolSize 426000 (bytes)
[8024]   64 OvlPoolK Handles: HandleSize 96 PoolSize 6160 (bytes)
[8024]   32 StmK Handles: HandleSize 176 PoolSize 5648 (bytes)
[8024] 
[8024] Dynamically allocated as needed:
[8024]  KLST_DEVINFO = 2596 bytes each
```

mcuee, 2023-02-19:
First run: unplug and then plug, run pr #1018 xusb example which is a success.

```
[5328] 
[5328] AllK required contiguous memory = 534200 (64bit)
[5328]   8 HotK Handles: HandleSize 2112 PoolSize 16912 (bytes)
[5328]   64 LstK Handles: HandleSize 64 PoolSize 4112 (bytes)
[5328]   1024 LstInfoK Handles: HandleSize 64 PoolSize 65552 (bytes)
[5328]   64 UsbK Handles: HandleSize 96 PoolSize 6160 (bytes)
[5328]   32 DevK Handles: HandleSize 112 PoolSize 3600 (bytes)
[5328]   4096 OvlK Handles: HandleSize 104 PoolSize 426000 (bytes)
[5328]   64 OvlPoolK Handles: HandleSize 96 PoolSize 6160 (bytes)
[5328]   32 StmK Handles: HandleSize 176 PoolSize 5648 (bytes)
[5328] 
[5328] Dynamically allocated as needed:
[5328]  KLST_DEVINFO = 2596 bytes each
libusb0-sys:[dispatch_pnp] IRP_MN_SURPRISE_REMOVAL: is-filter=Y usb\vid_0403&pid_6001&rev_0600
libusb0-sys:[set_filter_interface_key] updated interface registry with LUsb0 direct-access symbolic link. id=-1
libusb0-sys:[dispatch_pnp] IRP_MN_REMOVE_DEVICE: is-filter=Y usb\vid_0403&pid_6001&rev_0600
libusb0-sys:[dispatch_pnp] deleting device #1 usb\vid_0403&pid_6001&rev_0600
libusb0-sys:[unload] [unloading-driver] v1.2.6.0
libusb0-sys:[DriverEntry] [loading-driver] v1.2.6.0
libusb0-sys:dbg [debug_show_devices] [AttachedDevice #0] \Driver\FTDIBUS
libusb0-sys:[add_device] device #1 created for usb\vid_0403&pid_6001&rev_0600
libusb0-sys:[reg_get_properties] using default device interface guid.
libusb0-sys:dbg [reg_get_properties] reg-key-name=\REGISTRY\MACHINE\SYSTEM\ControlSet001\Enum\USB\VID_0403&PID_6001\FTDCRPIE\Device Parameters
libusb0-sys:[set_filter_interface_key] updated interface registry with LUsb0 direct-access symbolic link. id=1
libusb0-sys:[add_device] updated interface registry with LUsb0 direct-access symbolic link. id=0001
libusb0-sys:dbg [add_device] [filter-mode] id=#1 usb\vid_0403&pid_6001&rev_0600
libusb0-sys:[add_device] complete status=00000000
libusb0-sys:[dispatch_pnp] IRP_MN_START_DEVICE: is-filter=Y usb\vid_0403&pid_6001&rev_0600
libusb0-sys:dbg [on_start_complete] is-filter=Y usb\vid_0403&pid_6001&rev_0600
libusb0-sys:dbg [dispatch_pnp] IRP_MN_QUERY_CAPABILITIES: is-filter=Y usb\vid_0403&pid_6001&rev_0600
libusb0-sys:[set_configuration] setting configuration index 0 timeout=0libusb0-sys:[get_descriptor] buffer size: 18 type: 0001 recipient: 0000 index: 0000 language id: 0000 timeout: 5000
libusb0-sys:dbg [call_usbd_ex] status = 00000000h
libusb0-sys:[get_descriptor] buffer size: 9 type: 0002 recipient: 0000 index: 0000 language id: 0000 timeout: 5000
libusb0-sys:dbg [call_usbd_ex] status = 00000000h
libusb0-sys:[get_descriptor] buffer size: 32 type: 0002 recipient: 0000 index: 0000 language id: 0000 timeout: 5000
libusb0-sys:dbg [call_usbd_ex] status = 00000000h
libusb0-sys:[set_configuration] found interface 0
libusb0-sys:dbg [set_configuration] #1 usb\vid_0403&pid_6001&rev_0600 passing configuration request to target-device.libusb0-sys:dbg [call_usbd_ex] status = 00000000h
libusb0-sys:[update_pipe_info] interface 0
libusb0-sys:[update_pipe_info] EP81h maximum-packet-size=64 maximum-transfer-size=4194304
libusb0-sys:wrn [update_pipe_info] overriding EP81h maximum-transfer-size=65536
libusb0-sys:[update_pipe_info] EP02h maximum-packet-size=64 maximum-transfer-size=4194304
libusb0-sys:wrn [update_pipe_info] overriding EP02h maximum-transfer-size=65536
libusb0-sys:[get_descriptor] buffer size: 9 type: 0002 recipient: 0000 index: 0000 language id: 0000 timeout: 0
libusb0-sys:[get_descriptor] buffer size: 32 type: 0002 recipient: 0000 index: 0000 language id: 0000 timeout: 0
libusb0-sys:[control_transfer] [read] timeout=0 request_type=80h request=06h value=0F00h index=0000h length=0005h
libusb0-sys:dbg [call_usbd_ex] status = C0000001h
libusb0-sys:err [control_transfer] request failed: status: 0xc0000001, urb-status: 0xc0000004
libusb0-sys:[get_descriptor] buffer size: 9 type: 0002 recipient: 0000 index: 0000 language id: 0000 timeout: 0
libusb0-sys:[get_descriptor] buffer size: 32 type: 0002 recipient: 0000 index: 0000 language id: 0000 timeout: 0
libusb0-sys:[control_transfer] [read] timeout=0 request_type=80h request=06h value=0300h index=0000h length=0004h
libusb0-sys:dbg [call_usbd_ex] status = 00000000h
libusb0-sys:[control_transfer] 4 bytes transmitted
libusb0-sys:[control_transfer] [read] timeout=0 request_type=80h request=06h value=0301h index=0409h length=00FFh
libusb0-sys:dbg [call_usbd_ex] status = 00000000h
libusb0-sys:[control_transfer] 10 bytes transmitted
libusb0-sys:[control_transfer] [read] timeout=0 request_type=80h request=06h value=0300h index=0000h length=0004h
libusb0-sys:dbg [call_usbd_ex] status = 00000000h
libusb0-sys:[control_transfer] 4 bytes transmitted
libusb0-sys:[control_transfer] [read] timeout=0 request_type=80h request=06h value=0302h index=0409h length=00FFh
libusb0-sys:dbg [call_usbd_ex] status = 00000000h
libusb0-sys:[control_transfer] 14 bytes transmitted
libusb0-sys:[control_transfer] [read] timeout=0 request_type=80h request=06h value=0300h index=0000h length=0004h
libusb0-sys:dbg [call_usbd_ex] status = 00000000h
libusb0-sys:[control_transfer] 4 bytes transmitted
libusb0-sys:[control_transfer] [read] timeout=0 request_type=80h request=06h value=0303h index=0409h length=00FFh
libusb0-sys:dbg [call_usbd_ex] status = 00000000h
libusb0-sys:[control_transfer] 18 bytes transmitted
libusb0-sys:[control_transfer] [read] timeout=0 request_type=80h request=06h value=03EEh index=0000h length=0012h
libusb0-sys:dbg [call_usbd_ex] status = C0000001h
libusb0-sys:err [control_transfer] request failed: status: 0xc0000001, urb-status: 0xc0000004
libusb0-sys:[release_all_interfaces] releasing all interfaces bound to file object 0x54f9cec0
```

tormodvolden, 2023-02-19:
I guess the driver is not much involved in the enumeration. And if there is no DebugView output I suppose the driver is not trying to access the USB device either, so I still think the bad state is in the driver and not in the device. Can you widen the scope of the DebugView logging? The failing winusbx_open() is from a CreateFileA() system call, so it would be interesting to know what is happening on the kernel side.

tormodvolden, 2023-02-19:
I can only suggest you try narrowing it down by modifying your test program. Try omitting the BOS/OS descriptor requests that causes the stalls, to see if that makes a difference. Try opening the device again after closing it in the same program. Try opening and closing without device transfer, then a second normal run.

tormodvolden, 2023-02-19:
> After installation of the filter driver, it works fine without switching out FTDI vendor driver. But then it will fail the next run due to the same issue as the above.

Is the error the same "[87] The parameter is incorrect" in both cases, without filter driver and on second run with filter driver? Is it possible to see in e.g. device manager whether the filter driver is still active for the device?

mcuee, 2023-02-20:
@tormodvolden 

Thanks for the suggestions. I will slowly debug the issue and report my findings here. The issue is not only with xusb test application, I have found the same issues from libftdi examples.

mcuee, 2023-02-20:
Just want to show that PR #1018 is good for the CH320 with libusb0.sys filter (tested with 1.2.7.3 version).

git HEAD.
```
C:\work\libusb\libusb\mingw64\bin [master ≡ +2 ~0 -0 !]> .\xusb.exe 1a86:7523
Using libusb v1.0.26.11748

Opening device 1A86:7523...

Reading device descriptor:
            length: 18
      device class: 255
               S/N: 0
           VID:PID: 1A86:7523
         bcdDevice: 0264
   iMan:iProd:iSer: 0:2:0
          nb confs: 1

Reading BOS descriptor: libusb: warning [winusbx_submit_control_transfer] ControlTransfer failed: [31] A device attached to the system is not functioning.
libusb: error [libusb_get_bos_descriptor] failed to read BOS (-1)
no descriptor

Reading first configuration descriptor:
              total length: 39
         descriptor length: 9
             nb interfaces: 1
              interface[0]: id = 0
interface[0].altsetting[0]: num endpoints = 3
   Class.SubClass.Protocol: FF.01.02
       endpoint[0].address: 82
           max packet size: 0020
          polling interval: 00
       endpoint[1].address: 02
           max packet size: 0020
          polling interval: 00
       endpoint[2].address: 81
           max packet size: 0008
          polling interval: 01

Kernel driver attached for interface 0: -12

Claiming interface 0...

Reading string descriptors:
   String (0x02): "USB Serial"

Reading OS string descriptor:libusb: warning [winusbx_submit_control_transfer] ControlTransfer failed: [31] A device attached to the system is not functioning.
 no descriptor

Releasing interface 0...
Closing device...
```

git HEAD with PR #1018.
```
C:\work\libusb\libusb_pr1018 [master ↑3]> .\examples\xusb.exe 1a86:7523
Using libusb v1.0.26.11784

Opening device 1A86:7523...

Reading device descriptor:
            length: 18
      device class: 255
               S/N: 0
           VID:PID: 1A86:7523
         bcdDevice: 0264
   iMan:iProd:iSer: 0:2:0
          nb confs: 1

Reading BOS descriptor: no descriptor

Reading first configuration descriptor:
              total length: 39
         descriptor length: 9
             nb interfaces: 1
              interface[0]: id = 0
interface[0].altsetting[0]: num endpoints = 3
   Class.SubClass.Protocol: FF.01.02
       endpoint[0].address: 82
           max packet size: 0020
          polling interval: 00
       endpoint[1].address: 02
           max packet size: 0020
          polling interval: 00
       endpoint[2].address: 81
           max packet size: 0008
          polling interval: 01

Kernel driver attached for interface 0: (not supported)

Claiming interface 0...

Reading string descriptors:
   String (0x02): "USB Serial"

Reading OS string descriptor: no descriptor

Reading interface association descriptors (IADs) for first configuration:
    nb IADs: 0

Releasing interface 0...
Closing device...

```

mcuee, 2023-02-20:
>  Is it possible to see in e.g. device manager whether the filter driver is still active for the device?

Yes I can see libusb0.sys is still attached to the device.

And libusb-win32 test program still works. That test program only works with device with either libusb0.sys or libusbk.sys driver. This futher proves that libusb0.sys filter driver is still attaching to the device. It is just that the FTDI device is in a strange state that libusb Windows enumeration process will fail.

```
PS C:\work\libusb\libusb_pr1018> .\examples\xusb.exe 0403:6001
Using libusb v1.0.26.11784

Opening device 0403:6001...
libusb: error [winusbx_open] could not open device \\?\USB#VID_0403&PID_6001#FTDCRPIE#{A5DCBF10-6530-11D2-901F-00C04FB951ED} (interface 0): [87] The parameter is incorrect.
  Failed.

PS C:\work\libusb\libusb_pr1018> C:\work\libusb\libusb-win32\libusb-win32-bin-1.2.7.3\bin\amd64\testlibusb.exe

Dev #0: 0000 - 0000

  Dev #1: FTDI - US232R
```

mcuee, 2023-02-21:
Just FYI, libusb0.sys filter also works well with libusbK for Cypress FX3 USB SuperSpeed device.

```
PS C:\libusbK-dev-kit> .\kList.exe

Loading USB ID's maintained by Stephen J. Gowdy <linux.usb.ids@gmail.com>..

 1. Cypress FX3 USB BulkloopExample Device (Cypress) [Connected]
    Service              : CYUSB3
    ClassGUID            : {36FC9E60-C465-11CF-8056-444553540000}
    DeviceID             : USB\VID_04B4&PID_00F0\5&586B51A&0&13
    DeviceInterfaceGUID  : {F9F3FF14-AE21-48A0-8A25-8011A7A931D9}
    SymbolicLink         : \\?\usb#vid_04b4&pid_00f0#5&586b51a&0&13#{f9f3ff14-ae21-48a0-8a25-8011a7a931d9}
    DevicePath           : \\.\libusb0-0001
    SerialNumber         : 5&586B51A&0&13
    BusNumber            : 0
    DeviceAddress        : 13

Select device (1-1) :1

Loading driver api..
Getting descriptors..

-Device:
  bLength             :18
  bDescriptorType     :0x01
  bcdUSB              :0x0310
  bDeviceClass        :0x00 (Defined at Interface level)
  bDeviceSubClass     :0x00
  bDeviceProtocol     :0x00
  bMaxPacketSize0     :9
  idVendor            :0x04B4 (Cypress Semiconductor Corp.)
  idProduct           :0x00F0
  bcdDevice           :0x0000
  iManufacturer       :1 (Cypress)
  iProduct            :2 (FX3)
  iSerialNumber       :0
  bNumConfigurations  :1
!End Device
-Configuration:
  bLength             :9
  bDescriptorType     :0x02
  wTotalLength        :44
  bNumInterfaces      :1
  bConfigurationValue :0x01
  iConfiguration      :0
  bmAttributes        :0x80
  MaxPower            :50 (100ma)
 -Interface:
    bLength             :9
    bDescriptorType     :0x04
    bInterfaceNumber    :0x00
    bAlternateSetting   :0x00
    bNumEndpoints       :2
    bInterfaceClass     :0xFF (Vendor Specific Class)
    bInterfaceSubClass  :0x00
    bInterfaceProtocol  :0x00
    iInterface          :0
   -Endpoint:
      bLength             :7
      bDescriptorType     :0x05
      bEndpointAddress    :0x01
      bmAttributes        :0x02 (Bulk)
      wMaxPacketSize      :1024
      bInterval           :0x00
     -SS Companion:
        bLength             :6
        bDescriptorType     :0x30
        bMaxBurst           :0 (1 Packets Per Burst)
        MaxStreams          :0
[... 118 lines cut ...]
        Num Endpoints   : 2
        Priority        : 0
        Read Size       : 4096
        Write Size      : 4096
        Buffer Count    : 1
        Display Refresh : 1000 (ms)
        Transfer Timeout: 5000 (ms)
        Retry Count     : 0
        Verify Data     : Off

Bulk Read (Ep81h) Maximum Packet Size:1024
        Total Bytes     : 1886982144
        Total Transfers : 460689
        Avg. Bytes/sec  : 53627252.79
        Elapsed Time    : 35.19 seconds

Bulk Write (Ep01h) Maximum Packet Size:1024
        Total Bytes     : 1886982144
        Total Transfers : 460689
        Avg. Bytes/sec  : 53627252.79
        Elapsed Time    : 35.19 seconds

Press any key to exit..

```

mcuee, 2023-02-22:
So far the only issue I found is the FTDI device. Maybe the filter driver is not working well with the FTDI vendor driver. The strange thing is that only libusb Windows enumeration gets affected. libusb-win32 based applications are not affected,

mcuee, 2023-02-22:
> I guess the driver is not much involved in the enumeration. And if there is no DebugView output I suppose the driver is not trying to access the USB device either, so I still think the bad state is [...]

@tormodvolden 

This seems to be a plausible hypothesis. Unfortunately I am having issues with DebugView since it crashes.

mcuee, 2023-02-22:
Another interesting findings -- using Xiaomi USB/Bluetooth dual mode mouse USB receiver as the test device.

The output without the filter driver attached is normal.
```
C:\work\libusb\libusb_pr1018 [libusb0ctrlerror ≡]> .\examples\xusb 2717:5013
Using libusb v1.0.26.11784

Opening device 2717:5013...
libusb: warning [hid_open] could not open HID device in R/W mode (keyboard or mouse?) - trying without

Reading device descriptor:
            length: 18
      device class: 0
               S/N: 0
           VID:PID: 2717:5013
         bcdDevice: 0625
   iMan:iProd:iSer: 1:2:0
          nb confs: 1

Reading BOS descriptor: libusb: warning [_hid_get_descriptor] unsupported
libusb: error [libusb_get_bos_descriptor] failed to read BOS (-12)
no descriptor

Reading first configuration descriptor:
              total length: 34
         descriptor length: 9
             nb interfaces: 1
              interface[0]: id = 0
interface[0].altsetting[0]: num endpoints = 1
   Class.SubClass.Protocol: 03.01.02
       endpoint[0].address: 81
           max packet size: 0008
          polling interval: 02

Kernel driver attached for interface 0: (not supported)

Claiming interface 0...

Reading string descriptors:
   String (0x01): "MI"
   String (0x02): "Mi Wireless Mouse"

Reading OS string descriptor: no descriptor

Reading interface association descriptors (IADs) for first configuration:
    nb IADs: 0

Releasing interface 0...
Closing device...

```

With the filter driver, xusb failed to find the device.

```
C:\work\libusb\libusb_pr1018 [libusb0ctrlerror ≡]> .\examples\xusb 2717:5013
Using libusb v1.0.26.11784

Opening device 2717:5013...
libusb: error [winusbx_open] could not open device \\?\USB#VID_2717&PID_5013#5&E9F3E45&0&1#{A5DCBF10-6530-11D2-901F-00C04FB951ED} (interface 0): [31] A device attached to the system is not functioning.
  Failed.

```

mcuee, 2023-02-22:
Interestingly if I run libusbK kList utility, it works fine, but after that, the mouse will stop working.

```
C:\libusbK-dev-kit> .\kList.exe

Loading USB ID's maintained by Stephen J. Gowdy <linux.usb.ids@gmail.com>..

 1. WinUsb Device (WinUsb Device) [Connected]
    Service              : WINUSB
    ClassGUID            : {88BAE032-5A81-49F0-BC3D-A4FF138216D6}
    DeviceID             : USB\VID_0424&PID_2740\D&127B8A6A&0&5
    DeviceInterfaceGUID  : {b35924d6-3e16-4a9e-9782-5524a4b79bac}
    SymbolicLink         : \\?\usb#vid_0424&pid_2740#d&127b8a6a&0&5#{b35924d6-3e16-4a9e-9782-5524a4b79bac}
    DevicePath           : \\?\usb#vid_0424&pid_2740#d&127b8a6a&0&5#{b35924d6-3e16-4a9e-9782-5524a4b79bac}
    SerialNumber         : D&127B8A6A&0&5
    BusNumber            : 0
    DeviceAddress        : 5

 2. USB Input Device ((Standard system devices)) [Connected]
    Service              : HidUsb
    ClassGUID            : {745A17A0-74D3-11D0-B6FE-00A0C90F57DA}
    DeviceID             : USB\VID_2717&PID_5013\5&E9F3E45&0&1
    DeviceInterfaceGUID  : {F9F3FF14-AE21-48A0-8A25-8011A7A931D9}
    SymbolicLink         : \\?\usb#vid_2717&pid_5013#5&e9f3e45&0&1#{f9f3ff14-ae21-48a0-8a25-8011a7a931d9}
    DevicePath           : \\.\libusb0-0001
    SerialNumber         : 5&E9F3E45&0&1
    BusNumber            : 0
    DeviceAddress        : 1

Select device (1-2) :2

Loading driver api..
Getting descriptors..

-Device:
  bLength             :18
  bDescriptorType     :0x01
  bcdUSB              :0x0110
  bDeviceClass        :0x00 (Defined at Interface level)
  bDeviceSubClass     :0x00
  bDeviceProtocol     :0x00
  bMaxPacketSize0     :64
  idVendor            :0x2717 (Xiaomi Inc.)
  idProduct           :0x5013
  bcdDevice           :0x0625
  iManufacturer       :1 (MI)
  iProduct            :2 (Mi Wireless Mouse)
  iSerialNumber       :0
  bNumConfigurations  :1
!End Device
-Configuration:
  bLength             :9
  bDescriptorType     :0x02
  wTotalLength        :34
  bNumInterfaces      :1
  bConfigurationValue :0x01
  iConfiguration      :0
  bmAttributes        :0xA0 (Self-Powered)
  MaxPower            :50 (100ma)
 -Interface:
    bLength             :9
    bDescriptorType     :0x04
    bInterfaceNumber    :0x00
    bAlternateSetting   :0x00
    bNumEndpoints       :1
    bInterfaceClass     :0x03 (Human Interface Device)
    bInterfaceSubClass  :0x01 (Boot Interface Subclass)
    bInterfaceProtocol  :0x02 (Mouse)
    iInterface          :0
   -Hid:
[... 102 lines cut ...]
       !End Collection
        Usage Page          :0xFF02 (Simulation Controls)
        Usage               :0x002 (Automobile Simulation Device)
       *Collection            :0x01 (Application)
          Report ID           :0x06
          Usage               :0x002 (Automobile Simulation Device)
          Logical Minimum     :0x00
          Logical Maximum     :0x00FF
          Report Size         :0x08
          Report Count        :0x07
          Feature             :0x02
       !End Collection
     !End Report
   !End Hid
   -Endpoint:
      bLength             :7
      bDescriptorType     :0x05
      bEndpointAddress    :0x81
      bmAttributes        :0x03 (Interrupt)
      wMaxPacketSize      :8
      bInterval           :0x02
   !End Interface
 !End Configuration

```

tormodvolden, 2023-02-22:
When you test "libusb-win32 based applications" do you run them with the DLL built from https://github.com/mcuee/libusb-win32.git ?

I see that libusb winusbx_claim_interface() is doing quite some special-casing for filter driver interfaces. The libusb-win32 code is doing nothing of that. This is not directly related to the error we see because there it fails already when opening the device, but I am just curious on how this is working. Can you please apply this debug patch for further tests?
[debug-windows_open-path.patch.txt](https://github.com/libusb/libusb/files/10806488/debug-windows_open-path.patch.txt)

mcuee, 2023-02-23:
> When you test "libusb-win32 based applications" do you run them with the DLL built from https://github.com/mcuee/libusb-win32.git ?

Yes for the test using libusb-win32 `testlibusb` example. Take note `kList` is from libusbK and it does not go through `libusb0.dll`. libusbK library will directly talk to `libusb0.sys` through IOCTLs. In any case, I am using the official 1.2.7.3 release of libusb0.dll and libusb0.sys.

mcuee, 2023-02-23:
> I see that libusb winusbx_claim_interface() is doing quite some special-casing for filter driver interfaces. The libusb-win32 code is doing nothing of that. This is not directly related to the error [...]

It seems the special handling of libusb0.sys filter driver is not used. Probably as it mentioned, libusbK.dll now already hide the differences away.

Here are the debug logs for the first successful run of xusb and the second failed xusb run.

<details>
<summary> click for debug logs, first run is OK and second run is not OK </summary>

```
PS C:\work\libusb\libusb_pr1018> .\build\v143\x64\Release\xusb 0403:6001
Using libusb v1.0.26.11784

[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.000236] [000011d4] libusb: debug [libusb_init_context] created default context
[ 0.000349] [000011d4] libusb: debug [libusb_init_context] libusb v1.0.26.11784
[ 0.000514] [000011d4] libusb: debug [usbi_add_event_source] add HANDLE 00000000000000B4 events 0
[ 0.000690] [000011d4] libusb: debug [usbi_io_init] using timer for timeouts
[ 0.000816] [000011d4] libusb: debug [usbi_add_event_source] add HANDLE 00000000000000B8 events 0
[ 0.000933] [000011d4] libusb: debug [get_windows_version] Windows 8 (or later) 64-bit
[ 0.001047] [000011d4] libusb: debug [htab_create] using 1021 entries hash table
[ 0.003103] [000011d4] libusb: info [winusbx_init] WinUSB DLL available (with isoch support)
[ 0.004995] [000011d4] libusb: debug [winusbx_init] libusbK DLL found, version: 3.1.0.0
[ 0.007102] [000011d4] libusb: info [windows_init] UsbDk backend is not available
Opening device 0403:6001...
[ 0.007486] [000011d4] libusb: debug [libusb_get_device_list]
[ 0.007778] [000030bc] libusb: debug [windows_iocp_thread] I/O completion thread started
[ 0.015333] [000011d4] libusb: debug [winusb_get_device_list] allocating new device for session [5]
[ 0.015509] [000011d4] libusb: debug [winusb_get_device_list] allocating new device for session [6]
[ 0.015923] [000011d4] libusb: debug [get_api_type] driver(s): WUDFRd
[ 0.016064] [000011d4] libusb: debug [get_api_type] lower filter driver(s): WinUsb
[ 0.016101] [000011d4] libusb: debug [get_api_type] matched lower filter driver name against WinUSB
[ 0.016186] [000011d4] libusb: debug [winusb_get_device_list] allocating new device for session [1D]
[ 0.016365] [000011d4] libusb: debug [get_api_type] driver(s): BTHUSB
[ 0.016423] [000011d4] libusb: debug [get_api_type] lower filter driver(s): ibtusb
[ 0.016513] [000011d4] libusb: debug [winusb_get_device_list] allocating new device for session [3C]
[ 0.016726] [000011d4] libusb: debug [get_api_type] driver(s): usbccgp
[ 0.016776] [000011d4] libusb: debug [get_api_type] matched driver name against Composite API
[ 0.016858] [000011d4] libusb: debug [winusb_get_device_list] allocating new device for session [6C]
[ 0.017068] [000011d4] libusb: debug [get_api_type] driver(s): FTDIBUS
[ 0.017136] [000011d4] libusb: debug [get_api_type] upper filter driver(s): libusb0
[ 0.017186] [000011d4] libusb: debug [get_api_type] matched upper filter driver name against libusb0
[ 0.017284] [000011d4] libusb: debug [winusb_get_device_list] allocating new device for session [6D]
[ 0.017471] [000011d4] libusb: debug [get_api_type] driver(s): usbccgp
[ 0.017593] [000011d4] libusb: debug [get_api_type] matched driver name against Composite API
[ 0.017687] [000011d4] libusb: debug [winusb_get_device_list] allocating new device for session [6E]
[ 0.017896] [000011d4] libusb: debug [enumerate_hcd_root_hub] assigning HCD 'PCI\VEN_8086&DEV_A0ED&SUBSYS_15451025&REV_20\3&11583659&0&A0' bus number 1
[ 0.018029] [000011d4] libusb: debug [enumerate_hcd_root_hub] assigning HCD 'PCI\VEN_8086&DEV_9A13&SUBSYS_72708086&REV_01\3&11583659&0&68' bus number 2
[ 0.018798] [000011d4] libusb: debug [winusb_get_device_list] found existing device for session [3C]
[ 0.019025] [000011d4] libusb: debug [init_device] found 1 configurations (current config: 1)
[ 0.019129] [000011d4] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 200 bytes)
[ 0.019248] [000011d4] libusb: debug [init_device] (bus: 1, addr: 3, depth: 1, port: 10): 'USB\VID_8087&PID_0026\5&586B51A&0&10'
[ 0.019427] [000011d4] libusb: debug [winusb_get_device_list] extra GUID: {EBB53BA2-7560-401B-85E2-FEBA3F503F87}
[ 0.019533] [000011d4] libusb: debug [winusb_get_device_list] found existing device for session [6D]
[ 0.019685] [000011d4] libusb: debug [init_device] found 1 configurations (current config: 1)
[ 0.019787] [000011d4] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 32 bytes)
[ 0.019913] [000011d4] libusb: debug [init_device] (bus: 1, addr: 11, depth: 1, port: 1): 'USB\VID_0403&PID_6001\FTDCRPIE'
[ 0.020089] [000011d4] libusb: debug [winusb_get_device_list] extra GUID: {16126A05-3179-4589-9DB8-952AFFD297D5}
[ 0.020197] [000011d4] libusb: debug [winusb_get_device_list] found existing device for session [6C]
[ 0.020275] [000011d4] libusb: debug [init_device] found 1 configurations (current config: 1)
[ 0.020370] [000011d4] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 512 bytes)
[ 0.020411] [000011d4] libusb: debug [init_device] (bus: 1, addr: 2, depth: 1, port: 5): 'USB\VID_04F2&PID_B6DD\0001'
[... cut ...]
[ 0.028115] [00002994] libusb: debug [libusb_unref_device] destroy device 1.0
[ 0.028150] [00002994] libusb: debug [libusb_unref_device] destroy device 2.0
  Failed.
[ 0.028210] [00002994] libusb: debug [libusb_exit] destroying default context
[ 0.028251] [00003798] libusb: debug [windows_iocp_thread] I/O completion thread exiting
[ 0.028661] [00002994] libusb: debug [usbi_remove_event_source] remove HANDLE 00000000000000B4
[ 0.028705] [00002994] libusb: debug [usbi_remove_event_source] remove HANDLE 0000000000000054
```

</details>

mcuee, 2023-02-23:
> I can only suggest you try narrowing it down by modifying your test program. Try omitting the BOS/OS descriptor requests that causes the stalls, to see if that makes a difference. 

That does not help. I have removed BOS/OS/IAD descriptor but the issue is still there.

```
PS C:\work\libusb\libusb_pr1018> $Env:LIBUSB_DEBUG=0
PS C:\work\libusb\libusb_pr1018> .\build\v143\x64\Release\xusb 0403:6001
Using libusb v1.0.26.11784

Opening device 0403:6001...

Reading device descriptor:
            length: 18
      device class: 0
               S/N: 3
           VID:PID: 0403:6001
         bcdDevice: 0600
   iMan:iProd:iSer: 1:2:3
          nb confs: 1

Reading first configuration descriptor:
              total length: 32
         descriptor length: 9
             nb interfaces: 1
              interface[0]: id = 0
interface[0].altsetting[0]: num endpoints = 2
   Class.SubClass.Protocol: FF.FF.FF
       endpoint[0].address: 81
           max packet size: 0040
          polling interval: 00
       endpoint[1].address: 02
           max packet size: 0040
          polling interval: 00

Kernel driver attached for interface 0: (not supported)

Claiming interface 0...

Reading string descriptors:
   String (0x01): "FTDI"
   String (0x02): "US232R"
   String (0x03): "FTDCRPIE"

Releasing interface 0...
Closing device...
PS C:\work\libusb\libusb_pr1018> .\build\v143\x64\Release\xusb 0403:6001
Using libusb v1.0.26.11784

Opening device 0403:6001...
  Failed.
```

mcuee, 2023-02-23:
@tormodvolden 

I find out that using `testlibusb` example already can trigger the problem, no need to use xusb.

I was thinking `testlibusb` example worked alreays but that is not correct.

```
PS C:\work\libusb\libusb_pr1018> .\build\v143\x64\Release\testlibusb.exe
Dev (bus 1, device 3): 8087 - 0026 speed: 12M
Dev (bus 1, device 17): 0403 - 6001 speed: 12M
  Manufacturer:              FTDI
  Product:                   US232R
Dev (bus 1, device 2): 04F2 - B6DD speed: 480M
Dev (bus 1, device 4): 1C7A - 0575 speed: 480M
Dev (bus 1, device 0): 8086 - A0ED speed: 5G
Dev (bus 1, device 5): 046D - C52B speed: 12M
  Manufacturer:              Logitech
  Product:                   USB Receiver
Dev (bus 2, device 0): 8086 - 9A13 speed: 5G

PS C:\work\libusb\libusb_pr1018> .\build\v143\x64\Release\testlibusb.exe
Dev (bus 1, device 3): 8087 - 0026 speed: 12M
Dev (bus 1, device 17): 0403 - 6001 speed: 12M
Dev (bus 1, device 2): 04F2 - B6DD speed: 480M
Dev (bus 1, device 4): 1C7A - 0575 speed: 480M
Dev (bus 1, device 0): 8086 - A0ED speed: 5G
Dev (bus 1, device 5): 046D - C52B speed: 12M
  Manufacturer:              Logitech
  Product:                   USB Receiver
Dev (bus 2, device 0): 8086 - 9A13 speed: 5G

```

Interestingly sometimes first run will fail already.
```
[ 0.023857] [000018e8] libusb: debug [libusb_get_device_descriptor]
Dev (bus 1, device 19): 0403 - 6001 speed: 12M
[ 0.023909] [000018e8] libusb: debug [libusb_open] open 1.19
[ 0.023936] [000018e8] libusb: debug [winusbx_open] opening device path \\?\USB#VID_0403&PID_6001#FTDCRPIE#{A5DCBF10-6530-11D2-901F-00C04FB951ED} (interface 0)
[ 0.024242] [000018e8] libusb: error [winusbx_open] could not open device \\?\USB#VID_0403&PID_6001#FTDCRPIE#{A5DCBF10-6530-11D2-901F-00C04FB951ED} (interface 0): [5] Access is denied.
[ 0.024284] [000018e8] libusb: debug [libusb_open] open 1.19 returns -3
```

Then second run will be okay.
```
[ 0.026706] [00003790] libusb: debug [libusb_get_device_descriptor]
Dev (bus 1, device 19): 0403 - 6001 speed: 12M
[ 0.026760] [00003790] libusb: debug [libusb_open] open 1.19
[ 0.026787] [00003790] libusb: debug [winusbx_open] opening device path \\?\USB#VID_0403&PID_6001#FTDCRPIE#{A5DCBF10-6530-11D2-901F-00C04FB951ED} (interface 0)
[ 0.038997] [00003790] libusb: debug [libusb_submit_transfer] transfer 00000212EFB938D0
[ 0.039076] [00003790] libusb: debug [add_to_flying_list] arm timer for timeout in 1000ms (first in line)
[ 0.039126] [00003790] libusb: debug [libusb_claim_interface] interface 0
[ 0.039360] [00003790] libusb: debug [get_interface_details_filter] assigned libusb0 symbolic link \\.\libusb0-0001
[ 0.039408] [00003790] libusb: debug [winusbx_claim_interface] opening filter driver path \\.\libusb0-0001 (interface 0)
[ 0.041489] [00003790] libusb: debug [winusbx_claim_interface] claimed interface 0
[ 0.041547] [00003790] libusb: debug [windows_assign_endpoints] (re)assigned endpoint 81 to interface 0
[ 0.041597] [00003790] libusb: debug [windows_assign_endpoints] (re)assigned endpoint 02 to interface 0
[ 0.041655] [00003790] libusb: debug [auto_claim] auto-claimed interface 0 for control request
[ 0.041701] [00003790] libusb: debug [winusbx_submit_control_transfer] will use interface 0
[ 0.041659] [00002450] libusb: debug [windows_iocp_thread] ignoring overlapped 000000C3B393F600 for handle 00000212EFB98FE0
[ 0.041791] [00002450] libusb: debug [windows_iocp_thread] ignoring overlapped 000000C3B393F600 for handle 00000212EFB98FE0
[ 0.041832] [00002450] libusb: debug [windows_iocp_thread] ignoring overlapped 000000C3B393F600 for handle 00000212EFB98FE0
[ 0.041871] [00002450] libusb: debug [windows_iocp_thread] ignoring overlapped 000000C3B393F600 for handle 00000212EFB98FE0
[ 0.041915] [00002450] libusb: debug [windows_iocp_thread] ignoring overlapped 000000C3B393F600 for handle 00000212EFB98FE0
[ 0.041962] [00002450] libusb: debug [windows_iocp_thread] ignoring overlapped 000000C3B393F6F0 for handle 00000212EFB98FE0
[ 0.042007] [00002450] libusb: debug [windows_iocp_thread] ignoring overlapped 000000C3B393F830 for handle 00000212EFB98FE0
[ 0.041876] [00003790] libusb: debug [windows_force_sync_completion] transfer 00000212EFB938D0, length 4
[ 0.042102] [00003790] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[ 0.042145] [00003790] libusb: debug [handle_events] event sources modified, reallocating event data
[ 0.042106] [00002450] libusb: debug [windows_iocp_thread] transfer 00000212EFB938D0 completed, length 4
[... 82 lines cut ...]
[ 0.048226] [00002450] libusb: debug [windows_iocp_thread] transfer 00000212EFB938D0 completed, length 14
[ 0.048279] [00003790] libusb: debug [usbi_wait_for_events] WaitForMultipleObjects() returned 0
[ 0.048328] [00003790] libusb: debug [handle_event_trigger] event triggered
[ 0.048376] [00003790] libusb: debug [windows_handle_transfer_completion] handling transfer 00000212EFB938D0 completion with errcode 0, length 14
[ 0.048442] [00003790] libusb: debug [libusb_release_interface] interface 0
[ 0.048519] [00003790] libusb: debug [auto_release] auto-released interface 0
[ 0.048573] [00003790] libusb: debug [arm_timer_for_next_timeout] no timeouts, disarming timer
[ 0.048620] [00003790] libusb: debug [usbi_handle_transfer_completion] transfer 00000212EFB938D0 has callback 00007FF68C7F64E0
[ 0.048677] [00003790] libusb: debug [sync_transfer_cb] actual_length=14
[ 0.048721] [00003790] libusb: debug [libusb_free_transfer] transfer 00000212EFB938D0
[... cut ...]
Dev (bus 1, device 19): 0403 - 6001 speed: 12M
[ 0.026109] [00003b7c] libusb: debug [libusb_open] open 1.19
[ 0.026138] [00003b7c] libusb: debug [winusbx_open] opening device path \\?\USB#VID_0403&PID_6001#FTDCRPIE#{A5DCBF10-6530-11D2-901F-00C04FB951ED} (interface 0)
[ 0.026773] [00003b7c] libusb: error [winusbx_open] could not open device \\?\USB#VID_0403&PID_6001#FTDCRPIE#{A5DCBF10-6530-11D2-901F-00C04FB951ED} (interface 0): [87] The parameter is incorrect.
[ 0.026869] [00003b7c] libusb: debug [libusb_open] open 1.19 returns -1

```

mcuee, 2023-02-24:
> Try opening and closing without device transfer, then a second normal run.

This seems to be okay.

```
C:\work\libusb\libusb_pr1018 [libusb0ctrlerror ≡]> $Env:LIBUSB_DEBUG=3

C:\work\libusb\libusb_pr1018 [libusb0ctrlerror ≡ +1 ~1 -0 !]> git diff
diff --git a/examples/testlibusb.c b/examples/testlibusb.c
index 394cec52..2758712a 100644
--- a/examples/testlibusb.c
+++ b/examples/testlibusb.c
@@ -163,7 +163,7 @@ static void print_configuration(struct libusb_config_descriptor *config)
 static void print_device(libusb_device *dev, libusb_device_handle *handle)
 {
        struct libusb_device_descriptor desc;
-       unsigned char string[256];
+//     unsigned char string[256];
        const char *speed;
        int ret;
        uint8_t i;
@@ -191,7 +191,8 @@ static void print_device(libusb_device *dev, libusb_device_handle *handle)
                libusb_open(dev, &handle);

        if (handle) {
-               if (desc.iManufacturer) {
+               printf("openning device and closing device.\n");
+/*             if (desc.iManufacturer) {
                        ret = libusb_get_string_descriptor_ascii(handle, desc.iManufacturer, string, sizeof(string));
                        if (ret > 0)
                                printf("  Manufacturer:              %s\n", (char *)string);
@@ -208,7 +209,7 @@ static void print_device(libusb_device *dev, libusb_device_handle *handle)
                        if (ret > 0)
                                printf("  Serial Number:             %s\n", (char *)string);
                }
-       }
+*/     }

        if (verbose) {
                for (i = 0; i < desc.bNumConfigurations; i++) {

C:\work\libusb\libusb_pr1018 [libusb0ctrlerror ≡ +1 ~1 -0 !]> .\examples\testlibusb.exe
libusb: info [winusbx_init] WinUSB DLL available (with isoch support)
libusb: error [load_usbdk_helper_dll] Failed to load UsbDkHelper.dll: [126] The specified module could not be found.
libusb: info [windows_init] UsbDk backend is not available
Dev (bus 3, device 5): 0424 - 5744 speed: 5G
Dev (bus 3, device 6): 0BDA - 5413 speed: 480M
Dev (bus 3, device 3): 0BDA - 0413 speed: 5G
Dev (bus 3, device 14): 0424 - 2740 speed: 480M
openning device and closing device.
Dev (bus 3, device 7): 413C - B06E speed: 480M
openning device and closing device.
Dev (bus 1, device 5): 8087 - 0AAA speed: 12M
Dev (bus 3, device 8): 0424 - 2744 speed: 480M
Dev (bus 3, device 2): 0BDA - 5487 speed: 480M
Dev (bus 1, device 8): 0403 - 6001 speed: 12M
openning device and closing device.
Dev (bus 3, device 1): 0BDA - 0487 speed: 10G
Dev (bus 1, device 2): 047F - C056 speed: 12M
openning device and closing device.
Dev (bus 1, device 1): 2717 - 5013 speed: 12M
libusb: warning [hid_open] could not open HID device in R/W mode (keyboard or mouse?) - trying without
openning device and closing device.
Dev (bus 3, device 9): 0BDA - 402E speed: 480M
Dev (bus 1, device 0): 8086 - A36D speed: 5G
Dev (bus 3, device 12): 413C - 2107 speed: 1.5M
libusb: warning [hid_open] could not open HID device in R/W mode (keyboard or mouse?) - trying without
openning device and closing device.
Dev (bus 1, device 3): 0A5C - 5842 speed: 480M
Dev (bus 3, device 0): 8086 - 15F0 speed: 10G
Dev (bus 3, device 11): 1679 - 2001 speed: 480M
openning device and closing device.
Dev (bus 2, device 0): 8086 - 15DB speed: 5G
Dev (bus 3, device 4): 0BDA - 8153 speed: 5G
Dev (bus 3, device 10): 413C - B06F speed: 480M
openning device and closing device.
Dev (bus 3, device 13): 046D - C077 speed: 1.5M
libusb: warning [hid_open] could not open HID device in R/W mode (keyboard or mouse?) - trying without
openning device and closing device.
Dev (bus 1, device 4): 0BDA - 58FD speed: 480M

C:\work\libusb\libusb_pr1018 [libusb0ctrlerror ≡ +1 ~1 -0 !]> .\examples\xusb.exe 0403:6001
Using libusb v1.0.26.11784

libusb: info [winusbx_init] WinUSB DLL available (with isoch support)
libusb: error [load_usbdk_helper_dll] Failed to load UsbDkHelper.dll: [126] The specified module could not be found.
libusb: info [windows_init] UsbDk backend is not available
Opening device 0403:6001...

Reading device descriptor:
            length: 18
      device class: 0
               S/N: 3
           VID:PID: 0403:6001
         bcdDevice: 0600
   iMan:iProd:iSer: 1:2:3
          nb confs: 1

Reading BOS descriptor: no descriptor

Reading first configuration descriptor:
              total length: 32
         descriptor length: 9
             nb interfaces: 1
              interface[0]: id = 0
interface[0].altsetting[0]: num endpoints = 2
   Class.SubClass.Protocol: FF.FF.FF
       endpoint[0].address: 81
           max packet size: 0040
          polling interval: 00
       endpoint[1].address: 02
           max packet size: 0040
          polling interval: 00

Kernel driver attached for interface 0: (not supported)

Claiming interface 0...

Reading string descriptors:
   String (0x01): "FTDI"
   String (0x02): "US232R"
   String (0x03): "FTDCRPIE"

Reading OS string descriptor: no descriptor

Reading interface association descriptors (IADs) for first configuration:
    nb IADs: 0

Releasing interface 0...
Closing device...
```

mcuee, 2023-02-24:
Just some quick summary.

1) libusb-win32 device filter is not guaranteed to work with all the USB devices out there. It functions as a upper filter for the USB device and work with the underlying vendor device driver.

2) As of now, in terms of driver sequence, libusb-win32 filter driver seems to be higher than the USB HID driver.

3) Issue #1 with FTDI device, it could be due to the way libusb Windows special codes to handle the filter driver, it could also due to some aspects of the libusb-win32 filter + FTDI vendor driver interaction.

4) Issue #2 with Xiaomi dual mode Wireless Mouse USB receiver -- since libusbK based kList also cause the device to hang, we can say libusb-win32 device filter driver does not work with the device.

mcuee, 2023-02-24:
Another thing, this is not a regression. I checked libusb-1.0.13 release (first version to integrate support of libusb0.sys and libusbK.sys through libusbK.dll) and it has the issue too.

```
PS C:\work\libusb\libusb_release\libusb-1.0.13> .\examples\xusb.exe 0403:6001
Using libusbx v1.0.13.10575

Opening device 0403:6001...
libusbx: error [init_device] device '\\.\USB#VID_046D&PID_C52B&MI_01#6&226CB6DF&0&0001' is no longer connected!

Reading device descriptor:
            length: 18
      device class: 0
               S/N: 3
           VID:PID: 0403:6001
         bcdDevice: 0600
   iMan:iProd:iSer: 1:2:3
          nb confs: 1

Reading configuration descriptors:
             nb interfaces: 1
              interface[0]: id = 0
interface[0].altsetting[0]: num endpoints = 2
   Class.SubClass.Protocol: FF.FF.FF
       endpoint[0].address: 81
           max packet size: 0040
          polling interval: 00
       endpoint[1].address: 02
           max packet size: 0040
          polling interval: 00

Claiming interface 0...

Reading string descriptors:
   String (0x01): "FTDI"
   String (0x02): "US232R"
   String (0x03): "FTDCRPIE"
libusbx: warning [winusbx_submit_control_transfer] ControlTransfer failed: [31] A device attached to the system is not functioning.

Releasing interface 0...
Closing device...
PS C:\work\libusb\libusb_release\libusb-1.0.13> .\examples\xusb.exe 0403:6001
Using libusbx v1.0.13.10575

Opening device 0403:6001...
libusbx: error [init_device] device '\\.\USB#VID_046D&PID_C52B&MI_01#6&226CB6DF&0&0001' is no longer connected!
libusbx: error [winusbx_open] could not open device \\.\USB#VID_0403&PID_6001#FTDCRPIE#{A5DCBF10-6530-11D2-901F-00C04FB951ED} (interface 0): [87] The parameter is incorrect.
  Failed.
```

tormodvolden, 2023-02-26:
I am curious about where in the driver stack the "[87] The parameter is incorrect" arises. I've checked the libusb0.sys source and there is no code there returning this without a debug warning. As I understand it, this happens on opening the device path which is done by standard kernel calls (the USB libraries are not involved) and there is a IRP_MJ_CREATE passed down the driver stack.

mcuee, 2023-02-26:
> I am curious about where in the driver stack the "[87] The parameter is incorrect" arises. I've checked the libusb0.sys source and there is no code there returning this without a debug warning. As I [...]

I am also curious where this error comes from. Do you think there is a possibility that this comes from libusbK.dll?

tormodvolden, 2023-02-26:
I wouldn't suspect the DLL because the the libusb code is not calling any of its DLL functions here. It is doing a plain CreateFileA() which I suppose is in Kernel32.dll.

tormodvolden, 2023-02-26:
If you have a kernel debugger there some nice tools to investigate the driver stack, see e.g. "Looking at Driver Dispatch Routines" from https://www.microsoftpressstore.com/articles/article.aspx?p=2201309&seqNum=3

mcuee, 2023-02-26:
> If you have a kernel debugger there some nice tools to investigate the driver stack, see e.g. "Looking at Driver Dispatch Routines" from https://www.microsoftpressstore.com/articles/article.aspx?p=2201309&seqNum=3

Windows kernel driver debugger typically requires two machine setup and the debug link is a bit tricy. I have never tried to debug a driver myself.

tormodvolden, 2023-02-26:
Having more debug printing in libusb0.sys could also be helpful. Do you have the setup to rebuild it?

mcuee, 2023-02-26:
> Having more debug printing in libusb0.sys could also be helpful. Do you have the setup to rebuild it?

I do not have the capability to sign the driver, but @dontech has the necessary facility to carry out the code signing, if you suspect that there is an issue with libusb0.sys.

tormodvolden, 2023-02-26:
But you can debug with your self-built, unsigned driver on your machine?

mcuee, 2023-02-26:
> But you can debug with your self-built, unsigned driver on your machine?

No, at least I do not know a way to install unsigned driver now.

My two laptops have secure boot on and I do not want to disable that as of now.

mcuee, 2023-02-26:
@tormodvolden 
Still please help to post your modifications of libusb0.sys and I will see if I can think of a way to try it.

@dontech
Just wondering if you have some suggestion to debug this issue futher with libusb0.sys filter driver.

tormodvolden, 2023-02-26:
For a start, I had in mind simple debugging prints like in this patch: [driver_mj_debug.patch.txt](https://github.com/libusb/libusb/files/10833936/driver_mj_debug.patch.txt)

Yes, I can also not rule out that libusb-1.0 is doing something wrong. Or it can be an issue in the driver triggered by how libusb-1.0 is working, or a combination of issues in several components. For what I know, it could also be the FTDI driver doing something wrong, that only affects operation with filter drivers.

tormodvolden, 2023-02-26:
I was thinking I could at least compile-test my patches, using cross-building, but it fails on missing "winddk.h". I don't find the file in the unpacked WDK tree here: https://github.com/9176324/WinDDK/tree/master/7600.16385.1
Where does it come from?

mcuee, 2023-02-26:
> I was thinking I could at least compile-test my patches, using cross-building

@tormodvolden 

Hmm, the driver building only works with WDK, not MinGW.

Long long ago (1.2.1.x version in 2010), I was able to compile the driver with some patches, but the support has since been dropped.
Ref: https://sourceforge.net/projects/picusb/files/libusb-win32/

mcuee, 2023-02-27:
> For a start, I had in mind simple debugging prints like in this patch: [driver_mj_debug.patch.txt](https://github.com/libusb/libusb/files/10833936/driver_mj_debug.patch.txt)
>
> Yes, I can also not [...]

I agree with you. It will probably take me some time to try the patch as I need to figure out a way to debug the driver. I am thinking using VM but I have never tried that myself.

tormodvolden, 2023-02-27:
It builds in my branch, and without warnings. I am not sure the resulting .sys will actually work on a Windows computer though.

mcuee, 2023-02-27:
> It builds in my branch, and without warnings. I am not sure the resulting .sys will actually work on a Windows computer though.

Indeed. I can also build the driver using MSYS2 mingw64. Nice. I did not expect it to even build. 

The library portion has always been able to be built using MinGW (with some warnings).

```
$ make driver
gcc -c ./src/driver/abort_endpoint.c -o abort_endpoint.o -O2 -Wall -DWINVER=0x500  -Wno-unknown-pragmas -Wno-multichar -DLOG_APPNAME=\"libusb0-sys\" -DTARGETTYPE=DRIVER  -I./src -I./src/driver -I. -I /mingw64/include/ddk -I /mingw64/include
gcc -c ./src/driver/claim_interface.c -o claim_interface.o -O2 -Wall -DWINVER=0x500  -Wno-unknown-pragmas -Wno-multichar -DLOG_APPNAME=\"libusb0-sys\" -DTARGETTYPE=DRIVER  -I./src -I./src/driver -I. -I /mingw64/include/ddk -I /mingw64/include
gcc -c ./src/driver/clear_feature.c -o clear_feature.o -O2 -Wall -DWINVER=0x500  -Wno-unknown-pragmas -Wno-multichar -DLOG_APPNAME=\"libusb0-sys\" -DTARGETTYPE=DRIVER  -I./src -I./src/driver -I. -I /mingw64/include/ddk -I /mingw64/include
gcc -c ./src/driver/dispatch.c -o dispatch.o -O2 -Wall -DWINVER=0x500  -Wno-unknown-pragmas -Wno-multichar -DLOG_APPNAME=\"libusb0-sys\" -DTARGETTYPE=DRIVER  -I./src -I./src/driver -I. -I /mingw64/include/ddk -I /mingw64/include
gcc -c ./src/driver/get_configuration.c -o get_configuration.o -O2 -Wall -DWINVER=0x500  -Wno-unknown-pragmas -Wno-multichar -DLOG_APPNAME=\"libusb0-sys\" -DTARGETTYPE=DRIVER  -I./src -I./src/driver -I. -I /mingw64/include/ddk -I /mingw64/include
gcc -c ./src/driver/get_descriptor.c -o get_descriptor.o -O2 -Wall -DWINVER=0x500  -Wno-unknown-pragmas -Wno-multichar -DLOG_APPNAME=\"libusb0-sys\" -DTARGETTYPE=DRIVER  -I./src -I./src/driver -I. -I /mingw64/include/ddk -I /mingw64/include
gcc -c ./src/driver/get_interface.c -o get_interface.o -O2 -Wall -DWINVER=0x500  -Wno-unknown-pragmas -Wno-multichar -DLOG_APPNAME=\"libusb0-sys\" -DTARGETTYPE=DRIVER  -I./src -I./src/driver -I. -I /mingw64/include/ddk -I /mingw64/include
gcc -c ./src/driver/get_status.c -o get_status.o -O2 -Wall -DWINVER=0x500  -Wno-unknown-pragmas -Wno-multichar -DLOG_APPNAME=\"libusb0-sys\" -DTARGETTYPE=DRIVER  -I./src -I./src/driver -I. -I /mingw64/include/ddk -I /mingw64/include
gcc -c ./src/driver/ioctl.c -o ioctl.o -O2 -Wall -DWINVER=0x500  -Wno-unknown-pragmas -Wno-multichar -DLOG_APPNAME=\"libusb0-sys\" -DTARGETTYPE=DRIVER  -I./src -I./src/driver -I. -I /mingw64/include/ddk -I /mingw64/include
gcc -c ./src/driver/libusb_driver.c -o libusb_driver.o -O2 -Wall -DWINVER=0x500  -Wno-unknown-pragmas -Wno-multichar -DLOG_APPNAME=\"libusb0-sys\" -DTARGETTYPE=DRIVER  -I./src -I./src/driver -I. -I /mingw64/include/ddk -I /mingw64/include
gcc -c ./src/driver/pnp.c -o pnp.o -O2 -Wall -DWINVER=0x500  -Wno-unknown-pragmas -Wno-multichar -DLOG_APPNAME=\"libusb0-sys\" -DTARGETTYPE=DRIVER  -I./src -I./src/driver -I. -I /mingw64/include/ddk -I /mingw64/include
gcc -c ./src/driver/release_interface.c -o release_interface.o -O2 -Wall -DWINVER=0x500  -Wno-unknown-pragmas -Wno-multichar -DLOG_APPNAME=\"libusb0-sys\" -DTARGETTYPE=DRIVER  -I./src -I./src/driver -I. -I /mingw64/include/ddk -I /mingw64/include
gcc -c ./src/driver/reset_device.c -o reset_device.o -O2 -Wall -DWINVER=0x500  -Wno-unknown-pragmas -Wno-multichar -DLOG_APPNAME=\"libusb0-sys\" -DTARGETTYPE=DRIVER  -I./src -I./src/driver -I. -I /mingw64/include/ddk -I /mingw64/include
gcc -c ./src/driver/reset_endpoint.c -o reset_endpoint.o -O2 -Wall -DWINVER=0x500  -Wno-unknown-pragmas -Wno-multichar -DLOG_APPNAME=\"libusb0-sys\" -DTARGETTYPE=DRIVER  -I./src -I./src/driver -I. -I /mingw64/include/ddk -I /mingw64/include
gcc -c ./src/driver/set_configuration.c -o set_configuration.o -O2 -Wall -DWINVER=0x500  -Wno-unknown-pragmas -Wno-multichar -DLOG_APPNAME=\"libusb0-sys\" -DTARGETTYPE=DRIVER  -I./src -I./src/driver -I. -I /mingw64/include/ddk -I /mingw64/include
gcc -c ./src/driver/set_descriptor.c -o set_descriptor.o -O2 -Wall -DWINVER=0x500  -Wno-unknown-pragmas -Wno-multichar -DLOG_APPNAME=\"libusb0-sys\" -DTARGETTYPE=DRIVER  -I./src -I./src/driver -I. -I /mingw64/include/ddk -I /mingw64/include
gcc -c ./src/driver/set_feature.c -o set_feature.o -O2 -Wall -DWINVER=0x500  -Wno-unknown-pragmas -Wno-multichar -DLOG_APPNAME=\"libusb0-sys\" -DTARGETTYPE=DRIVER  -I./src -I./src/driver -I. -I /mingw64/include/ddk -I /mingw64/include
gcc -c ./src/driver/set_interface.c -o set_interface.o -O2 -Wall -DWINVER=0x500  -Wno-unknown-pragmas -Wno-multichar -DLOG_APPNAME=\"libusb0-sys\" -DTARGETTYPE=DRIVER  -I./src -I./src/driver -I. -I /mingw64/include/ddk -I /mingw64/include
gcc -c ./src/driver/transfer.c -o transfer.o -O2 -Wall -DWINVER=0x500  -Wno-unknown-pragmas -Wno-multichar -DLOG_APPNAME=\"libusb0-sys\" -DTARGETTYPE=DRIVER  -I./src -I./src/driver -I. -I /mingw64/include/ddk -I /mingw64/include
gcc -c ./src/driver/vendor_request.c -o vendor_request.o -O2 -Wall -DWINVER=0x500  -Wno-unknown-pragmas -Wno-multichar -DLOG_APPNAME=\"libusb0-sys\" -DTARGETTYPE=DRIVER  -I./src -I./src/driver -I. -I /mingw64/include/ddk -I /mingw64/include
gcc -c ./src/driver/power.c -o power.o -O2 -Wall -DWINVER=0x500  -Wno-unknown-pragmas -Wno-multichar -DLOG_APPNAME=\"libusb0-sys\" -DTARGETTYPE=DRIVER  -I./src -I./src/driver -I. -I /mingw64/include/ddk -I /mingw64/include
[... cut ...]
gcc -c ./tests/testlibusb_win.c -o testlibusb_win.4.o -O2 -Wall -DWINVER=0x500  -Wno-unknown-pragmas -Wno-multichar -DLOG_APPNAME=\"testlibusb-win\" -DTARGETTYPE=PROGRAMwindows  -I./src -I./src/driver -I.
windres  -I./src ./tests/testlibusb_win_rc.rc -o testlibusb_win_rc.4.o
gcc -O2 -Wall -DWINVER=0x500  -Wno-unknown-pragmas -Wno-multichar -DLOG_APPNAME=\"testlibusb-win\" -DTARGETTYPE=PROGRAMwindows -o testlibusb-win.exe -I./src  testlibusb_win.4.o testlibusb_win_rc.4.o -s -L. -lusb -lkernel32 -lgdi32 -luser32 -lnewdev -lsetupapi -lcomctl32 -lole32 -mwindows
```

tormodvolden, 2023-02-27:
> I can also build the driver from your branch, without warnings, using MSYS2 mingw64. Nice. I did not expect it to even build.

Excellent. Do you have the WDK? It would be good to verify that my branch doesn't break on the WDK compiler.

> The library portion has always been able to be built using MinGW (with some warnings).

Yes, there are many warnings that can be worked on. Most of the times they are innocent but if there are too many nobody will notice the possible real issues. BTW, I have updated the library build so it also builds on 32-bit again.

mcuee, 2023-02-27:
> > I can also build the driver from your branch, without warnings, using MSYS2 mingw64. Nice. I did not expect it to even build.
> 
> Excellent. Do you have the WDK? It would be good to verify that my branch doesn't break on the WDK compiler.

Yes I have. And your changes do not cause any problems with WDK build.

```
c:\work\libusb\libusb-win32-mingw\libusb\ddk_make>make.cmd "arch=x64" driver dll
Libusb-Win32 directory = c:\work\libusb\libusb-win32-mingw\libusb\
WARNING: x64 Native compiling isn't supported. Using cross compilers.
Launching OACR monitor
Starting dirs creation...Completed.
WinDDK ok. Target version = 0x502
Building (fre_wnet_AMD64) make_dll.bat..
Building (fre_wnet_AMD64) make_driver.bat..
Building (fre_wnet_AMD64) make_test.bat..
Building (fre_wnet_AMD64) make_test_win.bat..
Building (fre_wnet_AMD64) make_install_filter.bat..
Building (fre_wnet_AMD64) make_install_filter_win.bat..
[SafeCreateDir] .\x64

c:\work\libusb\libusb-win32-mingw\libusb\ddk_make>cd x64

c:\work\libusb\libusb-win32-mingw\libusb\ddk_make\x64>ls
install-filter-win.exe  install-filter.exe  libusb.lib  libusb0.dll  libusb0.sys  testlibusb-win.exe  testlibusb.exe
```

mcuee, 2023-03-06:
Debug version of libusbK 3.0.10.

First run of libusb testlibusb example -- DebugView64
```
Dev (bus 1, device 11): 0403 - 6001 speed: 12M
  Manufacturer:              FTDI
  Product:                   US232R

[23800] 
[23800] AllK required contiguous memory = 675016 (64bit)
[23800]   8 HotK Handles: HandleSize 2112 PoolSize 16912 (bytes)
[23800]   64 LstK Handles: HandleSize 64 PoolSize 4112 (bytes)
[23800]   2048 LstInfoK Handles: HandleSize 64 PoolSize 131088 (bytes)
[23800]   128 UsbK Handles: HandleSize 96 PoolSize 12304 (bytes)
[23800]   64 DevK Handles: HandleSize 112 PoolSize 7184 (bytes)
[23800]   2048 OvlK Handles: HandleSize 104 PoolSize 213008 (bytes)
[23800]   64 OvlPoolK Handles: HandleSize 96 PoolSize 6160 (bytes)
[23800]   32 StmK Handles: HandleSize 176 PoolSize 5648 (bytes)
[23800]   2048 IsochK Handles: HandleSize 136 PoolSize 278544 (bytes)
[23800] 
[23800] Dynamically allocated as needed:
[23800]  KLST_DEVINFO = 2596 bytes each
[23800] [ERR](libusbK.dll)[k_Init_Version] ErrorCode=00000001h failed get version request.
[23800] [ERR](libusbK.dll)[k_Init_Config] ErrorCode=00000001h ->k_Init_Version
[23800] [ERR](libusbK.dll)[UsbStack_Init] ErrorCode=00000001h ->Init_ConfigCB
[23800] (libusbK.dll)[k_Init_Version] libusb0.sys v1.2.6.0
[23800] (libusbK.dll)[k_Init_Version] libusb0.sys v1.2.6.0
[23800] (libusbK.dll)[k_Init_Version] libusb0.sys v1.2.6.0
[23800] (libusbK.dll)[k_Init_Version] libusb0.sys v1.2.6.0
```

Second run:
```
Dev (bus 1, device 11): 0403 - 6001 speed: 12M (not able to print out Vendor ID and Product ID).

[10384] 
[10384] AllK required contiguous memory = 675016 (64bit)
[10384]   8 HotK Handles: HandleSize 2112 PoolSize 16912 (bytes)
[10384]   64 LstK Handles: HandleSize 64 PoolSize 4112 (bytes)
[10384]   2048 LstInfoK Handles: HandleSize 64 PoolSize 131088 (bytes)
[10384]   128 UsbK Handles: HandleSize 96 PoolSize 12304 (bytes)
[10384]   64 DevK Handles: HandleSize 112 PoolSize 7184 (bytes)
[10384]   2048 OvlK Handles: HandleSize 104 PoolSize 213008 (bytes)
[10384]   64 OvlPoolK Handles: HandleSize 96 PoolSize 6160 (bytes)
[10384]   32 StmK Handles: HandleSize 176 PoolSize 5648 (bytes)
[10384]   2048 IsochK Handles: HandleSize 136 PoolSize 278544 (bytes)
[10384] 
[10384] Dynamically allocated as needed:
[10384]  KLST_DEVINFO = 2596 bytes each
```

mcuee, 2023-03-06:
DebugView64 does not work well. Here is the log from DebugView Win32.

Removal, insertion and then first run of testlibusb.exe 32bit,
```
UevFilter.SendProcessEventMessage: Process 4340 has started.
KTM:  TmCommitTransactionExt for tx fafe8cf0
KTM:  Notifying RM of 2 for tx fafe8cf0
KTM:  Notifying RM of 4 for tx fafe8cf0
KTM:  TmRollbackEnlistmentExt for tx fafe8cf0
KTM:  TmRollbackTransactionExt for tx fafe8cf0
[4340] 
[4340] AllK required contiguous memory = 421636 (32bit)
[4340]   8 HotK Handles: HandleSize 2084 PoolSize 16680 (bytes)
[4340]   64 LstK Handles: HandleSize 36 PoolSize 2312 (bytes)
[4340]   2048 LstInfoK Handles: HandleSize 36 PoolSize 73736 (bytes)
[4340]   128 UsbK Handles: HandleSize 56 PoolSize 7176 (bytes)
[4340]   64 DevK Handles: HandleSize 60 PoolSize 3848 (bytes)
[4340]   2048 OvlK Handles: HandleSize 60 PoolSize 122888 (bytes)
[4340]   64 OvlPoolK Handles: HandleSize 52 PoolSize 3336 (bytes)
[4340]   32 StmK Handles: HandleSize 100 PoolSize 3208 (bytes)
[4340]   2048 IsochK Handles: HandleSize 92 PoolSize 188424 (bytes)
[4340] 
[4340] Dynamically allocated as needed:
[4340]  KLST_DEVINFO = 2596 bytes each
KTM:  TmCommitTransactionExt for tx dc4f7d10
KTM:  Notifying RM of 2 for tx dc4f7d10
KTM:  Notifying RM of 4 for tx dc4f7d10
KTM:  TmRollbackEnlistmentExt for tx dc4f7d10
KTM:  TmRollbackTransactionExt for tx dc4f7d10
KTM:  TmCommitTransactionExt for tx d9e9cf0
KTM:  Notifying RM of 2 for tx d9e9cf0
KTM:  Notifying RM of 4 for tx d9e9cf0
KTM:  TmRollbackEnlistmentExt for tx d9e9cf0
KTM:  TmRollbackTransactionExt for tx d9e9cf0
UevFilter.SendProcessEventMessage: Process 13980 has started.
UevFilter.SendProcessEventMessage: Process 6668 has started.
libusb0-sys:[dispatch_pnp] IRP_MN_SURPRISE_REMOVAL: is-filter=Y usb\vid_0403&pid_6001&rev_0600
libusb0-sys:[set_filter_interface_key] updated interface registry with LUsb0 direct-access symbolic link. id=-1
libusb0-sys:[dispatch_pnp] IRP_MN_REMOVE_DEVICE: is-filter=Y usb\vid_0403&pid_6001&rev_0600
libusb0-sys:[dispatch_pnp] deleting device #1 usb\vid_0403&pid_6001&rev_0600
libusb0-sys:[unload] [unloading-driver] v1.2.6.0
UevFilter.SendProcessEventMessage: Process 8152 has started.
libusb0-sys:[DriverEntry] [loading-driver] v1.2.6.0
libusb0-sys:dbg [debug_show_devices] [AttachedDevice #0] \Driver\ACPI
libusb0-sys:dbg [debug_show_devices] [AttachedDevice #1] \Driver\CSDeviceControl
libusb0-sys:dbg [debug_show_devices] [AttachedDevice #2] \Driver\FTDIBUS
libusb0-sys:[add_device] device #1 created for usb\vid_0403&pid_6001&rev_0600
libusb0-sys:[reg_get_properties] using default device interface guid.
libusb0-sys:dbg [reg_get_properties] reg-key-name=\REGISTRY\MACHINE\SYSTEM\ControlSet001\Enum\USB\VID_0403&PID_6001\FTDCRPIE\Device Parameters
libusb0-sys:[set_filter_interface_key] updated interface registry with LUsb0 direct-access symbolic link. id=1
libusb0-sys:[add_device] updated interface registry with LUsb0 direct-access symbolic link. id=0001
libusb0-sys:dbg [add_device] [filter-mode] id=#1 usb\vid_0403&pid_6001&rev_0600
libusb0-sys:[add_device] complete status=00000000
libusb0-sys:[dispatch_pnp] IRP_MN_START_DEVICE: is-filter=Y usb\vid_0403&pid_6001&rev_0600
libusb0-sys:dbg [on_start_complete] is-filter=Y usb\vid_0403&pid_6001&rev_0600
libusb0-sys:dbg [dispatch_pnp] IRP_MN_QUERY_CAPABILITIES: is-filter=Y usb\vid_0403&pid_6001&rev_0600
libusb0-sys:dbg [dispatch_pnp] IRP_MN_QUERY_CAPABILITIES: is-filter=Y usb\vid_0403&pid_6001&rev_0600
libusb0-sys:dbg [dispatch_pnp] IRP_MN_QUERY_CAPABILITIES: is-filter=Y usb\vid_0403&pid_6001&rev_0600
UevFilter.SendProcessEventMessage: Process 6244 has started.
UevFilter.SendProcessEventMessage: Process 7376 has started.
UevFilter.SendProcessEventMessage: Process 6728 has started.
[6728] 
[6728] AllK required contiguous memory = 421636 (32bit)
[6728]   8 HotK Handles: HandleSize 2084 PoolSize 16680 (bytes)
[6728]   64 LstK Handles: HandleSize 36 PoolSize 2312 (bytes)
[6728]   2048 LstInfoK Handles: HandleSize 36 PoolSize 73736 (bytes)
[6728]   128 UsbK Handles: HandleSize 56 PoolSize 7176 (bytes)
[6728]   64 DevK Handles: HandleSize 60 PoolSize 3848 (bytes)
[6728]   2048 OvlK Handles: HandleSize 60 PoolSize 122888 (bytes)
[6728]   64 OvlPoolK Handles: HandleSize 52 PoolSize 3336 (bytes)
[... 44 lines cut ...]
UevFilter.SendProcessEventMessage: Process 8416 has started.
UevFilter.SendProcessEventMessage: Process 16620 has started.
```

Second run.
```
UevFilter.SendProcessEventMessage: Process 15908 has started.
[15908] 
[15908] AllK required contiguous memory = 421636 (32bit)
[15908]   8 HotK Handles: HandleSize 2084 PoolSize 16680 (bytes)
[15908]   64 LstK Handles: HandleSize 36 PoolSize 2312 (bytes)
[15908]   2048 LstInfoK Handles: HandleSize 36 PoolSize 73736 (bytes)
[15908]   128 UsbK Handles: HandleSize 56 PoolSize 7176 (bytes)
[15908]   64 DevK Handles: HandleSize 60 PoolSize 3848 (bytes)
[15908]   2048 OvlK Handles: HandleSize 60 PoolSize 122888 (bytes)
[15908]   64 OvlPoolK Handles: HandleSize 52 PoolSize 3336 (bytes)
[15908]   32 StmK Handles: HandleSize 100 PoolSize 3208 (bytes)
[15908]   2048 IsochK Handles: HandleSize 92 PoolSize 188424 (bytes)
[15908] 
[15908] Dynamically allocated as needed:
[15908]  KLST_DEVINFO = 2596 bytes each
UevFilter.SendProcessEventMessage: Process 2824 has started.
UevFilter.SendProcessEventMessage: Process 4092 has started.

```

[45 more replies]