libusb/libusb #1020: libusb_submit_transfer is caused restart

zklyzhoutao, 2021-11-04:
Hi, We use libuvc and libusb in the Android devices.
When We test the Android device of Google Pixel 4, The App is caused phone restart.
End We find libusb_submit_transfer is get stuck and cause phone restart.

libusb version 1.0.25;
libuvc version 0.06;
Android device: Google Pixel 4 and Android 11.

mcuee, 2021-11-04:
There is no libusb 1.0.25 yet. Do you mean libusb 1.0.24 or latest git?

libusb-1.0.24 does not work well with Android devices. You may try the latest git.
https://github.com/libusb/libusb/tree/master/android/examples

Or try this one.
https://github.com/libusb/libusb/pull/874

mcuee, 2021-11-04:
You may also create the ticket in libuvc, this may be an issue with libuvc and you may want to try latest git of libuvc. libuvc 0.06 is pretty old.

zklyzhoutao, 2021-11-04:
sorry, the libusb version is here.
https://github.com/xloem/libusb/tree/android-dev.
I have open the device, But submit_transfer is caused restart phone.

mcuee, 2021-11-04:
> sorry, the libusb version is here. https://github.com/xloem/libusb/tree/android-dev. I have open the device, But submit_transfer is caused restart phone.

You have to to generate the debug log to see if others can help you or not. I do not know much about Android and have no real interests in Android myself. Google seems to have done a really bad job on USB access.

xloem, 2021-11-04:
hi, android-dev is out of date, use the android branch

but yes logs + debugging help please, it's all a work in progress

zklyzhoutao, 2021-11-05:
Thanks!
I use the libusb of android branch,  https://github.com/xloem/libusb/tree/android
But it also caused phone(google pixel 4) restart.
And I debug the app, I find the function **submit_iso_transfer**  in the **file linux_usbfs.c** , the code `int r = ioctl(hpriv->fd, IOCTL_USBFS_SUBMITURB, urbs[i]);` is block, not return any result. But the function **submit_control_transfer** and the code `r = ioctl(hpriv->fd, IOCTL_USBFS_SUBMITURB, urb);` is ok. It's a Android bug or others?

mcuee, 2021-11-08:
isoc transfer may not be easy. Does your code work under normal Linux distributions?

If the code works under other Linux distros, but caused the Google Pixel phone to restart, you can report the bug to Google as it seems to be a Android bug (maybe even a Linux kernel bug with Google's specific modification for Androdi).

mcuee, 2022-02-06:
@zklyzhoutao Can you try your code with normal Linux distributions first and then other Android phones? If it works with other Android phone but not Pixel 4, maybe it is a specific problem under Pixel 4.