libusb/libusb #1021: Better detect Windows version in libusb debug log

mcuee, 2021-11-05:
Right now the Windows version detection in libusb debug logs seems to have some limitations.

For my testing under Windows 10, it will only detect as Windows 8.

It is better to differentiate Windows 8, 8.1, 10 and 11 at least. Not so sure about server versions of Windows.

a-v-s, 2021-12-16:
Windows 10 needs a manifest to be able to detect the version, otherwise it will report windows 8.
However, when being called from a library, does the library or the application require the manifest?

mcuee, 2021-12-20:
It is said you have to go through WMI objects.
https://docs.microsoft.com/en-us/answers/questions/547050/win32-api-to-detect-windows-11.html

mcuee, 2022-01-29:
There are some codes here from libwdi, we can probably reuse it.
https://github.com/pbatard/libwdi/blob/master/libwdi/libwdi.c#L184

tormodvolden, 2022-02-05:
Note that we already use the same code as libwdi, I think it is broken in the same way there if there is no "application manifest".
I think just printing "8 (or later)" is as much as we can reasonably do, and should do. Doing all the WMI trickery is too much, also since we only use the version for display purpose.

If libusb is used in an application with manifest (stating support for the actual version), I believe it will show the exact version (10 or 11).

mcuee, 2022-02-05:
It seems to me Zadig works fine without the clear application manifest file.

Ref: zadig source code:
https://github.com/pbatard/libwdi/tree/master/examples

Zadig debug log example under my Windows 11 laptop.
```
Zadig 2.7.765
Windows 11 Home, 64-bit (Build 22000.434)
ini file 'zadig.ini' not found in 'C:\work\libusb' - default parameters will be used
default driver set to 'WinUSB'
0 devices found.
7 devices found.
...
``1

mcuee, 2022-02-05:
But I think pull request #1060 is correct.

Ref: https://docs.microsoft.com/en-us/windows/win32/sysinfo/operating-system-version
> * For applications that have been manifested for Windows 8.1 or Windows 10. Applications not manifested for Windows 8.1 or Windows 10 will return the Windows 8 OS version value (6.2). To manifest your applications for Windows 8.1 or Windows 10, refer to [Targeting your application for Windows](https://docs.microsoft.com/en-us/windows/win32/sysinfo/targeting-your-application-at-windows-8-1).

tormodvolden, 2022-02-05:
> It seems to me Zadig works fine without an obvious application manifest file.
> 
> Ref: zadig source code: https://github.com/pbatard/libwdi/tree/master/examples

See examples/common_controls_and_elevation.manifest
+1 on the spineless morons :)

mcuee, 2022-02-05:
> See examples/common_controls_and_elevation.manifest +1 on the spineless morons :)

Ah I see. Thanks. Interesting that the manifest file does not need to mention Windows 11 yet it knows Windows 11.

tormodvolden, 2022-02-05:
Yes, I don't understand this. Can your binary have been built with a different manifest than the git one?

tormodvolden, 2022-02-05:
For our verification, can the manifest be extracted from the binary? I just found https://www.nirsoft.net/utils/resources_extract.html

mcuee, 2022-02-06:
> Yes, I don't understand this. Can your binary have been built with a different manifest than the git one?

I am using the official binary so it should be from the same manifest files.

mcuee, 2022-02-06:
> For our verification, can the manifest be extracted from the binary? I just found > https://www.nirsoft.net/utils/resources_extract.html

No this does not seem to work.

Zadig 2.7 download: https://zadig.akeo.ie/

mcuee, 2022-02-06:
@pbatard Sorry to disturb you. Maybe you can help explain how Zadig can detect Windows 11 version. Thanks.

mcuee, 2022-02-06:
@tormodvolden this is the commit which makes libwdi Zadig 2.7 able to detect Windows 11. Zadig 2.6 can not do that.
https://github.com/pbatard/libwdi/commit/1aaa794a8f934d1b796e3716039e0fad9b42e2a9

tormodvolden, 2022-02-06:
As I understand it, that commit doesn't help if the manifest doesn't mention Windows 11.

mcuee, 2022-02-07:
Let's see if Pete can help here. @pbatard Thanks.

pbatard, 2022-02-08:
Somehow I didn't get the pings.

To cut a long story short, not only were Microsoft not happy to completely cripple `GetversionEx()` unless you had a manifest with the relevant `supportedOS` GUID, but, with the release of Windows 11, [they didn't even bother adding a new `supportedOS` GUID for that OS](https://stackoverflow.com/questions/68240304/whats-the-supportedos-guid-for-windows-11), which means that `GetVerionEx()` will report `0xA0` (i.e. Windows 10) and not `0xB0` (Windows 11).

One really has to wonder why Microsoft went all the trouble of __finally__ setting the version reported by its API to a value that actually resembles the consumer facing one, only to drop the whole thing in the next version.

So, the __ONLY__ way you can detect Windows 11, when it is deliberately misreporting its version as Windows 10 even with a manifest, is through the Build Number. And that's why you see a [check for `dwBuildNumber < 20000` along with a fall through](https://github.com/pbatard/libwdi/blob/master/libwdi/libwdi.c#L259-L265) in libwdi's version detection code, because that's all Microsoft is leaving developers with, when you don't want to go through WMI or other cumbersome methods.

And for the record, the two reasons why Microsoft has chosen not to report an accurate version of Windows through `GetversionEx()` is first because they really don't want people writing apps that offer different features according to the version of Windows being run (that's the _moronic_ part) and also because they don't want to annoy corporate customers with something as pesky as __ACTUALLY MAINTAINING THEIR BLOODY LEGACY APPLICATIONS__ (i.e. the ones they have long fired the developers of, so that executives could post yet another year of record bonuses) in case these applications relied on detecting the Windows version to perform specific tasks.

And that last one is the _spineless_ part, because, whereas Microsoft had the opportunity and power to make corporate customers understand that applications that aren't maintained and/or break with something as minor as a new OS version __are a major liability__, they decided to just bend over backward and continue to promote this massive and dangerous lie that unmaintained software is okay to use. But of course, if Microsoft told it like it actually is, there's also a good chance that folks would start to realize why anything closed source is exceedingly problematic... which wouldn't exactly play into Microsoft's pocket.

So once again, Microsoft is collectively making tons of people waste their time with trying to figure out what the heck version of Windows the application is running on, for the benefit of people who would totally have the resources to handle the alternative... but pressured Microsoft into not doing so because it would impact their precious profitability margins.

tormodvolden, 2022-02-09:
@pbatard Thanks for sharing this. Are you sure about your 20000 threshold? I have seen 22000 in several places, e.g. here: https://docs.microsoft.com/en-us/windows/release-health/windows11-release-information

commit 0d880bf4 (2022-03-18): windows: Discern Windows 11 (if manifest allows)

Pre-releases of Windows 11 had build numbers from 20000.

Thanks to Pete Batard for extensive explanation.

Fixes #1021
commit f9a5cf5b (2022-03-18): windows: Add "or later" to debug print of Windows version 8

Windows 10 and 11 is detected as Windows 8 unless there is an
application manifest. We simply cannot tell by appropriate means.

Note that we currently only use the detected version to check for the
minimal version (Vista), so it doesn't matter much.

Closes #1021
closed 2022-03-18 by tormodvolden