libusb/libusb #1022: How to extract as depends/libusb

Tavagero, 2021-11-06:
Sorry to bothr you with such stupid question, but I havn't any experience with github. I want to try a project that require your project:
"Download the latest build (.7z file) from https://github.com/libusb/libusb/releases, and extract as depends/libusb (rename folder libusb-1.x.y to libusb if any)." 
I downloaded your project and extracted it, but I don't know how to do this as depends. Do I need to start the project in Visual Studio, because I don't see any way to start it or I need to create somehow project in VS with your project?

tormodvolden, 2021-11-06:
It would be useful if you link to the location where those instructions are, but I guess "depends" is a file directory that already exists, or you create it in the root of the other project.

Tavagero, 2021-11-06:
https://github.com/OpenKinect/libfreenect2/blob/master/README.md#windows--visual-studio

tormodvolden, 2021-11-06:
It is a pre-existing folder: https://github.com/OpenKinect/libfreenect2/tree/master/depends

closed 2021-11-07 by tormodvolden