libusb/libusb #1024: timeout error libusb_set_interface_alt_setting

gentooza, 2021-11-08:
Hi!

I write for an issue calling this function using heimdall, it happens always (and only) with some devices.
one issue and it's workaround commenting out this libusb function here: https://github.com/Benjamin-Dobell/Heimdall/issues/340

using Ubuntu 20.04 amd64 GNU/Linux

I paste here an example log:
```
Heimdall v1.4.2

Copyright (c) 2010-2017 Benjamin Dobell, Glass Echidna
http://www.glassechidna.com.au/

This software is provided free of charge. Copying and redistribution is
encouraged.

If you appreciate this software and you would like to support future
development please consider donating:
http://www.glassechidna.com.au/donate/

Initialising connection...
Detecting device...
[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.011476] [000013d2] libusb: debug [libusb_get_device_list] 
[ 0.011503] [000013d2] libusb: debug [discovered_devs_append] need to increase capacity
[ 0.011513] [000013d2] libusb: debug [libusb_get_device_descriptor] 
[ 0.011519] [000013d2] libusb: debug [libusb_get_device_descriptor] 
[ 0.011527] [000013d2] libusb: debug [libusb_get_device_descriptor] 
[ 0.011531] [000013d2] libusb: debug [libusb_get_device_descriptor] 
[ 0.011538] [000013d2] libusb: debug [libusb_open] open 1.5
[ 0.011575] [000013d2] libusb: debug [usbi_add_pollfd] add fd 10 events 4
[ 0.011583] [000013d2] libusb: debug [libusb_get_device_descriptor] 
[ 0.011591] [000013d2] libusb: debug [libusb_alloc_transfer] transfer 0x559dda0d7ea0
[ 0.011598] [000013d2] libusb: debug [libusb_submit_transfer] transfer 0x559dda0d7ea0
[ 0.011603] [000013d2] libusb: debug [add_to_flying_list] arm timerfd for timeout in 1000ms (first in line)
[ 0.011659] [000013d2] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[ 0.011675] [000013d2] libusb: debug [handle_events] poll fds modified, reallocating
[ 0.011685] [000013d2] libusb: debug [handle_events] poll() 3 fds with timeout in 60000ms
[ 0.015189] [000013d2] libusb: debug [handle_events] poll() returned 1
[ 0.015247] [000013d2] libusb: debug [reap_for_handle] urb type=2 status=0 transferred=10
[ 0.015264] [000013d2] libusb: debug [handle_control_completion] handling completion status 0
[ 0.015282] [000013d2] libusb: debug [disarm_timerfd] 
[ 0.015303] [000013d2] libusb: debug [usbi_handle_transfer_completion] transfer 0x559dda0d7ea0 has callback 0x7f2ffe983fc0
[ 0.015317] [000013d2] libusb: debug [sync_transfer_cb] actual_length=10
[ 0.015346] [000013d2] libusb: debug [libusb_free_transfer] transfer 0x559dda0d7ea0
[ 0.015370] [000013d2] libusb: debug [libusb_alloc_transfer] transfer 0x559dda0d8c90
[ 0.015381] [000013d2] libusb: debug [libusb_submit_transfer] transfer 0x559dda0d8c90
[ 0.015388] [000013d2] libusb: debug [add_to_flying_list] arm timerfd for timeout in 1000ms (first in line)
[ 0.015425] [000013d2] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[ 0.015450] [000013d2] libusb: debug [handle_events] poll() 3 fds with timeout in 60000ms
[ 0.019462] [000013d2] libusb: debug [handle_events] poll() returned 1
[ 0.019526] [000013d2] libusb: debug [reap_for_handle] urb type=2 status=0 transferred=40
[ 0.019547] [000013d2] libusb: debug [handle_control_completion] handling completion status 0
[ 0.019579] [000013d2] libusb: debug [disarm_timerfd] 
[ 0.019592] [000013d2] libusb: debug [usbi_handle_transfer_completion] transfer 0x559dda0d8c90 has callback 0x7f2ffe983fc0
[ 0.019600] [000013d2] libusb: debug [sync_transfer_cb] actual_length=40
[ 0.019613] [000013d2] libusb: debug [libusb_free_transfer] transfer 0x559dda0d8c90
      Manufacturer: "Samsung Electronics"
[ 0.019638] [000013d2] libusb: debug [libusb_alloc_transfer] transfer 0x559dda0d6070
[ 0.019652] [000013d2] libusb: debug [libusb_submit_transfer] transfer 0x559dda0d6070
[ 0.019667] [000013d2] libusb: debug [add_to_flying_list] arm timerfd for timeout in 1000ms (first in line)
[ 0.019713] [000013d2] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[ 0.019729] [000013d2] libusb: debug [handle_events] poll() 3 fds with timeout in 60000ms
[ 0.023982] [000013d2] libusb: debug [handle_events] poll() returned 1
[ 0.024046] [000013d2] libusb: debug [reap_for_handle] urb type=2 status=0 transferred=10
[ 0.024073] [000013d2] libusb: debug [handle_control_completion] handling completion status 0
[ 0.024120] [000013d2] libusb: debug [disarm_timerfd] 
[ 0.024144] [000013d2] libusb: debug [usbi_handle_transfer_completion] transfer 0x559dda0d6070 has callback 0x7f2ffe983fc0
[... 80 lines cut ...]
[10.338184] [000013d2] libusb: debug [libusb_exit] destroying default context
[10.338203] [000013d2] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[10.338221] [000013d2] libusb: debug [handle_events] poll fds modified, reallocating
[10.338261] [000013d2] libusb: debug [handle_events] poll() 2 fds with timeout in 0ms
[10.338276] [000013d2] libusb: debug [handle_events] poll() returned 0
[10.338289] [000013d2] libusb: debug [libusb_unref_device] destroy device 2.2
[10.338314] [000013d2] libusb: debug [libusb_unref_device] destroy device 2.1
[10.338339] [000013d2] libusb: debug [libusb_unref_device] destroy device 4.1
[10.338351] [000013d2] libusb: debug [libusb_unref_device] destroy device 1.5
[10.338362] [000013d2] libusb: debug [libusb_unref_device] destroy device 1.1
[10.338370] [000013d2] libusb: debug [libusb_unref_device] destroy device 3.1
[10.338379] [000013d2] libusb: debug [libusb_unref_device] destroy device 6.1
[10.338387] [000013d2] libusb: debug [libusb_unref_device] destroy device 5.3
[10.338406] [000013d2] libusb: debug [libusb_unref_device] destroy device 5.2
[10.338416] [000013d2] libusb: debug [libusb_unref_device] destroy device 5.1
[10.338424] [000013d2] libusb: debug [usbi_remove_pollfd] remove fd 7
[10.338456] [000013d2] libusb: debug [usbi_remove_pollfd] remove fd 9
[10.338568] [000013d4] libusb: debug [linux_udev_event_thread_main] udev event thread exiting
```

code related is here: https://github.com/Benjamin-Dobell/Heimdall/blob/master/heimdall/source/BridgeManager.cpp#L264
It can be of course a Heimdall issue, I'm not programmer of this project, but it seems to me a correct call to a libusb function.

cheers and Happy Hacking!
Joa

mcuee, 2021-11-08:
Have you tried latest libusb git head? It has many fixes including some important fixes for Linux.
https://github.com/libusb/libusb/issues/777

mcuee, 2021-11-08:
And some devices are broken that it may not support the call properly. If the device has only one alt setting, this call is not necesary.

gentooza, 2021-11-08:
> Have you tried latest libusb git head? It has many fixes including some important fixes for Linux. #777

no, but I will! version installed in ubuntu seems to be: 1.0.23-2build1

> And some devices are broken that it may not support the call properly. If the device has only one alt setting, this call is not necessary.

yeah, I think it's device related for sure, for now I'm bypassing this libusb call with no further issues, but as it's a timeout error if there was something I could do increasing timeout? or something else. :-/

thanks!

mcuee, 2021-11-08:
> no, but I will! version installed in ubuntu seems to be: 1.0.23-2build1

You may want to try out libusb 1.0.24 or libusb git just to make sure it is not related to libusb.

> yeah, I think it's device related for sure, for now I'm bypassing this libusb call with no further issues, but as it's a timeout error if there was something I could do increasing timeout? or something else. :-/

Most likely it will not help if the device has problem with this call.

If you want to make sure, you can use usbmon or Wireshark to capture the traffic.
https://www.kernel.org/doc/Documentation/usb/usbmon.txt
https://wiki.wireshark.org/CaptureSetup/USB

tormodvolden, 2021-11-08:
Please attach the output from `lsusb -v -d 04E8:685D`

gentooza, 2021-11-08:
> lsusb -v -d 04E8:685D

```
joa@joa-HP-Pavilion-15-Notebook-PC:~$ lsusb -v -d 04E8:685D

Bus 001 Device 006: ID 04e8:685d Samsung Electronics Co., Ltd GT-I9100 Phone [Galaxy S II] (Download mode)
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            2 Communications
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x04e8 Samsung Electronics Co., Ltd
  idProduct          0x685d GT-I9100 Phone [Galaxy S II] (Download mode)
  bcdDevice            0.01
  iManufacturer           2 Samsung Electronics
  iProduct                1 Mobile USB Modem 1.0
  iSerial                 3 00000000000000/0
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0043
    bNumInterfaces          2
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              474mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         2 Communications
      bInterfaceSubClass      2 Abstract (modem)
      bInterfaceProtocol      1 AT-commands (v.25ter)
      iInterface              0 
      CDC Header:
        bcdCDC               1.10
      CDC Call Management:
        bmCapabilities       0x00
        bDataInterface          1
      CDC ACM:
        bmCapabilities       0x00
      CDC Union:
        bMasterInterface        0
        bSlaveInterface         1 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               5
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass        10 CDC Data
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval              32
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x02  EP 2 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval              32
Device Qualifier (for other device speed):
  bLength                10
  bDescriptorType         6
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  bNumConfigurations      1
can't get debug descriptor: Resource temporarily unavailable
cannot read device status, Resource temporarily unavailable (11)
```

after this, device gets unresponsive, and I have to reboot it and reconnect. :-O

gentooza, 2021-11-08:
> ```
> joa@joa-HP-Pavilion-15-Notebook-PC:~$ lsusb -v -d 04E8:685D
>
> Bus 001 Device 006: ID 04e8:685d Samsung Electronics Co., Ltd GT-I9100 Phone [Galaxy S II] (Download mode)
> Device Descriptor: [...]

wow!
I've just realize kernel detects a wrong device model! Perhaps here's the problem :-/
model detected : GT-I9100 Phone [Galaxy S II] https://en.wikipedia.org/wiki/Samsung_Galaxy_S_II
real model: GT-S5570i [Galaxy mini plus] https://en.wikipedia.org/wiki/Samsung_Galaxy_Mini

They have in fact different cpus
https://wiki.postmarketos.org/wiki/Samsung_Galaxy_SII_(samsung-i9100)
https://wiki.postmarketos.org/wiki/Samsung_Galaxy_Pop_Plus_(samsung-s5570i)

tormodvolden, 2021-11-08:
Don't worry about the model being displayed. They use the same bootloader for many models, with the same USB ID. It was maybe first used on the 9100 model so that is what ended up in the USB ID string database.

If lsusb makes it hang, it is another indication that the Samsung bootloader is a bit flaky. Does Heimdall work if you instead of commenting out the libusb_set_interface() call, just make it ignore the timeout error?

tormodvolden, 2021-11-08:
Just FYI, the strings displayed for idVendor and idProduct come from the database on your OS. The strings for iManufacturer and iProduct come from the device (the bootloader).

gentooza, 2021-11-08:
> Don't worry about the model being displayed. They use the same bootloader for many models, with the same USB ID. It was maybe first used on the 9100 model so that is what ended up in the USB ID [...]

No! it returns error.

```
Heimdall v1.4.2

Copyright (c) 2010-2017 Benjamin Dobell, Glass Echidna
http://www.glassechidna.com.au/

This software is provided free of charge. Copying and redistribution is
encouraged.

If you appreciate this software and you would like to support future
development please consider donating:
http://www.glassechidna.com.au/donate/

Initialising connection...
Detecting device...
[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.010661] [0000328e] libusb: debug [libusb_get_device_list] 
[ 0.010688] [0000328e] libusb: debug [discovered_devs_append] need to increase capacity
[ 0.010698] [0000328e] libusb: debug [libusb_get_device_descriptor] 
[ 0.010704] [0000328e] libusb: debug [libusb_get_device_descriptor] 
[ 0.010708] [0000328e] libusb: debug [libusb_get_device_descriptor] 
[ 0.010712] [0000328e] libusb: debug [libusb_get_device_descriptor] 
[ 0.010718] [0000328e] libusb: debug [libusb_open] open 1.17
[ 0.081643] [0000328e] libusb: debug [usbi_add_pollfd] add fd 10 events 4
[ 0.081702] [0000328e] libusb: debug [libusb_get_device_descriptor] 
[ 0.081755] [0000328e] libusb: debug [libusb_alloc_transfer] transfer 0x55c17d5a6ea0
[ 0.081766] [0000328e] libusb: debug [libusb_submit_transfer] transfer 0x55c17d5a6ea0
[ 0.081776] [0000328e] libusb: debug [add_to_flying_list] arm timerfd for timeout in 1000ms (first in line)
[ 0.081835] [0000328e] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[ 0.081857] [0000328e] libusb: debug [handle_events] poll fds modified, reallocating
[ 0.081903] [0000328e] libusb: debug [handle_events] poll() 3 fds with timeout in 60000ms
[ 0.085968] [0000328e] libusb: debug [handle_events] poll() returned 1
[ 0.086030] [0000328e] libusb: debug [reap_for_handle] urb type=2 status=0 transferred=10
[ 0.086054] [0000328e] libusb: debug [handle_control_completion] handling completion status 0
[ 0.086085] [0000328e] libusb: debug [disarm_timerfd] 
[ 0.086109] [0000328e] libusb: debug [usbi_handle_transfer_completion] transfer 0x55c17d5a6ea0 has callback 0x7fb1a2313fc0
[ 0.086131] [0000328e] libusb: debug [sync_transfer_cb] actual_length=10
[ 0.086176] [0000328e] libusb: debug [libusb_free_transfer] transfer 0x55c17d5a6ea0
[ 0.086204] [0000328e] libusb: debug [libusb_alloc_transfer] transfer 0x55c17d5a7c90
[ 0.086224] [0000328e] libusb: debug [libusb_submit_transfer] transfer 0x55c17d5a7c90
[ 0.086246] [0000328e] libusb: debug [add_to_flying_list] arm timerfd for timeout in 1000ms (first in line)
[ 0.086295] [0000328e] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[ 0.086316] [0000328e] libusb: debug [handle_events] poll() 3 fds with timeout in 60000ms
[ 0.090458] [0000328e] libusb: debug [handle_events] poll() returned 1
[ 0.090531] [0000328e] libusb: debug [reap_for_handle] urb type=2 status=0 transferred=40
[ 0.090558] [0000328e] libusb: debug [handle_control_completion] handling completion status 0
[ 0.090578] [0000328e] libusb: debug [disarm_timerfd] 
[ 0.090604] [0000328e] libusb: debug [usbi_handle_transfer_completion] transfer 0x55c17d5a7c90 has callback 0x7fb1a2313fc0
[ 0.090625] [0000328e] libusb: debug [sync_transfer_cb] actual_length=40
[ 0.090652] [0000328e] libusb: debug [libusb_free_transfer] transfer 0x55c17d5a7c90
      Manufacturer: "Samsung Electronics"
[ 0.090697] [0000328e] libusb: debug [libusb_alloc_transfer] transfer 0x55c17d5a5070
[ 0.090718] [0000328e] libusb: debug [libusb_submit_transfer] transfer 0x55c17d5a5070
[ 0.090738] [0000328e] libusb: debug [add_to_flying_list] arm timerfd for timeout in 1000ms (first in line)
[ 0.090784] [0000328e] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[ 0.090802] [0000328e] libusb: debug [handle_events] poll() 3 fds with timeout in 60000ms
[ 0.094819] [0000328e] libusb: debug [handle_events] poll() returned 1
[ 0.094880] [0000328e] libusb: debug [reap_for_handle] urb type=2 status=0 transferred=10
[ 0.094899] [0000328e] libusb: debug [handle_control_completion] handling completion status 0
[ 0.094925] [0000328e] libusb: debug [disarm_timerfd] 
[ 0.094951] [0000328e] libusb: debug [usbi_handle_transfer_completion] transfer 0x55c17d5a5070 has callback 0x7fb1a2313fc0
[ 0.094968] [0000328e] libusb: debug [sync_transfer_cb] actual_length=10
[ 0.094995] [0000328e] libusb: debug [libusb_free_transfer] transfer 0x55c17d5a5070
[ 0.095022] [0000328e] libusb: debug [libusb_alloc_transfer] transfer 0x55c17d59c640
[ 0.095040] [0000328e] libusb: debug [libusb_submit_transfer] transfer 0x55c17d59c640
[... 137 lines cut ...]
[ 8.971792] [0000328e] libusb: debug [libusb_release_interface] interface 1
[ 8.971917] [0000328e] libusb: debug [libusb_close] 
[ 8.971942] [0000328e] libusb: debug [usbi_remove_pollfd] remove fd 10
[ 8.971967] [0000328e] libusb: debug [libusb_exit] 
[ 8.971979] [0000328e] libusb: debug [libusb_exit] destroying default context
[ 8.971989] [0000328e] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[ 8.972007] [0000328e] libusb: debug [handle_events] poll fds modified, reallocating
[... cut ...]
[ 8.972187] [0000328e] libusb: debug [libusb_unref_device] destroy device 6.1
[ 8.972205] [0000328e] libusb: debug [libusb_unref_device] destroy device 5.3
[ 8.972223] [0000328e] libusb: debug [libusb_unref_device] destroy device 5.2
[ 8.972241] [0000328e] libusb: debug [libusb_unref_device] destroy device 5.1
[ 8.972264] [0000328e] libusb: debug [usbi_remove_pollfd] remove fd 7
[ 8.972310] [0000328e] libusb: debug [usbi_remove_pollfd] remove fd 9
[ 8.972420] [0000328f] libusb: debug [linux_udev_event_thread_main] udev event thread exiting
```

cheers

tormodvolden, 2021-11-08:
OK. I don't think libusb should do anything different here, it is just the "messenger". Heimdall must try to workaround this by not calling libusb_set_interface(), either when it seems superfluous (no other alternate interface) or when it can identify a flaky bootloader version (this might be difficult if the USB descriptors are identical).

closed 2021-11-08 by mcuee