libusb/libusb #1025: Undefined reference error

BandaniM, 2021-11-08:
Hello Team,

I have compiled the libusb library using clang compiler in vs2019. But I'm getting a linker error when calling the libusb API's.
below are the errors:

**lld-link: error: undefined symbol: libusb_init
>>> referenced by release\main.obj:(main)

lld-link: error: undefined symbol: libusb_set_debug
>>> referenced by release\main.obj:(main)

lld-link: error: undefined symbol: libusb_get_device_list
>>> referenced by release\main.obj:(main)

lld-link: error: undefined symbol: libusb_free_device_list
>>> referenced by release\main.obj:(main)

lld-link: error: undefined symbol: libusb_exit
>>> referenced by release\main.obj:(main)

lld-link: error: undefined symbol: libusb_get_device_descriptor
>>> referenced by release\main.obj:(void __cdecl printdev(struct libusb_device *))

lld-link: error: undefined symbol: libusb_get_config_descriptor
>>> referenced by release\main.obj:(void __cdecl printdev(struct libusb_device *))

lld-link: error: undefined symbol: libusb_free_config_descriptor
>>> referenced by release\main.obj:(void __cdecl printdev(struct libusb_device *))
jom: D:\Qt_QML_Training\build-qtusb-CLANG_MSVC-Release\Makefile.Release [release\qtusb.exe] Error 1
jom: D:\Qt_QML_Training\build-qtusb-CLANG_MSVC-Release\Makefile [release] Error 2**

Build environment: Windows
Qt IDE, Clang compiler

Kindly let me know, how we can resolve this issue.

Thanks

Best Regards
Bandani Maharana

mcuee, 2021-11-09:
For clang you have to build libusb from the source code. The current libusb Windows binary use either MinGW or Microsoft compiler.

closed 2021-11-09 by mcuee

BandaniM, 2021-11-09:
Yes, I have built the source code using the clang compiler in VS2019. and it got built successfully. 
but still getting these errors while compiling. In VS2019 I have set the compiler as llvm-clang-cl (refer below image).
Is there any other way to compile the libusb for clang kindly let me know?
[image]

mcuee, 2021-11-09:
I think you have to use the same Clang compiler inside your QT installation, not using VS2019. Basically you need to create your own project files using QT or use autoconf/automake with autogen.sh.

mcuee, 2021-11-09:
In any case, you may get better support in QT community, if you can not get it to work.