libusb/libusb #1027: windows - [libusb_cancel_transfer] cancel transfer failed error -5

P5-2005, 2021-11-21:
hello,
is that possible fix this issue? on mac or linux all good only on windows when sending transfert then cancling cannot handle it :
using zdiag 2.7 libusbK and libusb 1.24, here debug :

> C:\Users\BH11040\Downloads\libusb-1.0.24(1)\examples\bin32>xusb.exe -d 05ac:1227
Using libusb v1.0.24.11584

[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.000340] [00000658] libusb: debug [libusb_init] created default context
[ 0.000516] [00000658] libusb: debug [libusb_init] libusb v1.0.24.11584
[ 0.000801] [00000658] libusb: debug [get_windows_version] Windows 8 64-bit
[ 0.000966] [00000658] libusb: debug [htab_create] using 1021 entries hash table
[ 0.010712] [00000658] libusb: info [winusbx_init] WinUSB DLL available (with isoch support)
[ 0.011272] [00000658] libusb: debug [winusbx_init] libusbK DLL found, version: 3.1.0.0
[ 0.016593] [00000658] libusb: info [windows_init] UsbDk backend is not available
[ 0.016824] [00000658] libusb: debug [usbi_add_event_source] add HANDLE 000000C8 events 0
[ 0.016959] [00000658] libusb: debug [usbi_io_init] using timer for timeouts
[ 0.017057] [00000658] libusb: debug [usbi_add_event_source] add HANDLE 00000218 events 0
Opening device 05AC:1227...
[ 0.017218] [00000658] libusb: debug [libusb_get_device_list]
[ 0.017626] [00002bd0] libusb: debug [windows_iocp_thread] I/O completion thread started
[ 0.029878] [00000658] libusb: debug [winusb_get_device_list] allocating new device for session [45]
[ 0.030508] [00000658] libusb: debug [get_api_type] driver(s): libusbK
[ 0.030672] [00000658] libusb: debug [get_api_type] matched driver name against libusbK
[ 0.030794] [00000658] libusb: debug [winusb_get_device_list] allocating new device for session [2B]
[ 0.031052] [00000658] libusb: debug [get_api_type] driver(s): BTHUSB
[ 0.031215] [00000658] libusb: debug [get_api_type] lower filter driver(s): ibtusb
[ 0.031325] [00000658] libusb: debug [winusb_get_device_list] allocating new device for session [2D]
[ 0.031610] [00000658] libusb: debug [get_api_type] driver(s): HidUsb
[ 0.031759] [00000658] libusb: debug [get_api_type] matched driver name against HID API
[ 0.031871] [00000658] libusb: debug [winusb_get_device_list] allocating new device for session [32]
[ 0.032124] [00000658] libusb: debug [get_api_type] driver(s): usbccgp
[ 0.032278] [00000658] libusb: debug [get_api_type] matched driver name against Composite API
[ 0.032392] [00000658] libusb: debug [winusb_get_device_list] allocating new device for session [33]
[ 0.032644] [00000658] libusb: debug [get_api_type] driver(s): usbccgp
[ 0.032795] [00000658] libusb: debug [get_api_type] matched driver name against Composite API
[ 0.032904] [00000658] libusb: debug [winusb_get_device_list] allocating new device for session [34]
[ 0.033262] [00000658] libusb: debug [enumerate_hcd_root_hub] assigning HCD 'PCI\VEN_8086&DEV_9DED&SUBSYS_08F91028&REV_30\3&11583659&0&A0' bus number 1
[ 0.034404] [00000658] libusb: debug [winusb_get_device_list] extra GUID: {850CD1BD-F257-4EC5-8805-D9E89D1D88C4}
[ 0.034498] [00000658] libusb: debug [winusb_get_device_list] found existing device for session [2B]
[ 0.034968] [00000658] libusb: debug [init_device] found 1 configurations (current config: 1)
[ 0.035321] [00000658] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 25 bytes)
[ 0.035699] [00000658] libusb: debug [init_device] (bus: 1, addr: 34, depth: 1, port: 7): 'USB\VID_05AC&PID_1227\CPID:8950_CPRV:20_CPFM:03_SCEP:10_BDID:0E_ECID:000003373C131D89_IBFL:00_SRTG:[IBOOT-1145.3]'
[ 0.036386] [00000658] libusb: debug [init_root_hub] root hub 'USB\ROOT_HUB30\4&AA20E8B&0&0' reports 18 ports
[ 0.036542] [00000658] libusb: debug [init_device] (bus: 1, addr: 0, depth: 0, port: 0): 'USB\ROOT_HUB30\4&AA20E8B&0&0'
[ 0.037010] [00000658] libusb: debug [winusb_get_device_list] found existing device for session [33]
[ 0.037356] [00000658] libusb: debug [init_device] found 1 configurations (current config: 1)
[ 0.037751] [00000658] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 59 bytes)
[ 0.038160] [00000658] libusb: debug [init_device] (bus: 1, addr: 3, depth: 1, port: 4): 'USB\VID_413C&PID_2113\5&6B19F92&0&4'
[ 0.038766] [00000658] libusb: debug [winusb_get_device_list] found existing device for session [34]
[ 0.039010] [00000658] libusb: debug [init_device] found 1 configurations (current config: 1)
[ 0.039398] [00000658] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 641 bytes)
[ 0.039802] [00000658] libusb: debug [init_device] (bus: 1, addr: 4, depth: 1, port: 5): 'USB\VID_0BDA&PID_5687\0001'
[ 0.040372] [00000658] libusb: debug [winusb_get_device_list] found existing device for session [32]
[ 0.040712] [00000658] libusb: debug [init_device] found 1 configurations (current config: 1)
[ 0.041195] [00000658] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 34 bytes)
[ 0.041877] [00000658] libusb: debug [init_device] (bus: 1, addr: 1, depth: 1, port: 1): 'USB\VID_413C&PID_301A\5&6B19F92&0&1'
[ 0.042530] [00000658] libusb: debug [winusb_get_device_list] found existing device for session [2D]
[ 0.042819] [00000658] libusb: debug [init_device] found 1 configurations (current config: 1)
[ 0.043253] [00000658] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 200 bytes)
[ 0.047436] [00000658] libusb: debug [init_device] (bus: 1, addr: 6, depth: 1, port: 10): 'USB\VID_8087&PID_0AAA\5&6B19F92&0&10'
[ 0.048228] [00000658] libusb: debug [winusb_get_device_list] setting HID interface for [32]:
[ 0.048346] [00000658] libusb: debug [set_hid_interface] interface[0] = \\?\HID#VID_413C&PID_301A#6&34546951&0&0000#{4D1E55B2-F16F-11CF-88CB-001111000030}
[ 0.048886] [00000658] libusb: debug [winusb_get_device_list] setting composite interface for [33]:
[ 0.049209] [00000658] libusb: debug [set_composite_interface] interface[1] = \\?\HID#VID_413C&PID_2113&MI_01&COL02#7&C67EF75&0&0001#{4D1E55B2-F16F-11CF-88CB-001111000030}
[ 0.049715] [00000658] libusb: debug [winusb_get_device_list] setting composite interface for [33]:
[ 0.050069] [00000658] libusb: debug [set_composite_interface] interface[1] already set - ignoring HID collection: HID\VID_413C&PID_2113&MI_01&COL01\7&C67EF75&0&0000
[ 0.050547] [00000658] libusb: debug [winusb_get_device_list] setting composite interface for [33]:
[ 0.050932] [00000658] libusb: debug [set_composite_interface] interface[0] = \\?\HID#VID_413C&PID_2113&MI_00#7&303F2D37&0&0000#{4D1E55B2-F16F-11CF-88CB-001111000030}\KBD
[... 13 lines cut ...]
               S/N: 4
           VID:PID: 05AC:1227
         bcdDevice: 0000
   iMan:iProd:iSer: 2:3:4
          nb confs: 1

Reading BOS descriptor: [ 0.062903] [00000658] libusb: debug [libusb_alloc_transfer] transfer 01207AB8
[ 0.063312] [00000658] libusb: debug [libusb_submit_transfer] transfer 01207AB8
[ 0.063709] [00000658] libusb: debug [add_to_flying_list] arm timer for timeout in 1000ms (first in line)
[ 0.064168] [00000658] libusb: debug [libusb_claim_interface] interface 0
[ 0.064629] [00002bd0] libusb: debug [windows_iocp_thread] transfer 0113F750 completed, length 24
[ 0.064689] [00000658] libusb: debug [winusbx_claim_interface] claimed interface 0
[ 0.065498] [00000658] libusb: debug [windows_assign_endpoints] no endpoints found for interface 0
[ 0.065945] [00000658] libusb: debug [auto_claim] auto-claimed interface 0 for control request
[ 0.066506] [00000658] libusb: debug [winusbx_submit_control_transfer] will use interface 0
[ 0.066979] [00000658] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[ 0.067461] [00000658] libusb: debug [handle_events] event sources modified, reallocating event data
[ 0.067813] [00000658] libusb: debug [usbi_wait_for_events] WaitForMultipleObjects() for 2 HANDLEs with timeout in 60000ms
[ 1.241314] [00000658] libusb: debug [usbi_wait_for_events] WaitForMultipleObjects() returned 1
[ 1.241445] [00000658] libusb: debug [libusb_cancel_transfer] transfer 01207AB8
[ 1.243017] [00000658] libusb: debug [libusb_cancel_transfer] cancel transfer failed error -5
[ 1.247013] [00000658] libusb: warning [handle_timeout] async cancel failed -5
[ 1.247398] [00000658] libusb: debug [arm_timer_for_next_timeout] no timeouts, disarming timer
[ 1.247753] [00000658] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[ 1.248246] [00000658] libusb: debug [usbi_wait_for_events] WaitForMultipleObjects() for 2 HANDLEs with timeout in 60000ms

P5-2005, 2021-11-21:
this when using winusb : 
```
C:\Users\BH11040\Downloads\libusb-1.0.24(1)\examples\bin32>xusb.exe -d 05ac:1227
Using libusb v1.0.24.11584

[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.000284] [0000280c] libusb: debug [libusb_init] created default context
[ 0.000456] [0000280c] libusb: debug [libusb_init] libusb v1.0.24.11584
[ 0.000727] [0000280c] libusb: debug [get_windows_version] Windows 8 64-bit
[ 0.000886] [0000280c] libusb: debug [htab_create] using 1021 entries hash table
[ 0.010719] [0000280c] libusb: info [winusbx_init] WinUSB DLL available (with isoch support)
[ 0.011250] [0000280c] libusb: debug [winusbx_init] libusbK DLL found, version: 3.1.0.0
[ 0.016999] [0000280c] libusb: debug [windows_init] UsbDk backend is available
[ 0.017185] [0000280c] libusb: debug [usbi_add_event_source] add HANDLE 000000C8 events 0
[ 0.017283] [0000280c] libusb: debug [usbi_io_init] using timer for timeouts
[ 0.017349] [0000280c] libusb: debug [usbi_add_event_source] add HANDLE 00000228 events 0
Opening device 05AC:1227...
[ 0.017474] [0000280c] libusb: debug [libusb_get_device_list]
[ 0.017837] [00003520] libusb: debug [windows_iocp_thread] I/O completion thread started
[ 0.028141] [0000280c] libusb: debug [winusb_get_device_list] allocating new device for session [45]
[ 0.028782] [0000280c] libusb: debug [get_api_type] driver(s): WINUSB
[ 0.028888] [0000280c] libusb: debug [get_api_type] upper filter driver(s): WUDFRd;AppleKmdfFilter
[ 0.029012] [0000280c] libusb: debug [get_api_type] matched driver name against WinUSB
[ 0.029110] [0000280c] libusb: debug [winusb_get_device_list] allocating new device for session [26]
[ 0.029455] [0000280c] libusb: debug [get_api_type] driver(s): BTHUSB
[ 0.030154] [0000280c] libusb: debug [get_api_type] lower filter driver(s): ibtusb
[ 0.030304] [0000280c] libusb: debug [winusb_get_device_list] allocating new device for session [2D]
[ 0.030609] [0000280c] libusb: debug [get_api_type] driver(s): HidUsb
[ 0.030780] [0000280c] libusb: debug [get_api_type] matched driver name against HID API
[ 0.030916] [0000280c] libusb: debug [winusb_get_device_list] allocating new device for session [32]
[ 0.031180] [0000280c] libusb: debug [get_api_type] driver(s): usbccgp
[ 0.031345] [0000280c] libusb: debug [get_api_type] matched driver name against Composite API
[ 0.031787] [0000280c] libusb: debug [winusb_get_device_list] allocating new device for session [33]
[ 0.032410] [0000280c] libusb: debug [get_api_type] driver(s): usbccgp
[ 0.032736] [0000280c] libusb: debug [get_api_type] matched driver name against Composite API
[ 0.033129] [0000280c] libusb: debug [winusb_get_device_list] allocating new device for session [34]
[ 0.033893] [0000280c] libusb: debug [enumerate_hcd_root_hub] assigning HCD 'PCI\VEN_8086&DEV_9DED&SUBSYS_08F91028&REV_30\3&11583659&0&A0' bus number 1
[ 0.034913] [0000280c] libusb: debug [winusb_get_device_list] extra GUID: {664BE590-54BD-4964-8A8C-6CD1314F6DC2}
[ 0.035048] [0000280c] libusb: debug [winusb_get_device_list] found existing device for session [26]
[ 0.035529] [0000280c] libusb: debug [init_device] found 1 configurations (current config: 1)
[ 0.035881] [0000280c] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 25 bytes)
[ 0.036291] [0000280c] libusb: debug [init_device] (bus: 1, addr: 21, depth: 1, port: 7): 'USB\VID_05AC&PID_1227\CPID:8950_CPRV:20_CPFM:03_SCEP:10_BDID:0E_ECID:000003373C131D89_IBFL:00_SRTG:[IBOOT-1145.3]'
[ 0.037117] [0000280c] libusb: debug [init_root_hub] root hub 'USB\ROOT_HUB30\4&AA20E8B&0&0' reports 18 ports
[ 0.037401] [0000280c] libusb: debug [init_device] (bus: 1, addr: 0, depth: 0, port: 0): 'USB\ROOT_HUB30\4&AA20E8B&0&0'
[ 0.037930] [0000280c] libusb: debug [winusb_get_device_list] found existing device for session [33]
[ 0.038272] [0000280c] libusb: debug [init_device] found 1 configurations (current config: 1)
[ 0.038665] [0000280c] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 59 bytes)
[ 0.039079] [0000280c] libusb: debug [init_device] (bus: 1, addr: 18, depth: 1, port: 4): 'USB\VID_413C&PID_2113\5&6B19F92&0&4'
[ 0.039694] [0000280c] libusb: debug [winusb_get_device_list] found existing device for session [34]
[ 0.039937] [0000280c] libusb: debug [init_device] found 1 configurations (current config: 1)
[ 0.040353] [0000280c] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 641 bytes)
[ 0.040765] [0000280c] libusb: debug [init_device] (bus: 1, addr: 19, depth: 1, port: 5): 'USB\VID_0BDA&PID_5687\0001'
[ 0.041347] [0000280c] libusb: debug [winusb_get_device_list] found existing device for session [32]
[ 0.041746] [0000280c] libusb: debug [init_device] found 1 configurations (current config: 1)
[ 0.042113] [0000280c] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 34 bytes)
[ 0.043099] [0000280c] libusb: debug [init_device] (bus: 1, addr: 1, depth: 1, port: 1): 'USB\VID_413C&PID_301A\5&6B19F92&0&1'
[ 0.043794] [0000280c] libusb: debug [winusb_get_device_list] found existing device for session [2D]
[ 0.044107] [0000280c] libusb: debug [init_device] found 1 configurations (current config: 1)
[ 0.044556] [0000280c] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 200 bytes)
[... cut ...]
[ 0.058649] [0000280c] libusb: debug [libusb_unref_device] destroy device 1.5
[ 0.059269] [0000280c] libusb: debug [libusb_unref_device] destroy device 1.0
  Failed.
[ 0.063499] [0000280c] libusb: debug [libusb_exit]
[ 0.063954] [0000280c] libusb: debug [libusb_exit] destroying default context
[ 0.064395] [0000280c] libusb: debug [usbi_remove_event_source] remove HANDLE 00000228
[ 0.064837] [0000280c] libusb: debug [usbi_remove_event_source] remove HANDLE 000000C8
[ 0.065293] [00003520] libusb: debug [windows_iocp_thread] I/O completion thread exiting
```

P5-2005, 2021-11-21:
another new debug :

```
C:\Users\BH11040\Downloads\libusb-1.0.24(1)\examples\bin32>xusb.exe -d 05ac:1227
Using libusb v1.0.24.11584

[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.000204] [000031bc] libusb: debug [libusb_init] created default context
[ 0.000287] [000031bc] libusb: debug [libusb_init] libusb v1.0.24.11584
[ 0.000456] [000031bc] libusb: debug [get_windows_version] Windows 8 64-bit
[ 0.000562] [000031bc] libusb: debug [htab_create] using 1021 entries hash table
[ 0.006280] [000031bc] libusb: info [winusbx_init] WinUSB DLL available (with isoch support)
[ 0.009091] [000031bc] libusb: debug [winusbx_init] libusbK DLL found, version: 3.1.0.0
[ 0.013067] [000031bc] libusb: debug [windows_init] UsbDk backend is available
[ 0.013268] [000031bc] libusb: debug [usbi_add_event_source] add HANDLE 000000C8 events 0
[ 0.013340] [000031bc] libusb: debug [usbi_io_init] using timer for timeouts
[ 0.013390] [000031bc] libusb: debug [usbi_add_event_source] add HANDLE 00000234 events 0
Opening device 05AC:1227...
[ 0.013489] [000031bc] libusb: debug [libusb_get_device_list]
[ 0.013895] [000004f0] libusb: debug [windows_iocp_thread] I/O completion thread started
[ 0.023600] [000031bc] libusb: debug [winusb_get_device_list] allocating new device for session [45]
[ 0.024208] [000031bc] libusb: debug [get_api_type] driver(s): BTHUSB
[ 0.024401] [000031bc] libusb: debug [get_api_type] lower filter driver(s): ibtusb
[ 0.024545] [000031bc] libusb: debug [winusb_get_device_list] allocating new device for session [2C]
[ 0.024851] [000031bc] libusb: debug [get_api_type] driver(s): libusbK
[ 0.025031] [000031bc] libusb: debug [get_api_type] matched driver name against libusbK
[ 0.025162] [000031bc] libusb: debug [winusb_get_device_list] allocating new device for session [31]
[ 0.025417] [000031bc] libusb: debug [get_api_type] driver(s): HidUsb
[ 0.025593] [000031bc] libusb: debug [get_api_type] matched driver name against HID API
[ 0.025781] [000031bc] libusb: debug [winusb_get_device_list] allocating new device for session [32]
[ 0.026064] [000031bc] libusb: debug [get_api_type] driver(s): usbccgp
[ 0.026246] [000031bc] libusb: debug [get_api_type] matched driver name against Composite API
[ 0.026364] [000031bc] libusb: debug [winusb_get_device_list] allocating new device for session [33]
[ 0.026648] [000031bc] libusb: debug [get_api_type] driver(s): usbccgp
[ 0.026833] [000031bc] libusb: debug [get_api_type] matched driver name against Composite API
[ 0.026985] [000031bc] libusb: debug [winusb_get_device_list] allocating new device for session [34]
[ 0.027426] [000031bc] libusb: debug [enumerate_hcd_root_hub] assigning HCD 'PCI\VEN_8086&DEV_9DED&SUBSYS_08F91028&REV_30\3&11583659&0&A0' bus number 1
[ 0.028704] [000031bc] libusb: debug [winusb_get_device_list] extra GUID: {9A2D6953-6228-491D-97E1-687DFD65B862}
[ 0.028851] [000031bc] libusb: debug [winusb_get_device_list] found existing device for session [31]
[ 0.029419] [000031bc] libusb: debug [init_device] found 1 configurations (current config: 1)
[ 0.029918] [000031bc] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 25 bytes)
[ 0.030369] [000031bc] libusb: debug [init_device] (bus: 1, addr: 31, depth: 1, port: 7): 'USB\VID_05AC&PID_1227\CPID:8950_CPRV:20_CPFM:03_SCEP:10_BDID:0E_ECID:000003373C131D89_IBFL:00_SRTG:[IBOOT-1145.3]'
[ 0.031444] [000031bc] libusb: debug [init_root_hub] root hub 'USB\ROOT_HUB30\4&AA20E8B&0&0' reports 18 ports
[ 0.031727] [000031bc] libusb: debug [init_device] (bus: 1, addr: 0, depth: 0, port: 0): 'USB\ROOT_HUB30\4&AA20E8B&0&0'
[ 0.032338] [000031bc] libusb: debug [winusb_get_device_list] found existing device for session [33]
[ 0.032664] [000031bc] libusb: debug [init_device] found 1 configurations (current config: 1)
[ 0.033037] [000031bc] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 59 bytes)
[ 0.033447] [000031bc] libusb: debug [init_device] (bus: 1, addr: 18, depth: 1, port: 4): 'USB\VID_413C&PID_2113\5&6B19F92&0&4'
[ 0.034060] [000031bc] libusb: debug [winusb_get_device_list] found existing device for session [34]
[ 0.034281] [000031bc] libusb: debug [init_device] found 1 configurations (current config: 1)
[ 0.034642] [000031bc] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 641 bytes)
[ 0.035046] [000031bc] libusb: debug [init_device] (bus: 1, addr: 19, depth: 1, port: 5): 'USB\VID_0BDA&PID_5687\0001'
[ 0.035586] [000031bc] libusb: debug [winusb_get_device_list] found existing device for session [32]
[ 0.035923] [000031bc] libusb: debug [init_device] found 1 configurations (current config: 1)
[ 0.036292] [000031bc] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 34 bytes)
[ 0.036706] [000031bc] libusb: debug [init_device] (bus: 1, addr: 1, depth: 1, port: 1): 'USB\VID_413C&PID_301A\5&6B19F92&0&1'
[ 0.037224] [000031bc] libusb: debug [winusb_get_device_list] found existing device for session [2C]
[ 0.038510] [000031bc] libusb: debug [init_device] found 1 configurations (current config: 1)
[ 0.041773] [000031bc] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 200 bytes)
[ 0.043474] [000031bc] libusb: debug [init_device] (bus: 1, addr: 5, depth: 1, port: 10): 'USB\VID_8087&PID_0AAA\5&6B19F92&0&10'
[... cut ...]
[ 2.893792] [000031bc] libusb: debug [usbi_wait_for_events] WaitForMultipleObjects() returned 1
[ 2.894755] [000031bc] libusb: debug [libusb_cancel_transfer] transfer 01349550
[ 2.896146] [000031bc] libusb: debug [arm_timer_for_next_timeout] no timeouts, disarming timer
[ 2.896928] [000031bc] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[ 2.897719] [000031bc] libusb: debug [usbi_wait_for_events] WaitForMultipleObjects() for 2 HANDLEs with timeout in 60000ms
```

tormodvolden, 2021-11-21:
Can you please try latest git of libusb?

P5-2005, 2021-11-21:
> Can you please try latest git of libusb?

okay i will, thanks for the reply

P5-2005, 2021-11-21:
here for latest git :
```
C:\Users\BH11040\Downloads\libusb\x64\Debug\examples>xusb.exe -d 05ac:1227
Using libusb v1.0.24.11675

[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.000319] [00003a50] libusb: debug [libusb_init] created default context
[ 0.000517] [00003a50] libusb: debug [libusb_init] libusb v1.0.24.11675
[ 0.000659] [00003a50] libusb: debug [usbi_add_event_source] add HANDLE 0000000000000088 events 0
[ 0.000857] [00003a50] libusb: debug [usbi_io_init] using timer for timeouts
[ 0.001033] [00003a50] libusb: debug [usbi_add_event_source] add HANDLE 000000000000008C events 0
[ 0.001249] [00003a50] libusb: debug [get_windows_version] Windows 8 64-bit
[ 0.001416] [00003a50] libusb: debug [htab_create] using 1021 entries hash table
[ 0.007061] [00003a50] libusb: info [winusbx_init] WinUSB DLL available (with isoch support)
[ 0.026908] [00003a50] libusb: debug [winusbx_init] libusbK DLL found, version: 3.1.0.0
[ 0.029512] [00003a50] libusb: info [windows_init] UsbDk backend is not available
Opening device 05AC:1227...
[ 0.029779] [00003a50] libusb: debug [libusb_get_device_list]
[ 0.030158] [000008a8] libusb: debug [windows_iocp_thread] I/O completion thread started
[ 0.039026] [00003a50] libusb: debug [winusb_get_device_list] allocating new device for session [45]
[ 0.039842] [00003a50] libusb: debug [get_api_type] driver(s): BTHUSB
[ 0.040014] [00003a50] libusb: debug [get_api_type] lower filter driver(s): ibtusb
[ 0.040081] [00003a50] libusb: debug [winusb_get_device_list] allocating new device for session [2C]
[ 0.040439] [00003a50] libusb: debug [get_api_type] driver(s): libusbK
[ 0.040655] [00003a50] libusb: debug [get_api_type] matched driver name against libusbK
[ 0.040852] [00003a50] libusb: debug [winusb_get_device_list] allocating new device for session [31]
[ 0.041364] [00003a50] libusb: debug [get_api_type] driver(s): HidUsb
[ 0.041567] [00003a50] libusb: debug [get_api_type] matched driver name against HID API
[ 0.041735] [00003a50] libusb: debug [winusb_get_device_list] allocating new device for session [32]
[ 0.042843] [00003a50] libusb: debug [get_api_type] driver(s): usbccgp
[ 0.043024] [00003a50] libusb: debug [get_api_type] matched driver name against Composite API
[ 0.043166] [00003a50] libusb: debug [winusb_get_device_list] allocating new device for session [33]
[ 0.043446] [00003a50] libusb: debug [get_api_type] driver(s): usbccgp
[ 0.043616] [00003a50] libusb: debug [get_api_type] matched driver name against Composite API
[ 0.043727] [00003a50] libusb: debug [winusb_get_device_list] allocating new device for session [34]
[ 0.044106] [00003a50] libusb: debug [enumerate_hcd_root_hub] assigning HCD 'PCI\VEN_8086&DEV_9DED&SUBSYS_08F91028&REV_30\3&11583659&0&A0' bus number 1
[ 0.045316] [00003a50] libusb: debug [init_root_hub] root hub 'USB\ROOT_HUB30\4&AA20E8B&0&0' reports 18 ports
[ 0.045512] [00003a50] libusb: debug [init_device] (bus: 1, addr: 0, depth: 0, port: 0): 'USB\ROOT_HUB30\4&AA20E8B&0&0'
[ 0.046113] [00003a50] libusb: debug [winusb_get_device_list] found existing device for session [33]
[ 0.047082] [00003a50] libusb: debug [init_device] found 1 configurations (current config: 1)
[ 0.047491] [00003a50] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 59 bytes)
[ 0.047979] [00003a50] libusb: debug [init_device] (bus: 1, addr: 2, depth: 1, port: 4): 'USB\VID_413C&PID_2113\5&6B19F92&0&4'
[ 0.048775] [00003a50] libusb: debug [winusb_get_device_list] found existing device for session [34]
[ 0.049016] [00003a50] libusb: debug [init_device] found 1 configurations (current config: 1)
[ 0.049506] [00003a50] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 641 bytes)
[ 0.050003] [00003a50] libusb: debug [init_device] (bus: 1, addr: 3, depth: 1, port: 5): 'USB\VID_0BDA&PID_5687\0001'
[ 0.050646] [00003a50] libusb: debug [winusb_get_device_list] found existing device for session [32]
[ 0.051155] [00003a50] libusb: debug [init_device] found 1 configurations (current config: 1)
[ 0.051626] [00003a50] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 34 bytes)
[ 0.052071] [00003a50] libusb: debug [init_device] (bus: 1, addr: 1, depth: 1, port: 1): 'USB\VID_413C&PID_301A\5&6B19F92&0&1'
[ 0.052877] [00003a50] libusb: debug [winusb_get_device_list] extra GUID: {992A1A1C-3A42-435A-8A5D-D36C7E181DE9}
[ 0.054294] [00003a50] libusb: debug [winusb_get_device_list] found existing device for session [31]
[ 0.055019] [00003a50] libusb: debug [init_device] found 1 configurations (current config: 1)
[ 0.055438] [00003a50] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 25 bytes)
[ 0.055862] [00003a50] libusb: debug [init_device] (bus: 1, addr: 16, depth: 1, port: 7): 'USB\VID_05AC&PID_1227\CPID:7000_CPRV:11_CPFM:03_SCEP:01_BDID:06_ECID:001A746024D04826_IBFL:1C_SRTG:[IBOOT-1992.0.0.1.19]'
[ 0.056522] [00003a50] libusb: debug [winusb_get_device_list] found existing device for session [2C]
[ 0.056878] [00003a50] libusb: debug [init_device] found 1 configurations (current config: 1)
[ 0.057412] [00003a50] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 200 bytes)
[... cut ...]
[ 0.171898] [00003a50] libusb: debug [libusb_unref_device] destroy device 1.16
[ 0.172280] [00003a50] libusb: debug [libusb_unref_device] destroy device 1.0
[ 0.172699] [00003a50] libusb: debug [libusb_exit] destroying default context
[ 0.173499] [000008a8] libusb: debug [windows_iocp_thread] I/O completion thread exiting
[ 0.175141] [00003a50] libusb: debug [usbi_remove_event_source] remove HANDLE 000000000000008C
[ 0.175291] [00003a50] libusb: debug [usbi_remove_event_source] remove HANDLE 0000000000000088
```

tormodvolden, 2021-11-21:
Thanks. From what I can see, latest git works as it should. Hopefully 1.0.25 will be released soon.

tormodvolden, 2021-11-22:
I was still wondering what originally went wrong here so as to attribute a specific fix. I guess this was fixed in commit b51c743 although symptoms were different than #844 .

closed 2021-11-22 by tormodvolden

P5-2005, 2022-01-02:
> I was still wondering what originally went wrong here so as to attribute a specific fix. I guess this was fixed in commit [b51c743](https://github.com/libusb/libusb/commit/b51c743e4210756a98f4f60c69a34745e4b27a55) although symptoms were different than #844 .

`[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.000352] [00002810] libusb: debug [libusb_init] created default context
[ 0.000598] [00002810] libusb: debug [libusb_init] libusb v1.0.24.11584
[ 0.000810] [00002810] libusb: debug [get_windows_version] Windows 8 64-bit
[ 0.000958] [00002810] libusb: debug [htab_create] using 1021 entries hash table
[ 0.005061] [00002810] libusb: info [winusbx_init] WinUSB DLL available (with isoch support)
[ 0.005734] [00002810] libusb: debug [winusbx_init] libusbK DLL found, version: 3.0.7.0
[ 0.010522] [00002810] libusb: debug [windows_init] UsbDk backend is available
[ 0.010731] [00002810] libusb: debug [usbi_add_event_source] add HANDLE 0000000000000138 events 0
[ 0.010853] [00002810] libusb: debug [usbi_io_init] using timer for timeouts
[ 0.010953] [00002810] libusb: debug [usbi_add_event_source] add HANDLE 0000000000000248 events 0
[ 0.011066] [00002810] libusb: debug [libusb_get_device_list]
[ 0.011202] [00003600] libusb: debug [windows_iocp_thread] I/O completion thread started
[ 0.026069] [00002810] libusb: debug [winusb_get_device_list] allocating new device for session [4A]
[ 0.026669] [00002810] libusb: debug [get_api_type] driver(s): WinUSB
[ 0.026899] [00002810] libusb: debug [get_api_type] matched driver name against WinUSB
[ 0.027055] [00002810] libusb: debug [winusb_get_device_list] allocating new device for session [2B]
[ 0.027414] [00002810] libusb: debug [get_api_type] driver(s): BTHUSB
[ 0.027633] [00002810] libusb: debug [get_api_type] lower filter driver(s): ibtusb
[ 0.027814] [00002810] libusb: debug [winusb_get_device_list] allocating new device for session [2F]
[ 0.028155] [00002810] libusb: debug [get_api_type] driver(s): HidUsb
[ 0.028361] [00002810] libusb: debug [get_api_type] matched driver name against HID API
[ 0.028514] [00002810] libusb: debug [winusb_get_device_list] allocating new device for session [34]
[ 0.028832] [00002810] libusb: debug [get_api_type] driver(s): usbccgp
[ 0.029036] [00002810] libusb: debug [get_api_type] matched driver name against Composite API
[ 0.029178] [00002810] libusb: debug [winusb_get_device_list] allocating new device for session [35]
[ 0.029984] [00002810] libusb: debug [get_api_type] driver(s): usbccgp
[ 0.030269] [00002810] libusb: debug [get_api_type] matched driver name against Composite API
[ 0.030708] [00002810] libusb: debug [winusb_get_device_list] allocating new device for session [36]
[ 0.031484] [00002810] libusb: debug [get_api_type] driver(s): HidUsb
[ 0.031757] [00002810] libusb: debug [get_api_type] matched driver name against HID API
[ 0.032175] [00002810] libusb: debug [winusb_get_device_list] allocating new device for session [37]
[ 0.032999] [00002810] libusb: debug [enumerate_hcd_root_hub] assigning HCD 'PCI\VEN_8086&DEV_9DED&SUBSYS_08F91028&REV_30\3&11583659&0&A0' bus number 1
[ 0.034587] [00002810] libusb: debug [init_root_hub] root hub 'USB\ROOT_HUB30\4&AA20E8B&0&0' reports 18 ports
[ 0.034774] [00002810] libusb: debug [init_device] (bus: 1, addr: 0, depth: 0, port: 0): 'USB\ROOT_HUB30\4&AA20E8B&0&0'
[ 0.038952] [00002810] libusb: debug [winusb_get_device_list] found existing device for session [35]
[ 0.041951] [00002810] libusb: debug [init_device] found 1 configurations (current config: 1)
[ 0.042313] [00002810] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 59 bytes)
[ 0.042860] [00002810] libusb: debug [init_device] (bus: 1, addr: 2, depth: 1, port: 4): 'USB\VID_413C&PID_2113\5&6B19F92&0&4'
[ 0.043724] [00002810] libusb: debug [winusb_get_device_list] found existing device for session [36]
[ 0.043889] [00002810] libusb: debug [init_device] found 1 configurations (current config: 1)
[ 0.044385] [00002810] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 641 bytes)
[ 0.044949] [00002810] libusb: debug [init_device] (bus: 1, addr: 3, depth: 1, port: 5): 'USB\VID_0BDA&PID_5687\0001'
[ 0.045654] [00002810] libusb: debug [winusb_get_device_list] found existing device for session [34]
[ 0.046019] [00002810] libusb: debug [init_device] found 1 configurations (current config: 1)
[ 0.046482] [00002810] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 34 bytes)
[ 0.047014] [00002810] libusb: debug [init_device] (bus: 1, addr: 1, depth: 1, port: 1): 'USB\VID_413C&PID_301A\5&6B19F92&0&1'
[ 0.047672] [00002810] libusb: debug [winusb_get_device_list] extra GUID: {714D8708-00B2-4C8E-B358-9B39D0212BCC}
[ 0.048007] [00002810] libusb: debug [winusb_get_device_list] found existing device for session [2B]
[ 0.048587] [00002810] libusb: debug [init_device] found 1 configurations (current config: 1)
[ 0.049101] [00002810] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 25 bytes)
[ 0.049588] [00002810] libusb: debug [init_device] (bus: 1, addr: 59, depth: 1, port: 7): 'USB\VID_05AC&PID_1227\CPID:8010_CPRV:11_CPFM:03_SCEP:01_BDID:0C_ECID:000D288C083BCD26_IBFL:3C_SRTG:[IBOOT-2696.0.0.1.33]'
[... cut ...]
[ 0.188417] [00002810] libusb: debug [libusb_alloc_transfer] transfer 000002153EFE0C38
[ 0.188866] [00002810] libusb: debug [libusb_submit_transfer] transfer 000002153EFE0C38
[ 0.189413] [00002810] libusb: debug [winusbx_submit_control_transfer] will use interface 0
[ 0.190051] [00002810] libusb: debug [libusb_cancel_transfer] transfer 000002153EFE0C38
[ 0.190183] [00003600] libusb: debug [windows_iocp_thread] transfer 000002153EFE0C38 completed, length 192
[ 0.191045] [00002810] libusb: debug [libusb_cancel_transfer] cancel transfer failed error -5`

tormodvolden, 2022-01-02:
Thanks. Is this fully solved in latest git? There is also issue #1036 which looks similar, but only shows up sometimes due to a race condition.

P5-2005, 2022-01-02:
> Thanks. Is this fully solved in latest git? There is also issue #1036 which looks similar, but only shows up sometimes due to a race condition.

the problem from my end is if i build from vcpkg the dll work for me but always v1.0.24.11584
and if i build with vs2022 the build success but not working, i dont know what im doing wrong