libusb/libusb #1029: Zero-length packet in libusb_bulk_transfer

JonasVautherin, 2021-11-23:
I first posted it on [StackOverflow](https://stackoverflow.com/questions/70085203/bulk-transfer-sending-too-much-multiple-of-usb-packet), but I realize it may be more pertinent here.

### Problem I am trying to solve

I am sending data over usb with `libusb_bulk_transfer`, with something like this:

```c
int sent = 0;
int bulk_result = libusb_bulk_transfer(handle, endpoint_out->bEndpointAddress, buffer,
                                       buffer_len, &sent, 5000);
```

and I receive those transfers on the other side in Kotlin (Android).

Most of the time, it works: I send a buffer of side, say, 289 bytes, and on the other side I receive 289 bytes.

Sometimes, however, I receive too much. Say I send 1536 bytes, and I receive 1719 bytes.

### My solution that does not work

My understanding (e.g. from [here](https://beyondlogic.org/usbnutshell/usb4.shtml#Bulk)) is that "A bulk transfer is considered complete when it has transferred the exact amount of data requested, transferred a packet less than the maximum endpoint size, or transferred a zero-length packet".

And because 1536 is a multiple of 64 (and all the wrong packets I receive are multiples of 64), I thought that this was my issue. So I went for sending a zero-length packet after I send a buffer that is a multiple of the maximum endpoint size. And I duly noted that the maximum endpoint size is not necessarily 64, so I wanted to detect it.

Here is my "solution":

```c++
int sent = 0;
int bulk_result = libusb_bulk_transfer(handle, endpoint_out->bEndpointAddress, buffer,
                                       buffer_len, &sent, 5000);

if (sent % get_usb_packet_size() == 0) {
  libusb_bulk_transfer(handle, endpoint_out->bEndpointAddress, nullptr, 0, &sent, 5000);
}
```

With the simple `get_usb_packet_size()` below, which happens to be 256:

```c++
int get_usb_packet_size() { return endpoint_out->wMaxPacketSize; }
```

Still, that does not seem to work! The return code of both `libusb_bulk_transfer` is 0 (success), the first one says it sent `buffer_len` bytes (as expected), and the second one says it sent 0 bytes (as expected).

But my receiver still receives packets that are longer than what is expected. I tried using 64 instead of 256 (therefore sending more zero-length packets), but I still get that same problem.

What am I missing? If the return code of the zero-length transfer is 0 (success), it should mark the end of my bulk transfer, shouldn't it?

mcuee, 2021-11-24:
What is the host OS and libusb version?

mcuee, 2021-11-24:
For Windows you need to use the latest git and it is new.
https://github.com/libusb/libusb/issues/777#issue-694718327

For macOS not so sure if the following is applicable or not.
https://github.com/libusb/libusb/pull/678

JonasVautherin, 2021-11-24:
> What is the host OS and libusb version?

Ubuntu 20.04 with libusb 1.0.23

mcuee, 2021-11-24:
> Ubuntu 20.04 with libusb 1.0.23

Can you try latest libusb git? Please also provide the debug log with environment variable LIBUSB_DEBUG set to 4.

diabolo38, 2021-11-24:
your understanding is correct if you transfer is a multiple of the endpoint max packet size it cannot bound a transfer and may cause troubles like you seeing.
I'm not too sure that sending a 0 byte packet is actually supported an doing what you expect ? 

i would advice to avoid use of ZLP  "Zero Len packet"   feature of the backend or device but managed it on your own by adding a dummy byte if this can be detected easily by receiving side. That way you  avoid all possible issue due to some weak (bad or no )support of ZLP in host and devices s/w stack.

JonasVautherin, 2021-11-24:
Here is the output with LIBUSB_DEBUG=4 for one of those "bigger" packets that arrive with the wrong side.

Original packet size (as per the header of the packet): 21504
Received packet size: 21561
Excess: 21561 - 21504 = 57

Log output:

```
[49.912275] [000778d8] libusb: debug [reap_for_handle] urb type=3 status=0 transferred=48
[49.912280] [000778d8] libusb: debug [handle_bulk_completion] handling completion status 0 of bulk urb 1/1
[49.912284] [000778d8] libusb: debug [handle_bulk_completion] last URB in transfer --> complete!
[49.912289] [000778d8] libusb: debug [disarm_timerfd] 
[49.912294] [000778d8] libusb: debug [usbi_handle_transfer_completion] transfer 0x7f62d40013b0 has callback 0x7f62e2c1cfc0
[49.912298] [000778d8] libusb: debug [sync_transfer_cb] actual_length=48
[49.912306] [000778d8] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[49.912311] [000778d8] libusb: debug [handle_events] poll() 13 fds with timeout in 60000ms
[49.912346] [000778d9] libusb: debug [libusb_free_transfer] transfer 0x7f62d40013b0
[49.915045] [000778db] libusb: debug [libusb_alloc_transfer] transfer 0x7f62d05213e0
[49.915059] [000778db] libusb: debug [libusb_submit_transfer] transfer 0x7f62d05213e0
[49.915064] [000778db] libusb: debug [add_to_flying_list] arm timerfd for timeout in 5000ms (first in line)
[49.915071] [000778db] libusb: debug [submit_bulk_transfer] need 1 urbs for new transfer with length 21504
[49.915090] [000778db] libusb: debug [libusb_handle_events_timeout_completed] another thread is doing event handling
[49.915823] [000778d9] libusb: debug [libusb_alloc_transfer] transfer 0x7f62d40013b0
[49.915835] [000778d9] libusb: debug [libusb_submit_transfer] transfer 0x7f62d40013b0
[49.915839] [000778d9] libusb: debug [submit_bulk_transfer] need 1 urbs for new transfer with length 57
[49.915852] [000778d9] libusb: debug [libusb_handle_events_timeout_completed] another thread is doing event handling
[49.938303] [000778d8] libusb: debug [handle_events] poll() returned 1
[49.938330] [000778d8] libusb: debug [reap_for_handle] urb type=3 status=0 transferred=21504
[49.938337] [000778d8] libusb: debug [handle_bulk_completion] handling completion status 0 of bulk urb 1/1
[49.938342] [000778d8] libusb: debug [handle_bulk_completion] last URB in transfer --> complete!
[49.938348] [000778d8] libusb: debug [arm_timerfd_for_next_timeout] next timeout originally 5000ms
[49.938356] [000778d8] libusb: debug [usbi_handle_transfer_completion] transfer 0x7f62d05213e0 has callback 0x7f62e2c1cfc0
[49.938361] [000778d8] libusb: debug [sync_transfer_cb] actual_length=21504
[49.938368] [000778d8] libusb: debug [reap_for_handle] urb type=3 status=0 transferred=57
[49.938372] [000778d8] libusb: debug [handle_bulk_completion] handling completion status 0 of bulk urb 1/1
[49.938377] [000778d8] libusb: debug [handle_bulk_completion] last URB in transfer --> complete!
[49.938381] [000778d8] libusb: debug [disarm_timerfd] 
[49.938386] [000778d8] libusb: debug [usbi_handle_transfer_completion] transfer 0x7f62d40013b0 has callback 0x7f62e2c1cfc0
[49.938390] [000778d8] libusb: debug [sync_transfer_cb] actual_length=57
[49.938403] [000778d8] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[49.938409] [000778d8] libusb: debug [handle_events] poll() 13 fds with timeout in 60000ms
[49.938410] [000778db] libusb: debug [libusb_free_transfer] transfer 0x7f62d05213e0
[49.938414] [000778d9] libusb: debug [libusb_free_transfer] transfer 0x7f62d40013b0
------ Packet is a multiple of 256 (21504) sending empty packet!
[49.938449] [000778db] libusb: debug [libusb_alloc_transfer] transfer 0x7f62d05213e0
[49.938455] [000778db] libusb: debug [libusb_submit_transfer] transfer 0x7f62d05213e0
[49.938460] [000778db] libusb: debug [add_to_flying_list] arm timerfd for timeout in 5000ms (first in line)
[49.938467] [000778db] libusb: debug [submit_bulk_transfer] need 1 urbs for new transfer with length 0
[49.938480] [000778db] libusb: debug [libusb_handle_events_timeout_completed] another thread is doing event handling
[49.946919] [000778d9] libusb: debug [libusb_alloc_transfer] transfer 0x7f62d40013b0
[49.946943] [000778d9] libusb: debug [libusb_submit_transfer] transfer 0x7f62d40013b0
[49.946953] [000778d9] libusb: debug [submit_bulk_transfer] need 1 urbs for new transfer with length 52
[49.946977] [000778d9] libusb: debug [libusb_handle_events_timeout_completed] another thread is doing event handling
[49.958754] [000778d8] libusb: debug [handle_events] poll() returned 1
[49.958776] [000778d8] libusb: debug [reap_for_handle] urb type=3 status=0 transferred=0
[49.958780] [000778d8] libusb: debug [handle_bulk_completion] handling completion status 0 of bulk urb 1/1
[49.958784] [000778d8] libusb: debug [handle_bulk_completion] last URB in transfer --> complete!
[49.958789] [000778d8] libusb: debug [arm_timerfd_for_next_timeout] next timeout originally 5000ms
[49.958796] [000778d8] libusb: debug [usbi_handle_transfer_completion] transfer 0x7f62d05213e0 has callback 0x7f62e2c1cfc0
[49.958800] [000778d8] libusb: debug [sync_transfer_cb] actual_length=0
[49.958814] [000778d8] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[49.958821] [000778d8] libusb: debug [handle_events] poll() 13 fds with timeout in 60000ms
[49.958823] [000778d9] libusb: debug [libusb_handle_events_timeout_completed] another thread is doing event handling
[... cut ...]
48 - 16384 - 5177 - 52
```

Note that 16384 + 5177 = 21561, which is indeed the received packet size (after my receiver detects that 16384 is not the full packet and therefore appends the following 5177).

So it all makes sense to me: 21504 is a multiple of 256, the next packet sent is 57, and the next is 0, so 21561 are received. _Except that I want the zero-length to be sent before the 57 packet_.

Given my code above, I don't understand why the 57 packet gets sent between my 21504 and my 0 packets. :thinking:

JonasVautherin, 2021-11-24:
Oh... what about two threads are sending in parallel? That would explain it :sweat_smile: :sweat_smile: :sweat_smile:

JonasVautherin, 2021-11-24:
Yeah I think that was it. Two threads were calling my send function, so sometimes one thread would not manage to send the zero-length packet right after its first packet.

Sorry for the inconvenience, and thanks a lot for the help!

closed 2021-11-24 by JonasVautherin