libusb/libusb #1033: Windows: make error in MinGW

tempodat, 2021-12-22:
Attempting to build from master in MinGW for bfgminer, getting unexplained make error.

Shell Transcript: 
```
$ ./autogen.sh --disable-debug-log --prefix=/MinGW
(...)
$ make
make  all-recursive
make[1]: Entering directory `/home/user/libusb-master'
Making all in libusb
make[2]: Entering directory `/home/user/libusb-master/libusb'
  CC     core.lo
  CC     descriptor.lo
  CC     hotplug.lo
  CC     io.lo
  CC     strerror.lo
  CC     sync.lo
  CC     os/events_windows.lo
  CC     os/threads_windows.lo
  RC       libusb-1.0.lo
  CC     os/windows_common.lo
os/windows_common.c: In function 'windows_error_str':
os/windows_common.c:68:30: error: 'FACILITY_SETUPAPI' undeclared (first use in this function)
   error_code = 0x80000000 | (FACILITY_SETUPAPI << 16) | (error_code & 0x0000FFFF);
                              ^~~~~~~~~~~~~~~~~
os/windows_common.c:68:30: note: each undeclared identifier is reported only once for each function it appears in
os/windows_common.c: In function 'windows_cancel_transfer':
os/windows_common.c:767:6: error: implicit declaration of function 'CancelIoEx' [-Werror=implicit-function-declaration]
  if (CancelIoEx(transfer_priv->handle, &transfer_priv->overlapped))
      ^~~~~~~~~~
cc1.exe: some warnings being treated as errors
make[2]: *** [os/windows_common.lo] Error 1
make[2]: Leaving directory `/home/user/libusb-master/libusb'
make[1]: *** [all-recursive] Error 1
make[1]: Leaving directory `/home/user/libusb-master'
make: *** [all] Error 2
```

mcuee, 2021-12-23:
Which MinGW distribution and gcc version are you using? Take note MinGW.org toolchain is not supported (but patches may be accepted if it is deemed to be okay).

We will recommend MinGW-w64 based distribution like MSYS2.

mcuee, 2021-12-24:
I have just build libusb git head with MSYS2 MinGW 32bit compiler and there is no issue building libusb (except a few warnings).

mcuee, 2021-12-24:
I will close this issue for now. Please switch to MinGW-w64 based toolchain like MSYS2.

closed 2021-12-24 by mcuee

Fraouf, 2022-01-27:
Hi mcuee,

were you able to solve you issue? I have the same error.

Thanks.

mcuee, 2022-01-27:
> were you able to solve you issue? I have the same error.

As mentioned, do not use MinGW.org MinGW. Use MSYS2 MinGW-w64.

MSYS2: https://www.msys2.org/