libusb/libusb #1034: Compiler warning messages for windows_winusb.c

mcuee, 2021-12-24:
There are some compiler warning messages generated for windows_winusb.c when using MSYS2 GCC compiler.

```
$ gcc -v
Using built-in specs.
COLLECT_GCC=C:\msys64\mingw32\bin\gcc.exe
COLLECT_LTO_WRAPPER=C:/msys64/mingw32/bin/../lib/gcc/i686-w64-mingw32/11.2.0/lto-wrapper.exe
Target: i686-w64-mingw32
Configured with: ../gcc-11.2.0/configure --prefix=/mingw32 --with-local-prefix=/mingw32/local --build=i686-w64-mingw32 --host=i686-w64-mingw32 --target=i686-w64-mingw32 --with-native-system-header-dir=/mingw32/i686-w64-mingw32/include --libexecdir=/mingw32/lib --enable-bootstrap --enable-checking=release --with-arch=i686 --with-tune=generic --enable-languages=c,lto,c++,fortran,ada,objc,obj-c++,jit --enable-shared --enable-static --enable-libatomic --enable-threads=posix --enable-graphite --enable-fully-dynamic-string --enable-libstdcxx-filesystem-ts --enable-libstdcxx-time --disable-libstdcxx-pch --disable-libstdcxx-debug --enable-lto --enable-libgomp --disable-multilib --disable-rpath --disable-win32-registry --disable-nls --disable-werror --disable-symvers --with-libiconv --with-system-zlib --with-gmp=/mingw32 --with-mpfr=/mingw32 --with-mpc=/mingw32 --with-isl=/mingw32 --with-pkgversion='Rev5, Built by MSYS2 project' --with-bugurl=https://github.com/msys2/MINGW-packages/issues --with-gnu-as --with-gnu-ld --with-boot-ldflags='-pipe -Wl,--dynamicbase,--nxcompat,--no-seh -Wl,--large-address-aware -Wl,--disable-dynamicbase -static-libstdc++ -static-libgcc' 'LDFLAGS_FOR_TARGET=-pipe -Wl,--dynamicbase,--nxcompat,--no-seh -Wl,--large-address-aware' --enable-linker-plugin-flags='LDFLAGS=-static-libstdc++\ -static-libgcc\ -pipe\ -Wl,--dynamicbase,--nxcompat,--no-seh\ -Wl,--large-address-aware\ -Wl,--stack,12582912' --disable-sjlj-exceptions --with-dwarf2
Thread model: posix
Supported LTO compression algorithms: zlib zstd
gcc version 11.2.0 (Rev5, Built by MSYS2 project)

$ ./autogen.sh
autoreconf: export WARNINGS=
autoreconf: Entering directory '.'
autoreconf: configure.ac: not using Gettext
autoreconf: running: aclocal --force -I m4
autoreconf: configure.ac: tracing
autoreconf: running: libtoolize --copy --force
libtoolize: putting auxiliary files in '.'.
libtoolize: copying file './ltmain.sh'
libtoolize: putting macros in AC_CONFIG_MACRO_DIRS, 'm4'.
libtoolize: copying file 'm4/libtool.m4'
libtoolize: copying file 'm4/ltoptions.m4'
libtoolize: copying file 'm4/ltsugar.m4'
libtoolize: copying file 'm4/ltversion.m4'
libtoolize: copying file 'm4/lt~obsolete.m4'
autoreconf: configure.ac: not using Intltool
autoreconf: configure.ac: not using Gtkdoc
autoreconf: running: aclocal --force -I m4
autoreconf: running: /usr/bin/autoconf --force
autoreconf: running: /usr/bin/autoheader --force
autoreconf: running: automake --add-missing --copy --force-missing
configure.ac:23: installing './compile'
configure.ac:27: installing './config.guess'
configure.ac:27: installing './config.sub'
configure.ac:26: installing './install-sh'
configure.ac:26: installing './missing'
Makefile.am: installing './INSTALL'
examples/Makefile.am: installing './depcomp'
autoreconf: './config.sub' is updated
autoreconf: './config.guess' is updated
autoreconf: Leaving directory '.'
configure: loading site script /etc/config.site
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.exe
checking for suffix of executables... .exe
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether the compiler supports GNU C... yes
checking whether gcc accepts -g... yes
checking for gcc option to enable C11 features... none needed
checking whether gcc understands -c and -o together... yes
checking for g++... g++
checking whether the compiler supports GNU C++... yes
checking whether g++ accepts -g... yes
checking for g++ option to enable C++11 features... none needed
checking for inline... inline
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for a race-free mkdir -p... /usr/bin/mkdir -p
checking for gawk... gawk
checking whether make sets $(MAKE)... yes
checking whether make supports the include directive... yes (GNU style)
checking whether make supports nested variables... yes
checking dependency style of gcc... gcc3
checking dependency style of g++... gcc3
checking build system type... i686-w64-mingw32
[... 147 lines cut ...]
  CC       ezusb.o
  CC       fxload.o
  CCLD     fxload.exe
  CC       hotplugtest.o
  CCLD     hotplugtest.exe
  CC       listdevs.o
  CCLD     listdevs.exe
  CC       sam3u_benchmark.o
  CCLD     sam3u_benchmark.exe
  CC       testlibusb.o
  CCLD     testlibusb.exe
  CC       xusb.o
  CCLD     xusb.exe
make[2]: Leaving directory '/c/work/libusb/libusb/examples'
Making all in tests
make[2]: Entering directory '/c/work/libusb/libusb/tests'
  CC       stress.o
  CC       testlib.o
  CCLD     stress.exe
make[2]: Leaving directory '/c/work/libusb/libusb/tests'
make[2]: Entering directory '/c/work/libusb/libusb'
make[2]: Leaving directory '/c/work/libusb/libusb'
make[1]: Leaving directory '/c/work/libusb/libusb'

```

mcuee, 2022-01-21:
The warning messgaes are still present in libusb-1.0.25rc1.

mcuee, 2022-01-22:
This is not a regression though. libusb-1.0.24 will have the same warning messages generated.

And some think it is a GCC compiler bug. So I think we can leave it as it is.
Ref: https://gitlab.gnome.org/GNOME/gtk/-/issues/3776
https://sourceware.org/bugzilla/show_bug.cgi?id=26687

mcuee, 2022-01-22:
libusb-1.0.24 release has the same warnings (using the latest up-to-date MSYS2 MinGW-w64 compiler).

```
MINGW64 /c/work/libusb/libusb-1.0.24
$ make
make  all-recursive
make[1]: Entering directory '/c/work/libusb/libusb-1.0.24'
Making all in libusb
make[2]: Entering directory '/c/work/libusb/libusb-1.0.24/libusb'
  CC       core.lo
  CC       descriptor.lo
  CC       hotplug.lo
  CC       io.lo
  CC       strerror.lo
  CC       sync.lo
  CC       os/events_windows.lo
  CC       os/threads_windows.lo
  RC       libusb-1.0.lo
  CC       os/windows_common.lo
  CC       os/windows_usbdk.lo
  CC       os/windows_winusb.lo
os/windows_winusb.c: In function '_hid_get_descriptor.constprop':
os/windows_winusb.c:3260:21: warning: array subscript 'struct libusb_interface_descriptor[0]' is partly outside array bounds of 'char[41]' [-Warray-bounds]
 3260 |         id->bLength = LIBUSB_DT_INTERFACE_SIZE;
os/windows_winusb.c:3228:14: note: while referencing 'tmp'
 3228 |         char tmp[HID_MAX_CONFIG_DESC_SIZE];
      |              ^~~
os/windows_winusb.c:3261:29: warning: array subscript 'struct libusb_interface_descriptor[0]' is partly outside array bounds of 'char[41]' [-Warray-bounds]
 3261 |         id->bDescriptorType = LIBUSB_DT_INTERFACE;
      |         ~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~
os/windows_winusb.c:3228:14: note: while referencing 'tmp'
 3228 |         char tmp[HID_MAX_CONFIG_DESC_SIZE];
      |              ^~~
os/windows_winusb.c:3262:30: warning: array subscript 'struct libusb_interface_descriptor[0]' is partly outside array bounds of 'char[41]' [-Warray-bounds]
 3262 |         id->bInterfaceNumber = 0;
      |         ~~~~~~~~~~~~~~~~~~~~~^~~
os/windows_winusb.c:3228:14: note: while referencing 'tmp'
 3228 |         char tmp[HID_MAX_CONFIG_DESC_SIZE];
      |              ^~~
os/windows_winusb.c:3263:31: warning: array subscript 'struct libusb_interface_descriptor[0]' is partly outside array bounds of 'char[41]' [-Warray-bounds]
 3263 |         id->bAlternateSetting = 0;
      |         ~~~~~~~~~~~~~~~~~~~~~~^~~
os/windows_winusb.c:3228:14: note: while referencing 'tmp'
 3228 |         char tmp[HID_MAX_CONFIG_DESC_SIZE];
      |              ^~~
os/windows_winusb.c:3264:27: warning: array subscript 'struct libusb_interface_descriptor[0]' is partly outside array bounds of 'char[41]' [-Warray-bounds]
 3264 |         id->bNumEndpoints = num_endpoints;
      |         ~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~
os/windows_winusb.c:3228:14: note: while referencing 'tmp'
 3228 |         char tmp[HID_MAX_CONFIG_DESC_SIZE];
      |              ^~~
os/windows_winusb.c:3265:29: warning: array subscript 'struct libusb_interface_descriptor[0]' is partly outside array bounds of 'char[41]' [-Warray-bounds]
 3265 |         id->bInterfaceClass = 3;
      |         ~~~~~~~~~~~~~~~~~~~~^~~
os/windows_winusb.c:3228:14: note: while referencing 'tmp'
 3228 |         char tmp[HID_MAX_CONFIG_DESC_SIZE];
      |              ^~~
os/windows_winusb.c:3266:32: warning: array subscript 'struct libusb_interface_descriptor[0]' is partly outside array bounds of 'char[41]' [-Warray-bounds]
 3266 |         id->bInterfaceSubClass = 0;
      |         ~~~~~~~~~~~~~~~~~~~~~~~^~~
os/windows_winusb.c:3228:14: note: while referencing 'tmp'
 3228 |         char tmp[HID_MAX_CONFIG_DESC_SIZE];
      |              ^~~
os/windows_winusb.c:3267:32: warning: array subscript 'struct libusb_interface_descriptor[0]' is partly outside array bounds of 'char[41]' [-Warray-bounds]
 3267 |         id->bInterfaceProtocol = 0;
      |         ~~~~~~~~~~~~~~~~~~~~~~~^~~
os/windows_winusb.c:3228:14: note: while referencing 'tmp'
 3228 |         char tmp[HID_MAX_CONFIG_DESC_SIZE];
      |              ^~~
os/windows_winusb.c:3268:24: warning: array subscript 'struct libusb_interface_descriptor[0]' is partly outside array bounds of 'char[41]' [-Warray-bounds]
[... 58 lines cut ...]
  CCLD     testlibusb.exe
  CC       xusb.o
  CCLD     xusb.exe
make[2]: Leaving directory '/c/work/libusb/libusb-1.0.24/examples'
Making all in tests
make[2]: Entering directory '/c/work/libusb/libusb-1.0.24/tests'
  CC       stress.o
  CC       testlib.o
  CCLD     stress.exe
make[2]: Leaving directory '/c/work/libusb/libusb-1.0.24/tests'
make[2]: Entering directory '/c/work/libusb/libusb-1.0.24'
make[2]: Leaving directory '/c/work/libusb/libusb-1.0.24'
make[1]: Leaving directory '/c/work/libusb/libusb-1.0.24'

MINGW64 /c/work/libusb/libusb-1.0.24
$ gcc -v
Using built-in specs.
COLLECT_GCC=C:\msys64\mingw64\bin\gcc.exe
COLLECT_LTO_WRAPPER=C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/11.2.0/lto-wrapper.exe
Target: x86_64-w64-mingw32
Configured with: ../gcc-11.2.0/configure --prefix=/mingw64 --with-local-prefix=/mingw64/local --build=x86_64-w64-mingw32 --host=x86_64-w64-mingw32 --target=x86_64-w64-mingw32 --with-native-system-header-dir=/mingw64/x86_64-w64-mingw32/include --libexecdir=/mingw64/lib --enable-bootstrap --enable-checking=release --with-arch=x86-64 --with-tune=generic --enable-languages=c,lto,c++,fortran,ada,objc,obj-c++,jit --enable-shared --enable-static --enable-libatomic --enable-threads=posix --enable-graphite --enable-fully-dynamic-string --enable-libstdcxx-filesystem-ts --enable-libstdcxx-time --disable-libstdcxx-pch --disable-libstdcxx-debug --enable-lto --enable-libgomp --disable-multilib --disable-rpath --disable-win32-registry --disable-nls --disable-werror --disable-symvers --with-libiconv --with-system-zlib --with-gmp=/mingw64 --with-mpfr= [... cut ...] py-va,--nxcompat,--default-image-base-high -Wl,--disable-dynamicbase -static-libstdc++ -static-libgcc' 'LDFLAGS_FOR_TARGET=-pipe -Wl,--dynamicbase,--high-entropy-va,--nxcompat,--default-image-base-high' --enable-linker-plugin-flags='LDFLAGS=-static-libstdc++\ -static-libgcc\ -pipe\ -Wl,--dynamicbase,--high-entropy-va,--nxcompat,--default-image-base-high\ -Wl,--stack,12582912'
Thread model: posix
Supported LTO compression algorithms: zlib zstd
gcc version 11.2.0 (Rev7, Built by MSYS2 project)
```

mcuee, 2023-01-22:
This issue has not been fixed in latest git HEAD.

```
$ ./autogen.sh
autoreconf-2.71: export WARNINGS=
autoreconf-2.71: Entering directory '.'
autoreconf-2.71: configure.ac: not using Gettext
autoreconf-2.71: running: aclocal --force -I m4
autoreconf-2.71: configure.ac: tracing
autoreconf-2.71: running: libtoolize --copy --force
libtoolize: putting auxiliary files in '.'.
libtoolize: copying file './ltmain.sh'
libtoolize: putting macros in AC_CONFIG_MACRO_DIRS, 'm4'.
libtoolize: copying file 'm4/libtool.m4'
libtoolize: copying file 'm4/ltoptions.m4'
libtoolize: copying file 'm4/ltsugar.m4'
libtoolize: copying file 'm4/ltversion.m4'
libtoolize: copying file 'm4/lt~obsolete.m4'
autoreconf-2.71: configure.ac: not using Intltool
autoreconf-2.71: configure.ac: not using Gtkdoc
autoreconf-2.71: running: aclocal --force -I m4
autoreconf-2.71: running: /usr/bin/autoconf-2.71 --force
autoreconf-2.71: running: /usr/bin/autoheader-2.71 --force
autoreconf-2.71: running: automake --add-missing --copy --force-missing
configure.ac:23: installing './compile'
configure.ac:26: installing './missing'
examples/Makefile.am: installing './depcomp'
autoreconf-2.71: Leaving directory '.'
configure: loading site script /etc/config.site
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.exe
checking for suffix of executables... .exe
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether the compiler supports GNU C... yes
checking whether gcc accepts -g... yes
checking for gcc option to enable C11 features... none needed
checking whether gcc understands -c and -o together... yes
checking for g++... g++
checking whether the compiler supports GNU C++... yes
checking whether g++ accepts -g... yes
checking for g++ option to enable C++11 features... none needed
checking for inline... inline
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for a race-free mkdir -p... /usr/bin/mkdir -p
checking for gawk... gawk
checking whether make sets $(MAKE)... yes
checking whether make supports the include directive... yes (GNU style)
checking whether make supports nested variables... yes
checking dependency style of gcc... gcc3
checking dependency style of g++... gcc3
checking build system type... x86_64-w64-mingw32
checking host system type... x86_64-w64-mingw32
checking how to print strings... printf
checking for a sed that does not truncate output... /usr/bin/sed
checking for grep that handles long lines and -e... /usr/bin/grep
checking for egrep... /usr/bin/grep -E
checking for fgrep... /usr/bin/grep -F
checking for ld used by gcc... C:/msys64/mingw64/x86_64-w64-mingw32/bin/ld.exe
checking if the linker (C:/msys64/mingw64/x86_64-w64-mingw32/bin/ld.exe) is GNU ld... yes
checking for BSD- or MS-compatible name lister (nm)... /mingw64/bin/nm -B
checking the name lister (/mingw64/bin/nm -B) interface... BSD nm
checking whether ln -s works... no, using cp -pR
checking the maximum length of command line arguments... 8192
checking how to convert x86_64-w64-mingw32 file names to x86_64-w64-mingw32 format... func_convert_file_msys_to_w32
checking how to convert x86_64-w64-mingw32 file names to toolchain format... func_convert_file_msys_to_w32
checking for C:/msys64/mingw64/x86_64-w64-mingw32/bin/ld.exe option to reload object files... -r
checking for file... file
[... 288 lines cut ...]
  CC       ezusb.o
  CC       fxload.o
  CCLD     fxload.exe
  CC       hotplugtest.o
  CCLD     hotplugtest.exe
  CC       listdevs.o
  CCLD     listdevs.exe
  CC       sam3u_benchmark.o
  CCLD     sam3u_benchmark.exe
  CC       testlibusb.o
  CCLD     testlibusb.exe
  CC       xusb.o
  CCLD     xusb.exe
make[2]: Leaving directory '/c/work/libusb/libusb/examples'
Making all in tests
make[2]: Entering directory '/c/work/libusb/libusb/tests'
  CC       stress.o
  CC       testlib.o
  CCLD     stress.exe
make[2]: Leaving directory '/c/work/libusb/libusb/tests'
make[2]: Entering directory '/c/work/libusb/libusb'
make[2]: Leaving directory '/c/work/libusb/libusb'
make[1]: Leaving directory '/c/work/libusb/libusb'

```

mcuee, 2023-12-21:
Just want to mention that the issue is still there in git HEAD. Please refer to the github action mingw64 build log
https://github.com/libusb/libusb/actions/runs/7255523522/job/19766296565

mcuee, 2023-12-26:
This issue is not a blocker for libusb-1.0.27 as it is also there in libusb-1.0.26 (not a regression).

mcuee, 2025-06-08:
@sonatique 
Just want to mention that this warning is still there in our CI build. Please take a look if it is worth fixing.
https://github.com/libusb/libusb/actions/runs/15379068006/job/43267492186

```
../../libusb/os/windows_winusb.c: In function 'init_root_hub':
../../libusb/os/windows_winusb.c:1012:9: warning: enumeration value 'LIBUSB_SPEED_SUPER_PLUS_X2' not handled in switch [-Wswitch-enum]
 1012 |         switch (speed) {
      |         ^~~~~~
In function '_hid_get_config_descriptor',
    inlined from '_hid_get_descriptor.isra' at ../../libusb/os/windows_winusb.c:3770:11:
../../libusb/os/windows_winusb.c:3583:21: warning: array subscript 'struct libusb_interface_descriptor[0]' is partly outside array bounds of 'char[41]' [-Warray-bounds=]
 3583 |         id->bLength = LIBUSB_DT_INTERFACE_SIZE;
../../libusb/os/windows_winusb.c: In function '_hid_get_descriptor.isra':
../../libusb/os/windows_winusb.c:3551:14: note: at offset 9 into object 'tmp' of size 41
../../libusb/os/windows_winusb.c:3551:14: note: at offset [27, 41] into object 'tmp' of size 41
 3551 |         char tmp[HID_MAX_CONFIG_DESC_SIZE];
      |              ^~~
../../libusb/os/windows_winusb.c:3551:14: note: at offset [34, 41] into object 'tmp' of size 41
In function '_hid_get_config_descriptor',
    inlined from '_hid_get_descriptor.isra' at ../../libusb/os/windows_winusb.c:3770:11:
../../libusb/os/windows_winusb.c:3610:34: warning: array subscript 'struct libusb_endpoint_descriptor[0]' is partly outside array bounds of 'struct libusb_endpoint_descriptor[1]' [-Warray-bounds=]
 3610 |                 ed->bmAttributes = 3;
      |                 ~~~~~~~~~~~~~~~~~^~~
../../libusb/os/windows_winusb.c: In function '_hid_get_descriptor.isra':
../../libusb/os/windows_winusb.c:3551:14: note: at offset [27, 41] into object 'tmp' of size 41
 3551 |         char tmp[HID_MAX_CONFIG_DESC_SIZE];
      |              ^~~
../../libusb/os/windows_winusb.c:3551:14: note: at offset [34, 41] into object 'tmp' of size 41
In function '_hid_get_config_descriptor',
    inlined from '_hid_get_descriptor.isra' at ../../libusb/os/windows_winusb.c:3770:11:
../../libusb/os/windows_winusb.c:3611:36: warning: array subscript 'struct libusb_endpoint_descriptor[0]' is partly outside array bounds of 'struct libusb_endpoint_descriptor[1]' [-Warray-bounds=]
 3611 |                 ed->wMaxPacketSize = hid_priv->output_report_size - 1;
      |                 ~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
../../libusb/os/windows_winusb.c: In function '_hid_get_descriptor.isra':
../../libusb/os/windows_winusb.c:3551:14: note: at offset [27, 41] into object 'tmp' of size 41
 3551 |         char tmp[HID_MAX_CONFIG_DESC_SIZE];
      |              ^~~
../../libusb/os/windows_winusb.c:3551:14: note: at offset [34, 41] into object 'tmp' of size 41
In function '_hid_get_config_descriptor',
    inlined from '_hid_get_descriptor.isra' at ../../libusb/os/windows_winusb.c:3770:11:
../../libusb/os/windows_winusb.c:3612:31: warning: array subscript 'struct libusb_endpoint_descriptor[0]' is partly outside array bounds of 'struct libusb_endpoint_descriptor[1]' [-Warray-bounds=]
 3612 |                 ed->bInterval = 10;
      |                 ~~~~~~~~~~~~~~^~~~
../../libusb/os/windows_winusb.c: In function '_hid_get_descriptor.isra':
../../libusb/os/windows_winusb.c:3551:14: note: at offset [27, 41] into object 'tmp' of size 41
 3551 |         char tmp[HID_MAX_CONFIG_DESC_SIZE];
      |              ^~~
../../libusb/os/windows_winusb.c:3551:14: note: at offset [34, 41] into object 'tmp' of size 41
```

mcuee, 2025-06-08:
@sonatique 
MSYS2 clang64 got different warnings.
https://github.com/libusb/libusb/actions/runs/15379068004/job/43267492179

```
In file included from ../../libusb/os/events_windows.c:24:
../../libusb/os/windows_common.h:163:10: warning: the current #pragma pack alignment value is modified in the included file [-Wpragma-pack]
  163 | #include <pshpack1.h>
      |          ^
D:/a/_temp/msys64/clang64/include/pshpack1.h:7:9: note: previous '#pragma pack' directive that modifies alignment is here
    7 | #pragma pack(push,1)
      |         ^
In file included from ../../libusb/os/events_windows.c:24:
../../libusb/os/windows_common.h:193:10: warning: the current #pragma pack alignment value is modified in the included file [-Wpragma-pack]
  193 | #include <poppack.h>
      |          ^
note: previous '#pragma pack' directive that modifies alignment is here
2 warnings generated.
  CC       os/windows_common.lo
  CC       os/windows_usbdk.lo
  CC       os/windows_winusb.lo
In file included from ../../libusb/os/windows_common.c:30:
../../libusb/os/windows_common.h:163:10: warning: the current #pragma pack alignment value is modified in the included file [-Wpragma-pack]
  163 | #include <pshpack1.h>
      |          ^
D:/a/_temp/msys64/clang64/include/pshpack1.h:7:9: note: previous '#pragma pack' directive that modifies alignment is here
    7 | #pragma pack(push,1)
      |         ^
In file included from ../../libusb/os/windows_common.c:30:
../../libusb/os/windows_common.h:[193](https://github.com/libusb/libusb/actions/runs/15379068004/job/43267492179#step:4:194):10: warning: the current #pragma pack alignment value is modified in the included file [-Wpragma-pack]
  193 | #include <poppack.h>
      |          ^
note: previous '#pragma pack' directive that modifies alignment is here
2 warnings generated.
In file included from ../../libusb/os/windows_usbdk.c:30:
In file included from ../../libusb/os/windows_usbdk.h:27:
../../libusb/os/windows_common.h:163:10: warning: the current #pragma pack alignment value is modified in the included file [-Wpragma-pack]
  163 | #include <pshpack1.h>
      |          ^
D:/a/_temp/msys64/clang64/include/pshpack1.h:7:9: note: previous '#pragma pack' directive that modifies alignment is here
    7 | #pragma pack(push,1)
      |         ^
In file included from ../../libusb/os/windows_usbdk.c:30:
In file included from ../../libusb/os/windows_usbdk.h:27:
../../libusb/os/windows_common.h:193:10: warning: the current #pragma pack alignment value is modified in the included file [-Wpragma-pack]
  193 | #include <poppack.h>
      |          ^
note: previous '#pragma pack' directive that modifies alignment is here
2 warnings generated.
In file included from ../../libusb/os/windows_winusb.c:35:
In file included from ../../libusb/os/windows_winusb.h:30:
../../libusb/os/windows_common.h:163:10: warning: the current #pragma pack alignment value is modified in the included file [-Wpragma-pack]
  163 | #include <pshpack1.h>
      |          ^
D:/a/_temp/msys64/clang64/include/pshpack1.h:7:9: note: previous '#pragma pack' directive that modifies alignment is here
    7 | #pragma pack(push,1)
      |         ^
In file included from ../../libusb/os/windows_winusb.c:35:
In file included from ../../libusb/os/windows_winusb.h:30:
../../libusb/os/windows_common.h:193:10: warning: the current #pragma pack alignment value is modified in the included file [-Wpragma-pack]
  193 | #include <poppack.h>
      |          ^
note: previous '#pragma pack' directive that modifies alignment is here
In file included from ../../libusb/os/windows_winusb.c:35:
../../libusb/os/windows_winusb.h:310:10: warning: the current #pragma pack alignment value is modified in the included file [-Wpragma-pack]
  310 | #include <pshpack1.h>
      |          ^
D:/a/_temp/msys64/clang64/include/pshpack1.h:7:9: note: previous '#pragma pack' directive that modifies alignment is here
    7 | #pragma pack(push,1)
      |         ^
In file included from ../../libusb/os/windows_winusb.c:35:
../../libusb/os/windows_winusb.h:404:10: warning: the current #pragma pack alignment value is modified in the included file [-Wpragma-pack]
  404 | #include <poppack.h>
      |          ^
note: previous '#pragma pack' directive that modifies alignment is here
../../libusb/os/windows_winusb.h:446:10: warning: the current #pragma pack alignment value is modified in the included file [-Wpragma-pack]
  446 | #include <pshpack1.h>
      |          ^
D:/a/_temp/msys64/clang64/include/pshpack1.h:7:9: note: previous '#pragma pack' directive that modifies alignment is here
    7 | #pragma pack(push,1)
      |         ^
In file included from ../../libusb/os/windows_winusb.c:35:
../../libusb/os/windows_winusb.h:456:10: warning: the current #pragma pack alignment value is modified in the included file [-Wpragma-pack]
  456 | #include <poppack.h>
      |          ^
note: previous '#pragma pack' directive that modifies alignment is here
../../libusb/os/windows_winusb.h:698:10: warning: the current #pragma pack alignment value is modified in the included file [-Wpragma-pack]
  698 | #include <pshpack1.h>
      |          ^
D:/a/_temp/msys64/clang64/include/pshpack1.h:7:9: note: previous '#pragma pack' directive that modifies alignment is here
    7 | #pragma pack(push,1)
      |         ^
In file included from ../../libusb/os/windows_winusb.c:35:
../../libusb/os/windows_winusb.h:707:10: warning: the current #pragma pack alignment value is modified in the included file [-Wpragma-pack]
  707 | #include <poppack.h>
      |          ^
note: previous '#pragma pack' directive that modifies alignment is here
../../libusb/os/windows_winusb.c:1012:10: warning: enumeration value 'LIBUSB_SPEED_SUPER_PLUS_X2' not explicitly handled in switch [-Wswitch-enum]
 1012 |         switch (speed) {
      |                 ^~~~~
9 warnings generated.
```

seanm, 2025-06-10:
> ../../libusb/os/windows_winusb.c:1012:9: warning: enumeration value 'LIBUSB_SPEED_SUPER_PLUS_X2' not handled in switch [-Wswitch-enum]

I am a fan of that warning personally, I suggest fixing them.

Building on Mac, there are no -Wswitch-enum warnings anywhere. But of course windows_winusb.c is not built on Mac.

> ../../libusb/os/windows_winusb.c:3583:21: warning: array subscript 'struct libusb_interface_descriptor[0]' is partly outside array bounds of 'char[41]' [-Warray-bounds=]

Ugh, that code is super sketchy. Those might be true positives, I'll investigate...

seanm, 2025-06-10:
This code is doing sketchy pointer arithmetic, seems to be assuming structure packing, and/or seems to be assuming various struct sizes match constants like `LIBUSB_DT_CONFIG_SIZE`.

I suspect the warning is a true positive and the stack is being overrun.

I don't have Windows though, so not easy for me to touch that code as I can't test.

mcuee, 2025-09-07:
Windows MSYS2 clang64 compiler warnings will be dealt in another issue (-Wpragma-pack).
* https://github.com/libusb/libusb/issues/1395

mcuee, 2025-09-07:
Interestingly I do not see the issue of `-Wswitch-enum` on my up-to-day Windows 11 24H2 MSYS2 ucrt64 gcc build.

```
UCRT64 /c/work/libusb/libusb/build_ucrt64
$ make
make  all-recursive
make[1]: Entering directory '/c/work/libusb/libusb/build_ucrt64'
Making all in libusb
make[2]: Entering directory '/c/work/libusb/libusb/build_ucrt64/libusb'
  CC       core.lo
  CC       descriptor.lo
  CC       hotplug.lo
  CC       io.lo
  CC       strerror.lo
  CC       sync.lo
  CC       os/events_windows.lo
  CC       os/threads_windows.lo
  RC       libusb-1.0.lo
  CC       os/windows_common.lo
  CC       os/windows_usbdk.lo
  CC       os/windows_winusb.lo
In function '_hid_get_config_descriptor',
    inlined from '_hid_get_descriptor.isra' at ../../libusb/os/windows_winusb.c:4265:11:
../../libusb/os/windows_winusb.c:4078:21: warning: array subscript 'struct libusb_interface_descriptor[0]' is partly outside array bounds of 'char[41]' [-Warray-bounds=]
 4078 |         id->bLength = LIBUSB_DT_INTERFACE_SIZE;
../../libusb/os/windows_winusb.c: In function '_hid_get_descriptor.isra':
../../libusb/os/windows_winusb.c:4046:14: note: at offset 9 into object 'tmp' of size 41
 4046 |         char tmp[HID_MAX_CONFIG_DESC_SIZE];
      |              ^~~
In function '_hid_get_config_descriptor',
    inlined from '_hid_get_descriptor.isra' at ../../libusb/os/windows_winusb.c:4265:11:
../../libusb/os/windows_winusb.c:4079:29: warning: array subscript 'struct libusb_interface_descriptor[0]' is partly outside array bounds of 'char[41]' [-Warray-bounds=]
 4079 |         id->bDescriptorType = LIBUSB_DT_INTERFACE;
      |         ~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~
../../libusb/os/windows_winusb.c: In function '_hid_get_descriptor.isra':
../../libusb/os/windows_winusb.c:4046:14: note: at offset 9 into object 'tmp' of size 41
 4046 |         char tmp[HID_MAX_CONFIG_DESC_SIZE];
      |              ^~~
In function '_hid_get_config_descriptor',
    inlined from '_hid_get_descriptor.isra' at ../../libusb/os/windows_winusb.c:4265:11:
../../libusb/os/windows_winusb.c:4080:30: warning: array subscript 'struct libusb_interface_descriptor[0]' is partly outside array bounds of 'char[41]' [-Warray-bounds=]
 4080 |         id->bInterfaceNumber = 0;
      |         ~~~~~~~~~~~~~~~~~~~~~^~~
../../libusb/os/windows_winusb.c: In function '_hid_get_descriptor.isra':
../../libusb/os/windows_winusb.c:4046:14: note: at offset 9 into object 'tmp' of size 41
 4046 |         char tmp[HID_MAX_CONFIG_DESC_SIZE];
      |              ^~~
In function '_hid_get_config_descriptor',
    inlined from '_hid_get_descriptor.isra' at ../../libusb/os/windows_winusb.c:4265:11:
../../libusb/os/windows_winusb.c:4081:31: warning: array subscript 'struct libusb_interface_descriptor[0]' is partly outside array bounds of 'char[41]' [-Warray-bounds=]
 4081 |         id->bAlternateSetting = 0;
      |         ~~~~~~~~~~~~~~~~~~~~~~^~~
../../libusb/os/windows_winusb.c: In function '_hid_get_descriptor.isra':
../../libusb/os/windows_winusb.c:4046:14: note: at offset 9 into object 'tmp' of size 41
 4046 |         char tmp[HID_MAX_CONFIG_DESC_SIZE];
      |              ^~~
In function '_hid_get_config_descriptor',
    inlined from '_hid_get_descriptor.isra' at ../../libusb/os/windows_winusb.c:4265:11:
../../libusb/os/windows_winusb.c:4082:27: warning: array subscript 'struct libusb_interface_descriptor[0]' is partly outside array bounds of 'char[41]' [-Warray-bounds=]
 4082 |         id->bNumEndpoints = num_endpoints;
      |         ~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~
../../libusb/os/windows_winusb.c: In function '_hid_get_descriptor.isra':
../../libusb/os/windows_winusb.c:4046:14: note: at offset 9 into object 'tmp' of size 41
 4046 |         char tmp[HID_MAX_CONFIG_DESC_SIZE];
      |              ^~~
In function '_hid_get_config_descriptor',
    inlined from '_hid_get_descriptor.isra' at ../../libusb/os/windows_winusb.c:4265:11:
../../libusb/os/windows_winusb.c:4083:29: warning: array subscript 'struct libusb_interface_descriptor[0]' is partly outside array bounds of 'char[41]' [-Warray-bounds=]
 4083 |         id->bInterfaceClass = 3;
      |         ~~~~~~~~~~~~~~~~~~~~^~~
[... 165 lines cut ...]
make[2]: Entering directory '/c/work/libusb/libusb/build_ucrt64/tests'
  CC       stress.o
  CC       testlib.o
  CCLD     stress.exe
  CC       stress_mt-stress_mt.o
  CCLD     stress_mt.exe
  CC       set_option.o
  CCLD     set_option.exe
  CC       init_context.o
  CCLD     init_context.exe
make[2]: Leaving directory '/c/work/libusb/libusb/build_ucrt64/tests'
make[2]: Entering directory '/c/work/libusb/libusb/build_ucrt64'
make[2]: Leaving directory '/c/work/libusb/libusb/build_ucrt64'
make[1]: Leaving directory '/c/work/libusb/libusb/build_ucrt64'

$ gcc -v
Using built-in specs.
COLLECT_GCC=C:\msys64\ucrt64\bin\gcc.exe
COLLECT_LTO_WRAPPER=C:/msys64/ucrt64/bin/../lib/gcc/x86_64-w64-mingw32/15.2.0/lto-wrapper.exe
Target: x86_64-w64-mingw32
Configured with: ../gcc-15.2.0/configure --prefix=/ucrt64 --with-local-prefix=/ucrt64/local --with-native-system-header-dir=/ucrt64/include --libexecdir=/ucrt64/lib --enable-bootstrap --enable-checking=release --with-arch=nocona --with-tune=generic --enable-mingw-wildcard --enable-languages=c,lto,c++,fortran,ada,objc,obj-c++,jit --enable-shared --enable-static --enable-libatomic --enable-threads=posix --enable-graphite --enable-fully-dynamic-string --enable-libstdcxx-backtrace=yes --enable-libstdcxx-filesystem-ts --enable-libstdcxx-ti [... cut ...] h-system-zlib --with-gmp=/ucrt64 --with-mpfr=/ucrt64 --with-mpc=/ucrt64 --with-isl=/ucrt64 --with-pkgversion='Rev8, Built by MSYS2 project' --with-bugurl=https://github.com/msys2/MINGW-packages/issues --with-gnu-as --with-gnu-ld --with-libstdcxx-zoneinfo=yes --disable-libstdcxx-debug --enable-plugin --with-boot-ldflags=-static-libstdc++ --with-stage1-ldflags=-static-libstdc++
Thread model: posix
Supported LTO compression algorithms: zlib zstd
gcc version 15.2.0 (Rev8, Built by MSYS2 project)
```

mcuee, 2026-05-20:
libusb-1.0.30 release github action:

1. `-Warray-bounds` warnings for MSYS2 mingw64 build (will happen for mingw32 and ucrt64 build as well).
https://github.com/libusb/libusb/actions/runs/26003909365/job/76431925300
```
../../libusb/os/windows_winusb.c: In function 'init_root_hub':
../../libusb/os/windows_winusb.c:1020:9: warning: enumeration value 'LIBUSB_SPEED_SUPER_PLUS_X2' not handled in switch [-Wswitch-enum]
 1020 |         switch (speed) {
      |         ^~~~~~
../../libusb/os/windows_winusb.c: In function 'winusb_get_device_string':
../../libusb/os/windows_winusb.c:2299:9: warning: enumeration value 'LIBUSB_DEVICE_STRING_COUNT' not handled in switch [-Wswitch-enum]
 2299 |         switch (string_type) {
      |              ^~~
../../libusb/os/windows_winusb.c:4187:14: note: at offset [34, 41] into object 'tmp' of size 41
In function '_hid_get_config_descriptor',
    inlined from '_hid_get_descriptor.isra' at ../../libusb/os/windows_winusb.c:4406:11:
../../libusb/os/windows_winusb.c:4245:38: warning: array subscript 'struct libusb_endpoint_descriptor[0]' is partly outside array bounds of 'struct libusb_endpoint_descriptor[1]' [-Warray-bounds=]
 4245 |                 ed->bEndpointAddress = HID_OUT_EP;
../../libusb/os/windows_winusb.c: In function '_hid_get_descriptor.isra':
../../libusb/os/windows_winusb.c:4187:14: note: at offset [27, 41] into object 'tmp' of size 41
 4187 |         char tmp[HID_MAX_CONFIG_DESC_SIZE];
      |              ^~~
../../libusb/os/windows_winusb.c:4187:14: note: at offset [34, 41] into object 'tmp' of size 41
In function '_hid_get_config_descriptor',
    inlined from '_hid_get_descriptor.isra' at ../../libusb/os/windows_winusb.c:4406:11:
../../libusb/os/windows_winusb.c:4246:34: warning: array subscript 'struct libusb_endpoint_descriptor[0]' is partly outside array bounds of 'struct libusb_endpoint_descriptor[1]' [-Warray-bounds=]
 4246 |                 ed->bmAttributes = 3;
      |                 ~~~~~~~~~~~~~~~~~^~~
../../libusb/os/windows_winusb.c: In function '_hid_get_descriptor.isra':
../../libusb/os/windows_winusb.c:4187:14: note: at offset [27, 41] into object 'tmp' of size 41
 4187 |         char tmp[HID_MAX_CONFIG_DESC_SIZE];
      |              ^~~
../../libusb/os/windows_winusb.c:4187:14: note: at offset [34, 41] into object 'tmp' of size 41
In function '_hid_get_config_descriptor',
    inlined from '_hid_get_descriptor.isra' at ../../libusb/os/windows_winusb.c:4406:11:
../../libusb/os/windows_winusb.c:4247:36: warning: array subscript 'struct libusb_endpoint_descriptor[0]' is partly outside array bounds of 'struct libusb_endpoint_descriptor[1]' [-Warray-bounds=]
 4247 |                 ed->wMaxPacketSize = hid_priv->output_report_size - 1;
      |                 ~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
../../libusb/os/windows_winusb.c: In function '_hid_get_descriptor.isra':
../../libusb/os/windows_winusb.c:4187:14: note: at offset [27, 41] into object 'tmp' of size 41
 4187 |         char tmp[HID_MAX_CONFIG_DESC_SIZE];
      |              ^~~
../../libusb/os/windows_winusb.c:4187:14: note: at offset [34, 41] into object 'tmp' of size 41
In function '_hid_get_config_descriptor',
    inlined from '_hid_get_descriptor.isra' at ../../libusb/os/windows_winusb.c:4406:11:
../../libusb/os/windows_winusb.c:4248:31: warning: array subscript 'struct libusb_endpoint_descriptor[0]' is partly outside array bounds of 'struct libusb_endpoint_descriptor[1]' [-Warray-bounds=]
 4248 |                 ed->bInterval = 10;
      |                 ~~~~~~~~~~~~~~^~~~
../../libusb/os/windows_winusb.c: In function '_hid_get_descriptor.isra':
../../libusb/os/windows_winusb.c:4187:14: note: at offset [27, 41] into object 'tmp' of size 41
 4187 |         char tmp[HID_MAX_CONFIG_DESC_SIZE];
      |              ^~~
../../libusb/os/windows_winusb.c:4187:14: note: at offset [34, 41] into object 'tmp' of size 41
```

2. `-Wswitch-enum` warning for MSYS2 clang64 and clangarm64 build.
https://github.com/libusb/libusb/actions/runs/26003909365/job/76431925300
https://github.com/libusb/libusb/actions/runs/26003909376/job/76432174158
```
../../libusb/os/windows_winusb.c:1020:10: warning: enumeration value 'LIBUSB_SPEED_SUPER_PLUS_X2' not explicitly handled in switch [-Wswitch-enum]
 1020 |         switch (speed) {
      |                 ^~~~~
../../libusb/os/windows_winusb.c:2299:10: warning: enumeration value 'LIBUSB_DEVICE_STRING_COUNT' not explicitly handled in switch [-Wswitch-enum]
 2299 |         switch (string_type) {
      |                 ^~~~~~~~~~~
```