libusb/libusb #1036: libusb_control_transfer() always waiting for timeout to expire

dmitry-zakablukov, 2021-12-29:
Good day.

I want to use libusb_control_transfer() with timeout set to 5 seconds. And documentation says that:
```
timeout - timeout  (in milliseconds) that this function should wait before giving up due  to no response being received. For an unlimited timeout, use value 0.
```

One may think that libusb_control_transfer() will return earlier if response is received before timeout is expired. But this is not true: this function is always waiting for full timeout. Here is some logs:
```
[Debug] [263.170213] [000022d4] libusb: debug [libusb_alloc_transfer] transfer 000001F61D686DF8
[Debug] [263.170619] [000022d4] libusb: debug [libusb_submit_transfer] transfer 000001F61D686DF8
[Debug] [263.170709] [000022d4] libusb: debug [add_to_flying_list] arm timer for timeout in 5000ms (first in line)
[Debug] [263.170800] [000022d4] libusb: debug [composite_submit_control_transfer] using interface 0
[Debug] [263.170889] [000022d4] libusb: debug [winusbx_submit_control_transfer] will use interface 1
[Debug] [263.171009] [000022d4] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[Debug] [263.171102] [000022d4] libusb: debug [usbi_wait_for_events] WaitForMultipleObjects() for 2 HANDLEs with timeout in 60000ms
[Debug] [268.176318] [000022d4] libusb: debug [usbi_wait_for_events] WaitForMultipleObjects() returned 1
[Debug] [268.176596] [000022d4] libusb: debug [libusb_cancel_transfer] transfer 000001F61D686DF8
[Debug] [268.176413] [00005788] libusb: debug [windows_iocp_thread] transfer 000001F61D686DF8 completed, length 15
[Debug] [268.176716] [000022d4] libusb: debug [libusb_cancel_transfer] cancel transfer failed error -5
[Warning] [268.176942] [000022d4] libusb: warning [handle_timeout] async cancel failed -5
[Debug] [268.177036] [000022d4] libusb: debug [arm_timer_for_next_timeout] no timeouts, disarming timer
[Debug] [268.177130] [000022d4] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[Debug] [268.177222] [000022d4] libusb: debug [usbi_wait_for_events] WaitForMultipleObjects() for 2 HANDLEs with timeout in 60000ms
[Debug] [268.177320] [000022d4] libusb: debug [usbi_wait_for_events] WaitForMultipleObjects() returned 0
[Debug] [268.177411] [000022d4] libusb: debug [handle_event_trigger] event triggered
[Debug] [268.177500] [000022d4] libusb: debug [windows_handle_transfer_completion] handling transfer 000001F61D686DF8 completion with errcode 0, length 15
[Debug] [268.177591] [000022d4] libusb: debug [arm_timer_for_next_timeout] no timeouts, disarming timer
[Debug] [268.177681] [000022d4] libusb: debug [usbi_handle_transfer_completion] transfer 000001F61D686DF8 has callback 00007FFC84151C50
[Debug] [268.177770] [000022d4] libusb: debug [sync_transfer_cb] actual_length=15
[Debug] [268.177858] [000022d4] libusb: debug [libusb_free_transfer] transfer 000001F61D686DF8
```

This log corresponds to this call:
```
libusb_control_transfer( handle, 40, 0, 0, 0, ptr, 15, 5000 );
```

As you can see, usbi_wait_for_events() call is waiting for 2 events, one of them is timer. And it is expired, because next call is libusb_cancel_transfer(). But transfer is failed to be canceled, because it is completed.

I think there should be a timer disarm somewhere in the code after transfer is completed.

tormodvolden, 2022-01-02:
Is this on latest git? Please consider attaching full logs, so that e.g. version and driver information will be included.

dmitry-zakablukov, 2022-01-10:
@tormodvolden , no, this issue for libusb built from commit 1e57e84.

I think I will close this issue, because the problem is likely with my usb device. Here is the test code:
```
int main(int argc, char *argv[])
{
    libusb_init(0);
    libusb_device_handle* device = libusb_open_device_with_vid_pid(0, VID, PID);

    int ret = libusb_claim_interface(device, 1);
    printf("ret = %i\n", ret);

    ret = libusb_set_interface_alt_setting(device, 1, 1);
    printf("ret = %i\n", ret);

    system( "timeout /t 1");

    ret = libusb_control_transfer(device, 0x40, 0, 0, 0, BUFFER, 15, 5 * 1000);
    printf("ret = %i\n", ret);

    return 0;
}
```

So I can observe that:
1. At the beginning the code above works as it should, all calls to libusb returns 0.
2. Then I perform some actions with usb device.
3. And after that the usb device is seems to be in a bugged state, because the code above will return -5 for `libusb_control_transfer()`.

It cannot be a race condition inside libusb, because I launch a new instance of the test program. There no other operations that can lead to a race conditions. And I looked into the libusb source code and failed to find any race conditions. It all looks like the Windows OS really doesn't get any response from the usb device within the timeout.

Hence, the problem is either in usb device or in a driver. More over, when I use libusbK driver, then I got -5 error for `libusb_control_transfer()`. But when I use WinUSB driver, I got the pipe error.

closed 2022-01-10 by dmitry-zakablukov