libusb/libusb #1038: device ref count goes out of sync for unfinished irq transfer

lerdmann, 2022-01-02:
To begin with: I have written an OS/2 backend for libusb. Since OS/2 knows nothing about events the way that Linux uses them, I have based my implementation on the "usbi_signal_transfer_completion" / "usbi_handle_transfer_completion" mechanism. I also was forced to start worker threads from libusb in order to implement asynchronous operation.

What I have observed is that "libusb_submit_transfer" increments the device ref count with every successfully submitted transfer and likewise, "usbi_handle_transfer_completion" will decrement the device ref count.

The problem is that there are programs that schedule an irq transfer, most likely to catch a user interaction like pressing a button on the USB device. When the program terminates and no user interaction took place, that irq transfer has been successfully submitted but not being terminated. On a "libusb_close"->"do_close", libusb determines which transfers are still pending and removes them from the transfer list, but nothing more. Likewise, on a "usbi_handle_disconnect".
However it is necessary to decrement the device ref count even if that transfer has not been executed to termination because only if the device ref count goes to 0 will libusb finally call the "destroy" backend function and delete the "dev" structure.

I have worked around the problem by doing a "libusb_unref_device" from the "close" backend function for all still pending transfers. That works but I consider that a kludge because it implies that the backend needs to know the internal delicacies of libusb but in my opinion, it should not.

I instead suggest to add a call to "libusb_unref_device" to "do_close" as well as "usbi_handle_disconnect" for each transfer still pending.

lerdmann, 2022-01-09:
I am proposing this fix:
```
static void do_close(struct libusb_context *ctx,
	struct libusb_device_handle *dev_handle)
{
	struct usbi_transfer *itransfer;
	struct usbi_transfer *tmp;

	/* remove any transfers in flight that are for this device */
	usbi_mutex_lock(&ctx->flying_transfers_lock);

	/* safe iteration because transfers may be being deleted */
	for_each_transfer_safe(ctx, itransfer, tmp) {
		struct libusb_transfer *transfer =
			USBI_TRANSFER_TO_LIBUSB_TRANSFER(itransfer);

		if (transfer->dev_handle != dev_handle)
			continue;

		usbi_mutex_lock(&itransfer->lock);
		if (!(itransfer->state_flags & USBI_TRANSFER_DEVICE_DISAPPEARED)) {
			usbi_err(ctx, "Device handle closed while transfer was still being processed, but the device is still connected as far as we know");

			if (itransfer->state_flags & USBI_TRANSFER_CANCELLING)
				usbi_warn(ctx, "A cancellation for an in-flight transfer hasn't completed but closing the device handle");
			else
				usbi_err(ctx, "A cancellation hasn't even been scheduled on the transfer for which the device is closing");
		}
		usbi_mutex_unlock(&itransfer->lock);

		/* remove from the list of in-flight transfers and make sure
		 * we don't accidentally use the device handle in the future
		 * (or that such accesses will be easily caught and identified as a crash)
		 */
		list_del(&itransfer->list);
                libusb_unref_device(itransfer->dev_handle->dev); /* <---- need to unref this device since obviously, this transfer was never properly finished */
		transfer->dev_handle = NULL;

		/* it is up to the user to free up the actual transfer struct.  this is
		 * just making sure that we don't attempt to process the transfer after
		 * the device handle is invalid
		 */
		usbi_dbg("Removed transfer %p from the in-flight list because device handle %p closed",
			 transfer, dev_handle);
	}
	usbi_mutex_unlock(&ctx->flying_transfers_lock);

	usbi_mutex_lock(&ctx->open_devs_lock);
	list_del(&dev_handle->list);
	usbi_mutex_unlock(&ctx->open_devs_lock);

	usbi_backend.close(dev_handle);
	libusb_unref_device(dev_handle->dev);
	usbi_mutex_destroy(&dev_handle->lock);
	free(dev_handle);
}

```

mcuee, 2022-02-05:
@lerdmann Please help to create a pull request to have better reviews. Thanks.

lerdmann, 2022-02-05:
Pardon my ignorance but how do I create a "pull request" ?

mcuee, 2022-02-05:
> Pardon my ignorance but how do I create a "pull request" ?

The following document should help.
https://docs.github.com/en/pull-requests/collaborating-with-pull-requests/proposing-changes-to-your-work-with-pull-requests/about-pull-requests

Or a simpler guide here.
https://opensource.com/article/19/7/create-pull-request-github

lerdmann, 2022-02-24:
Sorry, I won't have any time in the foreseeable future. I implemented my suggested fix and removed the kludge that I had added to the backend "close" function (the backend "close" function no longer calls "libusb_unref_dev") and that works.

tormodvolden, 2022-02-24:
If you don't want to do the whole pull request procedure, it would be helpful if you can at least run `git diff > mydiff.txt `after doing your changes and attach the mydiff.txt here.

lerdmann, 2022-02-25:
[core.zip](https://github.com/libusb/libusb/files/8142966/core.zip)
Here we go

benzea, 2022-03-02:
Note that #1073 (and #1078) resolve this issue. The difference though is that the device is only unref'ed when the transfer is free'ed. As such, the device will still leak if the application ends up leaking the transfer, which is reasonably likely to happen in this kind of scenario.

commit 7cc06ea5 (2022-03-16): io: Track device in usbi_transfer

transfer->dev_handle currently has the behaviour that it will be unset
if the device is closed. The sync API uses this fact to catch an error
case.

In other cases, transfer->dev_handle will keep its value, which means
that if the transfer lives longer than the device handle, the pointer
becomes invalid.

The transfer does however keep a reference to the device, which owns the
pointer to the context. As such, we can track this reference internal to
the transfer, and it is set while the transfer is in-flight.

With this, switch the logging infrastructure to us [... cut]
closed 2022-03-16 by tormodvolden