libusb/libusb #1039: USB devices with incorrect interface descriptors cause read access violations

Eekle, 2022-01-02:
I have a pair of USB speakers that seem to have an invalid USB descriptor.

Looking via USB Device Tree Viewer:

```     ------------------ Configuration Descriptor -------------------
bLength                  : 0x09 (9 bytes)
bDescriptorType          : 0x02 (Configuration Descriptor)
wTotalLength             : 0x0087 (135 bytes)
bNumInterfaces           : 0x03 (3 Interfaces)
bConfigurationValue      : 0x01 (Configuration 1)
iConfiguration           : 0x00 (No String Descriptor)
<snip>
```
and then later on...

```
        ---------------- Interface Descriptor -----------------
bLength                  : 0x09 (9 bytes)
bDescriptorType          : 0x04 (Interface Descriptor)
bInterfaceNumber         : 0x03 (Interface 3)
bAlternateSetting        : 0x00
bNumEndpoints            : 0x01 (1 Endpoint)
bInterfaceClass          : 0x03 (HID - Human Interface Device)
bInterfaceSubClass       : 0x00 (None)
bInterfaceProtocol       : 0x00 (None)
iInterface               : 0x00 (No String Descriptor)
Data (HexDump)           : 09 04 03 00 01 03 00 00 00  
```

This Interface Number is incorrect, i believe. It can't possibly be interface 3, when the configuration says there are only 3 interfaces total.

However libusb does not check this index is in bounds.

When doing a control transfer, we eventually end up in `auto_claim` where it iterates over the array of 32 possible descriptors. It finds a matching descriptor at interface index 3, and then passes that index down.

In `windows_assign_endpoints`, we then use that to index into the config description interface struct. This is an out of bounds access, and the value of `if_desc` is nonsense.

https://github.com/libusb/libusb/blob/f2b218b61867f27568ba74fa38e156e5f55ed825/libusb/os/windows_winusb.c#L606-L611

When `if_desc` is then accessed, the stack dies with a read access violation.

https://github.com/libusb/libusb/blob/f2b218b61867f27568ba74fa38e156e5f55ed825/libusb/os/windows_winusb.c#L526-L530

I appreciate that this is really a hardware issue, but it seems egregious to me that it blows up with an access violation rather than erroring out or anything that might be caught higher up.

I initially encountered this issue trying to work with `probe-rs`, which was silently failing because of this error.

---

OS: Win 11
LibUSB version: 1.0.24.0 (as in version.h)

tormodvolden, 2022-01-04:
Thanks for the report and analysis! Can you please attach the full descriptor dump and debug log? (You may grab a Windows lsusb executable from http://dfu-util.sourceforge.net -> releases -> 0.11 binaries.) So the interfaces have indices 0,1,3 or something? I guess there is an assumption of contiguous interface indices in some places and the distinction of interface number and interface index is not consistently maintained. We should definitely catch this and not go out of bounds.

Eekle, 2022-01-05:
I can't use that software. It's based on libusb so it breaks the exact same way :P I'm using USB Device Tree Viewer

```     =========================== USB Port6 ===========================

Connection Status        : 0x01 (Device is connected)
Port Chain               : 2-6
Properties               : 0x01
 IsUserConnectable       : yes
 PortIsDebugCapable      : no
 PortHasMultiCompanions  : no
 PortConnectorIsTypeC    : no
ConnectionIndex          : 0x06 (Port 6)
CompanionIndex           : 0
 CompanionHubSymLnk      : USB#ROOT_HUB30#5&e8a0ca7&0&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}
 CompanionPortNumber     : 0x02 (Port 2)
 -> CompanionPortChain   : 2-2

      ========================== Summary =========================
Vendor ID                : 0x041E (Creative Labs)
Product ID               : 0x3272
USB Version              : 1.10
Port maximum Speed       : High-Speed (Companion Port 2-2 supports SuperSpeed)
Device maximum Speed     : Full-Speed
Device Connection Speed  : Full-Speed
Self powered             : no
Demanded Current         : 300 mA
Used Endpoints           : 3

      ======================== USB Device ========================

        +++++++++++++++++ Device Information ++++++++++++++++++
Device Description       : USB Composite Device
Device Path              : \\?\USB#VID_041E&PID_3272#6&e9bcfe3&0&6#{a5dcbf10-6530-11d2-901f-00c04fb951ed} (GUID_DEVINTERFACE_USB_DEVICE)
Kernel Name              : \Device\USBPDO-3
Device ID                : USB\VID_041E&PID_3272\6&E9BCFE3&0&6
Hardware IDs             : USB\VID_041E&PID_3272&REV_0100 USB\VID_041E&PID_3272
Driver KeyName           : {36fc9e60-c465-11cf-8056-444553540000}\0025 (GUID_DEVCLASS_USB)
Driver                   : \SystemRoot\System32\drivers\usbccgp.sys (Version: 10.0.22000.1  Date: 2021-06-05)
Driver Inf               : C:\WINDOWS\inf\usb.inf
Legacy BusType           : PNPBus
Class                    : USB
Class GUID               : {36fc9e60-c465-11cf-8056-444553540000} (GUID_DEVCLASS_USB)
Service                  : usbccgp
Enumerator               : USB
Location Info            : Port_#0006.Hub_#0001
Location IDs             : PCIROOT(0)#PCI(0103)#PCI(0000)#USBROOT(0)#USB(6), ACPI(_SB_)#ACPI(PCI0)#ACPI(GPP2)#ACPI(PTXH)#ACPI(RHUB)#ACPI(POT6)
Container ID             : {056c0760-6b26-11ec-9f4d-0c9d92c75205}
Manufacturer Info        : (Standard USB Host Controller)
Capabilities             : 0x84 (Removable, SurpriseRemovalOK)
Status                   : 0x0180600A (DN_DRIVER_LOADED, DN_STARTED, DN_DISABLEABLE, DN_REMOVABLE, DN_NT_ENUMERATOR, DN_NT_DRIVER)
Problem Code             : 0
Address                  : 6
HcDisableSelectiveSuspend: 0
EnableSelectiveSuspend   : 0
SelectiveSuspendEnabled  : 0
EnhancedPowerMgmtEnabled : 0
IdleInWorkingState       : 0
WakeFromSleepState       : 0
Power State              : D0 (supported: D0, D3, wake from D0)
 Child Device 1          : Pebble V3 (USB Audio Device)
  Device Path 1          : \\?\USB#VID_041E&PID_3272&MI_00#7&8d971ac&2&0000#{6994ad04-93ef-11d0-a3cc-00a0c9223196}\global (AM_KSCATEGORY_AUDIO)
  Device Path 2          : \\?\USB#VID_041E&PID_3272&MI_00#7&8d971ac&2&0000#{65e8773e-8f56-11d0-a3b9-00a0c9223196}\global (AM_KSCATEGORY_RENDER)
  Kernel Name            : \Device\000000a4
  Device ID              : USB\VID_041E&PID_3272&MI_00\7&8D971AC&2&0000
  Class                  : MEDIA
  Driver KeyName         : {4d36e96c-e325-11ce-bfc1-08002be10318}\0003 (GUID_DEVCLASS_MEDIA)
  Service                : usbaudio
  Location               : 0002.0000.0000.006.000.000.000.000.000
  LocationPaths          : PCIROOT(0)#PCI(0103)#PCI(0000)#USBROOT(0)#USB(6)#USBMI(0)  PCIROOT(0)#PCI(0103)#PCI(0000)#USBROOT(0)#USB(6)#USB(6)  ACPI(_SB_)#ACPI(PCI0)#ACPI(GPP2)#ACPI(PTXH)#ACPI(RHUB)#ACPI(POT6)#USBMI(0)  ACPI(_SB_)#ACPI(PCI0)#ACPI(GPP2)#ACPI(PTXH)#ACPI(RHUB)#ACPI(POT6)#USB(6)
   Child Device 1        : Speakers (Pebble V3) (Audio Endpoint)
[... 274 lines cut ...]
bDescriptorType          : 0x05 (Endpoint Descriptor)
bEndpointAddress         : 0x82 (Direction=IN EndpointID=2)
bmAttributes             : 0x03 (TransferType=Interrupt)
wMaxPacketSize           : 0x0004 (4 bytes)
bInterval                : 0x01 (1 ms)
Data (HexDump)           : 07 05 82 03 04 00 01                              .......

      -------------------- String Descriptors -------------------
             ------ String Descriptor 0 ------
bLength                  : 0x04 (4 bytes)
bDescriptorType          : 0x03 (String Descriptor)
Language ID[0]           : 0x0409 (English - United States)
Data (HexDump)           : 04 03 09 04                                       ....
             ------ String Descriptor 1 ------
bLength                  : 0x10 (16 bytes)
bDescriptorType          : 0x03 (String Descriptor)
Language 0x0409          : "ACTIONS"
Data (HexDump)           : 10 03 41 00 43 00 54 00 49 00 4F 00 4E 00 53 00   ..A.C.T.I.O.N.S.
             ------ String Descriptor 2 ------
bLength                  : 0x14 (20 bytes)
bDescriptorType          : 0x03 (String Descriptor)
Language 0x0409          : "Pebble V3"
Data (HexDump)           : 14 03 50 00 65 00 62 00 62 00 6C 00 65 00 20 00   ..P.e.b.b.l.e. .
                           56 00 33 00                                       V.3.
 ```

tormodvolden, 2022-01-05:
Haha, true! So it has these interfaces:
```
bInterfaceNumber         : 0x00 (Interface 0)
bInterfaceNumber         : 0x01 (Interface 1) alternate setting 0
bInterfaceNumber         : 0x01 (Interface 1) alternate setting 1
bInterfaceNumber         : 0x03 (Interface 3)
```
And no interface with bInterfaceNumber 2...

Eekle, 2022-01-05:
Good eh?

mcuee, 2022-01-07:
This is a broken device as per USB specifications.

USB 2.0 Spec Page 244.
> Within each configuration, each interface descriptor contains fields that identify the interface number and the alternate setting. Interfaces are numbered from zero to one less than the number of concurrent interfaces supported by the configuration. Alternate settings range from zero to one less than the number of alternate settings for a specific interface. The default setting when a device is initially configured is alternate setting zero.

Eekle, 2022-01-07:
I'm afraid I don't have easy access to either of those platforms :(

Obviously this isn't at all my project, but I would have thought that invalid devices should *not* be allowed to lead to memory unsafety. In the wild people are going to have all kinds of stuff plugged in, and I don't think you can expect it all to be well made.

tormodvolden, 2022-01-07:
I suppose it is this device? https://us.creative.com/p/speakers/creative-pebble-v3

The device probably works on all platforms with their proprietary drivers. It is possible that libusb core also can deal with this interface numbering, but not the Windows backend. I fully agree this shouldn't cause a memory access error, even if not supported.

Eekle, 2022-01-07:
Correct device.

I can also confirm that it enumerates fine in Windows Device Manager, and obviously in the USB tree viewer I used to generate the report here (https://usb-device-tree-viewer.en.lo4d.com/windows).

lunixbochs, 2022-03-18:
I've made a PR #1093 to add bounds checking to interface numbers across libusb.

To trivially test the crash, I propose compiling a custom libusb which force-sets the latest interface number to be a bad index >= bNumInterfaces on every device.

tormodvolden, 2022-03-18:
@Eekle if you can test latest master and confirm that this has been fixed, it would be very much appreciated.

closed 2022-03-18 by tormodvolden

LemonPG, 2025-07-07:
[image]Check the system's IgnoreHWSerNum registry. If your USB vid pid exists here, it will ignore getting SerialNumber.

commit f5275f9a (2022-03-18): Add interface bounds checks in darwin and windows endpoint functions

A broken device with invalid interface numbering could cause
an out-of-bounds array access.

Fixes #1039
Closes #1093