libusb/libusb #1040: Can't import libusb using chipwhisperer on MacBook M1, Monterey

MrUndead13, 2022-01-03:
Good evening,
I'm trying to use a chipwhisperer on my MacBook M1 and it is using libusb to connect with USB.
When I try to connect to it with : 

import chipwhisperer as cw
scope = cw.scope()

I get : 
OSError: Could not import libusb dll. Try 
pip uninstall libusb1
pip install libusb1

I tried to uninstall it and reinstall it as well as doing the same with brew and libusb but nothing changed.

Could someone help me please ?

mcuee, 2022-01-04:
As pyusb is a seperate project, please go to pyusb side to ask the question. But anyway, you should search the existing pyusb github issues there befcore asking this question.

mcuee, 2022-01-04:
Just an example: https://github.com/pyusb/pyusb/issues/355

closed 2022-01-04 by mcuee

MrUndead13, 2022-01-04:
I've been told in pyusb GitHub that this belonged here ....

mcuee, 2022-01-05:
Take note I am on both libusb project and pyusn project. Now I saw what you ask in pyusb.

Apparently you mixed up python-libusb1 and libusb project.
https://github.com/pyusb/pyusb/issues/404

Jonas is saying that your question points to python-libusb1 and not pyusb.

mcuee, 2022-01-05:
Anyway, the answer should be similar to this one.
https://github.com/vpelletier/python-libusb1/issues/53

Or https://github.com/pyusb/pyusb/issues/355 as the issue is related to how you install the libusb under macOS.

MrUndead13, 2022-01-05:
I already installed libusb with homebrew, followed everything in both links, linked the intel's path to the homebrew's libusb path and even used your loop in zprofile, but nothing worked ....

I've seen in the activity monitor that when I run my Jupiter notebook, python3.9 use Intel's version instead of Apple's one. However I'm still looking for a way to change this.