libusb/libusb #1041: ARM64-hotplug  Specifying the device Type

J1n-CH1na, 2022-01-07:
`
void* monitor_hid_device(void* para){
    int res;
    libusb_hotplug_callback_handle callback_handle;
    res = libusb_hotplug_register_callback(NULL, LIBUSB_HOTPLUG_EVENT_DEVICE_ARRIVED | LIBUSB_HOTPLUG_EVENT_DEVICE_LEFT,
                                           0,
                                           LIBUSB_HOTPLUG_MATCH_ANY, LIBUSB_HOTPLUG_MATCH_ANY,USB_CLASS_HID,
                                           func, NULL, &callback_handle);
    if (LIBUSB_SUCCESS != res) {
        libusb_exit(NULL);
        return NULL;
    }
    while (1) {
        libusb_handle_events_completed(NULL, NULL);
        nanosleep(&(struct timespec){0, 10000000UL}, NULL);
    }
    libusb_hotplug_deregister_callback(NULL, callback_handle);
    libusb_exit(NULL);

    return NULL;
}
`
How to specify the argument about 'dev_class' ? I use the USB_CLASS_HID, LIBUSB_DT_HID and so on, but it doesn't work.

J1n-CH1na, 2022-01-07:
I used these MICRO :  LIBUSB_CLASS_HID  \  LIBUSB_DT_HID  \  USB_CLASS_HID  \  LIBUSB_DT_DEVICE.

mcuee, 2022-01-19:
@J1n-CH1na What is the OS used here? Can you post the device USB descriptors ("lsusb -vvv" output under Linux if you use Linux).

J1n-CH1na, 2022-01-23:
Thanks！
Ubuntu18.04  kernel 4.19.193 

lsusb -vvv （just my keyboard）

```
Bus 001 Device 003: ID 046a:0077 Cherry GmbH
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0
  bDeviceProtocol         0
  bMaxPacketSize0        64
  idVendor           0x046a Cherry GmbH
  idProduct          0x0077
  bcdDevice            1.11
  iManufacturer           1 CHERRY
  iProduct                2 Mechanical Keyboard
  iSerial                 0
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           66
    bNumInterfaces          2
    bConfigurationValue     1
    iConfiguration          0
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              400mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      79
         Report Descriptors:
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass         3 Human Interface Device
[... 612 lines cut ...]
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0004  1x 4 bytes
        bInterval              12
Hub Descriptor:
  bLength               9
  bDescriptorType      41
  nNbrPorts             1
  wHubCharacteristic 0x0009
    Per-port power switching
    Per-port overcurrent protection
  bPwrOn2PwrGood       10 * 2 milli seconds
  bHubContrCurrent      0 milli Ampere
  DeviceRemovable    0x00
  PortPwrCtrlMask    0xff
 Hub Port Status:
   Port 1: 0000.0503 highspeed power enable connect
Device Status:     0x0001
  Self Powered
```

J1n-CH1na, 2022-01-23:
I want to explain my question again.

It's about this API:
```
libusb_hotplug_register_callback(ctx, events,flags,vendor_id,product_id,**dev_class**,cb_fn,user_data,callback_handle);
```

I set different values for **dev_class**, just like LIBUSB_CLASS_HID, LIBUSB_DT_HID, USB_CLASS_HID, LIBUSB_DT_DEVICE, but it doesn't  work.Whether I plug in a keyboard or a mass-storage device, I can always execute cb_fn.

 Using  [ this example](https://libusb.sourceforge.io/api-1.0/libusb_hotplug.html), and you need modify something~

THANKS~

mcuee, 2022-01-23:
I can only say that "DO NOT USE LIBUSB" with a keyboard or mouse or things like that.

closed 2022-01-23 by mcuee

J1n-CH1na, 2022-01-28:
Thanks, and using "libusb/hidapi" is better for me. 
Best wishes~