libusb/libusb #1042: rework "list_del" so that "list_empty" can be used on that list element

lerdmann, 2022-01-07:
in libusbi.h, change:
`static inline void list_del(struct list_head *entry)
{
	entry->next->prev = entry->prev;
	entry->prev->next = entry->next;
	entry->next = entry->prev = NULL;
}
`
to
`static inline void list_del(struct list_head *entry)
{
	entry->next->prev = entry->prev;
	entry->prev->next = entry->next;
	entry->next = entry->prev = entry;  /* this effectively mirrors the list_init case */
}
`

For one, that will protect against a double call of "list_del". It will also allow use use "list_empty" to check for empty list on that element:
`#define list_empty(entry) ((entry)->next == (entry))
`
There are code paths where a list_del is called twice and it must be possible to check for emptyness after "list_del" was called (and not only after list_init was invoked).

lerdmann, 2022-01-09:
After some analysis of the core, I can see that "list_del(&itransfer->list)" is called from 2 different places:
1) from "remove_from_flying_list" in io.c, in the scope of usbi_handle_transfer_completion which in turn is normally called from backend function "handle_transfer_completion"
2) from "do_close" in core.c, in the scope of a lisbusb_close, shortly before backend function "close" is called.

If a transfer is still in progress while the "close" action is executed and terminates and is signalled after "close" is finished, then the core code will call "list_del" twice on the very same element.

diabolo38, 2022-01-10:
> 
>     2. from "do_close" in core.c, in the scope of a lisbusb_close, shortly before backend function "close" is called.
> 
i believe  it is documented or known that application shall not use close until it has flying transfer 
still it would be safer avoid  uncontroeld chrach to check for emptyness before del but it's meaningfull only in an atomic context

lerdmann, 2022-01-10:
I can only say that libuvc uses open and close. It's always used as a pair and that also makes sense because libusb_close will decrement the device handle that was incremented by lisbusb_open.
And I would expect other applications to do the same.
I don't know what you mean by "atomic context".

papr, 2022-09-02:
Hi 👋🏻 

I hope you all are having a good week! I am a libuvc user and it is the first time that I am looking into libusb details.

Specifically, I am running into an EXC_BAD_ACCESS in `list_del` (`entry->next` and `entry->prev` are both `NULL`) when calling `uvc_open` and `uvc_close` on a device handle directly after each other. I understand that this is not the normal use case but I struggle to understand which condition needs to be fulfilled between opening and closing the handle s.t. that I can call `uvc_close` safely.

I hope this is the right place to ask. Please let me know if it is not.

mcuee, 2022-09-02:
> I hope this is the right place to ask. Please let me know if it is not.

@papr I think it is okay if you think your issue is related.  

Please also post the debug log ( (with LIBUSB_DEBUG =4) and other necessary info if you think libusb has issues.
https://github.com/libusb/libusb/wiki/Troubleshooting

You can also try the changes proposed by the original poster to see if that helps your use case or not.

mcuee, 2022-09-02:
> If a transfer is still in progress while the "close" action is executed and terminates and is signalled after "close" is finished, then the core code will call "list_del" twice on the very same element.

Depending on the use of sync API or async API, you may want to check out the following two issues.
* https://github.com/libusb/libusb/issues/703
* https://github.com/libusb/libusb/issues/903

papr, 2022-09-02:
@mcuee Thank you for your kind words and the pointers! I have yet to read through the linked issues. Nonetheless, I have attached the code to reproduce the issue and the debug logs.

Turns out, the issue only reproduces if one open+closes two different devices. Open+close twice on the same device does not reproduce the issue (tried on different devices).

OS Version: macOS 12.4 (21F79)
Host: Apple M1 MacBook Air

[Test that reproduces the issue](https://github.com/pupil-labs/libuvc/blob/e65e1905bbf1b6326532aeb918c7dfc740faa89e/tests/test-enumerating-devices.c) - Iterates over available devices and tries to open+close them. Test will be built as part of the [top-level cmake file](https://github.com/pupil-labs/libuvc/blob/e65e1905bbf1b6326532aeb918c7dfc740faa89e/CMakeLists.txt#L130).

I have built libusb on current master (ff8fe9397d4c61f072d209703f854c9f62ed17e3) without any optimizations (`CFLAGS='-g -O0'`) and have ensured to link against when building the example via the `PKG_CONFIG_PATH` env var.

To debug, I run: `sudo lldb tests/test_enumerating_devices` (sudo, because I need to detach the default driver)

This is the [full log](https://gist.github.com/papr/1a67bbb8801648fe915e28171b6fe0ba). You can find the full backtrace at the bottom.