libusb/libusb #1043: Crash when closing rtlsdr device on Windows

jvde-github, 2022-01-07:
Good day!

I am developing a program using the [rtlsdr](https://github.com/osmocom/rtl-sdr) library with MSVC 2019 on Windows 11 and experience occasional crashes when closing down the SDR device. It is difficult to replicate but it happens frequent enough. The same behavior occurs for the rtl_sdr program that is provided with rtlsdr.  I am new to the USB interface and libusb but did an attempt to understand. After some inspection of the logs and debugging I have developed an idea.

Hypothesis: the calls to ```libusb_handle_events_timeout_completed``` (after the call to ```libusb_cancel_transfer```
in the function [```rtlsdr_read_async```](https://github.com/osmocom/rtl-sdr/blob/master/src/librtlsdr.c))  are not clearing all the trivial events resulting in an access violation from activity that occurs after ```libusb_free_transfer```.  

I noticed that ```libusb_handle_events_timeout_completed``` in libusb relies on a call to ```WaitForMultipleObjects``` with zero wait time in this case. This might not trigger a [wait state](https://docs.microsoft.com/en-us/windows/win32/api/synchapi/nf-synchapi-waitformultipleobjects) and different result in a different behavior from using a wait time of a minimum of 1 ms?

Can it be that because of the zero wait time used in this applications that libusb is missing to handle some obvious events? Is it therefore advisable not to use a zero timeout time? Or is it an idea to replace this part of ```handle_events````
````
	/* round up to next millisecond */
	if (tv->tv_usec % 1000)
		timeout_ms++;
````
to 
````
	/* round up to next millisecond */
	timeout_ms++;
````

With this change, the rtl-sdr programs run perfectly stable again from the limited testing I have done.

The way that this rtlsdr library closes the transfer seems to be fairly common but might not be best practice. I don't know.

Thank you and apologies if above does not make sense at all! Grateful for any ideas.

The log for a crash-scenario:
```
[ 1.901930] [0000368c] libusb: debug [libusb_free_transfer] transfer 0000014C8C488278
[ 1.902158] [0000368c] libusb: debug [libusb_free_transfer] transfer 0000014C8C488668
[ 1.902384] [0000368c] libusb: debug [libusb_free_transfer] transfer 0000014C8C487948 <---
[ 1.891864] [00003ee0] libusb: debug [windows_iocp_thread] transfer 0000014C8C488668 completed, length 14336
[ 1.902820] [00003ee0] libusb: debug [windows_iocp_thread] transfer 0000014C8C487948 completed, length 16896 <---
[ 2.544650] [000050f4] libusb: debug [libusb_submit_transfer] transfer 0000014C8C486C28
[ 2.545322] [000050f4] libusb: debug [add_to_flying_list] arm timer for timeout in 300ms (first in line)
[ 2.547119] [000050f4] libusb: debug [composite_submit_control_transfer] using interface 0
[ 2.547890] [000050f4] libusb: debug [winusbx_submit_control_transfer] will use interface 0
```

tormodvolden, 2022-01-07:
This with the libusb_cancel_transfer and the timeout reminds me about #1036. Can it be the same root cause?

jvde-github, 2022-01-09:
Thank you. It reads like a different issue. My problem is now resolved by adding a 1 ms Sleep in the rtlsdr library between the cancel and handle_event.

I am sharing the log for a run with a successful device close, which I achieve by changing the ```handle_events``` function as per above  (i.e. ensure a minimum of 1 ms timeout):
```
	/* round up to next millisecond */
	timeout_ms++;
````
The log for a cancel looks in this case like:
```
[ 3.146617] [0000462c] libusb: debug [libusb_cancel_transfer] transfer 000001C871BDC258
[ 3.146926] [0000462c] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[ 3.147226] [0000462c] libusb: debug [usbi_wait_for_events] WaitForMultipleObjects() for 2 HANDLEs with timeout in 1ms
--WINIOCP: lock requested
--WINIOCP: lock released
[ 3.147999] [00005144] libusb: debug [windows_iocp_thread] transfer 000001C871BDC258 completed, length 9728
[ 3.148279] [0000462c] libusb: debug [usbi_wait_for_events] WaitForMultipleObjects() returned 0
[ 3.148606] [0000462c] libusb: debug [handle_event_trigger] event triggered
[ 3.148708] [0000462c] libusb: debug [windows_handle_transfer_completion] handling transfer 000001C871BDC258 completion with errcode 995, length 9728
[ 3.149959] [0000462c] libusb: debug [windows_handle_transfer_completion] detected operation aborted
[ 3.150064] [0000462c] libusb: debug [usbi_handle_transfer_completion] transfer 000001C871BDC258 has callback 00007FF9342F4810
```
The ```WINIOCP: lock requested``` are statements added by myself to understand the running of the thread.

In case of a program crash with the latest libusb implementation, the log around a cancel looks like:
```
[ 2.185963] [00002480] libusb: debug [libusb_cancel_transfer] transfer 0000022EBA7D2FD8
[ 2.186189] [00002480] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[ 2.186408] [00002480] libusb: debug [usbi_wait_for_events] WaitForMultipleObjects() for 2 HANDLEs with timeout in 0ms
[ 2.186631] [00002480] libusb: debug [usbi_wait_for_events] WaitForMultipleObjects() returned 258
[ 2.186848] [00002480] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[ 2.187064] [00002480] libusb: debug [usbi_wait_for_events] WaitForMultipleObjects() for 2 HANDLEs with timeout in 0ms
[ 2.187294] [00002480] libusb: debug [usbi_wait_for_events] WaitForMultipleObjects() returned 258
[ 2.187515] [00002480] libusb: debug [libusb_free_transfer] transfer 0000022EBA7FA038
[ 2.187725] [00002480] libusb: debug [libusb_free_transfer] transfer 0000022EBA7B55F8
[ 2.187936] [00002480] libusb: debug [libusb_free_transfer] transfer 0000022EBA7D86D8
[ 2.188155] [00002480] libusb: debug [libusb_free_transfer] transfer 0000022EBA7D7A18
[ 2.188374] [00002480] libusb: debug [libusb_free_transfer] transfer 0000022EBA7D7208
[ 2.188590] [00002480] libusb: debug [libusb_free_transfer] transfer 0000022EBA7D6788
[ 2.188807] [00002480] libusb: debug [libusb_free_transfer] transfer 0000022EBA7BFEA8
[ 2.189028] [00002480] libusb: debug [libusb_free_transfer] transfer 0000022EBA7DE278
[ 2.189243] [00002480] libusb: debug [libusb_free_transfer] transfer 0000022EBA7FE448
[ 2.189454] [00002480] libusb: debug [libusb_free_transfer] transfer 0000022EBA7D6358
[ 2.189674] [00002480] libusb: debug [libusb_free_transfer] transfer 0000022EBA7D64A8
[ 2.189890] [00002480] libusb: debug [libusb_free_transfer] transfer 0000022EBA7BF6A8
[ 2.190134] [00002480] libusb: debug [libusb_free_transfer] transfer 0000022EBA7BF7F8
[ 2.190393] [00002480] libusb: debug [libusb_free_transfer] transfer 0000022EBA7D2E88
[ 2.190610] [00002480] libusb: debug [libusb_free_transfer] transfer 0000022EBA7D2FD8
[ 2.180882] [000050b4] libusb: debug [windows_iocp_thread] transfer 0000022EBA7DE278 completed, length 4096
--WINIOCP: lock requested
--WINIOCP: lock released
[ 2.191539] [000050b4] libusb: debug [windows_iocp_thread] transfer 0000022EBA7FE448 completed, length 3584
--WINIOCP: lock requested
--WINIOCP: lock released
[ 2.192206] [000050b4] libusb: debug [windows_iocp_thread] transfer 0000022EBA7D6358 completed, length 3072
--WINIOCP: lock requested
--WINIOCP: lock released
[ 2.192843] [000050b4] libusb: debug [windows_iocp_thread] transfer 0000022EBA7D64A8 completed, length 4096
--WINIOCP: lock requested
--WINIOCP: lock released
[ 2.193490] [000050b4] libusb: debug [windows_iocp_thread] transfer 0000022EBA7BF6A8 completed, length 7168
--WINIOCP: lock requested
--WINIOCP: lock released
[ 2.194134] [000050b4] libusb: debug [windows_iocp_thread] transfer 0000022EBA7BF7F8 completed, length 14848
--WINIOCP: lock requested
--WINIOCP: lock released
[ 2.194767] [000050b4] libusb: debug [windows_iocp_thread] transfer 0000022EBA7D2E88 completed, length 512
--WINIOCP: lock requested
--WINIOCP: lock released
[ 2.195404] [000050b4] libusb: debug [windows_iocp_thread] transfer 0000022EBA7D2FD8 completed, length 1536
(process 2760) exited with code -1073740940.
```

It seems that the winiocp thread is not processing events only until after the libusb_free calls.

The issue here can be resolved on the user side so perhaps not an libusb issue. But the behavior is a bit unexpected and subtle so wondering if a call like ```Sleep(1)``` is a worthwhile addition.

Hope this makes sense.

Kind regards and thanks for the great work on the library.

argilo, 2022-01-14:
> Hypothesis: the calls to `libusb_handle_events_timeout_completed` [...] are not clearing all the trivial events

While tracking down another issue in librtlsdr, I noticed that librtlsdr does not actually call `libusb_handle_events_timeout_completed`, due to a bug introduced in the CMake build. (It calls `libusb_handle_events_timeout` instead.) I sent a patch to the osmocom-sdr mailing list, but it's also posted here:

https://github.com/argilo/rtl-sdr/commit/2d18fe2db3f4b12f87cdce1788d73c01282d835a

Could that explain the crashes you're seeing?

jvde-github, 2022-01-14:
That is a great find on the delay in closing time! I suspect it is a different issue though as the call after the cancel_transfer is:
```
             libusb_handle_events_timeout_completed(dev->ctx, &zerotv, NULL);
```
So, for this particular call it will execute the same code in libusb with or without the override you identified (the NULL is stripped out and then added again by libusb). To be sure though I will properly test if your fix also solves the issue I identified.

The key problem is that librtlsdr does not ensure that all the callbacks are completed before freeing memory. The sleep(1) does trigger it on Windows but a proper fix might be needed in the future.

mcuee, 2022-01-19:
> The key problem is that librtlsdr does not ensure that all the callbacks are completed before freeing memory.

So this looks like a librtlsdr problem and not libusb problem, right?

jvde-github, 2022-01-19:
Yes, the issue can be closed. Many thanks,.

closed 2022-01-19 by jvde-github