libusb/libusb #1044: libusb_claim_interface throw exception "Access violation reading location 0xFFFFFFFFFFFFFFFF" on Windows

YueLu0116, 2022-01-16:
## Brief description
I want to use libusb to transfer h264 data from an Android device in [accessory mode](https://source.android.com/devices/accessories/aoa) to a Windows PC. When I tried to claim the interface, `libusb_claim_interface` threw a reading access violation exception. 

## Drivers
libusbK installed by zadig
[image]

## Demo and results
```cpp
#include "libusb-1.0/libusb.h"
int main()
{
	// test demo
	auto ret = libusb_init(nullptr);
	if (ret) {
		std::cerr << "libusb_init failed. Reason: " << ret << "\n";
		return -1;
	}
	libusb_set_option(nullptr, LIBUSB_OPTION_LOG_LEVEL, LIBUSB_LOG_LEVEL_DEBUG);
	libusb_device_handle* devHandle = nullptr;
        // Google Vendor ID and
        // product ID to use when in accessory mode
	devHandle = libusb_open_device_with_vid_pid(nullptr, 0x18D1, 0x2D00);
	if (!devHandle) {
		std::cerr << "Cannot open accessory\n";
		libusb_exit(nullptr);
		return -1;
	}
        // bInterfaceNumber can be seen in usbview.exe (see later)
	ret = libusb_claim_interface(devHandle, 0);
	if (ret) {
		std::cerr << "libusb claim interface failed. reason:" << ret << "\n";
		libusb_close(devHandle);
		libusb_exit(nullptr);
		return -1;
	}
	libusb_release_interface(devHandle, 0);
	libusb_close(devHandle);
	libusb_exit(nullptr);
}
``` 
When libusb_claim_interface is called:
```
Exception thrown at 0x00007FFF4064BF1C (libusb-1.0.dll) in Test.exe: 0xC0000005: Access violation reading location 0xFFFFFFFFFFFFFFFF.
```

The outputs with libusb debug log:
```
[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.181074] [0000011c] libusb: debug [libusb_get_device_list]
[ 0.183699] [00006938] libusb: debug [windows_iocp_thread] I/O completion thread started
[ 0.195960] [0000011c] libusb: debug [winusb_get_device_list] allocating new device for session [76]
[ 0.196077] [0000011c] libusb: debug [winusb_get_device_list] allocating new device for session [77]
[ 0.196214] [0000011c] libusb: debug [get_api_type] driver(s): BTHUSB
[ 0.196269] [0000011c] libusb: debug [get_api_type] lower filter driver(s): ibtusb
[ 0.196312] [0000011c] libusb: debug [winusb_get_device_list] allocating new device for session [10]
[ 0.196524] [0000011c] libusb: debug [get_api_type] driver(s): usbccgp
[ 0.196572] [0000011c] libusb: debug [get_api_type] matched driver name against Composite API
[ 0.196600] [0000011c] libusb: debug [winusb_get_device_list] allocating new device for session [58]
[ 0.196696] [0000011c] libusb: debug [get_api_type] driver(s): WUDFRd
[ 0.196747] [0000011c] libusb: debug [get_api_type] lower filter driver(s): WinUsb
[ 0.196777] [0000011c] libusb: debug [get_api_type] matched lower filter driver name against WinUSB
[ 0.196802] [0000011c] libusb: debug [winusb_get_device_list] allocating new device for session [59]
[ 0.196899] [0000011c] libusb: debug [get_api_type] driver(s): WINUSB
[ 0.196949] [0000011c] libusb: debug [get_api_type] matched driver name against WinUSB
[ 0.196976] [0000011c] libusb: debug [winusb_get_device_list] allocating new device for session [5A]
[ 0.197069] [0000011c] libusb: debug [get_api_type] driver(s): libusbK
[ 0.197113] [0000011c] libusb: debug [get_api_type] matched driver name against libusbK
[ 0.197141] [0000011c] libusb: debug [winusb_get_device_list] allocating new device for session [5B]
[... 127 lines cut ...]
bInterfaceNumber:                  0x00
bAlternateSetting:                 0x00
bNumEndpoints:                     0x02
bInterfaceClass:                   0xFF  -> Interface Class Unknown to USBView
bInterfaceSubClass:                0xFF
bInterfaceProtocol:                0x00
iInterface:                        0x05
     English (United States)  "Android Accessory Interface"

          ===>Endpoint Descriptor<===
bLength:                           0x07
bDescriptorType:                   0x05
bEndpointAddress:                  0x81  -> Direction: IN - EndpointID: 1
bmAttributes:                      0x02  -> Bulk Transfer Type
wMaxPacketSize:                  0x0200 = 0x200 max bytes
bInterval:                         0x00

          ===>Endpoint Descriptor<===
bLength:                           0x07
bDescriptorType:                   0x05
bEndpointAddress:                  0x01  -> Direction: OUT - EndpointID: 1
bmAttributes:                      0x02  -> Bulk Transfer Type
wMaxPacketSize:                  0x0200 = 0x200 max bytes
bInterval:                         0x00
```

YueLu0116, 2022-01-16:
Solved by install WinUsb...

YueLu0116, 2022-01-16:
Solved by install WinUsb driver.

closed 2022-01-16 by YueLu0116

mcuee, 2022-01-22:
Thanks for the updates. If you want to try out libusbk, please use libusb-1.0.25rc1 which has fixed some issues with libusbk.sys and libusb0.sys.