libusb/libusb #1045: github actions for Window

mcuee, 2022-01-21:
On Fri, Jan 21, 2022 at 5:23 PM Jef Driesen wrote:
On 20/01/2022 22:32, Ludovic Rousseau wrote:
> It would be nice to also have an automatic build for Windows. But I do
> not know Windows.
> Volunteers are welcome.

Have a look at the github action for my project:

https://github.com/libdivecomputer/libdivecomputer/blob/master/.github/workflows/build.yml

It's a very similar project compared to libusb and has libusb and hidapi as
dependencies. It supports Windows builds using both mingw and msvc, so you can
probably re-use parts for creating a libusb action. The mingw build even builds
libusb and hidapi from source.

Jef

tormodvolden, 2022-01-22:
In the meantime we have Windows builds through Appveyor though: https://ci.appveyor.com/project/dickens/libusb/builds/42292441

mcuee, 2025-09-09:
Close this now that #1353 has been merged.