libusb/libusb #1046: 1.0.25-rc1 test matrix

tormodvolden, 2022-01-21:
For as many platforms as possible we want to test building and a run of the RC. At least xusb or lsusb -v should be tested.
### Linux x86_64
- [x] build: gcc 9.3 OK
- [ ] run:
### Linux arm64
  - [ ] build: (tormod can do)
### Windows MinGW 64-bit
  - [x] build: gcc 9.3 OK
  - [ ] run:
### Windows MinGW 32-bit
  - [x] build: gcc 9.3 OK
  - [ ] run:
### macOS 12 Intel
  - [ ] build:
  - [ ] run:
### macOS 12 ARM
  - [ ] build:
  - [ ] run:
### macOS 10.13 Intel
   - [ ] build: (tormod can do)

mcuee, 2022-01-21:
I will try to test the following ove the next few days.

### Linux arm64
  - [ ] build: gcc 8.3 OK
  - [ ] run: xusb, lsusb, libftdi

### Linux arm64
  - [ ] build: gcc 8.3 OK
  - [ ] run: xusb, lsusb, libftdi
 
### Windows VS2019 64-bit
  - [x] build: gcc 9.3 OK
  - [ ] run: xusb, libftdi
  
### Windows VS2019 32-bit
  - [x] build: gcc 9.3 OK
  - [ ] run: xusb, libftdi
 
### Windows MinGW 64-bit
  - [x] build: MSYS2 MinGW-w64 64bit GCC 11.2.0 OK but with some warnings
  - [ ] run: xusb, libftdi
  
### Windows MinGW 32-bit
  - [x] build: MinGW-w64 32 bit GCC 11.2.0 OK but with some warnings
  - [ ] run: xusb, libftdi

### macOS 12 ARM64
  - [ ] build:
  - [ ] run:

mcuee, 2022-01-21:
Not so sure who will be able to test the following.

### OpenBSD 64bit (x86_64)
  - [ ] build:
  - [ ] run: 

### NetBSD 64bit (x86_64)
  - [ ] build: 
  - [ ] run: 

### Solaris 64 bit (x86_64)
  - [ ] build: 
  - [ ] run: 

### Haiku 64 bit (x86_64)
  - [ ] build: 
  - [ ] run:

hjelmn, 2022-01-22:
### macOS 10.12 x86_64
- [x] build: OK
- [x] run: xusb, stress, listdevs

mcuee, 2022-01-22:
@tormodvolden Just wondering how you build usbutils (lsusb) under Raspbian Buster to get it to depend on libusb-1.0.25rc1. 

Somehow I tried to build usbutils (lsusb) under Raspberry Pi OS Bullseye 32bit/64bit, it seems to be always dependant on libusb-1.0.24 release. Very strange. I have even overwrite the libusb libary under /usr/lib with libusb-1.0.25rc1.

mcuee, 2022-01-22:
Just installed Haiku r1beta3 using VirtualBox under Windows 11, it failed to build.

OS: https://www.haiku-os.org/get-haiku/r1beta3/
```
~/Desktop/build/libusb-1.0.25-rc1> make -i
make  all-recursive
make[1]: Entering directory '/boot/home/Desktop/build/libusb-1.0.25-rc1'
Making all in libusb
make[2]: Entering directory '/boot/home/Desktop/build/libusb-1.0.25-rc1/libusb'
  CC       core.lo
  CC       descriptor.lo
  CC       hotplug.lo
  CC       io.lo
  CC       strerror.lo
  CC       sync.lo
  CC       os/events_posix.lo
  CC       os/threads_posix.lo
  CXX      os/haiku_usb_backend.lo
In file included from ./libusbi.h:106,
                 from os/haiku_usb.h:25,
                 from os/haiku_usb_backend.cpp:27:
/boot/system/develop/tools/lib/gcc/x86_64-unknown-haiku/8.3.0/include/stdatomic.h:40:9: error: '_Atomic' does not name a type
 typedef _Atomic _Bool atomic_bool;
         ^~~~~~~
/boot/system/develop/tools/lib/gcc/x86_64-unknown-haiku/8.3.0/include/stdatomic.h:41:9: error: '_Atomic' does not name a type
 typedef _Atomic char atomic_char;
         ^~~~~~~
/boot/system/develop/tools/lib/gcc/x86_64-unknown-haiku/8.3.0/include/stdatomic.h:42:9: error: '_Atomic' does not name a type
 typedef _Atomic signed char atomic_schar;
         ^~~~~~~
/boot/system/develop/tools/lib/gcc/x86_64-unknown-haiku/8.3.0/include/stdatomic.h:43:9: error: '_Atomic' does not name a type
 typedef _Atomic unsigned char atomic_uchar;
         ^~~~~~~
/boot/system/develop/tools/lib/gcc/x86_64-unknown-haiku/8.3.0/include/stdatomic.h:44:9: error: '_Atomic' does not name a type
 typedef _Atomic short atomic_short;
         ^~~~~~~
/boot/system/develop/tools/lib/gcc/x86_64-unknown-haiku/8.3.0/include/stdatomic.h:45:9: error: '_Atomic' does not name a type
 typedef _Atomic unsigned short atomic_ushort;
         ^~~~~~~
/boot/system/develop/tools/lib/gcc/x86_64-unknown-haiku/8.3.0/include/stdatomic.h:46:9: error: '_Atomic' does not name a type
 typedef _Atomic int atomic_int;
         ^~~~~~~
/boot/system/develop/tools/lib/gcc/x86_64-unknown-haiku/8.3.0/include/stdatomic.h:47:9: error: '_Atomic' does not name a type
 typedef _Atomic unsigned int atomic_uint;
         ^~~~~~~
/boot/system/develop/tools/lib/gcc/x86_64-unknown-haiku/8.3.0/include/stdatomic.h:48:9: error: '_Atomic' does not name a type
 typedef _Atomic long atomic_long;
         ^~~~~~~
/boot/system/develop/tools/lib/gcc/x86_64-unknown-haiku/8.3.0/include/stdatomic.h:49:9: error: '_Atomic' does not name a type
 typedef _Atomic unsigned long atomic_ulong;
         ^~~~~~~
/boot/system/develop/tools/lib/gcc/x86_64-unknown-haiku/8.3.0/include/stdatomic.h:50:9: error: '_Atomic' does not name a type
 typedef _Atomic long long atomic_llong;
         ^~~~~~~
/boot/system/develop/tools/lib/gcc/x86_64-unknown-haiku/8.3.0/include/stdatomic.h:51:9: error: '_Atomic' does not name a type
 typedef _Atomic unsigned long long atomic_ullong;
         ^~~~~~~
/boot/system/develop/tools/lib/gcc/x86_64-unknown-haiku/8.3.0/include/stdatomic.h:52:9: error: '_Atomic' does not name a type
 typedef _Atomic __CHAR16_TYPE__ atomic_char16_t;
         ^~~~~~~
/boot/system/develop/tools/lib/gcc/x86_64-unknown-haiku/8.3.0/include/stdatomic.h:53:9: error: '_Atomic' does not name a type
 typedef _Atomic __CHAR32_TYPE__ atomic_char32_t;
         ^~~~~~~
/boot/system/develop/tools/lib/gcc/x86_64-unknown-haiku/8.3.0/include/stdatomic.h:54:9: error: '_Atomic' does not name a type
 typedef _Atomic __WCHAR_TYPE__ atomic_wchar_t;
         ^~~~~~~
/boot/system/develop/tools/lib/gcc/x86_64-unknown-haiku/8.3.0/include/stdatomic.h:55:9: error: '_Atomic' does not name a type
 typedef _Atomic __INT_LEAST8_TYPE__ atomic_int_least8_t;
         ^~~~~~~
/boot/system/develop/tools/lib/gcc/x86_64-unknown-haiku/8.3.0/include/stdatomic.h:56:9: error: '_Atomic' does not name a type
[... 445 lines cut ...]
          ^
dpfp.c: In function 'alloc_transfers':
dpfp.c:551:10: warning: integer overflow in expression '-2147483648' of type 'int' results in '-2147483648' [-Woverflow]
   return -ENOMEM;
          ^
dpfp.c:555:10: warning: integer overflow in expression '-2147483648' of type 'int' results in '-2147483648' [-Woverflow]
   return -ENOMEM;
          ^
make[2]: *** No rule to make target '../libusb/libusb-1.0.la', needed by 'dpfp'.  Stop.
make[2]: Leaving directory '/boot/home/Desktop/build/libusb-1.0.25-rc1/examples'
Makefile:480: recipe for target 'all-recursive' failed
make[1]: [all-recursive] Error 1 (ignored)
make[1]: Leaving directory '/boot/home/Desktop/build/libusb-1.0.25-rc1'

~/Desktop/build/libusb-1.0.25-rc1> 
~/Desktop/build/libusb-1.0.25-rc1> gcc -v
Using built-in specs.
COLLECT_GCC=gcc
COLLECT_LTO_WRAPPER=/boot/system/develop/tools/bin/../lib/gcc/x86_64-unknown-haiku/8.3.0/lto-wrapper
Target: x86_64-unknown-haiku
Configured with: /sources/gcc-8.3.0/configure --build=x86_64-unknown-haiku --prefix=/packages/gcc-8.3.0_2019_05_24-10/.self/develop/tools --libexecdir=/packages/gcc-8.3.0_2019_05_24-10/.self/develop/tools/lib --mandir=/packages/gcc-8.3.0_2019_05_24-10/.self/documentation/man --docdir=/packages/gcc-8.3.0_2019_05_24-10/.self/documentation/packages/gcc --enable-threads=posix --disable-nls --enable-shared --with-gnu-ld --with-gnu-as --enable-version-specific-runtime-libs --enable-languages=c,c++,fortran,objc --enable-lto --enable-frame-pointer --with-pkgversion=2019_05_24 --enable-__cxa-atexit --with-system-zlib --enable-checking=release --with-bug-url=http://dev.haiku-os.org/ --with-default-libstdcxx-abi=gcc4-compatible --enable-libssp --disable-multilib
Thread model: posix
gcc version 8.3.0 (2019_05_24) 

```

tormodvolden, 2022-01-22:
> @tormodvolden Just wondering how you build usbutils (lsusb) under Raspbian Buster to get it to depend on libusb-1.0.25rc1.

I didn't rebuild lsusb, I used LD_LIBRARY_PATH to make it run-time link with the new libusb shared object.

> Somehow I tried to build usbutils (lsusb) under Raspberry Pi OS Bullseye 32bit/64bit, it seems to be always dependant on libusb-1.0.24 release. Very strange. I have even overwrite the libusb libary under /usr/lib with libusb-1.0.25rc1.

Did you use the new pkg-config files? For test-building I build libusb and e.g. lsusb with a new, common pkg-config tree, i.e. use this when configuring both build trees:
`PKG_CONFIG_PATH=~/my-build-test/lib/pkgconfig ./configure --prefix ~/my-build-test`

tormodvolden, 2022-01-22:
> Just installed Haiku r1beta3 using VirtualBox under Windows 11, it failed to build.

I think it must be a Haiku bug if the system has stdatomic.h but not _Atomic. As an experiment you can try reverting commit eed8a371.

mcuee, 2022-01-22:
@leilchen Just wondering if you can test libusb-1.0.25rc1 on Solaris, either x86_64 or Sparc. Thanks.

mcuee, 2022-01-22:
Hopefully someone can test Windows on ARM64 as well. There seem to be quite some users running Windows on ARM64 using Raspberry Pi 4B 8GB version.
https://github.com/pbatard/libwdi/issues/155

mcuee, 2022-01-22:
There is building problem as well under NetBSD 7.2 GCC 

```
make  all-recursive
Making all in libusb
  CC       os/netbsd_usb.lo
In file included from os/netbsd_usb.c:33:0:
os/netbsd_usb.c: In function ‘_access_endpoint’:
os/netbsd_usb.c:564:11: error: implicit declaration of function ‘ITRANFER_CTX’; did you mean ‘ITRANSFER_CTX’? [-Werror=implicit-function-declaration]
  usbi_dbg(ITRANFER_CTX(itransfer), "endpoint %d mode %d", endpt, mode);
           ^
./libusbi.h:315:45: note: in definition of macro ‘_usbi_log’
 #define _usbi_log(ctx, level, ...) usbi_log(ctx, level, __func__, __VA_ARGS__)
                                             ^~~
os/netbsd_usb.c:564:2: note: in expansion of macro ‘usbi_dbg’
  usbi_dbg(ITRANFER_CTX(itransfer), "endpoint %d mode %d", endpt, mode);
  ^~~~~~~~
os/netbsd_usb.c:564:24: error: ‘itransfer’ undeclared (first use in this function); did you mean ‘transfer’?
  usbi_dbg(ITRANFER_CTX(itransfer), "endpoint %d mode %d", endpt, mode);
                        ^
./libusbi.h:315:45: note: in definition of macro ‘_usbi_log’
 #define _usbi_log(ctx, level, ...) usbi_log(ctx, level, __func__, __VA_ARGS__)
                                             ^~~
os/netbsd_usb.c:564:2: note: in expansion of macro ‘usbi_dbg’
  usbi_dbg(ITRANFER_CTX(itransfer), "endpoint %d mode %d", endpt, mode);
  ^~~~~~~~
os/netbsd_usb.c:564:24: note: each undeclared identifier is reported only once for each function it appears in
  usbi_dbg(ITRANFER_CTX(itransfer), "endpoint %d mode %d", endpt, mode);
                        ^
./libusbi.h:315:45: note: in definition of macro ‘_usbi_log’
 #define _usbi_log(ctx, level, ...) usbi_log(ctx, level, __func__, __VA_ARGS__)

./libusbi.h:315:45: note: in definition of macro ‘_usbi_log’
 #define _usbi_log(ctx, level, ...) usbi_log(ctx, level, __func__, __VA_ARGS__)
                                             ^~~
os/netbsd_usb.c:564:2: note: in expansion of macro ‘usbi_dbg’
  usbi_dbg(ITRANFER_CTX(itransfer), "endpoint %d mode %d", endpt, mode);
  ^~~~~~~~
os/netbsd_usb.c:564:24: error: ‘itransfer’ undeclared (first use in this function); did you mean ‘transfer’?
  usbi_dbg(ITRANFER_CTX(itransfer), "endpoint %d mode %d", endpt, mode);
                        ^
./libusbi.h:315:45: note: in definition of macro ‘_usbi_log’
 #define _usbi_log(ctx, level, ...) usbi_log(ctx, level, __func__, __VA_ARGS__)
                                             ^~~
os/netbsd_usb.c:564:2: note: in expansion of macro ‘usbi_dbg’
  usbi_dbg(ITRANFER_CTX(itransfer), "endpoint %d mode %d", endpt, mode);
  ^~~~~~~~
os/netbsd_usb.c:564:24: note: each undeclared identifier is reported only once for each function it appears in
  usbi_dbg(ITRANFER_CTX(itransfer), "endpoint %d mode %d", endpt, mode);
                        ^
./libusbi.h:315:45: note: in definition of macro ‘_usbi_log’
 #define _usbi_log(ctx, level, ...) usbi_log(ctx, level, __func__, __VA_ARGS__)
                                             ^~~
os/netbsd_usb.c:564:2: note: in expansion of macro ‘usbi_dbg’
  usbi_dbg(ITRANFER_CTX(itransfer), "endpoint %d mode %d", endpt, mode);
  ^~~~~~~~
cc1: some warnings being treated as errors
*** Error code 1 (ignored)
  CCLD     libusb-1.0.la
libtool:   error: 'os/netbsd_usb.lo' is not a valid libtool object
*** Error code 1 (ignored)
Making all in examples
  CC       dpfp.o
make[2]: don't know how to make ../libusb/libusb-1.0.la. Stop

make[2]: stopped in /home/mcuee/build/libusb-1.0.25-rc1/examples
*** Error code 1 (ignored)

localhost$ gcc -v
Using built-in specs.
COLLECT_GCC=gcc
COLLECT_LTO_WRAPPER=/usr/libexec/lto-wrapper
Target: x86_64--netbsd
Configured with: /usr/src/tools/gcc/../../external/gpl3/gcc/dist/configure --target=x86_64--netbsd --enable-long-long --enable-threads --with-bugurl=http://www.NetBSD.org/Misc/send-pr.html --with-pkgversion='NetBSD nb4 20200810' --with-system-zlib --without-isl --enable-__cxa_atexit --enable-libstdcxx-time=rt --enable-libstdcxx-threads --with-diagnostics-color=auto-if-env --with-tune=nocona --with-default-libstdcxx-abi=new --with-mpc-lib=/var/obj/mknative/amd64-x86_64/usr/src/external/lgpl3/mpc/lib/libmpc --with-mpfr-lib=/var/obj/mknative/amd64-x86_64/usr/src/external/lgpl3/mpfr/lib/libmpfr --with-gmp-lib=/var/obj/mknative/amd64-x86_64/usr/src/external/lgpl3/gmp/lib/libgmp --with-mpc-include=/usr/src/external/lgpl3/mpc/dist/src --with-mpfr-include=/usr/src/external/lgpl3/mpfr/dist/src --with-gmp-include=/usr/src/external/lgpl3/gmp/lib/libgmp/arch/x86_64 --enable-tls --disable-multilib --disable-libstdcxx-pch --build=x86_64--netbsd --host=x86_64--netbsd --with-sysroot=/var/obj/mknative/amd64-x86_64/usr/src/destdir.amd64
Thread model: posix
gcc version 7.5.0 (nb4 20200810) 

```

```

tormodvolden, 2022-01-22:
> 
> The problematic commit seems to be this one. I tend to think the commit is probably okay but netbsd_usb.c needs minor changes. [6cae9c6](https://github.com/libusb/libusb/commit/6cae9c6dbd74c0840848f343dd605c5ddcef1ad1)

It is just a typo, try replacing TRANFER with TRANSFER.

mcuee, 2022-01-22:
For Haiku, libusb-1.0.24 seems to build okay but xusb does not seem to work. 

As for libusb-1.0.25rc1, even if I revert back the commit eed8a371 it still failed to build.

```
~/Desktop/build/libusb-1.0.25-rc1> make -i
make  all-recursive
make[1]: Entering directory '/boot/home/Desktop/build/libusb-1.0.25-rc1'
Making all in libusb
make[2]: Entering directory '/boot/home/Desktop/build/libusb-1.0.25-rc1/libusb'
  CC       core.lo
In file included from core.c:23:
libusbi.h:372:2: error: unknown type name 'usbi_atomic_t'
  usbi_atomic_t hotplug_ready;
  ^~~~~~~~~~~~~
libusbi.h:484:2: error: unknown type name 'usbi_atomic_t'
  usbi_atomic_t refcnt;
  ^~~~~~~~~~~~~
libusbi.h:498:2: error: unknown type name 'usbi_atomic_t'
  usbi_atomic_t attached;
  ^~~~~~~~~~~~~
core.c: In function 'usbi_alloc_device':
core.c:712:2: error: implicit declaration of function 'usbi_atomic_store' [-Werror=implicit-function-declaration]
  usbi_atomic_store(&dev->refcnt, 1);
  ^~~~~~~~~~~~~~~~~
core.c: In function 'libusb_ref_device':
core.c:1168:11: error: implicit declaration of function 'usbi_atomic_inc'; did you mean 'usbi_io_init'? [-Werror=implicit-function-declaration]
  refcnt = usbi_atomic_inc(&dev->refcnt);
           ^~~~~~~~~~~~~~~
           usbi_io_init
core.c: In function 'libusb_unref_device':
core.c:1186:11: error: implicit declaration of function 'usbi_atomic_dec'; did you mean 'usbi_alloc_device'? [-Werror=implicit-function-declaration]
  refcnt = usbi_atomic_dec(&dev->refcnt);
           ^~~~~~~~~~~~~~~
           usbi_alloc_device
core.c: In function 'libusb_open':
core.c:1306:7: error: implicit declaration of function 'usbi_atomic_load'; did you mean 'usbi_mutex_lock'? [-Werror=implicit-function-declaration]
  if (!usbi_atomic_load(&dev->attached))
       ^~~~~~~~~~~~~~~~
       usbi_mutex_lock
cc1: some warnings being treated as errors
Makefile:649: recipe for target 'core.lo' failed
make[2]: [core.lo] Error 1 (ignored)
  CC       descriptor.lo
In file included from descriptor.c:22:
libusbi.h:372:2: error: unknown type name 'usbi_atomic_t'
  usbi_atomic_t hotplug_ready;
  ^~~~~~~~~~~~~
libusbi.h:484:2: error: unknown type name 'usbi_atomic_t'
  usbi_atomic_t refcnt;
  ^~~~~~~~~~~~~
libusbi.h:498:2: error: unknown type name 'usbi_atomic_t'
  usbi_atomic_t attached;
  ^~~~~~~~~~~~~
Makefile:649: recipe for target 'descriptor.lo' failed
make[2]: [descriptor.lo] Error 1 (ignored)
  CC       hotplug.lo
In file included from hotplug.c:22:
libusbi.h:372:2: error: unknown type name 'usbi_atomic_t'
  usbi_atomic_t hotplug_ready;
  ^~~~~~~~~~~~~
libusbi.h:484:2: error: unknown type name 'usbi_atomic_t'
  usbi_atomic_t refcnt;
  ^~~~~~~~~~~~~
libusbi.h:498:2: error: unknown type name 'usbi_atomic_t'
  usbi_atomic_t attached;
  ^~~~~~~~~~~~~
hotplug.c: In function 'usbi_hotplug_init':
hotplug.c:161:2: error: implicit declaration of function 'usbi_atomic_store' [-Werror=implicit-function-declaration]
  usbi_atomic_store(&ctx->hotplug_ready, 1);
[... 117 lines cut ...]
  ^~~~~~~~~~~~~
  sig_atomic_t
./libusbi.h:498:2: error: 'usbi_atomic_t' does not name a type; did you mean 'sig_atomic_t'?
  usbi_atomic_t attached;
  ^~~~~~~~~~~~~
  sig_atomic_t
Makefile:673: recipe for target 'os/haiku_usb_raw.lo' failed
make[2]: [os/haiku_usb_raw.lo] Error 1 (ignored)
  CXXLD    libusb_haiku.la
libtool:   error: 'os/haiku_usb_backend.lo' is not a valid libtool object
Makefile:591: recipe for target 'libusb_haiku.la' failed
make[2]: [libusb_haiku.la] Error 1 (ignored)
  CCLD     libusb-1.0.la
libtool:   error: 'core.lo' is not a valid libtool object
Makefile:584: recipe for target 'libusb-1.0.la' failed
make[2]: [libusb-1.0.la] Error 1 (ignored)
make[2]: Leaving directory '/boot/home/Desktop/build/libusb-1.0.25-rc1/libusb'
Making all in examples
make[2]: Entering directory '/boot/home/Desktop/build/libusb-1.0.25-rc1/examples'
make[2]: *** No rule to make target '../libusb/libusb-1.0.la', needed by 'dpfp'.  Stop.
make[2]: Leaving directory '/boot/home/Desktop/build/libusb-1.0.25-rc1/examples'
Makefile:480: recipe for target 'all-recursive' failed
make[1]: [all-recursive] Error 1 (ignored)
make[1]: Leaving directory '/boot/home/Desktop/build/libusb-1.0.25-rc1'
```

tormodvolden, 2022-01-22:
> As for libusb-1.0.25rc1, even if I revert back the commit [eed8a37](https://github.com/libusb/libusb/commit/eed8a371ea53939096ba94d44001e0637d042572) it still failed to build.

Was the revert successful? There should thereafter be `typedef long usbi_atomic_t;` inside libusb/libusbi.h

mcuee, 2022-01-22:
> > The problematic commit seems to be this one. I tend to think the commit is probably okay but netbsd_usb.c needs minor changes. [6cae9c6](https://github.com/libusb/libusb/commit/6cae9c6dbd74c0840848f343dd605c5ddcef1ad1)
> 
> It is just a typo, try replacing TRANFER with TRANSFER.

Okay, after fixing the obvious typo, there are still problems.

mcuee, 2022-01-22:
> 

Oops, you are right, I forget to rename the file.

tormodvolden, 2022-01-22:
netbsd_usb.c: Please try just commenting out the usbi_dbg() line 564, it is not needed.

mcuee, 2022-01-22:
> netbsd_usb.c: Please try just commenting out the usbi_dbg() line 564, it is not needed.

Yes, that fixed the issue. So this file needs two fixes, one is the typo error and then the other is to remove this line.

mcuee, 2022-01-22:
> Or change it in line 564 to `TRANSFER_CTX(transfer)`

Yes that also works for NetBSD.

tormodvolden, 2022-01-22:
> There are still some warnings under Haiku though for the examples. I think probably not so important for now.
> ```
> dpfp.c:320:10: warning: integer overflow in expression '-2147483648' of type 'int' results in '-2147483648' [-Woverflow]
>    return -ENOMEM;
> ```

Yes, this must be a weird haiku issue: https://github.com/haiku/haiku/blob/39dfe582a355d9f6d0e3608bbec23a4421af04e9/headers/os/support/Errors.h#L193 I don't know why it pops up here.

tormodvolden, 2022-01-22:
I have pushed the fix for NetBSD to git. The two Haiku issues are likely both upstream issues, I would prefer to not add workarounds in libusb if it can be fixed there.

mcuee, 2022-01-22:
@tormodvolden Just wondering if you have the simple fixes for the MinGW GCC side. Thanks.
* https://github.com/libusb/libusb/issues/1034

tormodvolden, 2022-01-22:
> Just wondering if you have the simple fixes for the MinGW GCC side.

The warnings in #1034 are new in gcc 11(?), I think the code is not doing anything wrong, but is maybe not expressed correctly.

tormodvolden, 2022-01-22:
The Haiku warnings: Some standards say that the errno macros like ENOMEM should be positive integers. We assume this in our code, but we could also change the code to set errno=ENOMEM and return -1 instead of returning -ENOMEM. The callers check for a negative return value so our exisiting code can go wrong on Haiku where the macros are negative numbers. That we get a warning at all is because ENOMEM on Haiku is a negative integer that cannot be negated. It is easier to explain with e.g. an 8 bit integer: We have 256 values, 127 positive, 1 zero, and 128 negative values, and -128 cannot be negated.

mcuee, 2022-01-22:
> > Just wondering if you have the simple fixes for the MinGW GCC side.
> 
> The warnings in #1034 are new in gcc 11(?), I think the code is not doing anything wrong, but is maybe not expressed correctly.

This is not a regression though, so probably not a big issue. libusb-1.0.24 release will generate the same warning messages.

```
MINGW64 /c/work/libusb/libusb-1.0.24
$ make
make  all-recursive
make[1]: Entering directory '/c/work/libusb/libusb-1.0.24'
Making all in libusb
make[2]: Entering directory '/c/work/libusb/libusb-1.0.24/libusb'
  CC       core.lo
  CC       descriptor.lo
  CC       hotplug.lo
  CC       io.lo
  CC       strerror.lo
  CC       sync.lo
  CC       os/events_windows.lo
  CC       os/threads_windows.lo
  RC       libusb-1.0.lo
  CC       os/windows_common.lo
  CC       os/windows_usbdk.lo
  CC       os/windows_winusb.lo
os/windows_winusb.c: In function '_hid_get_descriptor.constprop':
os/windows_winusb.c:3260:21: warning: array subscript 'struct libusb_interface_descriptor[0]' is partly outside array bounds of 'char[41]' [-Warray-bounds]
 3260 |         id->bLength = LIBUSB_DT_INTERFACE_SIZE;
os/windows_winusb.c:3228:14: note: while referencing 'tmp'
 3228 |         char tmp[HID_MAX_CONFIG_DESC_SIZE];
      |              ^~~
os/windows_winusb.c:3261:29: warning: array subscript 'struct libusb_interface_descriptor[0]' is partly outside array bounds of 'char[41]' [-Warray-bounds]
 3261 |         id->bDescriptorType = LIBUSB_DT_INTERFACE;
      |         ~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~
os/windows_winusb.c:3228:14: note: while referencing 'tmp'
 3228 |         char tmp[HID_MAX_CONFIG_DESC_SIZE];
      |              ^~~
os/windows_winusb.c:3262:30: warning: array subscript 'struct libusb_interface_descriptor[0]' is partly outside array bounds of 'char[41]' [-Warray-bounds]
 3262 |         id->bInterfaceNumber = 0;
      |         ~~~~~~~~~~~~~~~~~~~~~^~~
os/windows_winusb.c:3228:14: note: while referencing 'tmp'
 3228 |         char tmp[HID_MAX_CONFIG_DESC_SIZE];
      |              ^~~
os/windows_winusb.c:3263:31: warning: array subscript 'struct libusb_interface_descriptor[0]' is partly outside array bounds of 'char[41]' [-Warray-bounds]
 3263 |         id->bAlternateSetting = 0;
      |         ~~~~~~~~~~~~~~~~~~~~~~^~~
os/windows_winusb.c:3228:14: note: while referencing 'tmp'
 3228 |         char tmp[HID_MAX_CONFIG_DESC_SIZE];
      |              ^~~
os/windows_winusb.c:3264:27: warning: array subscript 'struct libusb_interface_descriptor[0]' is partly outside array bounds of 'char[41]' [-Warray-bounds]
 3264 |         id->bNumEndpoints = num_endpoints;
      |         ~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~
os/windows_winusb.c:3228:14: note: while referencing 'tmp'
 3228 |         char tmp[HID_MAX_CONFIG_DESC_SIZE];
      |              ^~~
os/windows_winusb.c:3265:29: warning: array subscript 'struct libusb_interface_descriptor[0]' is partly outside array bounds of 'char[41]' [-Warray-bounds]
 3265 |         id->bInterfaceClass = 3;
      |         ~~~~~~~~~~~~~~~~~~~~^~~
os/windows_winusb.c:3228:14: note: while referencing 'tmp'
 3228 |         char tmp[HID_MAX_CONFIG_DESC_SIZE];
      |              ^~~
os/windows_winusb.c:3266:32: warning: array subscript 'struct libusb_interface_descriptor[0]' is partly outside array bounds of 'char[41]' [-Warray-bounds]
 3266 |         id->bInterfaceSubClass = 0;
      |         ~~~~~~~~~~~~~~~~~~~~~~~^~~
os/windows_winusb.c:3228:14: note: while referencing 'tmp'
 3228 |         char tmp[HID_MAX_CONFIG_DESC_SIZE];
      |              ^~~
os/windows_winusb.c:3267:32: warning: array subscript 'struct libusb_interface_descriptor[0]' is partly outside array bounds of 'char[41]' [-Warray-bounds]
 3267 |         id->bInterfaceProtocol = 0;
      |         ~~~~~~~~~~~~~~~~~~~~~~~^~~
[... 51 lines cut ...]
  CCLD     dpfp_threaded.exe
  CC       ezusb.o
  CC       fxload.o
  CCLD     fxload.exe
  CC       hotplugtest.o
  CCLD     hotplugtest.exe
  CC       listdevs.o
  CCLD     listdevs.exe
  CC       sam3u_benchmark.o
  CCLD     sam3u_benchmark.exe
  CC       testlibusb.o
  CCLD     testlibusb.exe
  CC       xusb.o
  CCLD     xusb.exe
make[2]: Leaving directory '/c/work/libusb/libusb-1.0.24/examples'
Making all in tests
make[2]: Entering directory '/c/work/libusb/libusb-1.0.24/tests'
  CC       stress.o
  CC       testlib.o
  CCLD     stress.exe
make[2]: Leaving directory '/c/work/libusb/libusb-1.0.24/tests'
make[2]: Entering directory '/c/work/libusb/libusb-1.0.24'
make[2]: Leaving directory '/c/work/libusb/libusb-1.0.24'
make[1]: Leaving directory '/c/work/libusb/libusb-1.0.24'
```

mcuee, 2022-01-22:
> > Just wondering if you have the simple fixes for the MinGW GCC side.
> 
> The warnings in #1034 are new in gcc 11(?), I think the code is not doing anything wrong, but is maybe not expressed correctly.

Some think it is a GCC compiler bug. So we can leave it as it is.
Ref: https://gitlab.gnome.org/GNOME/gtk/-/issues/3776
https://sourceware.org/bugzilla/show_bug.cgi?id=26687

hjelmn, 2022-01-22:
> I think it must be a Haiku bug if the system has stdatomic.h but not _Atomic. As an experiment you can try reverting commit [...]

_Atomic is a compiler directive. We may need to add ```-std=c11```. I can look at the configury and see what is wrong.

tormodvolden, 2022-01-22:
With `make V=1` I see -std=gnu11 here on Linux, both for native and MinGW builds.

mcuee, 2022-01-23:
> With `make V=1` I see -std=gnu11 here on Linux, both for native and MinGW builds.

Same for MSYS2 MinGW-w64 compiler under Windows.

```
make[2]: Entering directory '/c/work/libusb/libusb_test/libusb-1.0.25-rc1/build64/libusb'
depbase=`echo core.lo | sed 's|[^/]*$|.deps/&|;s|\.lo$||'`;\
/bin/sh ../libtool  --tag=CC   --mode=compile gcc -DHAVE_CONFIG_H -I. -I../../libusb -I..    -std=gnu11 -mwin32 -fno-omit-frame-pointer -Wno-cast-function-type -Wall -Wextra -Wshadow -Wunused -Wwrite-strings -Werror=format-security -Werror=implicit-function-declaration -Werror=implicit-int -Werror=init-self -Werror=missing-prototypes -Werror=strict-prototypes -Werror=undef -Werror=uninitialized -fvisibility=hidden  -g -O2 -MT core.lo -MD -MP -MF $depbase.Tpo -c -o core.lo ../../libusb/core.c &&\
mv -f $depbase.Tpo $depbase.Plo
libtool: compile:  gcc -DHAVE_CONFIG_H -I. -I../../libusb -I.. -std=gnu11 -mwin32 -fno-omit-frame-pointer -Wno-cast-function-type -Wall -Wextra -Wshadow -Wunused -Wwrite-strings -Werror=format-security -Werror=implicit-function-declaration -Werror=implicit-int -Werror=init-self -Werror=missing-prototypes -Werror=strict-prototypes -Werror=undef -Werror=uninitialized -fvisibility=hidden -g -O2 -MT core.lo -MD -MP -MF .deps/core.Tpo -c ../../libusb/core.c  -DDLL_EXPORT -DPIC -o .libs/core.o
libtool: compile:  gcc -DHAVE_CONFIG_H -I. -I../../libusb -I.. -std=gnu11 -mwin32 -fno-omit-frame-pointer -Wno-cast-function-type -Wall -Wextra -Wshadow -Wunused -Wwrite-strings -Werror=format-security -Werror=implicit-function-declaration -Werror=implicit-int -Werror=init-self -Werror=missing-prototypes -Werror=strict-prototypes -Werror=undef -Werror=uninitialized -fvisibility=hidden -g -O2 -MT core.lo -MD -MP -MF .deps/core.Tpo -c ../../libusb/core.c -o core.o >/dev/null 2>&1
```

mcuee, 2022-01-23:
Same for Haiku.

```
~/Desktop/build/libusb-1.0.25-rc1> make V=1
make  all-recursive
make[1]: Entering directory '/boot/home/Desktop/build/libusb-1.0.25-rc1'
Making all in libusb
make[2]: Entering directory '/boot/home/Desktop/build/libusb-1.0.25-rc1/libusb'
depbase=`echo core.lo | sed 's|[^/]*$|.deps/&|;s|\.lo$||'`;\
/bin/sh ../libtool  --tag=CC   --mode=compile gcc -DHAVE_CONFIG_H -I. -I..    -std=gnu11  -Wall -Wextra -Wshadow -Wunused -Wwrite-strings -Werror=format-security -Werror=implicit-function-declaration -Werror=implicit-int -Werror=init-self -Werror=missing-prototypes -Werror=strict-prototypes -Werror=undef -Werror=uninitialized -fvisibility=hidden  -g -O2 -MT core.lo -MD -MP -MF $depbase.Tpo -c -o core.lo core.c &&\
mv -f $depbase.Tpo $depbase.Plo
libtool: compile:  gcc -DHAVE_CONFIG_H -I. -I.. -std=gnu11 -Wall -Wextra -Wshadow -Wunused -Wwrite-strings -Werror=format-security -Werror=implicit-function-declaration -Werror=implicit-int -Werror=init-self -Werror=missing-prototypes -Werror=strict-prototypes -Werror=undef -Werror=uninitialized -fvisibility=hidden -g -O2 -MT core.lo -MD -MP -MF .deps/core.Tpo -c core.c  -DPIC -o .libs/core.o
libtool: compile:  gcc -DHAVE_CONFIG_H -I. -I.. -std=gnu11 -Wall -Wextra -Wshadow -Wunused -Wwrite-strings -Werror=format-security -Werror=implicit-function-declaration -Werror=implicit-int -Werror=init-self -Werror=missing-prototypes -Werror=strict-prototypes -Werror=undef -Werror=uninitialized -fvisibility=hidden -g -O2 -MT core.lo -MD -MP -MF .deps/core.Tpo -c core.c -o core.o >/dev/null 2>&1
```

mcuee, 2022-01-23:
>

mcuee, 2022-01-23:
> I didn't rebuild lsusb, I used LD_LIBRARY_PATH to make it run-time link with the new libusb shared object.

Oh, yes this works.
```
pi@raspberrypi:~/build/libusb/usbutils $ export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/usr/local/lib
pi@raspberrypi:~/build/libusb/usbutils $ lsusb
[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.000131] [000023eb] libusb: debug [libusb_init] libusb v1.0.25.11688-rc1

```

mcuee, 2022-01-23:
> Did you use the new pkg-config files? For test-building I build libusb and e.g. lsusb with a new, common pkg-config tree, i.e. use this when configuring both build trees: PKG_CONFIG_PATH=~/my-build-test/lib/pkgconfig ./configure --prefix ~/my-build-test

I just install the new libusb-1.0.25rc1 to /usr/local and pkg-config will correctly point to the right location. But somehow lsusb is still insisting to link to /usr/lib and not /usr/local/lib. Kind of strange.

```
pi@raspberrypi:~/build/libusb/usbutils $ pkg-config --cflags libusb-1.0
-I/usr/local/include/libusb-1.0
pi@raspberrypi:~/build/libusb/usbutils $ pkg-config --libs libusb-1.0
-L/usr/local/lib -lusb-1.0

```

tormodvolden, 2022-01-23:
I guess you can at least override it: `LD_LIBRARY_PATH=/usr/local/lib ./lsusb`

mcuee, 2022-01-23:
> I guess you can at least override it: `LD_LIBRARY_PATH=/usr/local/lib ./lsusb`

Yes that works. I need to figure out why it does not work even after I install libusb-1.0.25rc1 to /usr/lib. Kind of strange. I did not encounter such issues under other Linux distros.

tormodvolden, 2022-01-23:
Check /etc/ld.so.conf (and the files it includes). Normally /usr/local/bin is added there by the distribution. These files are used to generate a cache file, which is looked up before resorting to the default paths /lib and /usr/lib while run-time linking.

tormodvolden, 2022-01-23:
> even after I install libusb-1.0.25rc1 to /usr/lib

I missed that earlier. In that case it sounds like your lsusb has libusb linked statically. Check with `ldd /usr/bin/lsusb`

jkunkee, 2022-01-24:
I'm happy to try it on ARM64 Windows, but I'm a newbie to what's needed here. Is what's needed is as simple as pulling the rc1 tarball off the releases page, pointing VS 2019 at it to build for ARM64, then running some set of commands to verify it's working?

I gather that could include tools that come out of the build itself (xusb, listdev, testlibusb, stress?) and a project that consumes it (libftdi 1.5 release (find_all, eeprom)). If you have suggestions or better instructions, I'm all ears; hardware-wise, though, the only WinUSB device I have is an RTL-SDR. (Hmm. My Cypress PSoC 3 dev board may currently be set up as a WinUSB device too...)

I'll see if I can't squeeze just xusb and listdev in sometime tomorrow night.

tormodvolden, 2022-01-24:
@jkunkee Yes, it should be as simple as that. The more the better but if you already can check that it builds (please report any warnings) and run some of the tools that would be great.

mcuee, 2022-01-25:
> Yes, it should be as simple as that. The more the better but if you already can check that it builds (please report any warnings) and run some of the tools that would be great.

I can test the build with my VS2019 installations. I do not see issues with building 1.0.25rc1 with ARM and ARM64.

```
Build started...
1>------ Build started: Project: libusb-1.0 (static), Configuration: Release ARM ------
2>------ Build started: Project: getopt, Configuration: Release ARM ------
3>------ Build started: Project: libusb-1.0 (dll), Configuration: Release ARM ------
3>core.c
2>getopt.c
1>core.c
2>getopt1.c
3>descriptor.c
1>descriptor.c
1>events_windows.c
3>events_windows.c
3>hotplug.c
1>hotplug.c
2>getopt_2019.vcxproj -> C:\work\libusb\libusb-1.0.25-rc1\ARM\Release\lib\getopt.lib
3>io.c
1>io.c
3>strerror.c
1>strerror.c
3>sync.c
1>sync.c
3>threads_windows.c
1>threads_windows.c
3>windows_common.c
1>windows_common.c
3>windows_usbdk.c
1>windows_usbdk.c
3>windows_winusb.c
1>windows_winusb.c
1>libusb_static_2019.vcxproj -> C:\work\libusb\libusb-1.0.25-rc1\ARM\Release\lib\libusb-1.0.lib
4>------ Build started: Project: dpfp, Configuration: Release ARM ------
5>------ Build started: Project: dpfp_threaded, Configuration: Release ARM ------
6>------ Build started: Project: fxload, Configuration: Release ARM ------
7>------ Build started: Project: hotplugtest, Configuration: Release ARM ------
8>------ Build started: Project: listdevs, Configuration: Release ARM ------
9>------ Build started: Project: sam3u_benchmark, Configuration: Release ARM ------
10>------ Build started: Project: stress, Configuration: Release ARM ------
11>------ Build started: Project: testlibusb, Configuration: Release ARM ------
12>------ Build started: Project: xusb, Configuration: Release ARM ------
3>   Creating library C:\work\libusb\libusb-1.0.25-rc1\msvc\..\ARM\Release\dll\libusb-1.0.lib and object C:\work\libusb\libusb-1.0.25-rc1\msvc\..\ARM\Release\dll\libusb-1.0.exp
3>Generating code
3>Finished generating code
4>dpfp.c
8>listdevs.c
5>dpfp.c
12>xusb.c
9>sam3u_benchmark.c
7>hotplugtest.c
6>ezusb.c
10>stress.c
3>libusb_dll_2019.vcxproj -> C:\work\libusb\libusb-1.0.25-rc1\ARM\Release\dll\libusb-1.0.dll
11>testlibusb.c
6>fxload.c
10>testlib.c
8>Generating code
4>Generating code
5>Generating code
12>Generating code
9>Generating code
7>Generating code
4>Finished generating code
8>Finished generating code
5>Finished generating code
10>Generating code
7>Finished generating code
11>Generating code
9>Finished generating code
12>Finished generating code
6>Generating code
10>Finished generating code
11>Finished generating code
6>Finished generating code
4>dpfp_2019.vcxproj -> C:\work\libusb\libusb-1.0.25-rc1\ARM\Release\examples\dpfp.exe
8>listdevs_2019.vcxproj -> C:\work\libusb\libusb-1.0.25-rc1\ARM\Release\examples\listdevs.exe
5>dpfp_threaded_2019.vcxproj -> C:\work\libusb\libusb-1.0.25-rc1\ARM\Release\examples\dpfp_threaded.exe
7>hotplugtest_2019.vcxproj -> C:\work\libusb\libusb-1.0.25-rc1\ARM\Release\examples\hotplugtest.exe
9>sam3u_benchmark_2019.vcxproj -> C:\work\libusb\libusb-1.0.25-rc1\ARM\Release\examples\sam3u_benchmark.exe
12>xusb_2019.vcxproj -> C:\work\libusb\libusb-1.0.25-rc1\ARM\Release\examples\xusb.exe
10>stress_2019.vcxproj -> C:\work\libusb\libusb-1.0.25-rc1\ARM\Release\tests\stress.exe
11>testlibusb_2019.vcxproj -> C:\work\libusb\libusb-1.0.25-rc1\ARM\Release\examples\testlibusb.exe
6>fxload_2019.vcxproj -> C:\work\libusb\libusb-1.0.25-rc1\ARM\Release\examples\fxload.exe
========== Build: 12 succeeded, 0 failed, 0 up-to-date, 0 skipped ==========

```

tormodvolden, 2022-01-25:
I am glad to see the rc-1 has already [been picked up](https://repology.org/project/libusb/history) by Debian experimental.

That will give us build testing on many other [hardware platforms](https://buildd.debian.org/status/package.php?p=libusb-1.0&suite=experimental) and hopefully also some real-life testing.

jkunkee, 2022-01-26:
Using VS 2022 with the Desktop C++ workload installed and the ARM64 MSVC toolchain added on ARM64 Windows 11, I was also able to build the entire MSVC solution with no issues.

With an admin CMD session on a Surface Pro X:

`stress` passes:

```
>C:\temp\libusb-1.0.25-rc1\ARM64\Debug\tests\stress.exe
Starting test run: init_and_exit...
Success (0)
Starting test run: get_device_list...
Success (0)
Starting test run: many_device_lists...
Success (0)
Starting test run: default_context_change...
Success (0)
---
Ran 4 tests
Passed 4 tests
Failed 0 tests
Error in 0 tests
Skipped 0 tests
```

`listdevs` doesn't seem to do much:

```
>C:\temp\libusb-1.0.25-rc1\ARM64\Debug\examples\listdevs.exe
1d6b:0004 (bus 1, device 0)
```

`xusb -i` reports "Failed.":

```
>C:\temp\libusb-1.0.25-rc1\ARM64\Debug\examples\xusb.exe -i
Using libusb v1.0.25.11688

Opening device 0000:0000...
libusb: warning [enumerate_hcd_root_hub] could not infer VID/PID of HCD root hub from 'ACPI\QCOM04A6\2'
  Failed.
```

Not sure if this was expected, I rebuilt with this in msvc/config.h (per the excellent `INSTALL_WIN.txt`):

```
/* Define to 1 to start with debug message logging enabled. */
/* #undef ENABLE_DEBUG_LOGGING */
#define ENABLE_DEBUG_LOGGING 1
```

and it makes `xusb -i` much, much chattier, mostly ignoring non-USB HID devices, but no helpful failure reason I could see:

```
>C:\temp\libusb-1.0.25-rc1\ARM64\Debug\examples\xusb.exe -i
Using libusb v1.0.25.11688

[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.000324] [00008694] libusb: debug [libusb_init] created default context
[ 0.000565] [00008694] libusb: debug [libusb_init] libusb v1.0.25.11688-rc1
[ 0.000764] [00008694] libusb: debug [usbi_add_event_source] add HANDLE 00000000000000A0 events 0
[ 0.001029] [00008694] libusb: debug [usbi_io_init] using timer for timeouts
[ 0.001200] [00008694] libusb: debug [usbi_add_event_source] add HANDLE 00000000000000A8 events 0
[ 0.001426] [00008694] libusb: debug [get_windows_version] Windows 8 64-bit
[ 0.001604] [00008694] libusb: debug [htab_create] using 1021 entries hash table
[ 0.011555] [00008694] libusb: info [winusbx_init] WinUSB DLL available (with isoch support)
[ 0.011971] [00008694] libusb: info [winusbx_init] libusbK DLL is not available
[ 0.016433] [00008694] libusb: info [windows_init] UsbDk backend is not available
Opening device 0000:0000...
[ 0.017043] [00008694] libusb: debug [libusb_get_device_list]
[... 25 lines cut ...]
[ 0.075567] [00008694] libusb: debug [winusb_get_device_list] unlisted ancestor for 'HID\MSHW0134&COL07\3&2466042B&0&0006' (non USB HID, newly connected, etc.) - ignoring
[ 0.075851] [00008694] libusb: debug [winusb_get_device_list] unlisted ancestor for 'HID\MSHW0125&COL01\4&1AA43153&0&0000' (non USB HID, newly connected, etc.) - ignoring
[ 0.076152] [00008694] libusb: debug [winusb_get_device_list] unlisted ancestor for 'HID\TARGET_KIP&CATEGORY_HID&COL04\4&229E758&3&0003' (non USB HID, newly connected, etc.) - ignoring
[ 0.076441] [00008694] libusb: debug [winusb_get_device_list] unlisted ancestor for 'HID\MSHW0134&COL08\3&2466042B&0&0007' (non USB HID, newly connected, etc.) - ignoring
[ 0.076839] [00008694] libusb: debug [winusb_get_device_list] unlisted ancestor for 'HID\MSHW0125&COL02\4&1AA43153&0&0001' (non USB HID, newly connected, etc.) - ignoring
[ 0.077101] [00008694] libusb: debug [winusb_get_device_list] unlisted ancestor for 'HID\CONVERTEDDEVICE&COL03\6&7F6DA4C&0&0002' (non USB HID, newly connected, etc.) - ignoring
[ 0.077444] [00008694] libusb: debug [winusb_get_device_list] unlisted ancestor for 'HID\CONVERTEDDEVICE&COL03\A&7B44228&0&0002' (non USB HID, newly connected, etc.) - ignoring
[ 0.078004] [00008694] libusb: debug [winusb_get_device_list] unlisted ancestor for 'HID\TARGET_KIP&CATEGORY_HID&COL05\4&229E758&3&0004' (non USB HID, newly connected, etc.) - ignoring
[ 0.078295] [00008694] libusb: debug [winusb_get_device_list] unlisted ancestor for 'HID\HID_DEVICE_SYSTEM_VHF\8&C0762BD&0&0000' (non USB HID, newly connected, etc.) - ignoring
[ 0.078601] [00008694] libusb: debug [winusb_get_device_list] unlisted ancestor for 'HID\HID_DEVICE_SYSTEM_VHF&COL01\4&2DB19FD5&0&0000' (non USB HID, newly connected, etc.) - ignoring
[ 0.078889] [00008694] libusb: debug [winusb_get_device_list] unlisted ancestor for 'HID\HID_DEVICE_SYSTEM_VHF&COL02\4&2DB19FD5&0&0001' (non USB HID, newly connected, etc.) - ignoring
[ 0.079200] [00008694] libusb: debug [winusb_get_device_list] unlisted ancestor for 'HID\CONVERTEDDEVICE&COL01\A&7B44228&0&0000' (non USB HID, newly connected, etc.) - ignoring
[ 0.079492] [00008694] libusb: debug [winusb_get_device_list] unlisted ancestor for 'HID\HID_DEVICE_SYSTEM_VHF&COL02\4&4F20D5D&0&0001' (non USB HID, newly connected, etc.) - ignoring
[ 0.079760] [00008694] libusb: debug [winusb_get_device_list] unlisted ancestor for 'HID\CONVERTEDDEVICE&COL01\6&7F6DA4C&0&0000' (non USB HID, newly connected, etc.) - ignoring
[ 0.080050] [00008694] libusb: debug [winusb_get_device_list] unlisted ancestor for 'HID\TARGET_KIP&CATEGORY_HID&COL01\4&18515015&3&0000' (non USB HID, newly connected, etc.) - ignoring
[ 0.081354] [00008694] libusb: debug [libusb_get_device_descriptor]
[ 0.081546] [00008694] libusb: debug [libusb_unref_device] destroy device 1.0
  Failed.
[ 0.081858] [00008694] libusb: debug [libusb_exit] destroying default context
[... cut ...] 64 and IA32 systems for at least a decade. As I understand it it's not actually a standard--and so, painful on a few different levels, though it's stable so INFs don't need to change--so the parsing will typically fail with ACPI-attached devices like this. While the ACPI table can be tweaked to convince acpi.sys to output _almost_ the classic format (by adding _HW, IIRC--fun story), I think the better solution is, _in time_, to start to allow for this increasingly common class of device strings.

mcuee, 2022-01-26:
Thanks a lot for the testing under ARM64.

Just wondering if you can attach a generic USB device to see how "listdev" and "xusb" work under your system. There is no Zardig for Windows on ARM64 yet so you are probably not able to switch driver. 

BTW, you do not need to rebuild libusb, rather you can use "xusb -d" to print out the debug message for xusb. Or you can use environmental variable LIBUSB_DEBUG=4 to get the debug info.

mcuee, 2022-01-26:
> Side note: [ 0.062403] [00008694] libusb: warning [enumerate_hcd_root_hub] could not infer VID/PID of HCD root hub from 'ACPI\QCOM04A6\2' suggests to me that this is expecting the PCI VID/PID string [...]

Good point. We may need to support this kind of HCD root hub down the road.

jkunkee, 2022-01-27:
Using the Pro X docked this time, I have a number of USB-attached devices including two mass-storage, two HID, and one known WinUSB device (funny enough, my Startech USB-DVI adapter).

`listdevs` exits with 0 and lists
```
045e:09ac (bus 1, device 6) path: 1.5
045e:09b4 (bus 1, device 4) path: 1.1
045e:09aa (bus 1, device 2) path: 3.1
1d6b:0005 (bus 1, device 0)
1d6b:0004 (bus 2, device 0)
045e:0a1b (bus 1, device 15) path: 1.1.3.5
04b4:00f3 (bus 1, device 14) path: 1.1.3.7
045e:07a5 (bus 1, device 11) path: 1.3
045e:09a0 (bus 1, device 9) path: 3.1.1
0951:16ae (bus 2, device 1) path: 2
046d:c077 (bus 1, device 7) path: 1.1.4
1a40:0201 (bus 1, device 10) path: 1.1.3
045e:09a8 (bus 1, device 8) path: 1.1.5
239a:8108 (bus 1, device 13) path: 1.1.3.3
045e:09b3 (bus 1, device 3) path: 1
045e:09a9 (bus 1, device 1) path: 3
04b4:0003 (bus 1, device 12) path: 1.1.3.1
1fc9:5002 (bus 1, device 5) path: 1.4
```

`xusb.exe -i` still fails, both 'has no driver' devices are being correctly detected as banged out in Device Manager:
```
Using libusb v1.0.25.11688

Opening device 0000:0000...
libusb: warning [enumerate_hcd_root_hub] could not infer VID/PID of HCD root hub from 'ACPI\QCOM04A6\2'
libusb: warning [enumerate_hcd_root_hub] could not infer VID/PID of HCD root hub from 'URS\VEN_QCOM&DEV_0497&SUBSYS_CLS08180&HOST\3&252BAFA3&1&0'
libusb: info [winusb_get_device_list] The following device has no driver: 'USB\VID_04B4&PID_00F3\0000000004BE'
libusb: info [winusb_get_device_list] libusb will not be able to access it
libusb: info [init_device] 0x4b4:0xf3 found 1 configurations but device isn't configured, forcing current configuration to 1
libusb: info [winusb_get_device_list] The following device has no driver: 'USB\VID_04B4&PID_0003&MI_02\8&2CDF079B&0&0002'
libusb: info [winusb_get_device_list] libusb will not be able to access it
  Failed.
```

Dangerous question time: what is `xusb` trying to do? I'll hold off on dumping the debug output here again for now.

`testlibusb.exe` seems to do well:
```
Dev (bus 1, device 6): 045E - 09AC speed: 12M
  Manufacturer:              Microsoft
  Product:                   Surface Dock 2
Dev (bus 1, device 4): 045E - 09B4 speed: 480M
Dev (bus 1, device 2): 045E - 09AA speed: 10G
Dev (bus 1, device 0): 1D6B - 0005 speed: 10G
Dev (bus 2, device 0): 1D6B - 0004 speed: 5G
Dev (bus 1, device 15): 045E - 0A1B speed: 12M
  Product:                   Surface Headphones USB
Dev (bus 1, device 14): 04B4 - 00F3 speed: 480M
Dev (bus 1, device 11): 045E - 07A5 speed: 12M
Dev (bus 1, device 9): 045E - 09A0 speed: 5G
Dev (bus 2, device 1): 0951 - 16AE speed: 5G
Dev (bus 1, device 7): 046D - C077 speed: 1.5M
  Manufacturer:              Logitech
  Product:                   USB Optical Mouse
Dev (bus 1, device 10): 1A40 - 0201 speed: 480M
Dev (bus 1, device 8): 045E - 09A8 speed: 480M
  Manufacturer:              Generic
  Product:                   USB Audio
Dev (bus 1, device 13): 239A - 8108 speed: 12M
  Manufacturer:              Adafruit
  Product:                   CircuitPython HID
Dev (bus 1, device 3): 045E - 09B3 speed: 480M
Dev (bus 1, device 1): 045E - 09A9 speed: 10G
Dev (bus 1, device 12): 04B4 - 0003 speed: 12M
Dev (bus 1, device 5): 1FC9 - 5002 speed: 12M
  Manufacturer:              NXP
  Product:                   PTN5002
```

`hotplugtest.exe` doesn't seem to work:
```
Hotplug capabilities are not supported on this platform
```

jkunkee, 2022-01-27:
(I do apologize that my lack of familiarity with the finer points of libusb leads me to post output instead of conclusions. I hope it's still useful.)

mcuee, 2022-01-27:
> (I do apologize that my lack of familiarity with the finer points of libusb leads me to post output instead of conclusions. I hope it's still useful.)

Thanks for the testing. For xusb please use "xusb -d vid:pid" for the WinUSB and HID device you have. Thanks.

tormodvolden, 2022-01-27:
The build tests and simple programs seem to go very well. Thanks to all your testing on an amazingly extensive list of platforms.
Are there any test results on more complex programs, e.g. using isochronous transfers? There is a streaming demo in examples/ for a SAM3U microcontroller but I don't have any of these.

mcuee, 2022-01-28:
> The build tests and simple programs seem to go very well. Thanks to all your testing on an amazingly extensive list of platforms. Are there any test results on more complex programs, e.g. using [...]

Problems with the isochronous transfer testing is to have a known-good FW. Right now we have some known good FW from Cypress for Cyusb EZ-USB FX2-LP and FX3. libusbk has some know good examples for Cypress EZ-USB FX3 as well. Unfortunately there is no corresponding demo program available for libusb yet. I have done some simple tests with pyusb or libusbdotnet -- which indirectly use libusb. But it is hard to debug the issues sometimes.

mcuee, 2022-01-28:
Based on my limited testing, libusb isoc works well under Linux and macOS, but not Windows. We know there are problems with libusb isoc especially the WinUSB isoc codes are not mature yet.

Ref: https://github.com/pyusb/pyusb/issues/380

jkunkee, 2022-01-28:
[still on Windows on Arm]

testlibusb -v looks like a wonderfully detailed report. No errors from what I can tell; happy to share the log if you're interested.

`xusb -i vid:pid` is able to talk to the Startech WinUSB device I mentioned earlier, but it still reports 'Failed.' (I'm slowly getting my head wrapped around what to expect here and there; thanks again for your patience.)

Normal log:
```
>ARM64\Debug\examples\xusb -i 1fc9:5002
Using libusb v1.0.25.11688

Opening device 1FC9:5002...
libusb: warning [enumerate_hcd_root_hub] could not infer VID/PID of HCD root hub from 'ACPI\QCOM04A6\2'
libusb: info [winusb_get_device_list] The following device has no driver: 'USB\VID_04B4&PID_0003&MI_02\8&2CDF079B&0&0002'
libusb: info [winusb_get_device_list] libusb will not be able to access it

Device properties:
        bus number: 1
         port path: 1->4 (from root hub)
             speed: 12 Mbit/s (USB FullSpeed)

Reading device descriptor:
            length: 18
      device class: 0
               S/N: 3
           VID:PID: 1FC9:5002
         bcdDevice: 0100
   iMan:iProd:iSer: 1:2:3
          nb confs: 1

Reading BOS descriptor: 3 caps
    USB 2.0 extension:
      attributes             : 06
    Container ID:
      {31700104-468c-4baf-dd47-895d071c00f5}
    Unknown BOS device capability 0d:

Reading first configuration descriptor:
              total length: 27
         descriptor length: 9
             nb interfaces: 1
              interface[0]: id = 0
interface[0].altsetting[0]: num endpoints = 0
   Class.SubClass.Protocol: FE.01.01

Kernel driver attached for interface 0: -12

Claiming interface 0...

Reading string descriptors:
   String (0x01): "NXP"
   String (0x02): "PTN5002"
   String (0x03): "000004cf7dc5"

Reading OS string descriptor:

Reading Extended Compat ID OS Feature Descriptor (wIndex = 0x0004):

  00000000  28 00 00 00 00 01 04 00 01 00 00 00 00 00 00 00  (...............
  00000010  00 01 57 49 4e 55 53 42 00 00 00 00 00 00 00 00  ..WINUSB........
  00000020  00 00 00 00 00 00 00 00                          ........

Reading Extended Properties OS Feature Descriptor (wIndex = 0x0005):
   Failed: Pipe error
Releasing interface 0...
Closing device...
```

Here's a snippet of the log from the segment that appears to me to be the issue:
```
Reading Extended Properties OS Feature Descriptor (wIndex = 0x0005):
[ 0.102108] [0000131c] libusb: debug [libusb_submit_transfer] transfer 000002182CC3F8A8
[ 0.102209] [0000131c] libusb: debug [add_to_flying_list] arm timer for timeout in 1000ms (first in line)
[ 0.102313] [0000131c] libusb: debug [libusb_claim_interface] interface 0
[ 0.102441] [0000131c] libusb: debug [auto_claim] auto-claimed interface 0 for control request
[ 0.102540] [0000131c] libusb: debug [winusbx_submit_control_transfer] will use interface 0
[ 0.102769] [0000131c] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[ 0.102868] [0000131c] libusb: debug [usbi_wait_for_events] WaitForMultipleObjects() for 2 HANDLEs with timeout in 60000ms
[ 0.102973] [00006624] libusb: debug [windows_iocp_thread] transfer 000002182CC3F8A8 completed, length 0
[ 0.103094] [0000131c] libusb: debug [usbi_wait_for_events] WaitForMultipleObjects() returned 0
[ 0.103188] [0000131c] libusb: debug [handle_event_trigger] event triggered
[ 0.103291] [0000131c] libusb: debug [windows_handle_transfer_completion] handling transfer 000002182CC3F8A8 completion with errcode 31, length 0
[ 0.103428] [0000131c] libusb: debug [windows_handle_transfer_completion] detected endpoint stall
[ 0.103532] [0000131c] libusb: debug [libusb_release_interface] interface 0
[ 0.103641] [0000131c] libusb: debug [auto_release] auto-released interface 0
[ 0.103749] [0000131c] libusb: debug [arm_timer_for_next_timeout] no timeouts, disarming timer
[ 0.103926] [0000131c] libusb: debug [usbi_handle_transfer_completion] transfer 000002182CC3F8A8 has callback 00007FF78A567D08
[ 0.104048] [0000131c] libusb: debug [sync_transfer_cb] actual_length=0
[ 0.104148] [0000131c] libusb: debug [libusb_free_transfer] transfer 000002182CC3F8A8
   Failed: Pipe error
```

Decimal 31 maps to (among other things) `ERROR_GEN_FAILURE`. At that point I'm not sure if it's a device bug or something further up the stack; I'll try and run the same test on my x64 machine to narrow it down (or up, as the case may be :).

mcuee, 2022-01-28:
@jkunkee Thanks for the updates. I believe the pipe error for xusb is okay if your device does not have the Extended Properties OS Feature Descriptor.

tormodvolden, 2022-01-28:
> Decimal 31 maps to (among other things) `ERROR_GEN_FAILURE`. At that point I'm not sure if it's a device bug or something further up the stack; I'll try and run the same test on my x64 machine to narrow it down (or up, as the case may be :).

If the device wouldn't have an OS string descriptor, xusb would just print "no descriptor" without error. In this case there is one, so xusb proceeds to fetch the MS WinUSB Feature Descriptors. Usually the former is there to hint about the latter, so both are present. When the latter fails, it can be a bug in xusb or the device. Please try the `-w` option that works around a limitation in WinUSB.

mcuee, 2022-01-28:
> Problems with the isochronous transfer testing is to have a known-good FW. Right now we have some known good FW from Cypress for Cyusb EZ-USB FX2-LP and FX3. libusbk has some know good examples for [...]

There is an existing issue for this topic here.
* https://github.com/libusb/libusb/issues/962

closed 2022-01-31 by tormodvolden

mcuee, 2022-02-01:
This test matrix is a good idea. We can have the same test matrix for the next release.