libusb/libusb #1047: Misc: To release 1.0.26

tormodvolden, 2022-01-21:
Ready to merge:

Must be checked:
- #1003 (possible API breakage)
 
Pending review:
- #1026 (solve API issue)

Pending review, may be postponed if not enough reviews:
- #1018 (libusb0.sys)

Nice to have but can probably be postponed:
- #678 (postpone unless reproduced issue)
- #965 (very useful feature enhancement, especially for Windows users, new APIs)

Merged main feature improvements:

Merged main bug fixes:

mcuee, 2022-01-22:
hotplug is the main feature missing for Windows. However, if we do not have Chris back in the project or another Windows developer to chime in, I think this will not happen in libusb-1.0.26 either.
* https://github.com/libusb/libusb/issues/86

mcuee, 2022-01-22:
This is probably good to be in libusb 1.0.26 as well.
* https://github.com/libusb/libusb/issues/1048

tormodvolden, 2022-01-22:
We should for a while after the 1.0.25 release avoid larger changes and rabbit holes that prevent us from releasing any regression fixes in due time. Or we have to use stable branches, but then we'd have even more to maintain and test.

For something like the Windows hotplug support we must be sure there are developers around that will stay available to timely fix any issues that arises after it lands. Anyway, until there is at least a draft of a pull request for it, it remains just a wishlist item with no relevance for our release milestones.

mcuee, 2022-01-23:
> For something like the Windows hotplug support we must be sure there are developers around that will stay available to timely fix any issues that arises after it lands. Anyway, until there is at [...]

Good point.

tormodvolden, 2022-02-24:
It seems like we should pretty soon release a 1.0.26 with only regression and build fixes, and leave API changes and more complicated changes for 1.0.27. Anyway we can still wait a bit to see if other regressions pop up (like the hotplug issue reported today).

mcuee, 2022-02-25:
I agree to release a 1.0.26 version with only regression and build fixes.

mcuee, 2022-03-17:
Thanks for the work done by @tormodvolden, we are closer to 1.0.26rc1 now.

tormodvolden, 2022-03-17:
More accurately, @benzea did all the work :) I think I will merge the "safe" items on the list as well, even if they are not strictly critical bug fixes. Possibly tag the RC in a few days, and if everything is good, release it around March 31 (two months after 1.0.25 already).

mcuee, 2022-03-18:
I also agree that #1071 and #1078 are ready to be merged. I like #1078 very much. It can be a highlight of 1.0.26 as a new feature.

lunixbochs, 2022-03-18:
Can I request my PR https://github.com/libusb/libusb/pull/1093 lands in the RC, as it's a memory safety / crash issue?

mcuee, 2022-03-20:
* #1093 has been merged.

tormodvolden, 2022-03-27:
With the RC out today, we can aim for final release around April 10th, if nothing complicated turns up. I hope to test some libkinect isochronous streaming with XBOX before then :)

mcuee, 2022-04-01:
#1102 should be safe to merge.

mcuee, 2022-04-01:
* https://github.com/libusb/libusb/pull/1101 seems to be okay as well.

mcuee, 2022-04-09:
So the main remaining thing is to merge #1119 and deal with #1110 -- both are for macOS.

sergeu90, 2022-04-10:
Hi all. a lot of thanks for library. When do you plan release 1.0.26 version? I saw in this conversation 10 april?

mcuee, 2022-04-10:
#1119 has been merged. Only pending #1110.

tormodvolden, 2022-04-10:
Any dates given are approximate and just estimation. But we are still on track :)

closed 2022-04-10 by tormodvolden

tormodvolden, 2022-04-10:
Like last time we will have a short post-release freeze before starting merging new things, just in case something critical shows up after release.

mcuee, 2022-04-10:
Two weeks freeze?