libusb/libusb #1049: MinGW 32-bit (cross-)build requires -static-libgcc to avoid DLL migraine

tormodvolden, 2022-01-25:
When cross-building MinGW 32-bit (gcc 9.3 on Ubuntu 20.04) the resulting DLL depends on various gcc DLLs, so if just installing the libusb.dll the program using it will fail to execute (_"libgcc_s_sjlj-1.dll was not found"_).

Adding the extra linker option `-static-libgcc` fixes this. However, this is not needed for 64-bit builds, so it is possible that there is a bug somewhere. This is not very new, but wasn't like this some years ago either.

This would better be rechecked with a newer MinGW toolchain.

mcuee, 2022-01-28:
Ah, I see the same problem with MSYS2 with 32bit build. And yes 64bit build is not affected.

mcuee, 2022-02-06:
> This is not very new, but wasn't like this some years ago either.

It seems to me this has been there for a while. Just windering which version you check does not cause this issue. 

My GCC version is pretty new under MSYS2.

```
$ gcc -v
Using built-in specs.
COLLECT_GCC=C:\msys64\mingw32\bin\gcc.exe
COLLECT_LTO_WRAPPER=C:/msys64/mingw32/bin/../lib/gcc/i686-w64-mingw32/11.2.0/lto-wrapper.exe
Target: i686-w64-mingw32
Configured with: ../gcc-11.2.0/configure --prefix=/mingw32 --with-local-prefix=/mingw32/local --build=i686-w64-mingw32 --host=i686-w64-mingw32 --target=i686-w64-mingw32 --with-native-system-header-dir=/mingw32/i686-w64-mingw32/include --libexecdir=/mingw32/lib --enable-bootstrap --enable-checking=release --with-arch=i686 --with-tune=generic --enable-languages=c,lto,c++,fortran,ada,objc,obj-c++,jit --enable-shared --enable-static --enable-libatomic --enable-threads=posix --enable-graphite --enable-fully-dynamic-string --enable-libstdcxx-filesystem-ts --enable-libstdcxx-time --disable-libstdcxx-pch --disable-libstdcxx-debug --enable-lto --enable-libgomp --disable-multilib --disable-rpath --disable-win32-registry --disable-nls --disable-werror --disable-symvers --with-libiconv --with-system-zlib --with-gmp=/mingw32 --with-mpfr=/mingw32 --with-mpc=/mingw32 --with-isl=/mingw32 --with-pkgversion='Rev7, Built by MSYS2 project' --with-bugurl=https://github.com/msys2/MINGW-packages/issues --with-gnu-as --with-gnu-ld --with-boot-ldflags='-pipe -Wl,--dynamicbase,--nxcompat,--no-seh -Wl,--large-address-aware -Wl,--disable-dynamicbase -static-libstdc++ -static-libgcc' 'LDFLAGS_FOR_TARGET=-pipe -Wl,--dynamicbase,--nxcompat,--no-seh -Wl,--large-address-aware' --enable-linker-plugin-flags='LDFLAGS=-static-libstdc++\ -static-libgcc\ -pipe\ -Wl,--dynamicbase,--nxcompat,--no-seh\ -Wl,--large-address-aware\ -Wl,--stack,12582912' --disable-sjlj-exceptions --with-dwarf2
Thread model: posix
Supported LTO compression algorithms: zlib zstd
gcc version 11.2.0 (Rev7, Built by MSYS2 project)
```

diabolo38, 2022-02-06:
As far as i remember i often see this in 32 and or 64 bit app/lib built with mingw/gcc  even with native windows built.
Mingw  built lib/dll are meant to be used in mingw/msys context  so isn't it  armless  ? The dependencies lib shall be present
User that wan't to build and distribute app/lib  with mingw ibsusb dll aside shall shall be aware of it and built their stuff with static gcc or even static libusb if to avoid any dependencies issue . Or just use the msvc version of the dll

tormodvolden, 2022-02-06:
What is a MinGW context? Isn't the point of MinGW that you build standalone binaries that "just works"? I admit this is not obvious for shared libraries. I find it practical to have an application and a libusb DLL and no more dependencies but see that this is a somewhat middle ground.

The question remains, why is this different on 64 bit?

diabolo38, 2022-02-06:
by mingw context i mean a machine where some mingw got installed so the gcc std lib dll is present.

yes it's strange that  64 bit one doesn't has it maybe some std library helper for 64 bit ops 
Another dif of gcc 32/64 is the exception model sjlj  vs SEH but that shoud=ld affect oly c++ 
if you look at dependencies symbols you should find out.

to come on inittial point anyway you right that it will avoid the migraine by adding static-gcc  in the lib build ;)

diabolo38, 2022-02-06:
i did on the 1.0.25 binary package 
obdjump -x dll 

```
        DLL Name: libgcc_s_sjlj-1.dll
        vma:  Hint/Ord Member-Name Bound-To
        238bc     121  __udivdi3
        238c8     122  __udivmoddi4
```
these are helper for 64 bit division

mcuee, 2022-02-07:
The idea is to have DLL interchangeability between MinGW32/64 and Visual Studio. Therefore it is good to remove the dependencies of MinGW helpers.

mcuee, 2022-02-07:
@diabolo38 It seems to me the 32bit MinGW DLL of libusb-1.0.25 Windows binary has dependancies on MinGW helpers but not the 64bit MinGW DLL.

Ref: https://stackoverflow.com/questions/11703000/finding-dlls-required-of-a-win-exe-on-linux-cross-compiled-with-mingw

```
/c/work/libusb/libusb_bin/libusb-1.0.25_bin/MinGW64/dll
$ strings libusb-1.0.dll | grep -i '\.dll$'
\%s.dll
libusb-1.0.dll
KERNEL32.dll
msvcrt.dll

/c/work/libusb/libusb_bin/libusb-1.0.25_bin/MinGW32/dll
$ strings libusb-1.0.dll | grep -i '\.dll$'
\%s.dll
libusb-1.0.dll
KERNEL32.dll
msvcrt.dll
libgcc_s_sjlj-1.dll
```

mcuee, 2022-02-07:
Older version of MinGW32 dll binary.

```
/c/work/libusb/libusb_bin/libusb-1.0.24_bin/MinGW32/dll
$ strings libusb-1.0.dll | grep -i '\.dll$'
libgcj-16.dll
\%s.dll
libusb-1.0.dll
KERNEL32.dll
msvcrt.dll
USER32.dll

/c/work/libusb/libusb_bin/libusb-1.0.23_bin/MinGW32/dll
$ strings libusb-1.0.dll | grep -i '\.dll$'
libgcj-16.dll
libusb-1.0.dll
ADVAPI32.dll
KERNEL32.dll
msvcrt.dll
USER32.dll

```

diabolo38, 2022-02-07:
if you used objdump -x like i did you will see what symbols is required in the gcc dll 
in the 1.0.25 i got from Sourceforge binary package it was clearly the divide 64 bit helper.

@mcuee Your investigation on past version show that even the base dll name is on toolschain dependent.
One more reason to avoid it  by use of static gcc lib at build

mcuee, 2022-02-07:
@diabolo38 Yes I agree that it is better to link to static gcc lib at build for MinGW toolchains.

mcuee, 2022-02-08:
> if you used objdump -x like i did you will see what symbols is required in the gcc dll in the 1.0.25 i got from Sourceforge binary package it was clearly the divide 64 bit helper.

Hmm, I can not get your result for the 64bit binary.
``` 
/c/work/libusb/libusb-1.0.25_bin/MinGW64/dll
$ objdump.exe -x libusb-1.0.dll | grep DLL
        DLL
 vma:            Hint    Time      Forward  DLL       First
        DLL Name: KERNEL32.dll
        DLL Name: msvcrt.dll
```

mcuee, 2022-02-08:
@diabolo38 using your objdump method, the old version of 32bit DLL does not seem to have an issue.

```
/c/work/libusb/libusb-1.0.24_bin/MinGW32/dll
$ objdump -x libusb-1.0.dll | grep DLL
        DLL
 vma:            Hint    Time      Forward  DLL       First
        DLL Name: KERNEL32.dll
        DLL Name: msvcrt.dll
        DLL Name: USER32.dll

 /c/work/libusb/libusb-1.0.23_bin/MinGW32/dll
$ objdump -x libusb-1.0.dll | grep DLL
        DLL
 vma:            Hint    Time      Forward  DLL       First
        DLL Name: ADVAPI32.dll
        DLL Name: KERNEL32.dll
        DLL Name: msvcrt.dll
        DLL Name: USER32.dll

/c/work/libusb/libusb-1.0.22_bin/MinGW32/dll
$ objdump -x libusb-1.0.dll | grep DLL
        DLL
 vma:            Hint    Time      Forward  DLL       First
        DLL Name: ADVAPI32.dll
        DLL Name: KERNEL32.dll
        DLL Name: msvcrt.dll
        DLL Name: USER32.dll

```

mcuee, 2022-02-08:
So the only issue seems to be with the 1.0.24 MinGW32 dll as per my testing. I am not so sure why the old version does not have such an issue, probably it is due to compiler differences.

tormodvolden, 2022-02-09:
In the binary package I posted on SF and GitHub I had myself compiled the MinGW binaries and it is quite possible I forgot to add the -static flag (it was already quite a mess to pull together the Windows package). However, they also get built in the appveyor service, from where I got the VS builds. The previous binary packages were maybe built with a local appveyor instance, from what I can guess from the paths used. Maybe they get the wanted DLL configuration there? For our convenience it should be possible to also download the appveyor MinGW build artifacts. I am just not sure how to configure this in the common "after_build" part. If I add a path that doesn't exists for some builds (the MinGW builds are piggy-backed on the VS15 build), then the 7z command and the whole job fails. There is probably a simple solution.

Anyway, it would be good if our build configuration does the right thing for us by default when building locally also.

mcuee, 2022-02-09:
> Anyway, it would be good if our build configuration does the right thing for us by default when building locally also.

Yes that will be the goal. Maybe the configure script needs to be changed to link to static gcc library.

diabolo38, 2022-02-09:
@mcuee  i used  objdump 2.27 without the grep dll , you will find the symbols from the dll  on the lines following the dll name itself

'''
 objdump -v 

'''

rom1v, 2022-02-10:
FYI, I encounter the same missing `libgcc_s_sjlj-1.dll` with the win32 `libusb-1.0.dll` release, but no problem with the win64 `libusb-1.0.dll` release: https://github.com/Genymobile/scrcpy/pull/3011

(I use the official dll from libusb 1.0.25 and cross-compile my project from Linux)

closed 2022-03-16 by tormodvolden

thewh1teagle, 2023-12-31:
I still get `libusb-1.0.dll` missing
I build like that:
```console
cmake.exe .. -G "MinGW Makefiles" -DCMAKE_CXX_FLAGS="-static-libgcc -static-libstdc++ -static"
mingw32-make.exe -j16
```

mcuee, 2023-12-31:
> I still get `libusb-1.0.dll` missing I build like that:
> 
> ```
> cmake.exe .. -G "MinGW Makefiles" -DCMAKE_CXX_FLAGS="-static-libgcc -static-libstdc++ -static"
> mingw32-make.exe -j16
> ```

If you use CMake, then you may want to look at the repo here.
https://github.com/libusb/libusb-cmake

You also need to provide more details of what you are doing. Your issue is apparently not related to libusb. So please do not continue here but rather create a discussion here.
https://github.com/libusb/libusb/discussions

thewh1teagle, 2023-12-31:
Ok, pasting here just to help others which encounter it too.
final solution to build it with static `libusb`:
```
cmake.exe .. -G "MinGW Makefiles" -DLIBUSB_LIBRARIES=/ucrt64/lib/libusb-1.0.a -DCMAKE_EXE_LINKER_FLAGS="-static -static-libgcc -static-libstdc++"
```
Note the line `-DLIBUSB_LIBRARIES=/ucrt64/lib/libusb-1.0.a`, change to the right path there to `libusb.a`

commit f9162d36 (2022-03-16): windows: Link gcc helpers statically on MinGW

Fixes #1049