libusb/libusb #1050: Better support of  HCD root hub on Windows on ARM

mcuee, 2022-01-26:
From: https://github.com/libusb/libusb/issues/1046

> Side note: [ 0.062403] [00008694] libusb: warning [enumerate_hcd_root_hub] could not infer VID/PID of HCD root hub from 'ACPI\QCOM04A6\2' suggests to me that this is expecting the PCI VID/PID string format that has been common for peripherals on x86_64 and IA32 systems for at least a decade. As I understand it it's not actually a standard--and so, painful on a few different levels, though it's stable so INFs don't need to change--so the parsing will typically fail with ACPI-attached devices like this. While the ACPI table can be tweaked to convince acpi.sys to output almost the classic format (by adding _HW, IIRC--fun story), I think the better solution is, in time, to start to allow for this increasingly common class of device strings.

Good point. We may need to support this kind of HCD root hub better down the road.

Currently the code is like this.
https://github.com/libusb/libusb/blob/master/libusb/os/windows_winusb.c#L1288
```
if (sscanf(dev_id, "PCI\\VEN_%04hx&DEV_%04hx%*s", &dev->device_descriptor.idVendor, &dev->device_descriptor.idProduct) != 2)
usbi_warn(ctx, "could not infer VID/PID of HCD root hub from '%s'", dev_id);
```

tormodvolden, 2022-01-26:
Is this format described anywhere? In the example above it looks like vendor is QCOM (Qualcomm?) and product could be 0x04A6. To not change our internal structures, if there are not too many vendors we could keep a table of vendor strings and an associated ID, or make a hash out of it, to stuff into our idVendor. I guess any clashes with real USB IDs it not an issue.

mcuee, 2022-01-28:
> Is this format described anywhere? In the example above it looks like vendor is QCOM (Qualcomm?) and product could be 0x04A6. To not change our internal structures, if there are not too many vendors we could keep a table of vendor strings and an associated ID, or make a hash out of it, to stuff into our idVendor. I guess any clashes with real USB PCI IDs is not an issue.

Unfortunately it does not seem to be documented. Hopefully more people will chime in to see what is shown in their system. The existing Intel/AMD x86/amd64 platform seems to follow the format "PCI\\VEN_%04hx&DEV_%04hx%*s". This does not seem to be true for Windows on ARM64.

mcuee, 2022-01-28:
Ref: https://docs.microsoft.com/en-us/windows-hardware/drivers/install/hardware-ids

All PCI/PCIE based USB controller will have the hardware ID which current Windows codes can deal with. The problem is for non PCI/PCIE based USB controller.

driver1998, 2022-09-12:
On my Snapdragon 8cx laptop, the USB Root Hub is indeed below a Qualcomm filter driver (ACPI\QCOM04A6).
[image]

My Raspberry Pi 3 and 4 currently doesn't have Windows installed, but on Pi4 the XHCI controller is a PCIe one exposed as an ACPI device in ACPI.
And Pi3 uses a fully custom USB stack with custom host controller driver and root hub driver. It is the case on both Windows IoT Core ARM32 and Desktop ARM64.

sonatique, 2022-10-04:
Could this https://github.com/libusb/libusb/pull/1202 help, maybe?

mcuee, 2022-10-05:
@sonatique 

Indeed I think PR #1202 is an interesting idea.
* #1202

tormodvolden, 2023-01-18:
Can somebody please test #1202 on Windows on ARM?

mcuee, 2023-01-20:
@driver1998 and @jkunlee

Just wondering if you can give #1202 a try on Windows on ARM64?

mcuee, 2023-01-20:
@jkunkee

Just wondering if you can give #1202 a try on Windows on ARM64? Thanks.

mcuee, 2023-02-17:
Some tests results from @ArminiusTux here in PR #1202.
* https://github.com/libusb/libusb/pull/1202

sonatique, 2023-10-19:
> @jkunkee
> 
> Just wondering if you can give #1202 a try on Windows on ARM64? Thanks.
> 
> Pre-built binary for ARM64 Windows. [#1202 (comment)](https://github.com/libusb/libusb/pull/1202#issuecomment-1398094170)

@mcuee : I tested your binary under Windows 11 ARM64 and I got the expected results with testlibusb and listdevs, but I also got expected results when running master (that I compiled for ARM64 by myself). 
So I guess I cannot reproduce the original issue, therefore not validating whether #1202 improve something or not in this particular case.

Droid-MAX, 2024-03-30:
https://github.com/Droid-MAX/SNANDer/issues/15

tormodvolden, 2024-03-30:
Try upgrading to libusb 1.0.27.

Droid-MAX, 2024-04-03:
When I updated libusb to version 1.0.27, the program output information about libusb became more.

Droid-MAX, 2024-04-03:
I have a vaguely bad feeling about this, could be the next potential supply chain poisoning similar to xz-utils?
And I found that programs built with the libusb-1.0.26 source code at some point in time always output some Libusb-related warning information, which had not been the case before. A warning message appears even after the upgrade to 1.0.27.

mcuee, 2024-04-03:
@Droid-MAX

Don't spam the github issues. Thanks.

I have re-opened the relevant issue.