libusb/libusb #1052: Known good devices and firmware to test libusb

mcuee, 2022-01-30:
Right now we have quite a few example program to carry out simple tests like listdevs, testlibusb and xusb but they have some limitations. For example they are not able to test isochronous transfer and they are not able to test performance regressions if any. 

We also have one example sam3u_benchmark for isoc transfer benchmark testing but the HW/FW combination is not so common.

So it will be good to provide some known good devices to carry out the tests. Virtual devices can be a good option as it can be integrated in CI/CD workflows.

mcuee, 2022-01-30:
Similar discussions for libusb based project.
1) pyusb
* https://github.com/pyusb/pyusb/issues/235

2) libusbdotnet
* https://github.com/LibUsbDotNet/LibUsbDotNet/issues/137

mcuee, 2022-01-30:
For HW devices, I will highly recommend the following US$49 Cypress FX3 development board. It provides support for Windows, macOS and Linux with quite some known good FW, including the support of isochronous transfer. The toolchain is based on ARM GCC.
https://www.cypress.com/documentation/development-kitsboards/cyusb3kit-003-ez-usb-fx3-superspeed-explorer-kit

mcuee, 2022-01-30:
Another possibility is to use the existing sam3u_benchmark example, and port the FW to more common devices like the STM32 development boards.

mcuee, 2022-01-30:
Yet another possibility is Linux USB gadget devices, using common hardware like Raspberry Pi Zero or Raspberry Pi 4/400, or using virtual device like a qemu VM.

Ref: 
* https://github.com/pyusb/pyusb/issues/235
*

mcuee, 2022-04-01:
libusb-1.0.26 will add the umockdev tests under Linux which is a good thing.

mcuee, 2022-04-01:
Cypress FX2LP based development boards are also a good buy. For example, CyStream FW can be a good testing FW for high speed bulk and isoc transfer.
Ref: https://github.com/libusb/libusb/issues/1104

mcuee, 2022-04-01:
Another option is those MCU development boards, especially ARM Cortex M (M0+, M3, M4, M33, M7, etc) based. 
STM32 boards are popular.

Of course PIC, AVR based USB boards can also be used.

tormodvolden, 2022-04-01:
I modified an Audio Speaker Full-Speed example for STM32 (just removed class and audio descriptors) and used it for outwards isochronous transfer testing, with a modified version of sam3u_benchmark.c that generates a saw-tooth signal (observable on an oscilloscope). It makes for a basic functionality test. Would be nice to add inwards transfers too. It seems like the sam3u_benchmark firmware only does iso IN, but bulk OUT/IN.

Otherwise I find USB Gadget on a Raspberry interesting but haven't looked at it yet.

mcuee, 2022-04-01:
@tormodvolden I think the sam3u_benchmark firmware is prety old. it will be good to have another sample application to support both ISOC IN and OUT.

As for USB Gadget on Raspberry Pi, it is pretty good. You can also try out python-functionfs. It is awesome. Please check out the discussion here. It is just that I do not have a good cross-platform C program to test it out (bulk and isoc). I can only use linux specific code (usbtest) or Windows specific code (libusbk kbench). I used to use libusbdotnet as the cross-platform app to test them but it has not been updated to support isoc well enough.
https://github.com/vpelletier/python-functionfs/issues/20

mcuee, 2022-04-01:
Other than sam3u_benchmark, I think the following example code can be a potential starting point for the host application to test isoc loop/sourcesink or bulk loop/sourcesink type of testing FW. Just need to give more command line option (like xusb) to make it more generic.
Ref: https://github.com/walac/isochronous

Or the Cypress example (LGPL 2.1 so no issue). It is using libcyusb which is a think wrapper of libusb. Need to strip the Cypress specific stuff though ("/etc/cyusb.conf" stuff).
https://github.com/hantianjz/cy_fx3_sdk/blob/master/cyusb_linux_1.0.5/src/09_cyusb_performance.cpp
https://github.com/hantianjz/cy_fx3_sdk/blob/master/cyusb_linux_1.0.5/lib/libcyusb.cpp

tormodvolden, 2022-04-03:
I added isochronous IN transfers to my STM32 firmware. This was less straightforward since I couldn't find any example code, so the reference manual had to be consulted for double-buffer toggle bit details. BTW, I discovered that those who wrote ST's own Audio Speaker example hadn't read it so carefully :) My typical go-to framework libopencm3 doesn't support isochronous transfers (https://github.com/libopencm3/libopencm3/issues/679).

Thanks for the reference to walac's test code. Both this and sam3u_benchmark have their advantages so we may try to combine their features. See also https://github.com/walac/isochronous/pull/1

mcuee, 2022-04-03:
Nice.

Interestingly long time ago I was trying isoc transfer -- at that time (2007), the host driver under Windows was pretty difficult but I was able to use libusb-win32 (which works with isoc transfer for full speed device well enough). 
https://www.microchip.com/forums/m270049.aspx?print=true

mcuee, 2022-04-10:
I will close the following issue and keep the discussion here.
* https://github.com/libusb/libusb/issues/1053

Linux has the following code which is good to test certain device like Gadget Zero.
https://github.com/torvalds/linux/blob/master/drivers/usb/misc/usbtest.c

But Gadget Zero type of LoopBack or SourceSink FW can be tested with something similar to libusbK kBench or libusbdotnet BenchmarkCon.
* https://github.com/libusb/libusb/issues/962

mcuee, 2022-04-10:
Just an update on the iso-test from Wander.

Tomord Volden has extend the iso-test codes from Wander Lairson Costa to be more generic.
https://github.com/tormodvolden/isochronous

He has also extend the sam3u_benchmark example to stm32_isoc here (along with mod FW for STM32F1)
https://libusb.sourceforge.io/stm32_isoc/

mcuee, 2022-04-10:
There are also quite some python code out there based on pyusb to test loopback or sourcesink type of FW.

mcuee, 2022-04-14:
> I added isochronous IN transfers to my STM32 firmware. This was less straightforward since I couldn't find any example code, so the reference manual had to be consulted for double-buffer toggle bit details. BTW, I discovered that those who wrote ST's own Audio Speaker example hadn't read it so carefully :) My typical go-to framework libopencm3 doesn't support isochronous transfers ([libopencm3/libopencm3#679](https://github.com/libopencm3/libopencm3/issues/679)).
> 
> EDIT: The STM32 firmware and stm32_isoc test program: https://libusb.sourceforge.io/stm32_isoc/

@tormodvolden Just FYI I can build the FW using Keil MDK community edition v5.36 (no code size limit, ARM CC compiler). I need to dig out my STM32F1 evaluation boards to see if it works on my hardware.

ST has already long moved away the old USB FS device stack so it is probably not maintained for quite sometime.
New driver as part of STM32 Cube F1 package in github.
https://github.com/STMicroelectronics/STM32CubeF1/blob/master/Drivers/STM32F1xx_HAL_Driver/Src/stm32f1xx_ll_usb.c

But the Audio Speaker example is gone.
https://github.com/STMicroelectronics/STM32CubeF1/tree/master/Projects/STM3210E_EVAL/Applications/USB_Device

mcuee, 2022-04-14:
@tormodvolden Yes the FW works fine on my STM3210E-Eval board -- I need to test using libusbk kBench program but not stm32_isoc under Windows.

```
PS C:\work\libusb\libusb_test\stm32_isoc> C:\libusbK-dev-kit\kBench.exe list notestselect intf=1 altf=1 loop buffersize=176 buffercount=3 log
device-count=2
1. STMicroelectronics STLink dongle (USB\VID_0483&PID_3748\6&8694159&0&4) [WinUSB]
2. STM32 Speaker (Composite Parent) (USB\VID_0483&PID_5730\TEST1) [WinUSB]
Select device (1-2) :2

opened STM32 Speaker (Composite Parent) (USB\VID_0483&PID_5730\TEST1)..
Loop Test Information
        Driver          : WinUSB
        Vid / Pid       : 0483h / 5730h
        DevicePath      : \\?\usb#vid_0483&pid_5730#test1#{fcdeda77-a713-62ce-7bdf-18efd02f84df}
        Device Speed    : Low/Full
        Interface #     : 01h
        Alt Interface # : 01h
        Num Endpoints   : 2
        Priority        : 0
        Read Size       : 176
        Write Size      : 176
        Buffer Count    : 3
        Display Refresh : 1000 (ms)
        Transfer Timeout: 5000 (ms)
        Retry Count     : 0
        Verify Data     : Off

Isochronous Read (Ep82h) Maximum Packet Size: 22
Isochronous Write (Ep01h) Maximum Packet Size: 22

While the test is running:
Press 'Q' to quit
Press 'T' for test details
Press 'I' for status information
Press 'R' to reset averages

Press 'Q' to exit, any other key to begin..
Avg. Bytes/s: 43954.59 Transfers: 242 Bytes/s: 43954.59 ISO-Packets (Total/Good/Bad):1984/1984/0
Avg. Bytes/s: 43989.50 Transfers: 494 Bytes/s: 44024.28 ISO-Packets (Total/Good/Bad):4000/4000/0
Avg. Bytes/s: 44103.74 Transfers: 744 Bytes/s: 44335.03 ISO-Packets (Total/Good/Bad):6000/6000/0
Avg. Bytes/s: 44000.00 Transfers: 996 Bytes/s: 43696.55 ISO-Packets (Total/Good/Bad):8016/8016/0
Avg. Bytes/s: 44000.00 Transfers: 1246 Bytes/s: 44000.00 ISO-Packets (Total/Good/Bad):10016/10016/0
Avg. Bytes/s: 44058.82 Transfers: 1498 Bytes/s: 44352.00 ISO-Packets (Total/Good/Bad):12032/12032/0
Avg. Bytes/s: 44000.00 Transfers: 1750 Bytes/s: 43653.54 ISO-Packets (Total/Good/Bad):14048/14048/0
Avg. Bytes/s: 44044.00 Transfers: 2002 Bytes/s: 44352.00 ISO-Packets (Total/Good/Bad):16064/16064/0
Avg. Bytes/s: 44000.00 Transfers: 2254 Bytes/s: 43653.54 ISO-Packets (Total/Good/Bad):18080/18080/0
waiting for Ep82h thread..
stopped Ep82h thread.   ExitCode=0
stopped Ep01h thread.   ExitCode=0
Loop Test Information
        Driver          : WinUSB
        Vid / Pid       : 0483h / 5730h
        DevicePath      : \\?\usb#vid_0483&pid_5730#test1#{fcdeda77-a713-62ce-7bdf-18efd02f84df}
        Device Speed    : Low/Full
        Interface #     : 01h
        Alt Interface # : 01h
        Num Endpoints   : 2
        Priority        : 0
        Read Size       : 176
        Write Size      : 176
        Buffer Count    : 3
        Display Refresh : 1000 (ms)
        Transfer Timeout: 5000 (ms)
        Retry Count     : 0
        Verify Data     : Off

Isochronous Read (Ep82h) Maximum Packet Size: 22
        Total Bytes     : 198528
        Total Transfers : 1128
        Avg. Bytes/sec  : 21982.95
[... 71 lines cut ...]
0000150 0706 0908 0b0a 0d0c 0f0e 1110 1312 1514
0000160 1000 0302 0504 0706 0908 0b0a 0d0c 0f0e
0000170 1110 1312 1514 1100 0302 0504 0706 0908
0000180 0b0a 0d0c 0f0e 1110 1312 1514 1200 0302
0000190 0504 0706 0908 0b0a 0d0c 0f0e 1110 1312
00001a0 1514 1300 0302 0504 0706 0908 0b0a 0d0c
00001b0 0f0e 1110 1312 1514 1400 0302 0504 0706
00001c0 0908 0b0a 0d0c 0f0e 1110 1312 1514 1500
00001d0 0302 0504 0706 0908 0b0a 0d0c 0f0e 1110
00001e0 1312 1514 1600 0302 0504 0706 0908 0b0a
00001f0 0d0c 0f0e 1110 1312 1514 1700 0302 0504
0000200 0706 0908 0b0a 0d0c 0f0e 1110 1312 1514
0000210 0000 0302 0504 0706 0908 0b0a 0d0c 0f0e
0000220 1110 1312 1514 0100 0302 0504 0706 0908
0000230 0b0a 0d0c 0f0e 1110 1312 1514 0200 0302
0000240 0504 0706 0908 0b0a 0d0c 0f0e 1110 1312
0000250 1514 0300 0302 0504 0706 0908 0b0a 0d0c
0000260 0f0e 1110 1312 1514 0400 0302 0504 0706
0000270 0908 0b0a 0d0c 0f0e 1110 1312 1514 0500
0000280 0302 0504 0706 0908 0b0a 0d0c 0f0e 1110
0000290 1312 1514 0600 0302 0504 0706 0908 0b0a
00002a0 0d0c 0f0e 1110 1312 1514 0700 0302 0504
00002b0 0706 0908 0b0a 0d0c 0f0e 1110 1312 1514

```

mcuee, 2022-04-14:
stm32_isoc is kind of working but the IN transfer message is kind of confusing. xfr length(bulk) should be 0.
`request endpoint:82 length:176, xfr length (bulk):96, total (isoch):176`

```
$ ./stm32_isoc.exe -v -o -n176
num_iso_pack = 8
request endpoint:01 length:176, xfr length (bulk):0, total (isoch):176
request endpoint:01 length:176, xfr length (bulk):0, total (isoch):176
request endpoint:01 length:176, xfr length (bulk):0, total (isoch):176
request endpoint:01 length:176, xfr length (bulk):0, total (isoch):176
request endpoint:01 length:176, xfr length (bulk):0, total (isoch):176
request endpoint:01 length:176, xfr length (bulk):0, total (isoch):176
request endpoint:01 length:176, xfr length (bulk):0, total (isoch):176
request endpoint:01 length:176, xfr length (bulk):0, total (isoch):176
request endpoint:01 length:176, xfr length (bulk):0, total (isoch):176
request endpoint:01 length:176, xfr length (bulk):0, total (isoch):176
request endpoint:01 length:176, xfr length (bulk):0, total (isoch):176
request endpoint:01 length:176, xfr length (bulk):0, total (isoch):176
request endpoint:01 length:176, xfr length (bulk):0, total (isoch):176
request endpoint:01 length:176, xfr length (bulk):0, total (isoch):176
request endpoint:01 length:176, xfr length (bulk):0, total (isoch):176
request endpoint:01 length:176, xfr length (bulk):0, total (isoch):176
request endpoint:01 length:176, xfr length (bulk):0, total (isoch):176
request endpoint:01 length:176, xfr length (bulk):0, total (isoch):176
request endpoint:01 length:176, xfr length (bulk):0, total (isoch):176
request endpoint:01 length:176, xfr length (bulk):0, total (isoch):176
request endpoint:01 length:176, xfr length (bulk):0, total (isoch):176
request endpoint:01 length:176, xfr length (bulk):0, total (isoch):176
request endpoint:01 length:176, xfr length (bulk):0, total (isoch):176
request endpoint:01 length:176, xfr length (bulk):0, total (isoch):176
request endpoint:01 length:176, xfr length (bulk):0, total (isoch):176
request endpoint:01 length:176, xfr length (bulk):0, total (isoch):176
request endpoint:01 length:176, xfr length (bulk):0, total (isoch):176
request endpoint:01 length:176, xfr length (bulk):0, total (isoch):176
request endpoint:01 length:176, xfr length (bulk):0, total (isoch):176
request endpoint:01 length:176, xfr length (bulk):0, total (isoch):176
request endpoint:01 length:176, xfr length (bulk):0, total (isoch):176
request endpoint:01 length:176, xfr length (bulk):0, total (isoch):176
request endpoint:01 length:176, xfr length (bulk):0, total (isoch):176
request endpoint:01 length:176, xfr length (bulk):0, total (isoch):176
request endpoint:01 length:176, xfr length (bulk):0, total (isoch):176
request endpoint:01 length:176, xfr length (bulk):0, total (isoch):176
request endpoint:01 length:176, xfr length (bulk):0, total (isoch):176
request endpoint:01 length:176, xfr length (bulk):0, total (isoch):176
request endpoint:01 length:176, xfr length (bulk):0, total (isoch):176
request endpoint:01 length:176, xfr length (bulk):0, total (isoch):176
request endpoint:01 length:176, xfr length (bulk):0, total (isoch):176
request endpoint:01 length:176, xfr length (bulk):0, total (isoch):176
request endpoint:01 length:176, xfr length (bulk):0, total (isoch):176
request endpoint:01 length:176, xfr length (bulk):0, total (isoch):176
request endpoint:01 length:176, xfr length (bulk):0, total (isoch):176
request endpoint:01 length:176, xfr length (bulk):0, total (isoch):176
request endpoint:01 length:176, xfr length (bulk):0, total (isoch):176
request endpoint:01 length:176, xfr length (bulk):0, total (isoch):176
47 transfers (total 8272 bytes) in 1004 milliseconds => 8239 bytes/sec
request endpoint:01 length:176, xfr length (bulk):0, total (isoch):176

$ ./stm32_isoc.exe -v -n176
num_iso_pack = 8
request endpoint:82 length:176, xfr length (bulk):96, total (isoch):176
 aa aa aa aa aa aa aa aa  aa aa aa aa aa aa aa aa
 aa aa aa aa aa aa bb bb  bb bb bb bb bb bb bb bb
 bb bb bb bb bb bb bb bb  bb bb bb bb aa aa aa aa
 aa aa aa aa aa aa aa aa  aa aa aa aa aa aa aa aa
 aa aa bb bb bb bb bb bb  bb bb bb bb bb bb bb bb
 bb bb bb bb bb bb bb bb  aa aa aa aa aa aa aa aa
 aa aa aa aa aa aa aa aa  aa aa aa aa aa aa bb bb
 bb bb bb bb bb bb bb bb  bb bb bb bb bb bb bb bb
 bb bb bb bb aa aa aa aa  aa aa aa aa aa aa aa aa
 aa aa aa aa aa aa aa aa  aa aa bb bb bb bb bb bb
[... 123 lines cut ...]
 aa aa aa aa aa aa bb bb  bb bb bb bb bb bb bb bb
 bb bb bb bb bb bb bb bb  bb bb bb bb aa aa aa aa
 aa aa aa aa aa aa aa aa  aa aa aa aa aa aa aa aa
 aa aa bb bb bb bb bb bb  bb bb bb bb bb bb bb bb
 bb bb bb bb bb bb bb bb  aa aa aa aa aa aa aa aa
 aa aa aa aa aa aa aa aa  aa aa aa aa aa aa bb bb
 bb bb bb bb bb bb bb bb  bb bb bb bb bb bb bb bb
 bb bb bb bb aa aa aa aa  aa aa aa aa aa aa aa aa
 aa aa aa aa aa aa aa aa  aa aa bb bb bb bb bb bb
 bb bb bb bb bb bb bb bb  bb bb bb bb bb bb bb bb
12 transfers (total 2112 bytes) in 386 milliseconds => 5471 bytes/sec
request endpoint:82 length:176, xfr length (bulk):96, total (isoch):176
 aa aa aa aa aa aa aa aa  aa aa aa aa aa aa aa aa
 aa aa aa aa aa aa bb bb  bb bb bb bb bb bb bb bb
 bb bb bb bb bb bb bb bb  bb bb bb bb aa aa aa aa
 aa aa aa aa aa aa aa aa  aa aa aa aa aa aa aa aa
 aa aa bb bb bb bb bb bb  bb bb bb bb bb bb bb bb
 bb bb bb bb bb bb bb bb  aa aa aa aa aa aa aa aa
 aa aa aa aa aa aa aa aa  aa aa aa aa aa aa bb bb
 bb bb bb bb bb bb bb bb  bb bb bb bb bb bb bb bb
 bb bb bb bb aa aa aa aa  aa aa aa aa aa aa aa aa
 aa aa aa aa aa aa aa aa  aa aa bb bb bb bb bb bb
 bb bb bb bb bb bb bb bb  bb bb bb bb bb bb bb bb

```

tormodvolden, 2022-04-14:
> @tormodvolden Just FYI I can build the FW using Keil MDK community edition v5.36 (no code size limit, ARM CC compiler).

Note that you don't need any proprietary tools to build this, just install gcc-arm-none-eabi and run `make` in the gcc folder.

And yes, as you saw, the later "cube" stuff doesn't have the Audio examples. It would have better support for newer microcontroller generations, but for the old F1 series the old library is good enough. I hope libopencm3 soon will get better iso support anyway.

tormodvolden, 2022-04-14:
> stm32_isoc is kind of working with WinUSB driver but the IN transfer message is kind of confusing. xfr length(bulk) should be 0

Don't at all look at this number for non-bulk transfers. Our API says that this number doesn't make any sense for isochronous transfers.

tormodvolden, 2022-04-14:
> Speed under macOS is a bit slower though at around 16000Bytes/sec.

To get full streaming speed we may to queue up more transfers, it is just too late to initiate a new transfer only when the last one has finished, there is no guarantee the program will get CPU time just at the right moment.

mcuee, 2022-04-14:
> > @tormodvolden Just FYI I can build the FW using Keil MDK community edition v5.36 (no code size limit, ARM CC compiler).
> 
> Note that you don't need any proprietary tools to build this, just install gcc-arm-none-eabi and run `make` in the gcc folder.

It seems to missing some linker scripts related to stm32. I will check again.

BTW, there is another alternative USB stack for stm32 which is actually named libusb_stm32. But it does not have isoc samples as of now.
Ref: https://github.com/dmitrystu/libusb_stm32

mcuee, 2022-04-14:
libfreenect is a good way to test libusb isochronous transfer but it requires the special HW.

libuvc may be more generic and just need a USB webcam (UVC compliant) to test. But the main git repo seem to have some problems with Windows (getting better since there is vcpkg port now).
https://github.com/libuvc/libuvc

There is a Windows centric fork of libuvc here but it is not as up to date.
https://github.com/pupil-labs/libuvc

mcuee, 2022-04-14:
USB Audio card is also readily available (some of them are very cheap). And there are quite some discussions of using libusb to drive it (mostly under Android), but somehow I could not find a good repo of using libusb to drive USB Audio devices.

So far I only find this one which seems to be Linux only.
https://github.com/Themaister/libmaru

mcuee, 2022-04-15:
> It seems to missing some stm32 specific files.
> 
> ```
> make: *** No rule to make target `startup_stm32f10x_md.o', needed by `audio.elf'.  Stop.
> ```

After changing the CMSIS gcc include directory and I can also build it but there is still an error.
```

mcuee@mcuees-Mac-mini gcc % pwd
/Users/mcuee/build/libusb/libusb_test/stm32_isoc/stm32_isoc_FW/en.stsw-stm32121_v4.1.0/STM32_USB-FS-Device_Lib_V4.1.0/Projects/Audio_Speaker/gcc

mcuee@mcuees-Mac-mini gcc % make -i
(as) startup_stm32f10x_md.o
(cc) stm32f10x_exti.o
(cc) main.o
(cc) hw_config.o
(cc) stm32_it.o
(cc) usb_desc.o
(cc) usb_istr.o
(cc) usb_prop.o
(cc) usb_pwr.o
(cc) usb_endp.o
(cc) usb_core.o
(cc) usb_init.o
(cc) usb_regs.o
(cc) usb_int.o
(cc) usb_mem.o
(cc) usb_sil.o
(cc) stm32f10x_gpio.o
(cc) stm32f10x_spi.o
(cc) stm32f10x_tim.o
(cc) stm32f10x_rcc.o
(cc) misc.o
(cc) system_stm32f10x.o
(ld) audio.elf
/Applications/ARM/bin/../lib/gcc/arm-none-eabi/10.2.1/../../../../arm-none-eabi/bin/ld: /Applications/ARM/bin/../lib/gcc/arm-none-eabi/10.2.1/../../../../arm-none-eabi/lib/thumb/v7-m/nofp/libc.a(lib_a-init.o): in function `__libc_init_array':
init.c:(.text.__libc_init_array+0x1e): undefined reference to `_init'
collect2: error: ld returned 1 exit status
make: [audio.elf] Error 1 (ignored)
(cp) audio.bin
arm-none-eabi-objcopy: 'audio.elf': No such file
make: [audio.bin] Error 1 (ignored)

```

tormodvolden, 2022-04-15:
> However the binary is not working as I am using a different chip -- I think I need to change the gcc.ld file to reflect the changes or add some BSP files to the build process.

The only board-specific detail for the USB part (which is what we care most about) should be which pin is used for the USB D+ pull-up, on your board it is PB14. But if you bridge JP14 pin 1 and 2 it will always be pulled up (like the board I was using), so please try that first.

Pin PB14 is selected in inc/platform_config.h if USE_STM3210E_EVAL is set in the Makefile. However this option changes things a bit. The "E" board has a proper audio interface with I2S. The "B" board uses PWM to generate the audio signal. And I excluded all the I2S files and dependencies from the Makefile. You can try this patch [0101-Adapt-for-STM3210E_EVAL-board.patch.txt](https://github.com/libusb/libusb/files/8495010/0101-Adapt-for-STM3210E_EVAL-board.patch.txt).

tormodvolden, 2022-04-15:
> After changing the CMSIS gcc include directory

Which version of the library is this? In my 4.0.0 copy there is only a gcc_ride folder but no gcc folder.

tormodvolden, 2022-04-15:
> /Applications/ARM/bin/../lib/gcc/arm-none-eabi/10.2.1/../../../../arm-none-eabi/bin/ld: [...]

I am using 9.2.1 - maybe some flags need to be added for 10.2.1. I will try a newer version also one day.

mcuee, 2022-04-15:
> > After changing the CMSIS gcc include directory
> 
> Which version of the library is this? In my 4.0.0 copy there is only a gcc_ride folder but no gcc folder.

4.1.0, one minor version newer than yours, probably the latest as I just downloaded from ST website.

tormodvolden, 2022-04-15:
I updated the "E" board patch to instead properly build the I2S (and I2C) code. Now you can listen to the data that you stream out :)

[0101-v2-Adapt-for-STM3210E_EVAL-board.patch.txt](https://github.com/libusb/libusb/files/8496523/0101-v2-Adapt-for-STM3210E_EVAL-board.patch.txt)

mcuee, 2022-04-16:
@tormodvolden Thanks. I will try this later. I have quite some STM32 evaluation boards with me (not new ones, only old ones with F0/F1/F2/F3/F4) and I will spend some time to get familar with the basics of the USB device stack.

FYI only.
STM32Cube F0/F1 has the audio driver here but not a ready made project. The good thing is that it does have both IN and OUT endpoint configured.
https://github.com/STMicroelectronics/STM32CubeF0/blob/master/Middlewares/ST/STM32_USB_Device_Library/Class/AUDIO/Src/usbd_audio.c
https://github.com/STMicroelectronics/STM32CubeF1/blob/master/Middlewares/ST/STM32_USB_Device_Library/Class/AUDIO/Src/usbd_audio.c

STM32Cube F2/F4 (I have the STM3220G-Eval and STM32F4 discovery board with F407 but not the F469) have the driver and the Audio sample with both IN and OUT.
https://github.com/STMicroelectronics/STM32CubeF2/tree/master/Projects/STM322xG_EVAL/Applications/USB_Device/AUDIO_Standalone
https://github.com/STMicroelectronics/STM32CubeF4/tree/master/Projects/STM32469I-Discovery/Applications/USB_Device/AUDIO_Standalone

mcuee, 2022-04-17:
> I updated the "E" board patch to instead properly build the I2S (and I2C) code. Now you can listen to the data that you stream out :)
> 
> [0101-v2-Adapt-for-STM3210E_EVAL-board.patch.txt](https://github.com/libusb/libusb/files/8496523/0101-v2-Adapt-for-STM3210E_EVAL-board.patch.txt)

Yes the patch is good. I can build the gcc project and the result binary file works fine. Thanks.

mcuee, 2022-05-08:
I will close this issue and convert the contents to the wiki.

closed 2022-05-08 by mcuee

mcuee, 2022-05-08:
Wiki FAQ page updated:
https://github.com/libusb/libusb/wiki/FAQ#Testing_libusb