libusb/libusb #1053: Examples: to port Linux usbtest.c using libusb

mcuee, 2022-01-30:
It will be great to port Linux usbtest.c (at least part of it) to libusb as it can be used to test different types of USB transfer including isochronous transfer.
https://github.com/torvalds/linux/blob/master/drivers/usb/misc/usbtest.c

diabolo38, 2022-01-30:
As far as i remember usbtest ans testusb https://github.com/torvalds/linux/blob/master/tools/usb/testusb.c  (module and user space side ) are tightly linked and requires gadget zero "loopback+sink" device  to work . Also i kind of remember gagdet 0  was a multiple config device so not window friendly only first config will be usable (bulk ?)  :/ It's a long time i din't look at the   framework so i may said crap sorry for that :( 
They were good tools for testing and bringing up  usb stack for device or host controler  , but would it benefit to libusb ? if my memory not bad it's basically doing bulk ,iso or intr in/out  to a loopback or souce sink test device?

Any extra test add to test coverage and is always welcome but is it worth the effort ?

mcuee, 2022-02-01:
You are right about gadget zero being a dual configuration device. 

gadget zero or other Linux USB gadget is one of the possible test device mentioned here.
https://github.com/libusb/libusb/issues/1052

And you may not need real hardware to use Linux USB gadget with dummy_hcd, so that is an advantage for automatic testing under CI/CD workflow.

mcuee, 2022-04-10:
Close this ticket and we can have further discussion in the following two tickets.
* https://github.com/libusb/libusb/issues/1052
* https://github.com/libusb/libusb/issues/962

closed 2022-04-10 by mcuee