libusb/libusb #1059: Segfault on Homebrew libusb-1.0.25 libusb_free_transfer()

jj1bdx, 2022-02-05:
See https://github.com/jj1bdx/airspy-fmradion/issues/35

I've traced that the segfault (SIGSEGV) was occurred at:
https://github.com/airspy/airspyhf/blob/1.6.8/libairspyhf/src/airspyhf.c#L170
The segfault is observed for both Intel (x86_64) and M1 (arm64), but not always reproducible (i.e., sometimes the segfault doesn't happen).

Replacing/rolling back libusb to Homebrew's libusb@1.0.24 solved this segfault issue.

The segfault example (of terminating the code with CTRL-C):

```
Stopping by getting signal: Interrupt

libusb: debug [libusb_free_transfer] transfer 0x7f8098307b38
zsh: segmentation fault  airspy-fmradion -m fm -E15 -t airspyhf -c freq=89700000 -b 0.1 -P -
```

A similar segfault issue is also reported for openocd at https://github.com/Homebrew/homebrew-core/pull/94207#issuecomment-1029206795

mcuee, 2022-02-05:
@jj1bdx Please post the full debug log. Thanks.

BTW, the openocd issue mentioned in the https://github.com/Homebrew/homebrew-core/pull/94207#issuecomment-1029206795 is a known openocd bug which has been fixed in openocd git head.

Ref:
1) openocd commit
 https://github.com/openocd-org/openocd/commit/cff0e417da58adef1ceef9a63a99412c2cc87ff3
2) libusb issue
* https://github.com/libusb/libusb/issues/928

jj1bdx, 2022-02-05:
Let me post the execution logs of `airspyhf_info`, which also caused the same segfault (not always but sometimes).

* [airspyhf_info source code](https://github.com/airspy/airspyhf/blob/1.6.8/tools/src/airspyhf_info.c)
* `airspyhf_close()` calls `free_transfer()` at https://github.com/airspy/airspyhf/blob/1.6.8/libairspyhf/src/airspyhf.c#L155 and the `free_transfer() calls `libusb_free_transfer()`.

The first one was the log with segfault:
[airspyhf_info-log-segfault.txt](https://github.com/libusb/libusb/files/8007948/airspyhf_info-log-segfault.txt)
The second one was the log of normal termination:
[airspyhf_info-log-noerror.txt](https://github.com/libusb/libusb/files/8007947/airspyhf_info-log-noerror.txt)

jj1bdx, 2022-02-05:
And execution results of `LIBUSB_DEBUG=4 build/airspy-fmradion -m fm -E20 -t airspyhf -c freq=89700000`, manually interrupted by CTRL-C (SIGINT):
The log with segfault:
[airspy-fmradion-log-segfault.txt](https://github.com/libusb/libusb/files/8007991/airspy-fmradion-log-segfault.txt)
The log of normal termination:
[airspy-fmradion-log-noerror.txt](https://github.com/libusb/libusb/files/8007990/airspy-fmradion-log-noerror.txt)

starvald, 2022-02-05:
arch just updated to 1.0.25 and having an problem with my airspy devices also.

[libusb-debuglog.txt](https://github.com/libusb/libusb/files/8008218/libusb-debuglog.txt)

tormodvolden, 2022-02-05:
Airspy is freeing the USB context at https://github.com/airspy/airspyhf/blob/1.6.8/libairspyhf/src/airspyhf.c#L988 and calling libusb_free_transfer() from the following line.

hjelmn, 2022-02-05:
Yeah. Looks like we are exposing usage bugs in multiple pieces of software.

jj1bdx, 2022-02-06:
A quick patch for [the airspyhf driver 1.6.8](https://github.com/airspy/airspyhf/tree/1.6.8). Also available [here](https://github.com/jj1bdx/airspyhf/commit/c55a305a79eea70cf61ea22a081b29d37805b9fc#diff-a314d35f0907b50ed4c2908ddd8db5fcddda3ac6515e4fc51cbd2c241ce88bfb). Looks like this patch has solved the issue. 

```
Subject: [PATCH] fix: change freeing sequence in airspyhf_close()

* A workaround to fix https://github.com/libusb/libusb/issues/1059
* Original bug is exposed here: https://github.com/jj1bdx/airspy-fmradion/issues/35
* libusb_free_transfer() called from airspyhf_close caused SIGSEGV in libusb 1.0.25
* libusb_exit()'s context should not be freed before freeing the transfers
* In this fix, the airspyhf_open_exit() calling point is moved after all the other objects are freed except the device itself
---
 libairspyhf/src/airspyhf.c | 2 +-
 1 file changed, 1 insertion(+), 1 deletion(-)

diff --git a/libairspyhf/src/airspyhf.c b/libairspyhf/src/airspyhf.c
index 08f8bfb..5a0a457 100644
--- a/libairspyhf/src/airspyhf.c
+++ b/libairspyhf/src/airspyhf.c
@@ -985,11 +985,11 @@ int ADDCALL airspyhf_close(airspyhf_device_t* device)
 		pthread_cond_destroy(&device->consumer_cv);
 		pthread_mutex_destroy(&device->consumer_mp);

-		airspyhf_open_exit(device);
 		free_transfers(device);
 		free(device->supported_samplerates);
 		free(device->samplerate_architectures);
 		iq_balancer_destroy(device->iq_balancer);
+		airspyhf_open_exit(device);
 		free(device);
 	}

--
2.35.1
```

mcuee, 2022-02-06:
> Yeah. Looks like we are exposing usage bugs in multiple pieces of software.

Maybe we can try to document such changes better in the future. But this may not be a bad thing either as it uncovers application bugs.

jj1bdx, 2022-02-06:
I guess I can conclude that the bug in this issue was caused by Airspy HF+ driver, not by libusb. I'll close this issue. Thanks for the help to @mcuee @tormodvolden and everyone!

tormodvolden, 2022-02-06:
@jj1bdx You call your quick patch a "workaround" - do you mean this isn't the correct fix?

jj1bdx, 2022-02-06:
@tormodvolden I did my best to fix the bug and I've tested in my applicable machines/OSes and the patch is working correctly (without causing SIGSEGV anymore), but I'm not sure whether this is the really best one because I'm not the principal authors/devs of the Airspy HF+ driver. I believe only the principal devs can determine whether this fix is really correct or not.

starvald, 2022-02-06:
it fixed it for me.

jj1bdx, 2022-02-06:
FYI: for Airspy R2/mini, this has already been fixed in [this commit](https://github.com/airspy/airspyone_host/commit/41c439f16818d931c4d0f8a620413ea5131c0bd6).

jj1bdx, 2022-02-06:
[The patch is merged](https://github.com/airspy/airspyhf/commit/646141f59101b0fe0fc0b624f8eee13e2de42b44).

photron, 2022-02-15:
This bug (or a variation of it) is not fixed, the underlying problem is still there in libusb. Commit 6cae9c6dbd74c0840848f343dd605c5ddcef1ad1 made libusb_free_transfer assume that a libusb_transfer would have a valid dev_handle (and therefore a valid ctx) set. This might not be the case in all circumstances. For example

`libusb_free_transfer(libusb_alloc_transfer(0));`

segfaults in libusb 1.0.25. In my application I run into this because I allocate a bunch of transfers, but might never submit them before freeing them again.

PR #1073 works around this problem, by making TRANSFER_CTX deal with a libusb_transfer without a valid dev_handle.

hjelmn, 2022-02-15:
Indeed. There are two separate signatures. One was not a bug but the other was. We still to fix the one that is an actual bug.

mcuee, 2022-02-23:
In this case, this will only be closed until #1073 is merged.
* #1073

tormodvolden, 2022-03-16:
The referenced PR has been merged to master so this should be fixed now. Please reopen if necessary.

closed 2022-03-16 by tormodvolden