libusb/libusb #1065: Access denied (insufficient permissions) on Windows if already open by another process

rom1v, 2022-02-08:
[`adb`](https://developer.android.com/studio/command-line/adb) is an Android tool to connect to an Android device. It runs a daemon, which opens all Android devices over USB.

I wrote a sample which just opens a specific device (adapt the `VID` and `PID` if you want to test with your device):

<details><summary>source code</summary>

```c
#include <assert.h>
#include <stdio.h>

#include <libusb-1.0/libusb.h>

#define VID 0x18d1
#define PID 0x4ee2

libusb_device *find_device(uint16_t vid, uint16_t pid) {
    libusb_device **list;
    ssize_t count = libusb_get_device_list(NULL, &list);
    assert(count != -1);

    libusb_device *result = NULL;

    for (size_t i = 0; i < (size_t) count; ++i) {
        libusb_device *device = list[i];

        struct libusb_device_descriptor desc;
        int r = libusb_get_device_descriptor(device, &desc);
        if (r < 0)
            continue;
        if (desc.idVendor == VID && desc.idProduct == PID) {
            result = libusb_ref_device(device);
            break;
        }
    }

    libusb_free_device_list(list, 1);
    return result;
}

int main() {
    int r = libusb_init(NULL);
    assert(!r);

    libusb_device *device = find_device(VID, PID);
    if (!device) {
        fprintf(stderr, "Device %04x:%04x not found\n", VID, PID);
        return 1;
    }

    libusb_device_handle *handle;
    r = libusb_open(device, &handle);
    if (r) {
        fprintf(stderr, "error: %s\n", libusb_strerror(r));
    } else {
        printf("Device %04x:%04x ok\n", VID, PID);
        libusb_close(handle);
    }

    libusb_exit(NULL);
    return 0;
}
```

```bash
# native compilation (Linux)
gcc test.c -o test -lusb-1.0

# cross-compilation from Linux for Windows
x86_64-w64-mingw32-gcc test.c -o test.exe -Ilibusb-1.0.25/include -Llibusb-1.0.25/MinGW64/dll -lusb-1.0
```

</details>

On Linux, I can open the device whether the adb daemon is running or not:

```
$ ./test
Device 18d1:4ee2 ok

$ adb start-server
* daemon not running; starting now at tcp:5037
* daemon started successfully

$ ./test
Device 18d1:4ee2 ok

$ adb kill-server

$ ./test
Device 18d1:4ee2 ok
```

However, on Windows, it only works if adb daemon is not running:

```
> test
Device 18d1:4ee2 ok

> adb start-server
* daemon not running; starting now at tcp:5037
* daemon started successfully

> test
error: Access denied (insufficient permissions)

> adb kill-server

> test
Device 18d1:4ee2 ok
```

I guess Windows refuses to open the device because it is already open by another process (the adb daemon).

Is it an intrinsic limitation from the Windows operating system, or is there any way to be able to connect even if the adb daemon is running?

a1ien, 2022-02-08:
https://github.com/libusb/libusb/wiki/Windows#known-restrictions

closed 2022-02-08 by rom1v

diabolo38, 2022-02-09:
> Is it an intrinsic limitation from the Windows operating system, or is there any way to be able to connect even if the adb daemon is running?

If the device is used via winusb yes it's  the known imitation 
you may overcome it if adb use itslef libusb  by replacing autodetect or installaed winusb backend driver by linubsk one 
But it will be very  risky and unsafe to do that i'll discourage it unless you kwon what you'r doing . 
You may brake the binding to other usb function such as modem and file transfer (mtp) over usb .