libusb/libusb #1070: darwin: clock_gettime usage is not guarded by HAVE_CLOCK_GETTIME

parafin, 2022-02-11:
PR #1035 (commit c1ed5895438f2ae3767162f74388b09a16ab0800) introduced into os/darwin_usb.c a clock_gettime usage which is not guarded by HAVE_CLOCK_GETTIME. This breaks compatibility with older macOS versions (see bug #836 for reference). This has to either be fixed or support for macOS <10.12 officially dropped.

mcuee, 2022-02-14:
Please help to create a pull request to fix the issue. Thanks for the help.

hjelmn, 2022-02-14:
Will drop support without a PR. 10.11 is 6 yeas old. I do not plan to support that old an OS but happy to take a PR to fix.

parafin, 2022-02-14:
I'm fine with dropping support, I won't do a PR to fix this myself, but maybe author of original PR (@agoodney) wants to do it. On the other hand there are ways to workaround this problem, e.g. see https://github.com/macports/macports-legacy-support

I would suggest to explicitly mention that macOS <10.12 is not supported (maybe in release notes?) if you are to go this way.

hjelmn, 2022-02-14:
Unfortunately not. Documentation says it is available from 10.12+

Looks like we still have code for older releases. See:

https://github.com/libusb/libusb/blob/master/libusb/os/darwin_usb.c#L2336

Probably should just use those functions to fix the issue. I plan to clean them out at the point when we definitely will not support 10.11 and earlier.

agoodney, 2022-02-14:
To confirm: usbi_get_monotonic_time() needs to be guarded in darwin_reenumerate_device() with HAVE_CLOCK_GETTIME.

Should I make this work an update to my previous closed PR, or start a new PR?

tormodvolden, 2022-02-14:
@agoodney Thanks a lot for working on this. I think a new PR makes sense.

Just want to chime in with what I saw when building on older macOS, as mentioned in https://github.com/libusb/libusb/pull/1056#issuecomment-1030634488
`checking whether OS X target version is 10.12 or later... yes`
when building on 10.4.11 - I wonder if something is wrong about that check (maybe only for really old macos).

agoodney, 2022-02-14:
Should I do this against libusb/libusb or against tormodvolden/libusb?

libusb/libusb doesn't have the tweaks you did to clean up my PR, e.g: 

#define DARWIN_REENUMERATE_TIMEOUT_US (10 * USEC_PER_SEC)

tormodvolden, 2022-02-15:
Yes, against my branch in #1056 would be good, because that one is queued up for inclusion.

seanm, 2022-02-21:
I was just upgrading to libusb 1.0.25 and ran into this issue.  I still need old macOS support so can help with a patch or testing.

Another API you could maybe use is `mach_absolute_time()`, it has existed forever.

tormodvolden, 2022-02-21:
Isn't it as straight-forward as just using usbi_get_monotonic_time(), i.e. never use clock_gettime() directly?

seanm, 2022-02-21:
Yes, seems like it to me.  The usual pattern for such things is to create your own wrapper function then implement it per-platform.

agoodney, 2022-02-21:
So move clock_gettime() inside usbi_get_monotonic() time? Yes, that makes more sense.

seanm, 2022-02-21:
If no one has started a patch for this, I could do it...

tormodvolden, 2022-02-21:
@agoodney , usbi_get_monotonic() uses clock_gettime() if it is available.

I added this to my PR #1056 . Testing on old and new macOS is welcome.

agoodney, 2022-02-21:
I must not have pulled the latest version of your branch or something because I didn't see clock_gettime() inside of usbi_get_monotonic().

I'll update my work to just call usbi_get_monotonic()

tormodvolden, 2022-02-21:
@agoodney No, I didn't modify usbi_get_monotonic. You don't need to update it, I have added it to #1056 already.

seanm, 2022-02-21:
The fallback code for old macOS seems unnecessarily complex... I think I'll whip up a PR to complement your work @tormodvolden ...

seanm, 2022-02-21:
>Ok, I updated my PR against your https://github.com/libusb/libusb/pull/1056 so it calls usbi_get_monotoic_time()

Which PR? Link?

agoodney, 2022-02-21:
My PR is against this branch 

https://github.com/tormodvolden/libusb/tree/get_endpoints_IfV550

tormodvolden, 2022-02-21:
@agoodney Sorry, I didn't see you had already started working on it, I didn't get any notification. Your PR is anyway against an outdated HEAD of my branch. For future situations like this, it is better to file the PR on the main libusb project, which also doesn't prevent you from including the commits from someone else's branch.

agoodney, 2022-02-21:
Do you need me to do anything at this point?

Thanks for the git advice. I've still got a lot to learn about git collaboration on big, public projects like this.

tormodvolden, 2022-02-21:
> The fallback code for old macOS seems unnecessarily complex... I think I'll whip up a PR to complement your work @tormodvolden ...

It does look overly complex at first sight, but after reading #836 I get the impression it is not trivial either. But please let us know if it can be simplified.

tormodvolden, 2022-02-21:
@agoodney If you can do any testing of #1056 on older macOS that would be great. Thanks for your contribution and no worries about the PR confusion, it was more of a GitHub workflow issue than a git issue, their features are usually helpful but sometimes things go through the cracks. And it got unusually busy here after the release. It was dead quiet during 10 days of RC, you might wonder if people actually test them...

@seanm and @parafin We tested macOS back to 10.12 during our RC testing #1046. I'll ping you about older versions next time :)

seanm, 2022-02-21:
>It was dead quiet during 10 days of RC, you might wonder if people actually test them...

In a company environment (where I use libusb), the next 10 days are often booked solid.  I didn't have a chance to try the RCs beforehand alas.

I take a stab at it and share my PR here...

seanm, 2022-02-22:
Here is a draft: https://github.com/libusb/libusb/pull/1080

parafin, 2022-02-22:
> @seanm and @parafin We tested macOS back to 10.12 during our RC testing #1046. I'll ping you about older versions next time :)

Thanx for the offer:) Sadly I don't really have a way to test things on old macOS versions. I noticed this regression thanks to macports guys, see here: https://github.com/macports/macports-ports/pull/13868
Previous clock_gettime issue was reported to darktable bug tracker by a user, so again not caught by me, but just forwarded to you (with analysis).

agoodney, 2022-03-07:
@tormodvolden I can now build/test on MacOS 10.5, so when you have an RC for 1.0.26 I'll test it there.

closed 2022-03-16 by tormodvolden

hjelmn, 2022-03-16:
Weird. c11 support should work fine with older targets. I can take a look.

hjelmn, 2022-03-16:
Ah. I see. No thread local on 10.5.  Guess 10.5 support is dead then.

agoodney, 2022-03-16:
I was able to build 1.0.25 on a 10.5 VM, I'll check this RC

agoodney, 2022-03-17:
I was able to build libusb/master on a 10.5 VM, caveat is that I have MacPorts installed clang v7.1.0

The latest Apple supplied Xcode for 10.5 is using gcc 4.0.1, so I don't think non-MacPort builds have worked for awhile.

I'll see if I can get a 10.6 VM up and running.

I'm willing to help out with things like this, is there a better place for dev chat? Slack? IRC?

seanm, 2022-03-21:
Does it work to build with Xcode 8 + macOS 10.11.5 but set the deployment target to be older?

agoodney, 2022-03-22:
I'll check, I had to install pkg-config (see #1096) and there is a missing #define on 10.11.5 I'm still tracking down.

seanm, 2022-03-22:
>and there is a missing #define on 10.11.5 I'm still tracking down

I just built git master on macOS 10.10.5 with Xcode 7.2.1, didn't see any missing define...

agoodney, 2022-03-22:
I'll start a new issue for it

tormodvolden, 2022-03-28:
@seanm @parafin @agoodney The RC is out :) Please add tested platforms and applications to #1099.

commit 02dc5493 (2022-03-16): darwin: Use usbi_get_monotonic_time() wrapper

On macOS < 10.12 we cannot use clock_gettime().

Closes #1070