libusb/libusb #1072: Can't link against 1.0.25 in OS X Monterey

majenkotech, 2022-02-13:
Code that works perfectly on 1.0.24 in Monterey won't work with 1.0.25:

```
Undefined symbols for architecture x86_64:
  "_SecTaskCopyValueForEntitlement", referenced from:
      _darwin_detach_kernel_driver in libusb-1.0.a(darwin_usb.o)
  "_SecTaskCreateFromSelf", referenced from:
      _darwin_detach_kernel_driver in libusb-1.0.a(darwin_usb.o)
ld: symbol(s) not found for architecture x86_64
```

majenkotech, 2022-02-13:
Fix: need to now add `-framework Security`

closed 2022-02-13 by majenkotech

zhangguofu, 2024-09-12:
May I ask where to add -framework Security?

seanm, 2024-09-12:
Security.framework is already part of libusb.xcodeproj. How are you building?

zhangguofu, 2024-09-13:
> Security.framework is already part of libusb.xcodeproj. How are you building?

I'm using cmake & make to do the jod, and it has been fixed as the snap-shot shows below:
[image]

by the way, why the libusb latest version is not adopted for macOS platform? do you know the story behind?

mcuee, 2024-09-13:
Not so sure what you mean.

Homebrew has libusb 1.0.27.
https://github.com/Homebrew/homebrew-core/blob/ce0f397aaa10126e9c962f87ed546f07c33eac77/Formula/lib/libusb.rb

Same for MacPorts.
https://ports.macports.org/port/libusb/details/

seanm, 2024-09-13:
> by the way, why the libusb latest version is not adopted for macOS platform? do you know the story behind?

I'm not sure what you mean here. (I suspect English in not your first language...)

For me at least, libusb works on all versions of macOS, at least for the things I use it for.

Do you mean why does macOS itself not include libusb?  Apple rarely includes third party software, so that's not surprising, and is unlikely to change.