libusb/libusb #1075: Version 1.0.25 may break mtp?

soerengrunewald, 2022-02-17:
Output with version 1.0.25: Actually Fedora package 1.0.25-3 (including PR #1058 and PR #1073)
```
Feb 17 14:39:59 has4116x.desy.de kernel: usb 2-1.1: USB disconnect, device number 21
Feb 17 14:39:59 has4116x.desy.de kernel: usb 2-1.1: new high-speed USB device number 22 using ehci-pci
Feb 17 14:39:59 has4116x.desy.de kernel: usb 2-1.1: New USB device found, idVendor=22b8, idProduct=2e82, bcdDevice= 3.18
Feb 17 14:39:59 has4116x.desy.de kernel: usb 2-1.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Feb 17 14:39:59 has4116x.desy.de kernel: usb 2-1.1: Product: Moto G (5)
Feb 17 14:39:59 has4116x.desy.de kernel: usb 2-1.1: Manufacturer: motorola
Feb 17 14:39:59 has4116x.desy.de kernel: usb 2-1.1: SerialNumber: XXXXXXXXXX
Feb 17 14:40:02 has4116x.desy.de gvfsd[182995]: PTP_ERROR_IO: failed to open session, trying again after resetting USB interface
Feb 17 14:40:02 has4116x.desy.de gvfsd[182995]: LIBMTP libusb: Attempt to reset device
Feb 17 14:40:02 has4116x.desy.de kernel: usb 2-1.1: reset high-speed USB device number 22 using ehci-pci
Feb 17 14:40:03 has4116x.desy.de gvfsd[182995]: inep: usb_get_endpoint_status(): Resource temporarily unavailable
Feb 17 14:40:03 has4116x.desy.de gvfsd[182995]: outep: usb_get_endpoint_status(): Resource temporarily unavailable
Feb 17 14:40:03 has4116x.desy.de kernel: usb 2-1.1: reset high-speed USB device number 22 using ehci-pci
Feb 17 14:40:03 has4116x.desy.de gvfsd[182995]: LIBMTP PANIC: failed to open session on second attempt
Feb 17 14:40:04 has4116x.desy.de gvfsd[182995]: Device 0 (VID=22b8 and PID=2e82) is a Motorola Moto G (ID2).
```
Output with version 1.0.24
```
Feb 17 15:44:18 has4116x.desy.de kernel: usb 2-1.1: new high-speed USB device number 39 using ehci-pci
Feb 17 15:44:18 has4116x.desy.de kernel: usb 2-1.1: New USB device found, idVendor=22b8, idProduct=2e82, bcdDevice= 3.18
Feb 17 15:44:18 has4116x.desy.de kernel: usb 2-1.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Feb 17 15:44:18 has4116x.desy.de kernel: usb 2-1.1: Product: Moto G (5)
Feb 17 15:44:18 has4116x.desy.de kernel: usb 2-1.1: Manufacturer: motorola
Feb 17 15:44:18 has4116x.desy.de kernel: usb 2-1.1: SerialNumber: XXXXXXXXXX
Feb 17 15:44:18 has4116x.desy.de gvfsd[228005]: Device 0 (VID=22b8 and PID=2e82) is a Motorola Moto G (ID2).
Feb 17 15:44:18 has4116x.desy.de gvfsd[228005]: Android device detected, assigning default bug flags
Feb 17 15:44:18 has4116x.desy.de gvfsd[228005]: Received event PTP_EC_DevicePropChanged in session 1
```

soerengrunewald, 2022-02-17:
It looks like PR #1058 causes the issue.

mcuee, 2022-02-18:
Just wondering how do you come to the conclusion that #1058 is the causing the issue here? Do you meant that plain 1.0.25 release is okay whereas it will fail once #1058 is applied?

benzea, 2022-02-18:
I would have expected the original version of #1073 (which was included in libusb1-1.0.25-3) broke mtp. It isn't entirely clear from the log though, but that patch resulted in all synchronous transfers to fail. This includes anything that tries to resolve descriptors pretty much.

mcuee, 2022-02-24:
@soerengrunewald Please help to try the latest #1073. Thanks.
* #1073

soerengrunewald, 2022-03-16:
Sorry for the late update. @benzea updated the [patch](https://src.fedoraproject.org/rpms/libusb1/c/07d18ab2e1c34fee14c1fbcef456dc9e06772d83?branch=f35) in the F35 repo and that works. So I would assume #1073 is ok.

mcuee, 2022-03-16:
Close this in this case.

cptX, 2023-04-22:
Hi, the last months I'm receiving the following error in linux mint 20.3 and I cannot connect my mobile phone with USB.

Apr 22 11:14:17 i7 gvfsd[5334]: PTP_ERROR_IO: failed to open session, trying again after resetting USB interface
Apr 22 11:14:17 i7 gvfsd[5334]: LIBMTP libusb: Attempt to reset device
Apr 22 11:14:17 i7 kernel: [  508.292956] usb 1-11: reset high-speed USB device number 19 using xhci_hcd
Apr 22 11:14:17 i7 kernel: [  508.568991] usb 1-11: reset high-speed USB device number 19 using xhci_hcd
Apr 22 11:14:25 i7 gvfsd[5334]: LIBMTP PANIC: failed to open session on second attempt
Apr 22 11:14:25 i7 kernel: [  516.406151] usb 1-11: USB disconnect, device number 19
Apr 22 11:14:26 i7 gvfsd[5334]: Device 0 (VID=2717 and PID=ff40) is a Xiaomi Mi-2s (id2) (MTP).
Apr 22 11:14:27 i7 kernel: [  518.220706] usb 1-11: new high-speed USB device number 21 using xhci_hcd
Apr 22 11:14:27 i7 kernel: [  518.374627] usb 1-11: New USB device found, idVendor=2717, idProduct=ff40, bcdDevice= 2.23
Apr 22 11:14:27 i7 kernel: [  518.374633] usb 1-11: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Apr 22 11:14:27 i7 kernel: [  518.374637] usb 1-11: Product: Redmi Note 9T
Apr 22 11:14:27 i7 kernel: [  518.374641] usb 1-11: Manufacturer: Xiaomi

Is this relevant to the current issue? What can I do to fix it?

mcuee, 2023-04-22:
@cptX 

Please make sure you update your libmtp and libusb version first to see if the issue is still there or not. 

If you still see the issue, then your issue is different from this one. In that case, please create a new issue.