libusb/libusb #1077: Is concurrent async "in" transfers order guaranted?

sonatique, 2022-02-18:
Hello, 
I guess the answer is "yes", but I didn't find a clear official answer despite all my seaches. Sorry if I missed it.

Say I submitted N async transfers for the same "in" endpoint, all with infinite timeout and same buffer size (say 65536 bytes). N is typically 16, but my question holds for any N>1.

I use the event handling thread technique (https://libusb.sourceforge.io/api-1.0/group__libusb__asyncio.html#eventthread) running a single loop in a dedicated thread for processing event.

Say that for any reason, this dedicated loop needs a large amount of time in between two consecutive libusb_handle_events calls (could be because the actual callback takes a long time or because there is some CPU cycle starvation at some point).

Say that during this period of time, a lot of data is received, such that several transfer buffers are filled and are now ready to be processed.

What will happen on the next libusb_handle_events call from within the dedicated thread? Is there a guarantee that the first transfer object to be processed will contain the first data receieved, and then that the next one (on the next loop iteration) will contains the next data?

In other words: is there a mechanism that guarantee that transfers are registered in the "event queue" (whatever it is) in the order they are filled and a guarantee that each call to libusb_handle_events will retreive them is that order?

Thanks a lot.

mcuee, 2022-06-02:
@hjelmn @tormodvolden Sorry I can not answer this one. Please help to answer. Thanks.

tormodvolden, 2022-09-08:
I don't know either but I guess this is down to details in how the OS is processing the transfers and presenting events to the libusb code in your application. Is this a general question or do you have a specific OS and driver that you are targeting? What kind of transfers? What problem do you want to solve, or avoid?

mcuee, 2022-09-12:
@sonatique 

Please help to answer the query from Tormod. Thanks.

sonatique, 2022-09-12:
@tormodvolden Thank you for jumping in. As far as I can understand, this behavior shall not be dependent on the OS implementation details, or it would basically defeat the portability purpose of libusb, in my opinion. Either the order is guaranteed for all implementations (and your application can rely on it) or it is not guarantee (and you cannot rely on it, though it may happen that it is implemented that way by chance on some given OS). This should be made clear, I think.
I am targeting all possible OS / driver with my application. The transfer type is mainly bulk (though the question should apply to all types).
So now: a little bit more background: take the “sam3u_benchmark.c” example code. You’ll see that the current code only submits one transfer at a time, resubmitting it every time it completes. There is this note in a comment, however:
“NOTE: To reach maximum possible performance the program must submit *multiple* transfers here, not just one.
When only one transfer is submitted there is a gap in the bus schedule from when the transfer completes until a new transfer is submitted by the callback. This causes some jitter for isochronous transfers and loss of throughput for bulk transfers. 
This is avoided by queueing multiple transfers in advance, so that the host controller is always kept busy, and will schedule more transfers on the bus while the callback is running for transfers which have completed on the bus.”
So: I tried to follow this advice, though there is not a single tested example of it in libusb repository.
I faced many issues when going from a little proof of concept to a real complex application. One of it was that I observed disorder: some chunk of data were received before others, in a different order than they were sent on the bus. Say I had a first transfer completing with “4 5 6” in the buffer, then another one with “0 1 2 3” in its buffer, when “0 1 2 3 4 5 6” has been actually sent on the bus.
This was unexpected to me, so I asked this question here. 
Later I found out that the issue was occurring because I was simultaneously i) having several submitted transfers with one properly handled single “event loop thread” to process them and ii) using libusb built-in synchronous Control Transfer functions.
Libusb built-in synchronous Control Transfer function are actually implemented by submitting one async transfer and waiting for completion from within the caller thread. So: during some period of time, I had 2 concurrent event handler threads: the caller thread of the Control Transfer would actually process some completed “data” transfer I submitted, and my event handler thread would process the control transfer completion. Of course, when you have more than one thread processing completions of several submitted transfers, the overall order cannot be guaranteed.
As a result, I rewrote all Control Transfer functions I needed to manually submit transfers in a way that only my event thread would process their completion. Therefore, at every instant I only have one single event thread. Since then, I did not observe disorder anymore.
Rewriting everything was some heavy work and I felt I really was in an undocumented territory: on one hand, some comment in an example program encouraged me to use multiple transfer submissions, but on the other hand, doing it for real “naively” triggers a lot of issue for which there is no help / documentation / example.
So: after I found out that the disorder I observed was caused by multiple event threads acting concurrently, I decided to leave my question open here since I still think that the behavior shall be better documented, and that what is guarantee and what is not shall be better described.
In conclusion: as of today, I managed by myself to not have issues anymore, but given the benefit of using multiple transfers, I think libusb should go beyond a simple comment burden deep in an example code and really provide full working example and improved documentation related to this.
I hope this helps, thanks again.

tormodvolden, 2022-09-12:
Thanks for the good explanation and context.

How I have seen this is that your application is not guaranteed to receive the transfer completions in order, but the transfer requests will always be sent to the device in the order you submit them. When you receive the transfer completion, it is accompanied by the user_data pointer that you provided in the request so that you can identify and match it to your submitted requests and thus e.g. copy the received data into the right buffer location of your application.

I can imagine how mixing synchronous and synchronous functions ends up with several event handlers although I haven't looked at it. Maybe we should warn about this. So far our sync functions documentation says "Another issue is that by tying up the thread with that single transaction there is no possibility of performing I/O with multiple endpoints and/or multiple devices simultaneously, unless you resort to creating one thread per transaction."

Other relevant documentation is https://libusb.sourceforge.io/api-1.0/group__libusb__asyncio.html#asyncevent which explains how to use an event handling thread, but it doesn't mention possible interference with sync functions. It communicates there should be only one event handling thread though.

Anyway, if we say the completion event order is not guaranteed, and you have to rely on user_data, the issue with multiple event threads that you experienced is not really a problem.

For the example application, please submit one, that would be nice to have. We have been discussing isochronous transfer benchmark tools where this would be needed, I guess related to the sam3u example (#962 ?).

tormodvolden, 2022-09-12:
The user_data pointer that you provided to e.g. libusb_fill_bulk_stream_transfer() will be in the libusb_transfer struct that is handed to your callback on completion.

tormodvolden, 2022-09-12:
> As far as I understand, the order in which you submit the transfer doesn’t really matter: there is no “First submitted, first to be used” policy anywhere, so I could as well have used numbering 3, 2, 4, 1,

Are you saying that if you submit them in order 1,2,3,4, you cannot be sure the device will receive them in that order? I thought that was guaranteed but I could be wrong. Anyway that shouldn't be a problem, unless your device needs to receive and process the requests in the defined order. Is this the case?

tormodvolden, 2022-09-12:
> Since I cannot rely on user data to order USB fragments, I have to rely on the order in which I receive them to order small buffers into the larger buffer.

Yes, you can rely on user_data to order the completed transfers. This is exactly how you can use it. If you submitted requests with user_data 1,2,3,4 you can now read out the user_data from the completed transfers and copy the data buffer into large_buffer[i] where _i_ is derived from user_data.

sonatique, 2022-09-12:
> Are you saying that if you submit them in order 1,2,3,4, you cannot be sure the device will receive them in that order? I thought that was guaranteed but I could be wrong. Anyway that shouldn't be a [...]

We are talking about "in" transfer here exclusively. I don't think the concept of "the device will receive them" makes sense. I understand that when submitting "in" transfer, all that happen is that there are some structure that are registered at the low level OS system in order to receive incomming data. Nothing is really sent to the device (appart from USB poll). If you plug a USB sniffer you won't see anything when submitting in transfers.

For out transfers however, yes, I guess the order is guranteed.

martinling, 2022-09-12:
There is plenty of code out there which assumes that multiple transfers submitted _on the same endpoint_ will be completed in order. Two examples that I'm familiar with which do this are libhackrf, and the fx2lafw driver in sigrok. It would be very surprising to find that this code is incorrect. Certainly to the best of my knowledge this approach has never resulted in out-or-order data being seen.

In both those examples, the transfers all use the same `user_data` pointer, with no per-transfer differences, and the order of completion is assumed to be the order of submission.

I wouldn't expect completion ordering to match submission ordering for transfers on different endpoints, but it would be quite a surprise for libusb to announce that there is no ordering guarantee at all.

sonatique, 2022-09-13:
@tormodvolden @martinling : thank you for your explanations and interest in this subject.
@martinling : in my case, the fact that the first submitted in transfer is the first to be callback-ed for when data arrives on the bus is not important (it is a good thing to know that this is guaranteed anyway). 
My concern is about whether the first transfer that have its callback called will contains the first bytes actually transferred on the bus, for a given endpoint and assuming there is a single thread doing the event loop.
As I understand from @martinling "algorithm" this should be true. 
I can see that since order of completion is guaranteed to be the order of submission, altogether we have: first submitted transfer contains the first part of the transmitted data when it completes. If we’re sure that there is only one thread processing all events, then this is sufficient, we don’t need to use User Data to re-order things.
However, if at some point we have more than one thread processing the even loop (such as having one main loop + a certain number of “sync calls” that introduces new thread into the game, then order could be jammed somehow, but using UserData to track order of submission would help sorting things out.
I hope I got everything right, thanks again!

martinling, 2022-09-13:
Hi @sonatique,

Something you may not be aware of is that there is an event handling lock, which can only be held by one thread at a time. When you call `libusb_handle_events` it takes this lock, and releases it when it returns. So even if you have multiple threads handling events (e.g. jumping in to make synchronous calls), the overall ordering of completions should be the same.

That's a very rough explanation only - this is all explained in a lot more detail here:
https://libusb.sourceforge.io/api-1.0/libusb_mtasync.html

sonatique, 2022-09-13:
> Something you may not be aware of is that there is an event handling lock, which can only be held by one thread at a time. When you call `libusb_handle_events` it takes this lock, and releases it [...]

Yes, thank you @martinling . I am aware of this, thanks. I wish we could talk directly, it would make my point much easier to explain. Actually, now that the orderering is asserted, I am very well aware that the reminder of my issue in not really libusb-1.0 business, but let me explain:
Say that, in my implementation, in order to minimize the time spent in the callback function, all what the callback does is putting (could be copy or not) the received bytes into a queue. This queue will be processed by another thread, such that the libusb event handler thread can run as fast as possible.

Now: it there is only one single event handler thread, and since callbacks are made such as to maintain the order in which data is received on the bus, then the same order will be preserved in the queue mentionned before.

But, if there is 2 (or more) threads at some instant, for instance the regular dedicated event handler + another one caused by a "sync" call: one of those thread (say A) will receive the fisrt of the next available data chunk, and then sligtly later, another one  (say B) of those will receive the second next available data chunk. 

But from now on, nothing will gurantee that thread A will put its received bytes in the queue before thread B. One need to use an additional mechanism (such as numbers is User Data) in order to reorder chunks.

I decided to avoid this by rewriting the libusb built-in control transfer function with async transfer in a way that only my dedicated event handler thread process their callback. Its one way of handling the issue, there are other of course.

martinling, 2022-09-14:
@sonatique I think that in the scenario you describe, the queue ordering is still guaranteed.

Your event handler thread and the various threads making sync calls cannot all hold the lock at the same time. Only one of them can actually be holding the lock and handling events. And your callbacks are called as part of the event handling.

Take a random example - say your dedicated thread (call it D) handles the completion of transfer 1. Then thread A comes in and takes over event handling and handles the completion of transfer 2. Meanwhile thread B makes another sync call, but A is already handling events and ends up handling the completion of transfer 3. Then B takes over and handles transfer 4, but then handling reverts back to D again for transfer 5.

Your callback ends up being called from thread D, then thread A twice, then B, then D again. But all those calls remain strictly sequenced because of the lock. The callback is called for transfer 1, then 2, then 3, then 4, then 5. And each call has to return before the next call can be made. The calls jump around between threads but they're not concurrent. If your callback puts the transfer data in a queue, then the queue will contain data from transfers 1-5 in order, regardless of what happens with the threads.

sonatique, 2022-10-04:
@martinling Thanks a lot for you reply and sorry for my delayed answer.

Given current implemention, I am not sure that we can say that "only one [thread] can actually be holding the lock and handling events". I might be wrong, but looking at the code of "usbi_handle_transfer_completion" in io.c, I only see lock being held around the state change (setting state_flags).

I think that there is no lock around the actual transfer->callback (in which I would "add to queue" my things).
From there, I don't think we can say that all those calls in you example will remain strictly sequenced.
I see nothing preventing thread B to execute its "transfer->callback(...)" while thread A is still inside its own. 
So at one given instant I think I may have 2 (or more) threads concurrently executing my "add to queue" method. Even if I add synchronization mechanism to ensure my queue operations intergrity, there is no way to ensure that thread A will be able to "add to queue" before thread B, event if it started to execute these lines of code before.

I guess there is no lock around transfer->callback on purpose: if one callback function would take very long (or get dead-locked), this would prevent all other pending callback to be executed. Since the API (in particular the standard sync API) permit to submit and wait concurrently and independantly for several calls, I guess having a lock around the callback call would just not work in the general case.

As it is now, you may have one sync call blocked or taking very long while other are still happily executed on other threads.

For this reason, submitting several concurrent read transfers does not play nice (in the general case) with having multiple threads allowed to handle events. Especially, I figured out that calling the sync API from a random thread while having a dedicated thread handling all callbacks of my concurrently submitted transfers just screw everything, because the random thread could handle completion of my transfer concurrently with the dedicated thread, messing with the order, then go back to handling the result of the sync call. Another issue is the dedicated thread handling the result of the sync call, spending a long time doing it instead of taking care of my tranfers.
That's why I had to rewrite all my sync call for using only the dedicated thread. 

What do you think, am I completly wrong? Thanks!

martinling, 2022-10-20:
@sonatique It sounds like you may be right, which would be bad news as it means a lot of code out there is subtly wrong!

If so, I wonder if there's some better way to fix this. It's painful to have to use a custom implementation of synchronous transfers just to enforce which thread does your event handling. And using `user_data` to reorder transfers is not great either, because it means you have to allocate extra space, and copy data around, to implement reordering.

@tormodvolden any thoughts?

Do we perhaps need some new API that lets you do:
```c
// In dedicated handler thread - bind the event handling to stay with this thread.
libusb_bind_event_handling(ctx);
while (!stop) {
    // All event handling for this device will be done here, so no transfer callbacks can run concurrently.
    libusb_handle_events(ctx);
}
libusb_unbind_event_handling(ctx);
```
...with the synchronous API functions modified such that they will play along, and not try to take over handling themselves?

mcuee, 2022-10-20:
@tormodvolden I have added the `core` label and keep the `Documentation` label. This is one area I feel libusb needs better documentation and better examples. I have never been able to understand this well myself. Please share your thoughts about this issue and the proposal from Martin.

benzea, 2022-10-20:
Hmm, I wonder why a reordering is occuring though. The events should be firing in the correct order (libusb internal locking), it is just random which thread ends up calling the callback. If reordering occurs, then I would expect that something else is going wrong in the application logic (which is completely plausible as e.g. an event on a separate thread might be queued while it is immediately processed if it is on the correct thread).

> It's painful to have to use a custom implementation of synchronous transfers just to enforce which thread does your event handling.
> ...
> Do we perhaps need some new API that lets you do: ... libusb_bind_event_handling(ctx); ...

I think you can already do that by using `libusb_try_lock_events` together with `libusb_handle_events_locked` (or the related functions for full event loop integration).

martinling, 2022-10-20:
> Hmm, I wonder why a reordering is occuring though. The events should be firing in the correct order (libusb internal locking), it is just random which thread ends up calling the callback.

The theory that @sonatique put forward was that re-ordering and concurrency of callbacks can occur because the callbacks are not called with the lock held.

However, I'm looking at the code now and I think that @sonatique has misunderstood the locking. They mentioned `usbi_handle_transfer_completion` and saw locking only around the flag change in that function. But that is only the individual transfer lock, not the event waiters lock. The event locking occurs outside that function, further up the call stack.

The `usbi_handle_transfer_completion` function is called from the backend's `.handle_events` hook, which is called from the common `handle_events` helper in `io.c`. That `handle_events` helper is documented to assume that no other thread is concurrently running it. And it is called in only one place: inside `libusb_handle_events_timeout_completed`, which calls it with the event waiters lock held. All the other variants of `libusb_handle_events_*` are implemented as wrappers around `libusb_handle_events_timeout_completed`.

So I think what I said earlier in [this comment](https://github.com/libusb/libusb/issues/1077#issuecomment-1246078645) is correct, and that two callbacks can never run concurrently, or out of order, despite the fact that the event handling may jump around between threads.

Yet @sonatique was apparently seeing something like this happen, and needed to fix their code to prevent it. It would be good to know more about that, but it may just be a bug on the application side.

I think it might still be useful to have a way to force event handling to stay in a specific thread, though - e.g. if a callback needs to access thread-local variables, thread-owned resources, or to stay pinned to a core for performance reasons.

> I think you can already do that by using `libusb_try_lock_events` together with `libusb_handle_events_locked` (or the related functions for full event loop integration).

Yes, but that means you need to implement the polling part of the event loop yourself, and that's currently only possible on operating systems that use file descriptors. On Windows, event sources are represented as a `HANDLE`, not a file descriptor, and there's no `poll` - so `libusb_get_pollfds` just returns NULL. There's no equivalent Windows-specific function to get event handles for use with `WaitForMultipleObjects`, which is the equivalent API there.

There was a proposal [here](https://github.com/libusb/libusb/issues/252#issuecomment-299331116) in #252 to add a cross-platform API to access other types of event sources, but it hasn't happened as yet. Most projects have just decided to run a separate handler thread rather than trying to integrate libusb into their own event loop. Sigrok hung onto the latter approach for many years, but eventually gave up due to this limitation, and runs a handler thread in the background on Windows only.

martinling, 2022-10-20:
Oh, sorry, I'm wrong about that last bit - you can indeed use `libusb_handle_events_locked` to force handling to a specific thread, without doing your own polling, by passing it a non-zero timeout.

sonatique, 2022-10-21:
Thank you everybody! I think we had a very interesting discussion here, at least from my point of view because I learned a lot. Thank you all for your time and efforts in investigating and replying. 

I will reconsider all my code with the new light that this discussion shined on this matter, and try to find the real reason behind my issues, since it is quite clear now that they cannot come from libusb itself.

I still think that a good working reference example showing how to properly use multiple submitted bulk "in" transfers would help newcomers and make a statment about the fact that this is really a supported practice.

Thanks again. Of course I will come back here if I happen to find anything new relevant to the subjet.

martinling, 2022-10-21:
Yes, queuing up multiple transfers definitely needs some explanatory documentation. This was also raised recently in #1205.

And as this discussion shows, what guarantees exist are far from clear, even to those working on the project.

Although we've confirmed that the event waiters lock prevents concurrent callbacks, it's still not 100% clear whether the order in which completions are handled is guaranteed to be the order in which they were submitted and executed on the bus. I think that for transfers submitted on the same endpoint the ordering ought to be maintained, but I'd like to be certain about whether that's guaranteed across all backends and situations.

lujnan, 2023-04-21:
The root cause is that in different threads, synchronous and asynchronous APIs are mixed
In the composite device, even if the dup(fd) method is used, there is no way to prevent different threads from returning to reap_for_handle, and the thread will be scheduled by the system at this point in time, which leads to sequence disorder