libusb/libusb #1079: Crash when closing a device with a pending sync transfer

jacobly0, 2022-02-21:
I have observed a segfault on [this line](https://github.com/libusb/libusb/blob/4622bfcf44db373c53502e3fe873bd611e8332f6/libusb/sync.c#L39) indicating that `transfer->dev_handle` is `NULL`.  The [logic](https://github.com/libusb/libusb/blob/4622bfcf44db373c53502e3fe873bd611e8332f6/libusb/sync.c#L59) in `sync_transfer_wait_for_completion` suggests that `dev_handle` being `NULL` is normal if the device was closed while the transfer was in progress, and indeed, it is set to `NULL` [here](https://github.com/libusb/libusb/blob/4622bfcf44db373c53502e3fe873bd611e8332f6/libusb/core.c#L1419).

This bug appears to be introduced by 6cae9c6dbd74c0840848f343dd605c5ddcef1ad1.

Let me know if further information is needed or if I should work on a reproducer.

hjelmn, 2022-02-21:
Duplicate bug.

mcuee, 2022-02-22:
@jacobly0 Can you try the following two pull requests applied?
* https://github.com/libusb/libusb/pull/1058
* https://github.com/libusb/libusb/pull/1073

jacobly0, 2022-02-23:
I've been running continuous transfers for 40 hours with those changes and I haven't seen the issue happen again, the last time it happened after about 36 hours.

skgsergio, 2022-02-27:
I was facing this very same issue with another project and I've applied the patches linked https://github.com/libusb/libusb/issues/1079#issuecomment-1047489305 over 1.0.25 release and it is working again.

tormodvolden, 2022-03-16:
The above has been merged to master so this should be fixed now. Please reopen if necessary.

closed 2022-03-16 by tormodvolden