libusb/libusb #1081: 1.0.25 Crashed when calling TRANSFER_CTX(transfer)

dspsir, 2022-02-23:
TRANSFER_CTX means transfer->dev_handle->xxxxxx. When transfer->dev_handle is equal to NULL, this Macro will cause crash. do_sync_bulk_transfer() in sync.c called libusb_free_transfer() is sufferring from this problem. Please fix it as soon as possible.

benzea, 2022-02-23:
Duplicate, see https://github.com/libusb/libusb/issues/1079#issuecomment-1047489305

mcuee, 2022-02-24:
@dspsir Yes, this is a duplicate. Please try the following two pull requests applied on top of 1.0.25 release. Thanks.

* https://github.com/libusb/libusb/pull/1058
* https://github.com/libusb/libusb/pull/1073

tormodvolden, 2022-03-16:
This should be fixed in master now. Please reopen if the issue is not fixed.

closed 2022-03-16 by tormodvolden