libusb/libusb #1082: hotplug enumerate regression in 1.0.25

lunixbochs, 2022-02-23:
With https://github.com/libusb/libusb/commit/6929b8270170a76e9af30d0a9e95c2081f4372b0 and roughly this code, the enumerated device list is sent twice to the hotplug callback on initialization.

Testing on macOS 12, M1, with libusb 1.0.25.
In 1.0.24 and prior, the list of devices was correctly sent only one time to the callback.

```
static int hotplug_event(
        struct libusb_context *ctx,
        struct libusb_device *dev,
        libusb_hotplug_event event,
        void *user) {
    ...
}

    libusb_hotplug_callback_handle handle = 0;
    int rc = libusb_hotplug_register_callback(eye_uctx.ctx,
        LIBUSB_HOTPLUG_EVENT_DEVICE_ARRIVED | LIBUSB_HOTPLUG_EVENT_DEVICE_LEFT,
        LIBUSB_HOTPLUG_ENUMERATE,
        LIBUSB_HOTPLUG_MATCH_ANY, LIBUSB_HOTPLUG_MATCH_ANY, LIBUSB_HOTPLUG_MATCH_ANY,
        hotplug_event, cb, &handle);
```

mcuee, 2022-02-24:
Please help to post the debug log by setting LIBUSB_DEBUG=4 and then run your application. Thanks. Does this happen on other OS like Windows or Linux?

gsurkov, 2022-02-24:
I can confirm the same issue on linux (Arch Linux, 5.16.10, libusb 1.0.25).
The code I am using to register the hotplug:
```c++
const auto events = libusb_hotplug_event(LIBUSB_HOTPLUG_EVENT_DEVICE_ARRIVED | LIBUSB_HOTPLUG_EVENT_DEVICE_LEFT);
const auto err = libusb_hotplug_register_callback(nullptr, events, LIBUSB_HOTPLUG_ENUMERATE, info.vendorID(),
                          info.productID(), LIBUSB_HOTPLUG_MATCH_ANY, libusbHotplugCallback, this, nullptr);
```
If I remove the `LIBUSB_HOTPLUG_ENUMERATE` flag, the callback gets activated only once, but then it stops working on the earlier versions.
Version 1.0.24 works just fine, only one enumerate event if the device was already connected and only one arrived event if it was connected during the application run time.

Below is the output with `LIBUSB_DEBUG=4` and libusb = 1.0.25.
```
[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.000021] [0002d815] libusb: debug [libusb_init] created default context
[ 0.000031] [0002d815] libusb: debug [libusb_init] libusb v1.0.25.11692
[ 0.000041] [0002d815] libusb: debug [usbi_add_event_source] add fd 13 events 1
[ 0.000051] [0002d815] libusb: debug [usbi_io_init] using timer for timeouts
[ 0.000057] [0002d815] libusb: debug [usbi_add_event_source] add fd 14 events 1
[ 0.000068] [0002d815] libusb: debug [get_kernel_version] reported kernel version is 5.16.10-arch1-1
[ 0.000087] [0002d815] libusb: debug [op_init] found usbfs at /dev/bus/usb
[ 0.000109] [0002d815] libusb: debug [op_init] max iso packet length is (likely) 98304 bytes
[ 0.000128] [0002d815] libusb: debug [op_init] sysfs is available
[ 0.000397] [0002d81b] libusb: debug [linux_udev_event_thread_main] udev event thread entering
[ 0.007806] [0002d815] libusb: debug [linux_get_device_address] getting address for device: usb1 detached: 0
[ 0.007817] [0002d815] libusb: debug [linux_get_device_address] scan usb1
[ 0.007851] [0002d815] libusb: debug [linux_get_device_address] bus=1 dev=1
[ 0.007856] [0002d815] libusb: debug [linux_enumerate_device] busnum 1 devaddr 1 session_id 257
[ 0.007860] [0002d815] libusb: debug [linux_enumerate_device] allocating new device for 1/1 (session 257)
[ 0.007938] [0002d815] libusb: debug [linux_get_device_address] getting address for device: 1-1 detached: 0
[ 0.007943] [0002d815] libusb: debug [linux_get_device_address] scan 1-1
[ 0.007968] [0002d815] libusb: debug [linux_get_device_address] bus=1 dev=2
[ 0.007973] [0002d815] libusb: debug [linux_enumerate_device] busnum 1 devaddr 2 session_id 258
[ 0.007978] [0002d815] libusb: debug [linux_enumerate_device] allocating new device for 1/2 (session 258)
[ 0.008006] [0002d815] libusb: debug [linux_get_parent_info] dev 0x55bc3dbf2f90 (1-1) has parent 0x55bc3dc2de80 (usb1) port 1
[ 0.008083] [0002d815] libusb: debug [linux_get_device_address] getting address for device: 1-1.2 detached: 0
[ 0.008089] [0002d815] libusb: debug [linux_get_device_address] scan 1-1.2
[ 0.008116] [0002d815] libusb: debug [linux_get_device_address] bus=1 dev=17
[ 0.008121] [0002d815] libusb: debug [linux_enumerate_device] busnum 1 devaddr 17 session_id 273
[ 0.008126] [0002d815] libusb: debug [linux_enumerate_device] allocating new device for 1/17 (session 273)
[ 0.008152] [0002d815] libusb: debug [linux_get_parent_info] dev 0x55bc3dc2dd40 (1-1.2) has parent 0x55bc3dbf2f90 (1-1) port 2
[ 0.008219] [0002d815] libusb: debug [linux_get_device_address] getting address for device: 1-1.4 detached: 0
[ 0.008224] [0002d815] libusb: debug [linux_get_device_address] scan 1-1.4
[ 0.008254] [0002d815] libusb: debug [linux_get_device_address] bus=1 dev=82
[ 0.008259] [0002d815] libusb: debug [linux_enumerate_device] busnum 1 devaddr 82 session_id 338
[ 0.008262] [0002d815] libusb: debug [linux_enumerate_device] allocating new device for 1/82 (session 338)
[ 0.008289] [0002d815] libusb: debug [linux_get_parent_info] dev 0x55bc3dbf36c0 (1-1.4) has parent 0x55bc3dbf2f90 (1-1) port 4
[ 0.008357] [0002d815] libusb: debug [linux_get_device_address] getting address for device: 1-1.6 detached: 0
[ 0.008362] [0002d815] libusb: debug [linux_get_device_address] scan 1-1.6
[ 0.008388] [0002d815] libusb: debug [linux_get_device_address] bus=1 dev=5
[ 0.008392] [0002d815] libusb: debug [linux_enumerate_device] busnum 1 devaddr 5 session_id 261
[ 0.008396] [0002d815] libusb: debug [linux_enumerate_device] allocating new device for 1/5 (session 261)
[ 0.008421] [0002d815] libusb: debug [linux_get_parent_info] dev 0x55bc3dbf30d0 (1-1.6) has parent 0x55bc3dbf2f90 (1-1) port 6
[ 0.008482] [0002d815] libusb: debug [linux_get_device_address] getting address for device: usb3 detached: 0
[ 0.008487] [0002d815] libusb: debug [linux_get_device_address] scan usb3
[ 0.008512] [0002d815] libusb: debug [linux_get_device_address] bus=3 dev=1
[ 0.008516] [0002d815] libusb: debug [linux_enumerate_device] busnum 3 devaddr 1 session_id 769
[ 0.008521] [0002d815] libusb: debug [linux_enumerate_device] allocating new device for 3/1 (session 769)
[ 0.008608] [0002d815] libusb: debug [linux_get_device_address] getting address for device: 3-1 detached: 0
[ 0.008614] [0002d815] libusb: debug [linux_get_device_address] scan 3-1
[ 0.008640] [0002d815] libusb: debug [linux_get_device_address] bus=3 dev=2
[ 0.008644] [0002d815] libusb: debug [linux_enumerate_device] busnum 3 devaddr 2 session_id 770
[... cut ...]
[ 0.579096] [0002d815] libusb: debug [arm_timer_for_next_timeout] no timeouts, disarming timer
[ 0.579102] [0002d815] libusb: debug [usbi_handle_transfer_completion] transfer 0x55bc3fb8d078 has callback 0x7f2d97b3c540
[ 0.579108] [0002d815] libusb: debug [sync_transfer_cb] actual_length=26
[ 0.579113] [0002d815] libusb: debug [libusb_free_transfer] transfer 0x55bc3fb8d078
[ 0.579120] [0002d815] libusb: debug [libusb_close]  
[ 0.579127] [0002d815] libusb: debug [usbi_remove_event_source] remove fd 34
[DEVREG] Detected new device: VID_0x483:PID_0x5740
```

benzea, 2022-03-01:
Rather than doing the change from  https://github.com/libusb/libusb/commit/6929b8270170a76e9af30d0a9e95c2081f4372b0, one could also guard `usbi_hotplug_exit` itself by testing whether `hotplug_ready` is set. Then calling `usbi_hotplug_exit` unconditionally in `libusb_init` is fine.

----

Note, that applications are (currently) required to handle this corner case. libusb tried to suppress the events for the initial enumeration. However, there is *no* guarantee that the application will not receive two notifications anyway. There are race conditions for hotplug events occurring between libusb_init and registering the callback.

Quoting the documentation:

> ... This means that when using [LIBUSB_HOTPLUG_ENUMERATE](https://libusb.sourceforge.io/api-1.0/group__libusb__hotplug.html#gga95ce2ec85dec177854c423cea2853c52a01a1c0c9f4e6852cb6df713f2e8a7f6b), your callback may be called twice for the arrival of the same device, once from [libusb_hotplug_register_callback()](https://libusb.sourceforge.io/api-1.0/group__libusb__hotplug.html#ga5ab3955e2110a3099497a66256fb7fab) and once from [libusb_handle_events()](https://libusb.sourceforge.io/api-1.0/group__libusb__poll.html#ga4989086e3f0327f3886a4c474ec7c327); and/or your callback may be called for the removal of a device for which an arrived call was never made.

To avoid that, libusb would need to track the events on a per callback level.

benzea, 2022-03-01:
OK, pushed the proposed patch to fix this into #1078 and included a regression test for this specific issue. No regression test was added for #989 as it is impossible to inject an error into the right location using umockdev.

commit 0846456f (2022-03-16): core: Suppress hotplug events during initial enumeration

The initial enumeration should not result in hotplug events to be fired.
This is just a convenience though, API users still need to be prepared
to be notified a second time for a device that was plugged in between
libusb_init and libusb_hotplug_register_callback.

This regressed with commit 6929b82 ("Fix segmentation fault in
libusb_init() if usbi_backend.init() fails"). This commit avoids the
mentioned segmentation fault by avoiding to clean up the hotplug code if
it was not yet initialised.

Fixes #1082
Closes #1090
References #989
closed 2022-03-16 by tormodvolden