libusb/libusb #1083: LIBUSB_ERROR_NOT_FOUND on Windows only if USB debugging is disabled

rom1v, 2022-02-23:
An Android device has a different vid:pid depending on whether USB debugging is enabled or not.

For example, on a Nexus 5:
 - without USB debugging: `18d1:4ee1`
 - with USB debugging: `18d1:4ee2`

Here is a minimal sample:

<details><summary>source</summary>

```c
#include <assert.h>
#include <stdio.h>
#include <unistd.h>

#include <libusb-1.0/libusb.h>

#define VID 0x18d1

int main() {
    int r = libusb_init(NULL);
    assert(!r);

    libusb_device **list;
    ssize_t count = libusb_get_device_list(NULL, &list);
    assert(count != -1);

    libusb_device *device = NULL;
    struct libusb_device_descriptor desc;

    for (size_t i = 0; i < (size_t) count; ++i) {
        int r = libusb_get_device_descriptor(list[i], &desc);
        if (r < 0)
            continue;
        if (desc.idVendor == VID) {
            device = libusb_ref_device(list[i]);
            break;
        }
    }

    libusb_free_device_list(list, 1);

    if (!device) {
        fprintf(stderr, "Device %04x:xxxx not found\n", VID);
        goto end;
    }

    printf("Device %04x:%04x ok\n", VID, desc.idProduct);

    libusb_device_handle *handle;
    r = libusb_open(device, &handle);
    if (r) {
        fprintf(stderr, "error: %s\n", libusb_strerror(r));
        goto end;
    }

    char buffer[128];
    r = libusb_get_string_descriptor_ascii(handle, desc.iSerialNumber,
                                           (unsigned char *) buffer,
                                           sizeof(buffer) - 1);
    libusb_close(handle);
    if (r < 0) {
        fprintf(stderr, "serial: libusb error: %s", libusb_strerror(r));
        goto end;
    }

    buffer[r] = '\0';

    printf("serial: [%s]\n", buffer);

end:
    libusb_unref_device(device);
    libusb_exit(NULL);
    return 0;
}

```

```
# native compilation (Linux)
gcc test.c -o test -lusb-1.0

# cross-compilation from Linux for Windows
x86_64-w64-mingw32-gcc test.c -o test.exe -Ilibusb-1.0.25/include -Llibusb-1.0.25/MinGW64/dll -lusb-1.0
```

</details>

On Linux, I can open the device and read the serial in both modes:

```
$ ./test
Device 18d1:4ee2 ok
serial: [05f5e60a0ae518e5]

$ ./test
Device 18d1:4ee1 ok
serial: [05f5e60a0ae518e5]
```

On Windows, I can open the device in both cases. However, I can only read the serial when USB debugging is enabled:

```
> test.exe
Device 18d1:4ee2 ok
serial: [05f5e60a0ae518e5]

> test.exe
Device 18d1:4ee1 ok
serial: libusb error: Entity not found
```

If I enable warnings:

```c
libusb_set_option(NULL, LIBUSB_OPTION_LOG_LEVEL, LIBUSB_LOG_LEVEL_WARNING);
```

Then I get these errors:
```
Device 18d1:4ee1 ok
libusb: error [winusbx_claim_interface] could not access interface 0: [1] Fonction incorrecte.
libusb: warning [winusbx_claim_interface] failed to auto-claim interface 0 (required to claim 1 with WinUSB): [1] Fonction incorrecte.
libusb: warning [winusbx_claim_interface] failed to auto-claim interface 0 (required to claim 2 with WinUSB): [1] Fonction incorrecte.
libusb: warning [winusbx_claim_interface] failed to auto-claim interface 0 (required to claim 3 with WinUSB): [1] Fonction incorrecte.
libusb: warning [winusbx_claim_interface] failed to auto-claim interface 0 (required to claim 4 with WinUSB): [1] Fonction incorrecte.
libusb: warning [winusbx_claim_interface] failed to auto-claim interface 0 (required to claim 5 with WinUSB): [1] Fonction incorrecte.
libusb: warning [winusbx_claim_interface] failed to auto-claim interface 0 (required to claim 6 with WinUSB): [1] Fonction incorrecte.
libusb: warning [winusbx_claim_interface] failed to auto-claim interface 0 (required to claim 7 with WinUSB): [1] Fonction incorrecte.
libusb: warning [winusbx_claim_interface] failed to auto-claim interface 0 (required to claim 8 with WinUSB): [1] Fonction incorrecte.
libusb: warning [winusbx_claim_interface] failed to auto-claim interface 0 (required to claim 9 with WinUSB): [1] Fonction incorrecte.
libusb: warning [winusbx_claim_interface] failed to auto-claim interface 0 (required to claim 10 with WinUSB): [1] Fonction incorrecte.
libusb: warning [winusbx_claim_interface] failed to auto-claim interface 0 (required to claim 11 with WinUSB): [1] Fonction incorrecte.
libusb: warning [winusbx_claim_interface] failed to auto-claim interface 0 (required to claim 12 with WinUSB): [1] Fonction incorrecte.
libusb: warning [winusbx_claim_interface] failed to auto-claim interface 0 (required to claim 13 with WinUSB): [1] Fonction incorrecte.
libusb: warning [winusbx_claim_interface] failed to auto-claim interface 0 (required to claim 14 with WinUSB): [1] Fonction incorrecte.
libusb: warning [winusbx_claim_interface] failed to auto-claim interface 0 (required to claim 15 with WinUSB): [1] Fonction incorrecte.
libusb: warning [winusbx_claim_interface] failed to auto-claim interface 0 (required to claim 16 with WinUSB): [1] Fonction incorrecte.
libusb: warning [winusbx_claim_interface] failed to auto-claim interface 0 (required to claim 17 with WinUSB): [1] Fonction incorrecte.
libusb: warning [winusbx_claim_interface] failed to auto-claim interface 0 (required to claim 18 with WinUSB): [1] Fonction incorrecte.
libusb: warning [winusbx_claim_interface] failed to auto-claim interface 0 (required to claim 19 with WinUSB): [1] Fonction incorrecte.
libusb: warning [winusbx_claim_interface] failed to auto-claim interface 0 (required to claim 20 with WinUSB): [1] Fonction incorrecte.
libusb: warning [winusbx_claim_interface] failed to auto-claim interface 0 (required to claim 21 with WinUSB): [1] Fonction incorrecte.
libusb: warning [winusbx_claim_interface] failed to auto-claim interface 0 (required to claim 22 with WinUSB): [1] Fonction incorrecte.
libusb: warning [winusbx_claim_interface] failed to auto-claim interface 0 (required to claim 23 with WinUSB): [1] Fonction incorrecte.
libusb: warning [winusbx_claim_interface] failed to auto-claim interface 0 (required to claim 24 with WinUSB): [1] Fonction incorrecte.
libusb: warning [winusbx_claim_interface] failed to auto-claim interface 0 (required to claim 25 with WinUSB): [1] Fonction incorrecte.
libusb: warning [winusbx_claim_interface] failed to auto-claim interface 0 (required to claim 26 with WinUSB): [1] Fonction incorrecte.
libusb: warning [winusbx_claim_interface] failed to auto-claim interface 0 (required to claim 27 with WinUSB): [1] Fonction incorrecte.
libusb: warning [winusbx_claim_interface] failed to auto-claim interface 0 (required to claim 28 with WinUSB): [1] Fonction incorrecte.
libusb: warning [winusbx_claim_interface] failed to auto-claim interface 0 (required to claim 29 with WinUSB): [1] Fonction incorrecte.
libusb: warning [winusbx_claim_interface] failed to auto-claim interface 0 (required to claim 30 with WinUSB): [1] Fonction incorrecte.
libusb: warning [winusbx_claim_interface] failed to auto-claim interface 0 (required to claim 31 with WinUSB): [1] Fonction incorrecte.
libusb: error [auto_claim] could not auto-claim any interface
serial: libusb error: Entity not found
```

It looks like #761, but I'm not sure it's the same situation (Windows 10, libusb 1.0.25, WinUSB I guess).

For reference, the real use case: https://github.com/Genymobile/scrcpy/issues/3050

Thank you for your help.

diabolo38, 2022-02-23:
it seam theirs os descriptor and/or IAD t make some interface to be associated to winusb 
and maybe overs one associated to standard class driver (mass storage, MTP .. serial..)  and that acquire interface 0 and cause failure to use any interfaces (what i guess from log)

I remember alike issue and one solution trial could be to replace default driver to force "winusb" for the overall device
but that could cause issue and unability to use the standard  one and to revert back :/

mcuee, 2022-02-24:
#761 is only for device with libusb0.sys driver so this issue is different.
* #761

mcuee, 2022-02-24:
Please post the output of "lsusb -vvv" for your device. If your device is using IAD as mention by michel, then you may want to try the following pull request.
* https://github.com/libusb/libusb/pull/965

mcuee, 2022-02-24:
From https://github.com/Genymobile/scrcpy/issues/3050, this is not a libusb issue. Apparently it is not using WinUSB driver any more without USB debugging.
Ref: https://github.com/Genymobile/scrcpy/issues/3050#issuecomment-1049206532

closed 2022-02-24 by mcuee