libusb/libusb #1086: winusb: cannot use device only handled by GEN pass

shuffle2, 2022-02-27:
I have following test code:
```c++
#include <cstdint>
#include <cstdio>

#include "c:/src/libusb/libusb/libusb.h"
#pragma comment(lib, "c:/src/libusb/build/v143/x64/Debug/lib/libusb-1.0.lib")

int main() {
	libusb_context* ctx{};
	int rv = libusb_init(&ctx);
	if (rv != LIBUSB_SUCCESS) {
		printf("libusb_init:%d\n", rv);
		return 1;
	}

	while (true) {
		libusb_device** devlist{};
		ssize_t num_devs = libusb_get_device_list(ctx, &devlist);
		for (ssize_t i = 0; i < num_devs; i++) {
			auto dev = devlist[i];
			libusb_device_descriptor desc{};
			rv = libusb_get_device_descriptor(dev, &desc);
			if (rv != LIBUSB_SUCCESS) {
				continue;
			}
			if (desc.idVendor == 0x2ca3 && desc.idProduct == 0x40) {
				libusb_device_handle* dev_handle{};
				rv = libusb_open(dev, &dev_handle);
				printf("libusb_open:%d\n", rv);
				if (rv == LIBUSB_SUCCESS) {
					puts("OK");
					return 0;
				}
				libusb_close(dev_handle);
			}
		}
		libusb_free_device_list(devlist, 1);
	}

	libusb_exit(ctx);
	return 0;
}
```

I am trying to interact with this 2ca3/0040 device (which works from linux/libusb). I enabled debug logging and added a debug log:
```diff
diff --git a/libusb/os/windows_winusb.c b/libusb/os/windows_winusb.c
index a03d6a5..d6ccc30 100644
--- a/libusb/os/windows_winusb.c
+++ b/libusb/os/windows_winusb.c
@@ -1771,6 +1771,7 @@ static int winusb_get_device_list(struct libusb_context *ctx, struct discovered_
                                if (priv->path != NULL)
                                        break;
                                // Take care of API initialization
+                               usbi_info(ctx, "%s %s api %d", passname[MIN(pass, EXT_PASS)], dev_interface_path, api);
                                priv->path = dev_interface_path;
                                dev_interface_path = NULL;
                                priv->apib = &usb_api_backend[api];
```

You can see the wanted device never has API assigned for it:
```
[ 2.280780] [000039b0] libusb: debug [libusb_get_device_list]  
[ 2.292117] [000039b0] libusb: debug [winusb_get_device_list] #### PROCESSING HUBs {F18A0E88-C30C-11D0-8815-00A0C906BED8}
[ 2.292189] [000039b0] libusb: debug [winusb_get_device_list] PRO: USB\VID_05AC&PID_1006\MSFT20000000000000
[ 2.292264] [000039b0] libusb: debug [winusb_get_device_list] allocating new device for session [4]
[ 2.292316] [000039b0] libusb: info [winusb_get_device_list] HUB \\?\USB#VID_05AC&PID_1006#MSFT20000000000000#{F18A0E88-C30C-11D0-8815-00A0C906BED8} api 1
[ 2.292367] [000039b0] libusb: debug [winusb_get_device_list] PRO: USB\VID_0451&PID_8142\MSFT201E0F084136CA
[ 2.292432] [000039b0] libusb: debug [winusb_get_device_list] allocating new device for session [5]
[... 196 lines cut ...]
[ 3.617404] [000039b0] libusb: debug [libusb_get_device_descriptor]  
[ 3.617437] [000039b0] libusb: debug [libusb_get_device_descriptor]  
[ 3.617469] [000039b0] libusb: debug [libusb_get_device_descriptor]  
[ 3.617508] [000039b0] libusb: debug [libusb_get_device_descriptor]  
[ 3.617692] [000039b0] libusb: debug [libusb_get_device_descriptor]  
[ 3.617758] [000039b0] libusb: debug [libusb_get_device_descriptor]  
[ 3.617791] [000039b0] libusb: debug [libusb_get_device_descriptor]  
[ 3.617824] [000039b0] libusb: debug [libusb_get_device_descriptor]  
[ 3.617857] [000039b0] libusb: debug [libusb_unref_device] destroy device 5.2
[ 3.617899] [000039b0] libusb: debug [libusb_unref_device] destroy device 1.0
[ 3.617940] [000039b0] libusb: debug [libusb_unref_device] destroy device 2.1
[ 3.617979] [000039b0] libusb: debug [libusb_unref_device] destroy device 3.0
[ 3.618021] [000039b0] libusb: debug [libusb_unref_device] destroy device 2.3
[ 3.618059] [000039b0] libusb: debug [libusb_unref_device] destroy device 5.1
[ 3.618091] [000039b0] libusb: debug [libusb_unref_device] destroy device 5.0
[ 3.618131] [000039b0] libusb: debug [libusb_unref_device] destroy device 2.7
[ 3.618168] [000039b0] libusb: debug [libusb_unref_device] destroy device 4.0
[ 3.618209] [000039b0] libusb: debug [libusb_unref_device] destroy device 2.6
[ 3.618246] [000039b0] libusb: debug [libusb_unref_device] destroy device 2.4
[ 3.618284] [000039b0] libusb: debug [libusb_unref_device] destroy device 2.5
[ 3.618318] [000039b0] libusb: debug [libusb_unref_device] destroy device 2.2
[ 3.618358] [000039b0] libusb: debug [libusb_unref_device] destroy device 2.0
```

Any idea what is the limitation here? is there some possibility for fix?

shuffle2, 2022-02-28:
The problem appears to be that `SetupDiEnumDeviceInfo` returns `ERROR_NO_MORE_ITEMS` for `{BF157A72-62F4-46A5-BE56-ADDFDA92170F}`...

mcuee, 2022-02-28:
I believe the following may be the main reason. What is this device?

```debug [winusb_get_device_list] allocating new device for session [116]
debug [init_device] 0x2ca3:0x40 found 1 configurations (not configured)
```

shuffle2, 2022-02-28:
Sorry, that's a bit confusing because I was reinstalling the driver with zadig a few times, which regenerated the GUID which shows up as the "extra GUID". The GUID listed there matches the value in the `DeviceInterfaceGUIDs` registry key.

This is a device which I've used "Create New Device" option of zadig to install winusb (https://github.com/pbatard/libwdi/wiki/Zadig#Advanced_usage). I wonder if that has something to do with it?

shuffle2, 2022-02-28:
Unfortunately for me it seems this is actually some problem between windows and the device. The PNP state of the device never makes it to the "present" state (even tho it is visible in device manager, the symbolic link is not openable). So I'll just close this.

closed 2022-02-28 by shuffle2

shuffle2, 2022-02-28:
p.s. actually, i've found that when libusb-win32 is installed on the device via zadig, it works... :/

mcuee, 2022-02-28:
Yes that is possible. There are non-compliant device which only works with libusb0.sys (WDM based) and not WinUSB or libusbK (KMDF based driver).

shuffle2, 2022-02-28:
KMDF is a wrapper around WDM, there's no intrinsic reason that a KMDF driver cannot "reach below the abstraction layer" and attain equivalent behavior. But yea, it's not a libusb problem.