libusb/libusb #1087: Windows: multiple processes can claim the same interface

maksimFOD, 2022-03-04:
I have a single device (with selected vendor id) connected to PC. libusbK is configured as the backend. I made an application, which does the following: lists (and later clears) device list, finds the first (which is actually the only) device with predefined VID, claims its only interface, sends a packet, receives some data, and waits for user's input. Upon receiving the input, it sends the packet again, receives some data, closes the interface and device, and exits. 
I launch the first instance of the app, all works well and stalls while waiting for input. Then I launch the second instance. I expected the seconds instance to fail, as the only suitable interface is already claimed. But the second app manages to claim the interface again, send and receive data. I go back to the first instance, allow it to proceed, and it also works normally, the app exits. As expected, the second instance exits normally, too.

My question: how do I know, that the interface is already claimed? And moreover, according to libusbK, my second process should have failed at claiming the interface. What am I missing?
Here's my code I used:
``` cpp
#include <iostream>
#include "libusb.h"
#include "my_constants.h" //defines VID, endpoints and packet

libusb_device_handle* claim_interface_by_VID(uint16_t vendor_id, int configuration, int interface){
    libusb_device_handle *selected_device{NULL};
    libusb_device **devs;
    ssize_t cnt;
    cnt = libusb_get_device_list(NULL, &devs);
    if (cnt < 0){
        std::cout << "error enumerating devices\n";
    }
    int r;
    for (int i=0; devs[i];i++){
        struct libusb_device_descriptor desc;
		r = libusb_get_device_descriptor(devs[i], &desc);
		if (r<0){
            continue;
		}
        if (desc.idVendor == vendor_id){
            std::cout << "Found device with the vendor id\n";
            r = libusb_open(devs[i], &selected_device);
            if (r!=0){
                selected_device = NULL;
                continue;
            }
            r = libusb_set_configuration(selected_device, configuration);
            if(r!=0){
                std::cout<<"Failed to set configuration";
                libusb_close(selected_device);
                selected_device = NULL;
                continue;
            }
            r = libusb_claim_interface(selected_device, interface);
            if (r!=0){
                std::cout<<"Failed to claim interface";
                libusb_close(selected_device);
                selected_device = NULL;
                continue;
            }
            break; //stop at the first successful attempt
        }
    }
    libusb_free_device_list(devs, 1);
    return selected_device;
}
void release_interface_and_close_dev(libusb_device_handle *dev_handle, int interface){
    libusb_release_interface(dev_handle, interface);
    libusb_close(dev_handle);
}

int write(libusb_device_handle *dev_handle, uint8_t endpoint_address, uint8_t *data_buf, int cnt){
    int bytes_written{0};
    int r = libusb_bulk_transfer(dev_handle, endpoint_address, data_buf, cnt, &bytes_written, 1000);
    std::cout<<"sent data to handle: "+std::to_string((uint64_t) dev_handle)+", endpoint: "+std::to_string(endpoint_address)+", response: "+std::to_string(r)+", bytes written:"+std::to_string(bytes_written)+"\n";
    if (r<0){
        return r;
    }else{
        return bytes_written;
    }

}

int read(libusb_device_handle *dev_handle, uint8_t endpoint_address, uint8_t *data_buf, uint16_t cnt){
    int bytes_read{0};
    int r = libusb_bulk_transfer(dev_handle, endpoint_address, data_buf, cnt, &bytes_read, 1000);
    std::cout<<"sent data to handle: "+std::to_string((uint64_t) dev_handle)+", endpoint: "+std::to_string(endpoint_address)+", response: "+std::to_string(r)+", bytes read:"+std::to_string(bytes_read)+"\n";
    if (r<0){
        return r;
    }else{
        return bytes_read;
    }
}

int speak_to_the_device(libusb_device_handle *dev_h){
    //packet_ is defined elsewhere
    int bytesWritten = write(dev_h, EP_wr, packet_, sizeof(packet_));
    if (bytesWritten!=sizeof(packet_)){
        return 2;
    }
    uint8_t receive_buff[1023];
    int bytesRead = read(dev_h, EP_rd, receive_buff, 1023);
    if (bytesRead>16){
        std::cout<<"Description:\n";
        std::string content = std::string((char*)receive_buff, bytesRead);
        std::cout<<content<<"\n";
    }else if (bytesRead>0) {
        for (int i=0; i<bytesRead;i++){
            std::cout<<std::to_string(receive_buff[i])<<",";
        }
    }
    return 0;
}
int main()
{
    libusb_device_handle *dev_handle;
    libusb_init(NULL);
    dev_handle = claim_interface_by_VID(VID, 1, 0);
    std::cout<<"Dev handle:"+std::to_string((uint64_t) dev_handle)+"\n";
    if (dev_handle==NULL){
        std::cout<<"Failed to get the handle\n";
        libusb_exit(NULL);
        return 1;
    }

    if (speak_to_the_device(dev_handle)!=0){
        return 2;
    }
    //stall:
    std::string s;
    std::cin>>s;

    //speak to the device again:
    if (speak_to_the_device(dev_handle)!=0){
        return 4;
    }

    release_interface_and_close_dev(dev_handle, 0);
    libusb_exit(NULL);
    //stall again, so that I can read the output:
    std::cin>>s;
}
```

maksimFOD, 2022-03-04:
I forgot to mention, that I am using the pre-built MingW binary of libusb-1.0.dll from libusb-1.0.25.

mcuee, 2022-03-04:
Which driver are you using?

Ref: https://github.com/libusb/libusb/wiki/Windows#known-restrictions

_WinUSB does not support multiple concurrent applications (as per the Microsoft Windows Hardware Drivers documentation). libusbk driver allows this but it may have the limitation that you can claim the interface multiple times (https://github.com/libusb/libusb/issues/807). libusb-win32 driver will also allow this but it is not recommended to be used due to multiple issues reported._

maksimFOD, 2022-03-04:
Somehow, I managed to miss this limitation while reading that article. It is very much unfortunate, that my knowledge is insufficient to develop such features in libusbK. :(

According to that same restriction list, "WinUSB does not support multiple concurrent applications". I made some experiments with WinUSB, and it works as expected in this experiment. However, if I try to claim an interface and try to claim it again inside the same application with a different handle (which fails as expected), then bulk write with the first handle results in access violation. Is this expected?

I could work around all of these issues, if there was a mechanism to know, if an interface is already claimed by someone. Did I miss it, or it really is not provided by libusb?

diabolo38, 2022-03-07:
If you have an interface opened use it ;) add locking (system wide or app only)  as needed
to me it's the best to do  to avoid troubles  under mac/linux  where you may have different behavior.
Even multiple open/claim  with libusk on window is a nice feature that enable  cooperative (multiple app or kind of user space driver in a share dll or lib)  it's not portable avoid it.

maksimFOD, 2022-03-08:
> If you have an interface opened use it ;) add locking (system wide or app only) as needed to me it's the best to do to avoid troubles under mac/linux where you may have different behavior. Even [...]

I know, that implementing my own locking system could help me, but here's the catch: I am writing an interlayer code for Matlab, and there potentially can be several Matlab sessions running concurrently. Matlab only offers a mechanism to place a lock inside a single session, not between them. Then I see two potential mechanisms for producing system-wide locking: a file in a specified path, or having a server, from which Matlab sessions ask for resources. The first seems easy, but it is difficult to ensure, that unplugged devices are cleared automatically, or there must be some periodic check if a handle is still valid. The server would naturally solve the problem, but I was hoping to make this system as simple as possible, to make it easily portable between computers. 
On the other hand, I already have an interface from Matlab to PyUSB, which works as expected (not-so-portable, though). Furthermore, I have also implemented what I need using libusb0.1 (libusb-win32). I was hoping to upgrade the lib to something more modern, and here we are discussing issues with that :)

diabolo38, 2022-03-08:
i have similar implementation it's a shared dll with libusbk opening same device  in multiple app  instance but takign a lock (window system mutex or shared files up to you) in a controlled/cooperative way.  This is trivial to do,  but it's not gonna work but on window+libusbk or old liusbwin32  backend

If you have os  portability concern then you don't have no choice than some forme of service/Server centralized access. 

device disconnect detection can be done using hotplug (not on window without with extra specific doe as of today hotplug not yet fully supported). If you devices as some bulk/intr  say in end  point  that you can read rom and wan't send/read any data until you ask him. Then you do a request on that end point that with infinite timeout, your either  get the data when you need it or a failure when device will go off.

mcuee, 2022-03-17:
I will close this as this is a known limitation.

sonatique, 2023-10-12:
Please have a look at : https://github.com/libusb/libusb/pull/1323