libusb/libusb #1089: 1.0.25 breaks the cnijfilter2 Canon printer driver

gombasg, 2022-03-11:
After upgrading libusb on my Debian desktop, printing over USB started to crash like:

```
cnijlgmon3[157019]: segfault at 38 ip 00007f198bfabf00 sp 00007ffe592533c8 error 4 in libpthread-2.33.so[7f198bfa8000+f000]
```

The cnijfilter2 driver from Canon has this code:

```
err = libusb_init(&g_context);
[...]
numbdev = libusb_get_device_list(NULL, &g_list)
```

Due to calling `libusb_init()` with a non-NULL argument, the default context is not allocated - this has changed between 1.0.24 and 1.0.25. When `libusb_get_device_list()` is then called with a NULL context, it crashes with SIGSEGV when referencing `&ctx->usb_devs_lock`, due to the default context being NULL.

I don't know if cnijfilter2 should be considered broken or not, but it is unlikely to be fixed (Canon does not update drivers for old printers, new version of the driver is seemingly only offered when a new model requires it), so restoring the behavior of 1.0.24 and making `libusb_init()` to always initialize the default context would help people using Canon printers.

mcuee, 2022-03-13:
cnijfilter2 should be considered broken.
Ref: https://github.com/libusb/libusb/issues/1059

mcuee, 2022-03-17:
I will close this for now.

tormodvolden, 2022-03-17:
gutenberg has the same issue: https://bugzilla.redhat.com/show_bug.cgi?id=2055504 but will likely be fixed promptly there.

We could consider applying @benzea's patch https://src.fedoraproject.org/rpms/libusb1/blob/rawhide/f/0001-core-Install-first-context-as-implicit-default.patch temporarily for a release or two.

benzea, 2022-03-18:
> We could consider applying @benzea's patch https://src.fedoraproject.org/rpms/libusb1/blob/rawhide/f/0001-core-Install-first-context-as-implicit-default.patch temporarily for a release or two.

If you don't want to merge the patch here, then a good alternative may be to mention this bug in the ChangeLog. Distributions can then carry the patch downstream as long as they see fit.

mcuee, 2022-03-19:
I think it is okay to applying the patch as there is a warning message so that the developer got a chance to fix it.

tormodvolden, 2022-03-19:
It can be advantageous that everybody is using the same code. And less work for everybody if we apply it upstream. And certainly less end-user frustration. I just hope the application developers react and get it fixed the right way nonetheless.

tormodvolden, 2022-03-20:
OK, the patch has been tested in Fedora (which seems to get a lot of real-life testing judging from bug reports) for almost a month so I will include it.

commit 6b29aeb9 (2022-03-20): core: Install first context as implicit default (and warn about its use)

There was a behaviour change in libusb 1.0.25 which triggers issues when
the API is misused. This caused for instance gutenprint to crash, see
https://bugzilla.redhat.com/show_bug.cgi?id=2055504

This seems to affect several applications "out in the wild".

For now, work around this by installing an implicit default. But, change
the code to log an error in case this "feature" is being used.

This will allow some grace time for developers to fix their
applications, before we at a later point revert to the stricter
behavio [... cut]
closed 2022-03-20 by tormodvolden