libusb/libusb #1092: Can I assume libusb_bulk_transfer will never modify the data passed when the endpoint is of type 'OUT'?

marcobergamin, 2022-03-17:
Specifically, I would like to know if I can safely const_cast away the "const" from a const buffer when I pass it to the libusb_bulk_transfer function when the endpoint address is of type 'out'.

If yes, should be documented in the public API?

Thanks,
Marco

mcuee, 2022-06-02:
I believe this is the case. 
@tormodvolden @hjelmn Please correct me if I am wrong. Thanks.

mcuee, 2022-10-22:
@marcobergamin 

Maybe you want to state what you want to do and the context of the question. Thanks.

tormodvolden, 2022-12-19:
It depends on what you mean by "safely" :) I think you will see in the code that libusb itself is not writing to the buffer, but is also passing it directly to the underlying OS libraries and drivers (instead of passing a copy). I am pretty sure they will also not write to the buffer, on the other hand they may require that you don't modify the buffer before the transfer is completed. But what I am saying here cannot be 100% safely trusted. Maybe someone can confirm this for at least some of the platforms/drivers.