libusb/libusb #1094: Misc: To release 1.0.27

tormodvolden, 2022-03-20:
Ready to merge:

Pending review:
- #1026 (solve API issue)
 - will need follow-up #1003 (possible API breakage)

Pending review, may be postponed if not enough reviews:
- #1018 (libusb0.sys)

Nice to have:
- #965 (very useful feature enhancement for Windows users, new APIs)

Probably postpone or drop:
- #678 (postpone unless reproduced issue)

mcuee, 2022-04-09:
This is a very popular pull request for better Android support. Not so sure if we can just pull it in as long as it does not affect other platforms (especially Linux).
* https://github.com/libusb/libusb/pull/874

mcuee, 2022-04-09:
I am also not so sure how we deal with new platforms like this.
- https://github.com/libusb/libusb/pull/1057

tormodvolden, 2022-04-09:
I think #1026 must be settled before merging #874. Both #874 and #1057 need some work (and ideally more user feedback). They can be be merged in any release when they are ready (like any PR), but I wouldn't let 1.0.27 drag on waiting for these to finalize. IMO the only important thing (that I know about now) for 1.0.27 is #1026, and all other items are bonus points.

mcuee, 2022-04-09:
Is it possible to create a branch for #874 (once it is rebased to git head)? I think #1057 is still far away but #874 seems to be more popular.

tormodvolden, 2022-04-09:
If the people involved in 874 are maintaining their branches, I don't see enough utility in maintaining an intermediate branch, at least for what my own contribution is concerned. And if they are not... who is gonna maintain and develop the new branch? :) OTOH 1057 is a non-invasive patch so easier to merge, but I'd like to see more testing and reviews from people using it, to make sure it is mature before landing in master. Anyway I see these questions as relevant for the individual PRs and not so much for 1.0.27. If #1026 is done and 874 gets adapted to this and otherwise is finalized in time, I would be happy to see it included in 1.0.27.

mcuee, 2022-04-09:
You know, somehow it is a "feeling good factor" for me if we have major features in a new release, other than bug fixes. Bug fixes are always important, but you know, it is just a feeling thingy.

Good to hear that #1057 is a non-invasive patch and easier to be merged when more people are testing and using it.

#874 is popular but I understand it probably needs more efforts from the contributors and testers.

mcuee, 2022-04-12:
One suggestion for the testing matrix is to get the developers of the main libusb language bindings to test it out. Not so sure how many of them will be on board though.

mcuee, 2022-04-25:
@tormodvolden #1125 has not been merged.

tormodvolden, 2022-05-08:
There were a couple of unfortunate issues in 1.0.26 that are now fixed up up in master (thanks @hjelmn). I don't know to which degree this have been affecting users (on macOS in particular). For instance homebrew hasn't patched their 1.0.26 version. (Fedora is still on their patched 1.0.25.) There haven't been much comments in our bug tracker other than the original reports. Has anyone seen reports elsewhere? Do we agree there is no need for a fix-up release at this point and that we rather see a "normal" 1.0.27 release some further months into the future?

mcuee, 2022-05-08:
Maybe @hjelmn can comment on the severity of the issue under macOS.

1) Nothing much from Homebrew
https://github.com/Homebrew/homebrew-core/issues

2) Nothing much from macports either (not so sure if this is the right place to check).
From macports trac I do not see much reports related to libusb-1.0.26 itself,
https://trac.macports.org/search?changeset=on&milestone=on&portsummarysearch=on&ticket=on&wiki=on&q=libusb&page=1&noquickjump=1

mcuee, 2022-05-20:
@tormodvolden 
Interestingly that there are no much complaints from macOS users. I guess we do not need to rush for 1.0.27 release as of now.

cyanfish, 2022-08-31:
I'll note that #1117 should also fix some macOS SANE scanning issues in #919.

martinling, 2022-10-18:
I've just submitted #1208, following the discussion on #1069, and it would be great if that feature finally made it into a release. [Sigrok](https://sigrok.org) has been patching libusb for RAW_IO for something like ten years now!

mcuee, 2022-10-30:
@tormodvolden and @hjelmn 

I think we can merge #1127, #1208 and create an RC.

Actually I would like to merge #965 and #1181 as well for Windows

tormodvolden, 2022-11-13:
I additionally asked on the [mailing list](https://sourceforge.net/p/libusb/mailman/message/37734720/) for review of the various API changes:

mcuee, 2022-11-14:
@tormodvolden 
I am not so sure if it is feasible to have a Christmas release for 1.0.27. It may be difficult at the current speed. 

But on the other hand, we can probbably have a release at that time for whatever is ready. What is your thought?

tormodvolden, 2022-11-14:
I am not sure I have heard about any Christmas release plan :) At least for me the very end of the year is often busy so not the best time for proper releases. I still think 1026 is the most important criteria for 1.0.27, and I think it can land within a couple weeks (hopefully awaiting feedback from the mailing list), together with other PRs that are ready.  So a December release is possible but I prefer releasing when it is ready and not by calendar.

Do you mean something like a beta release around that time?

mcuee, 2022-11-14:
> Do you mean something like a beta release around that time?

1.0.27 RC1 will be great.

CraigHutchinson, 2022-11-14:
> Is it possible to create a branch for #874 (once it is rebased to git head)? I think #1057 is still far away but #874 seems to be more popular.

PR https://github.com/libusb/libusb/pull/1164 is a rebase of #874

mcuee, 2022-11-14:
> PR #1164 is a rebase of #874

Yes, that is a good effort. Hopefully more Android people will go to #1164 and post their review or successful story.
* #1164

mcuee, 2022-12-25:
> Christmas release is just a wish from my side.
> 
> I have asked a few project like HIDAPI and avrdude to see if a Christmas release is feasible or not and the answer is positive from the lead developer.

So looks like we have to wait a few month before the 1.0.27 release due to lack of developers.

mcuee, 2023-01-22:
@tormodvolden 

Wonderful job at integrating the PRs. Thanks a lot for the efforts!

mcuee, 2023-09-17:
Contributors are more than welcome for libusb projects now that we are short of developers and reviewers which makes 1.0.27 release not happening yet.

* https://github.com/libusb/libusb/issues/947

sonatique, 2023-09-19:
What about https://github.com/libusb/libusb/pull/1270 ?

mcuee, 2023-10-19:
I am also interested in #1270 and a few others.

* #1258 
* #1270 

Other PRs -- no real libusb code changes, just github action scripts.
* #1234
* #1294

mcuee, 2023-11-05:
Thanks to @tormodvolden, #1270 has been merged.

tormodvolden, 2023-11-05:
I would be willing to target a 1.0.27 release soon. There has been a long time, and many small useful things are in the tree and should be released to the users. That would mean just including what is ready, reverting the unresolved WinUSB raw_io stuff, and otherwise leave other things to another release, that doesn't have to be so far away either. Almost all outstanding things are Windows-only, and we cannot allow those to hold back things for other platforms.

mcuee, 2023-11-05:
> I would be willing to target a 1.0.27 release soon. There has been a long time, and many small useful things are in the tree and should be released to the users. That would mean just including what [...]

Totally agree with you. Thanks a lot for the great work!

martinling, 2023-11-05:
I am frustrated to see the proposal to drop the RAW_IO support.

I wrote the first version of this over a year ago, in #1208. Due to previous discussions, this was engineered to fit within specific API requirements desired at the time: that it should involve no new functions, and all be done through the `libusb_set_option` interface.

Some months after that version was already reviewed and merged, I was then told that new functions were now preferred instead, so I proposed a new API in #1297 and asked for feedback on it.

Over three months later, I have yet to get any response as to whether this new API is acceptable to proceed with. There was one response, after two months asking me to restructure the branch into two commits, which I did the the very next day - but then nothing.

And now the proposal is to revert the whole effort because it is "not ready".

I have my employer's support to do whatever work is needed on this, and remain ready to do so, but it seems to be impossible to make any progress, because of the lack of response as to what work is required, or what solution would be acceptable.

Is there anything I can do to move this forwards?

tormodvolden, 2023-11-05:
We are only "dropping" the RAW_IO libusb_set_option implementation that I suppose we all agree is not the way to go. We are not "reverting any effort". We cannot make a release with this and then change the API again shortly after with the new implementation. So we'll take the revert part that you prepared, make a release, and then we can continue working on RAW_IO again following your new proposal. Last time I looked at it you sounded unsure about some details, and we are not rushing this in just before a release. It is an important API change so it is important to get it right, and that the core developers agree on this. I would love to hear from hjelmn (or even Chris) about it.

martinling, 2023-11-06:
The only detail I am unsure about is what API the libusb project will accept for this feature.

I've made multiple proposals, including the latest one in #1297 for which I've added a couple of possible variations, and I'm still awaiting an answer or even any feedback.

Either "yes, this version is good, please move forward with it" or "no, please change X or satisfy condition Y" would allow me to make progress. With neither I cannot do anything further.

sonatique, 2023-11-06:
May I take this opportunity to ask for clarification regarding who, in the end, is in charge of taking decisions related to API and mid-term / long-term  general "shape" of libusb-1.0?

Personally I feel that it would be better if 1-2 persons would carry the vision and take high-level decisions of this kind (based on community input when relevant). 

As opposed to implementation details, API definition is something that seems to be difficult to define "bottom-up" given that it is more a matter of taste and global vision rather than pure technical pros / cons that could be rationally debated and leading to an objective best choice (in the ideal case).

I feel that this difficulty to converge to decisions regarding API is also what put a brake on other efforts, for instance for https://github.com/libusb/libusb/pull/1258. 

Maybe having a stronger voice for API decision would sometimes help to keep a things on track more efficiently and have ready-to-merge PR more rapidly.

mcuee, 2023-11-08:
> May I take this opportunity to ask for clarification regarding who, in the end, is in charge of taking decisions related to API and mid-term / long-term general "shape" of libusb-1.0?

@sonatique 
In the end, the active main developers have the say. As of now, it will be @hjelmn and @tormodvolden. Unfortunately Nathan seems to be busy now.

mcuee, 2023-11-09:
@sonatique and @martinling 

BTW, libusb team will welcome new contributors who are willing to step up as maintainers. Just wondering if you are interested. Thanks.

sonatique, 2023-11-16:
@mcuee , thanks for asking, I am flattered by your proposal. I find libusb to be a very interesting and stimulating project to work on, but I don't know what is implied by being a maintainer, could you tell me in 2 words?. 
The amount of time I am able dedicate to open source projet is very variable, as you guess already I think therefore that could be an issue, I don't known.

mcuee, 2023-11-16:
> @mcuee , thanks for asking, I am flattered by your proposal. I find libusb to be a very interesting and stimulating project to work on, but I don't know what is implied by being a maintainer, could [...]

Basically do what @tormodvolden does now -- **review the PRs and issues, create a PR **, when you got the time.

I think it should not take a lot of time unless you want to take on the big task like libusb Windows hotplug support. :-)

mcuee, 2023-12-05:
@tormodvolden 

> The status now is that #1266 is pretty much ready, @hjelmn hasn't commented on #1297 so it looks like that one must wait till after 1.0.27 like suggested earlier. The CI stress_mt failure on macOS #1358 seems only for "legacy" macOS 12.

I agree.

> This could mean a real RC is possible later this week, and a release before the holidays if nothing else shows up.

That will be great.

> Putting together Windows binaries is some work. Do we have volunteers or have someone found build scripts that I suppose have been used for 1.0.24 and earlier?

For MinGW the following script is used. I think you need a multilib enabled toolchain (not using MSYS2 mingw).
https://github.com/libusb/libusb/blob/master/.private/bm.sh

tormodvolden, 2023-12-09:
We are now in freeze for 1.0.27. Of course, any regression fixes will still be considered, as well as e.g. trivial build fixes. I will do the formal [RC work](https://github.com/libusb/libusb/wiki/Maintainers'-corner#user-content-RCRelease_work) tomorrow.

tormodvolden, 2023-12-10:
I have the uploaded 1.0.27-rc1 tarball to SourceForge and GitHub Releases. Let the test party begin! There is a test matrix in  #1372.

kloczek, 2023-12-10:
Is it any sense to use still sf.net? 🤔

tormodvolden, 2023-12-10:
What is your reason for questioning our use of SourceForge? Maybe you want to open a separate ticket? We have always used SourceForge for our release files. BTW, contrary to Github you have full control and more flexibility there. We also have a git mirror at GitLab, and although we have our issue and merge request tracker on GitHub we are in principle not stuck to one provider.

mcuee, 2023-12-20:
> What is your reason for questioning our use of SourceForge? Maybe you want to open a separate ticket? We have always used SourceForge for our release files. BTW, contrary to Github you have full [...]

I think we should stick to using github (main development location) and Sourceforge (mailing list and release) at the same time.

kloczek, 2023-12-20:
> What is your reason for questioning our use of SourceForge?

SF interface dos not allow many operations which are possible to do using github/gitlab.
For example it is not possible to download exact commit patch over SF interface which is crucial on integrate exact commits changes with packaging build procedures.
SF interface in issues tracker do not track of referencie between tickets/repos/commits.

seanm, 2024-01-02:
@tormodvolden I think these should be considered for 1.0.27 too:

- https://github.com/libusb/libusb/pull/1418
- https://github.com/libusb/libusb/pull/1413
- https://github.com/libusb/libusb/pull/1412
- https://github.com/libusb/libusb/pull/1409
- https://github.com/libusb/libusb/pull/1408

mcuee, 2024-01-03:
Now the idea is to fix Regression Issue #1386 and then release 1.0.27. But along the way, minor fixes can be included if @tormodvolden or @hjelmn agrees.
https://github.com/libusb/libusb/pull/1399#issuecomment-1873828540

seanm, 2024-01-04:
>But along the way, minor fixes can be included if @tormodvolden or @hjelmn agrees.

What do you mean by "minor" here?  Do you mean that we should be conservative in the scope of changes, since we are close to release?  If so, I generally agree.  But at the same time, some of the things I have identified are significant bugs. Finding such things is the point of testing this rc, is it not? :)

mcuee, 2024-01-04:
> > But along the way, minor fixes can be included if @tormodvolden or @hjelmn agrees.
> 
> What do you mean by "minor" here? Do you mean that we should be conservative in the scope of changes, since we are close to release? If so, I generally agree. 

Yes, that is what I mean. 

Regression from 1.0.26 release -- better to be fixed.
Minor fixes which should not cause issues -- can be merged
Feature enhancement -- to be delayed.

> But at the same time, some of the things I have identified are significant bugs. Finding such things is the point of testing this rc, is it not? :)

Yes, if the fix is deemed to be good and not causing potential problems, they should be incorporated.

If the fix is deemed to be too complex and the bug is a known issue (not a regression), then the fix may be delayed.

If the fix is just nice to have, then they may not be incorporated.

mcuee, 2024-01-06:
Good news:
PR #1427 seems to fix the regression issue #1386.

* #1386 
* #1427

seanm, 2024-01-08:
@tormodvolden think you could start merging some of my PRs?  It's hard to do any integration testing until then...

tormodvolden, 2024-01-08:
I hope to get to them tomorrow or the next day. See the updated list above for what I will target. In the meantime you can always merge them in a branch of your own and make a "fake" PR to check the CI.

tormodvolden, 2024-01-08:
I just now pushed my own "staging" branch where I had dumbly merged all those PRs. But I will do some cherry-picking and fix-ups before I rebase it onto master.

mcuee, 2024-01-20:
It seems to me that we are more or less fine now with 1.0.27 release, other than a few minor fixes to be considered.

tormodvolden, 2024-01-28:
I am building an RC2, soon to hit the repos. This will (hopefully) be a short release candidate, with the final release in a few days if nothing pops up.

tormodvolden, 2024-01-28:
The only remaining change for the release is bumping the libtool library version to (4, 0, 4). Our wiki (and libtool documentation) says to do that only for the final release but not in the RCs.

mcuee, 2024-01-29:
> The only remaining change for the release is bumping the libtool library version to (4, 0, 4). Our wiki (and libtool documentation) says to do that only for the final release but not in the RCs.

Reference:
https://github.com/libusb/libusb/wiki/Maintainers'-corner

tormodvolden, 2024-01-31:
Nothing has been reported, so I might get to release it later today.

tormodvolden, 2024-01-31:
Note that for a week or two after the release I won't commit anything in master unless there are regression fixes. This is so that we can quickly release a fix-up release without getting entangled in other changes (innocent as they may seem!).

Neustradamus, 2024-01-31:
Good job @tormodvolden!

Links:
- https://github.com/libusb/libusb/releases
- https://github.com/libusb/libusb/releases/tag/v1.0.27

seanm, 2024-02-01:
> Note that for a week or two after the release I won't commit anything in master unless there are regression fixes. This is so that we can quickly release a fix-up release without getting entangled in other changes (innocent as they may seem!).

But this is why git has branches... why hold up master?  If a 1.0.27.1 is needed, you can always branch off 1.0.27's tag...

tormodvolden, 2024-02-01:
That git allows branches, doesn't mean we use them for releases. We have a clean, linear git history, and this works fine for us. At the rate we are developing, it is not a problem to freeze master a bit.

[7 more replies]

closed 2024-05-26 by tormodvolden