libusb/libusb #1097: TARGET_OS_OSX undefined on 10.11.5

agoodney, 2022-03-22:
Attempting to build using the "native" compiler for 10.11.5 (i.e clang 8 provided by Xcode 8.2.1) gives:
```
os/darwin_usb.c:2449:5: error: 'TARGET_OS_OSX' is not defined, evaluates to 0 [-Werror,-Wundef]
#if TARGET_OS_OSX
```
This is supposed to be defined in TargetConditionals.h, but if you look at the TargetConditionals.h that is in /usr/include you'll see TARGET_OS_OSX is not defined. However, changing to TARGET_OS_MAC does fix the problem.

Interestingly, if you look in the TargetCondionals.h inside the Xcode SDK (/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.12.sdk/usr/include/TargetConditionals.h)

So perhaps it is because Xcode only has the 10.12 SDK and I'm building on 10.11?

FWIW, TARGET_OS_MAC is present going back to at least 10.6. Perhaps this is another thing to catch/set if the MacOS version is <10.12?

Steps to reproduce:

OS: 10.11.5
Xcode: 8.2.1
manually download and build autoconf, automake, libtool and pkg-config
Start with clean libusb-master:

./autogen.sh
make

seanm, 2022-03-22:
On my 10.11 box I see the same difference you do between the two TargetConditionals.h files.  Weird.  I guess one is older.  From the comments in the file, it seems `TARGET_OS_MAC` is a superset of `TARGET_OS_OSX`.

What's the point of that `#if TARGET_OS_OSX` anyway?  It was added in 6a902c3b.  Perhaps @osy can elaborate...

I suspect it's probably safe/correct to change `TARGET_OS_OSX` -> `TARGET_OS_MAC`

tormodvolden, 2022-03-22:
Maybe it was meant to be `#ifdef TARGET_OS_OSX` and the presence of TARGET_OS_OSX coincides with the presence of Security/Security.h ?

agoodney, 2022-03-22:
It looks like TARGET_OS_OSX works from inside Xcode. I am able to build libusb.1.0.0.dylib using the Xcode project.

seanm, 2022-03-22:
> It looks like TARGET_OS_OSX works from inside Xcode. I am able to build libusb.1.0.0.dylib using the Xcode project.

Yeah, because like you said, there are two different TargetConditionals.h.

osy, 2022-03-23:
> On my 10.11 box I see the same difference you do between the two TargetConditionals.h files. Weird. I guess one is older. From the comments in the file, it seems `TARGET_OS_MAC` is a superset of `TARGET_OS_OSX`.
> 
> What's the point of that `#if TARGET_OS_OSX` anyway? It was added in [6a902c3](https://github.com/libusb/libusb/commit/6a902c3b207a017cd699ac8bb6659abf32eb0c99). Perhaps @osy can elaborate...
> 
> I suspect it's probably safe/correct to change `TARGET_OS_OSX` -> `TARGET_OS_MAC`

Hi, TARGET_OS_OSX and TARGET_OS_MAC are quite different. Please see https://stackoverflow.com/a/49560690

By using TARGET_OS_MAC it is enabled on all *os while TARGET_OS_OSX will exclude iOS.

osy, 2022-03-23:
I guess on older OSX sdk they don’t have TARGET_OS_OSX so you may have to guard it all in another ifdef and define a new conditional…

tormodvolden, 2022-03-23:
Would it be fine if we do like this for now?
`#if defined(TARGET_OS_OSX) && TARGET_OS_OSX == 1`
From the original posting I understand that TARGET_OS_OSX is not defined on 10.11.5 but it has Security/Security.h .
So my suggested change will take it to the `else` clause there instead, but that may also work fine?

On the other hand, if the intention is to use the supplied header if it is present, and otherwise just do best effort and hope the external symbols will be resolved when linking, it would be better to probe for the header file by adding a AC_CHECK_HEADERS in configure.ac, instead of relying on that TARGET_OS_OSX being 1 means that the header is present.

tormodvolden, 2022-03-23:
Actually, this is better if we can assume TARGET_OS_OSX is always defined (but 0) when building for iOS etc:
`#if !defined(TARGET_OS_OSX) || TARGET_OS_OSX == 1`

tormodvolden, 2022-03-23:
It would be good if someone can test my suggestion above. It could be the smallest change to improve things. If we can clarify this quickly, I can include it in 1.0.26 but we cannot be stuck on this for long. I think AC_CHECK_HEADERS might be a better solution but I don't know exactly why the header include vs extern declaration was done with TARGET_OS_OSX. Is it copy-paste from another project or found by trial and error?

agoodney, 2022-03-23:
```
#if !defined(TARGET_OS_OSX) || TARGET_OS_OSX == 1
```
Tested on clean libusb-master on 10.11.5 with "native" Xcode compiler, confirmed to work.

tormodvolden, 2022-03-23:
> Tested on clean libusb-master on 10.11.5 with "native" Xcode compiler, confirmed to work.

Thanks! And our appveyor CI confirms that it builds on 11.6.4 both with plain Makefile build and with Xcode.

If somebody can elaborate on the header file include at a later point, it would be very welcome, but any code improvement may wait till next release.

commit 780044d4 (2022-03-23): darwin: Fix Makefile build on macOS 10.11 and older

TARGET_OS_OSX is not defined when building with the clang 8 provided by
Xcode 8.2.1 on 10.11.5 (unless the Xcode project is used).

If it is not defined we'll just assume we have a "OSX" target.

Closes #1097
closed 2022-03-23 by tormodvolden