libusb/libusb #1099: 1.0.26-rc1 test matrix

tormodvolden, 2022-03-27:
For as many platforms as possible we want to test building and a run of the RC. At least xusb or lsusb -v should be tested. Please use the libusb-1.0.26-rc1.tar.bz2 tarball, and not a git checkout. Use the included configure script, no autogen or bootstrap needed.

Also more application testing is welcome, in particular those who had issues in 1.0.25.

## Major platforms

### Linux x86_64
- [ ] build: gcc 9.3 (Ubuntu 20.04), gcc 11.1 (Arch Linux), gcc 11.2 (Debian unstable)
- [ ] run: xusb, lsusb, libftdi
### Linux arm64 (Raspbian buster)
- [ ] build: gcc 8.3 OK
- [ ] run: xusb, lsusb, dfu-util
### Windows 64-bit
- [ ] build:
- [ ] run: dfu-util, lsusb
### Windows 32-bit
- [ ] build:
- [ ] run: dfu-util, lsusb - test on 64-bit if no 32-bit available
### macOS 12 Intel
- [ ] build: Apple clang
- [ ] run: xusb, stress, listdevs
### macOS 12 ARM64
- [ ] build: Apple clang 13.0.0 OK
- [ ] run: xusb, lsusb, libftdi
### macOS 11.6.1 Intel
- [ ] build: Apple clang LLVM version 10.0.0
- [ ] run: lsusb, dfu-util, pyusb

## Other platforms

### NetBSD
### OpenBSD
### Haiku

## Applications
- [ ] isochronous streaming with libfreenect on XBOX

mcuee, 2022-03-28:
Let me try the following. 

### macOS 12.1 ARM64 (mac Mini M1)
  - [ ] build: Apple clang
  - [ ] run: xusb, listdev, testlibusb, stress (from libusb project), lsusb, libftdi 1.5 release (find_all, eeprom) 

### macOS 12.1 x86_64 using Rosetta Translation, not native (mac Mini M1)
  - [ ] build:Apple clang
  - [ ] run: xusb, listdev, testlibusb, stress (from libusb project), lsusb, libftdi 1.5 release (find_all, eeprom) 
  
### Linux x86_64 (Ubuntu 20.04)
  - [ ] build: gcc 9.3
  - [ ] run: xusb, listdev, testlibusb, stress (from libusb project); lsusb, libftdi 1.5 release (find_all, eeprom)

### Linux ARM32 (Raspberry Pi 400 running Raspberry Pi OS Bullseye 32 bit)
  - [ ] build: gcc 10.2.1
  - [ ] run: xusb, listdev, testlibusb, stress (from libusb project), lsusb, libftdi 1.5 release (find_all, eeprom)

### Linux ARM64 (Raspberry Pi 400 running Raspberry Pi OS Bullseye 64 bit)
  - [ ] build: gcc 10.2.1
  - [ ] run: xusb, listdev, testlibusb, stress (from libusb project), lsusb,  libftdi 1.5 release (find_all, eeprom)
 
### Windows VS2019 64 bit (Windows 11 x64 laptop)
  - [ ] build: 
  - [ ] run: xusb, listdev, testlibusb, stress (from libusb project); libftdi 1.5 release (find_all, eeprom)
  
### Windows VS2019 32 bit (Windows 11 x64 laptop)
  - [ ] build: 
  - [ ] run: xusb, listdev, testlibusb, stress (from libusb project); libftdi 1.5 release (find_all, eeprom)
 
### Windows MinGW 64 bit (Windows 11 x64 laptop)
  - [ ] build: MSYS2 MinGW-w64 64bit GCC 11.2.0
  - [ ] run: xusb, listdev, testlibusb, stress (from libusb project)
  
### Windows MinGW 32 bit (Windows 11 x64 laptop)
  - [ ] build: MinGW-w64 32 bit GCC 11.2.0
  - [ ] run: xusb, listdev, testlibusb, stress (from libusb project)

I will carry out the tests on the following platforms as well using Virtualbox VMs.

### OpenBSD 64 bit (x86_64) -- OpenBSD 7.0 release
  - [ ] build: OpenBSD clang 11.1.0 OK
  - [ ] run: xusb, listdev, testlibusb, stress (from libusb project)
### NetBSD 64 bit (x86_64) -- NetBSD 9.2 release
  - [ ] build: NetNSD GCC 7.5.0 
  - [ ] run: xusb, listdev, testlibusb, stress (from libusb project) 
### Haiku 64 bit (x86_64) -- r1beta3 release
  - [ ] build: gcc 8.3.0
  - [ ] run:

mcuee, 2022-03-28:
Not so sure who will be able to test the following. But we should just proceed with the release if nobody chimed in to test the following platforms.

### Windows on ARM 32 bit
  - [ ] build: VS2019
  - [ ] run: 
### Windows on ARM 64 bit
  - [x] build: VS2019
  - [ ] run: 
### Solaris 64 bit (x86_64)
  - [ ] build: 
  - [ ] run:

mcuee, 2022-03-28:
This may be specific to usbutils under macOS (Homebrew formular usbutils).

```
mcuee@mcuees-Mac-mini libusb-1.0.26-rc1 % brew reinstall libusb
==> Cloning https://github.com/libusb/libusb.git
Updating /Users/mcuee/Library/Caches/Homebrew/libusb--git
From https://github.com/libusb/libusb
   4622bfc..78de7dc  master     -> origin/master
==> Checking out branch master
Already on 'master'
Your branch is behind 'origin/master' by 25 commits, and can be fast-forwarded.
  (use "git pull" to update your local branch)
HEAD is now at 78de7dc libusb 1.0.26-rc1
==> Reinstalling libusb 
==> ./autogen.sh
==> ./configure --prefix=/opt/homebrew/Cellar/libusb/HEAD-78de7dc
==> make install
🍺  /opt/homebrew/Cellar/libusb/HEAD-78de7dc: 22 files, 595KB, built in 17 seconds
==> Running `brew cleanup libusb`...
Disable this behaviour by setting HOMEBREW_NO_INSTALL_CLEANUP.
Hide these hints with HOMEBREW_NO_ENV_HINTS (see `man brew`).

mcuee@mcuees-Mac-mini libusb-1.0.26-rc1 % brew reinstall usbutils
==> Downloading https://mirrors.edge.kernel.org/pub/linux/utils/usb/usbutils/usb
Already downloaded: /Users/mcuee/Library/Caches/Homebrew/downloads/efeb221070cbeb34bf109ccc7027b9cf51afc64ddb820b93ed88a018fc0e92b1--usbutils-007.tar.xz
==> Reinstalling mikhailai/misc/usbutils 
==> ./configure --prefix=/opt/homebrew/Cellar/usbutils/007
==> make install AM_LDFLAGS= SUBDIRS= bin_SCRIPTS= man_MANS=lsusb.8
🍺  /opt/homebrew/Cellar/usbutils/007: 13 files, 821KB, built in 8 seconds
==> Running `brew cleanup usbutils`...
Disable this behaviour by setting HOMEBREW_NO_INSTALL_CLEANUP.
Hide these hints with HOMEBREW_NO_ENV_HINTS (see `man brew`).

mcuee@mcuees-Mac-mini libusb-1.0.26-rc1 % lsusb
libusb: warning [libusb_init] installing new context as implicit default
Bus 002 Device 006: ID 0bda:5411 Realtek Semiconductor Corp. 
Bus 002 Device 005: ID 05e3:0749 Genesys Logic, Inc. 
Bus 002 Device 004: ID 0bda:0411 Realtek Semiconductor Corp. 
Bus 002 Device 003: ID 0b95:1790 ASIX Electronics Corp. 
Bus 002 Device 002: ID 0bda:0411 Realtek Semiconductor Corp. 
Bus 002 Device 001: ID 0bda:5411 Realtek Semiconductor Corp. 
libusb: warning [libusb_exit] device 2.6 still referenced
libusb: warning [libusb_exit] device 2.5 still referenced
libusb: warning [libusb_exit] device 2.4 still referenced
libusb: warning [libusb_exit] device 2.3 still referenced
libusb: warning [libusb_exit] device 2.2 still referenced
libusb: warning [libusb_exit] device 2.1 still referenced```

tormodvolden, 2022-03-28:
usbutils 007 is from 2013... Didn't we get latest usbutils properly into homebrew recently? Or are you somehow stuck with mikhailai's version?

mcuee, 2022-03-28:
MSYS2 MinGW gcc 11.2 still have the same warning as in 1.0.25 release. But I think the understanding is that it seems to be a gcc issue. So we do not need to fix this.

Ref: 
* https://github.com/libusb/libusb/issues/1034

```
$ make
make  all-recursive
make[1]: Entering directory '/c/work/libusb/libusb-1.0.26-rc1/build64'
Making all in libusb
make[2]: Entering directory '/c/work/libusb/libusb-1.0.26-rc1/build64/libusb'
  CC       core.lo
  CC       descriptor.lo
  CC       hotplug.lo
  CC       io.lo
  CC       strerror.lo
  CC       sync.lo
  CC       os/events_windows.lo
  CC       os/threads_windows.lo
  RC       libusb-1.0.lo
  CC       os/windows_common.lo
  CC       os/windows_usbdk.lo
  CC       os/windows_winusb.lo
../../libusb/os/windows_winusb.c: In function 'winusbx_copy_transfer_data':
../../libusb/os/windows_winusb.c:97:32: warning: implicit conversion from 'enum libusb_error' to 'enum libusb_transfer_status' [-Wenum-conversion]
   97 |                         return LIBUSB_ERROR_ACCESS;     \
      |                                ^~~~~~~~~~~~~~~~~~~
../../libusb/os/windows_winusb.c:97:32: note: in definition of macro 'CHECK_WINUSBX_AVAILABLE'
   97 |                         return LIBUSB_ERROR_ACCESS;     \
      |                                ^~~~~~~~~~~~~~~~~~~
../../libusb/os/windows_winusb.c: In function '_hid_get_descriptor.constprop':
../../libusb/os/windows_winusb.c:3358:21: warning: array subscript 'struct libusb_interface_descriptor[0]' is partly outside array bounds of 'char[41]' [-Warray-bounds]
 3358 |         id->bLength = LIBUSB_DT_INTERFACE_SIZE;
../../libusb/os/windows_winusb.c:3326:14: note: while referencing 'tmp'
 3326 |         char tmp[HID_MAX_CONFIG_DESC_SIZE];
      |              ^~~
../../libusb/os/windows_winusb.c:3359:29: warning: array subscript 'struct libusb_interface_descriptor[0]' is partly outside array bounds of 'char[41]' [-Warray-bounds]
 3359 |         id->bDescriptorType = LIBUSB_DT_INTERFACE;
      |         ~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~
../../libusb/os/windows_winusb.c:3326:14: note: while referencing 'tmp'
 3326 |         char tmp[HID_MAX_CONFIG_DESC_SIZE];
      |              ^~~
../../libusb/os/windows_winusb.c:3360:30: warning: array subscript 'struct libusb_interface_descriptor[0]' is partly outside array bounds of 'char[41]' [-Warray-bounds]
 3360 |         id->bInterfaceNumber = 0;
      |         ~~~~~~~~~~~~~~~~~~~~~^~~
../../libusb/os/windows_winusb.c:3326:14: note: while referencing 'tmp'
 3326 |         char tmp[HID_MAX_CONFIG_DESC_SIZE];
      |              ^~~
../../libusb/os/windows_winusb.c:3361:31: warning: array subscript 'struct libusb_interface_descriptor[0]' is partly outside array bounds of 'char[41]' [-Warray-bounds]
 3361 |         id->bAlternateSetting = 0;
      |         ~~~~~~~~~~~~~~~~~~~~~~^~~
../../libusb/os/windows_winusb.c:3326:14: note: while referencing 'tmp'
 3326 |         char tmp[HID_MAX_CONFIG_DESC_SIZE];
      |              ^~~
../../libusb/os/windows_winusb.c:3362:27: warning: array subscript 'struct libusb_interface_descriptor[0]' is partly outside array bounds of 'char[41]' [-Warray-bounds]
 3362 |         id->bNumEndpoints = num_endpoints;
      |         ~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~
../../libusb/os/windows_winusb.c:3326:14: note: while referencing 'tmp'
 3326 |         char tmp[HID_MAX_CONFIG_DESC_SIZE];
      |              ^~~
../../libusb/os/windows_winusb.c:3363:29: warning: array subscript 'struct libusb_interface_descriptor[0]' is partly outside array bounds of 'char[41]' [-Warray-bounds]
 3363 |         id->bInterfaceClass = 3;
      |         ~~~~~~~~~~~~~~~~~~~~^~~
../../libusb/os/windows_winusb.c:3326:14: note: while referencing 'tmp'
 3326 |         char tmp[HID_MAX_CONFIG_DESC_SIZE];
      |              ^~~
../../libusb/os/windows_winusb.c:3364:32: warning: array subscript 'struct libusb_interface_descriptor[0]' is partly outside array bounds of 'char[41]' [-Warray-bounds]
 3364 |         id->bInterfaceSubClass = 0;
      |         ~~~~~~~~~~~~~~~~~~~~~~~^~~
../../libusb/os/windows_winusb.c:3326:14: note: while referencing 'tmp'
[... 56 lines cut ...]
  CCLD     dpfp_threaded.exe
  CC       ezusb.o
  CC       fxload.o
  CCLD     fxload.exe
  CC       hotplugtest.o
  CCLD     hotplugtest.exe
  CC       listdevs.o
  CCLD     listdevs.exe
  CC       sam3u_benchmark.o
  CCLD     sam3u_benchmark.exe
  CC       testlibusb.o
  CCLD     testlibusb.exe
  CC       xusb.o
  CCLD     xusb.exe
make[2]: Leaving directory '/c/work/libusb/libusb-1.0.26-rc1/build64/examples'
Making all in tests
make[2]: Entering directory '/c/work/libusb/libusb-1.0.26-rc1/build64/tests'
  CC       stress.o
  CC       testlib.o
  CCLD     stress.exe
make[2]: Leaving directory '/c/work/libusb/libusb-1.0.26-rc1/build64/tests'
make[2]: Entering directory '/c/work/libusb/libusb-1.0.26-rc1/build64'
make[2]: Leaving directory '/c/work/libusb/libusb-1.0.26-rc1/build64'
make[1]: Leaving directory '/c/work/libusb/libusb-1.0.26-rc1/build64'
```

tormodvolden, 2022-03-28:
Yes, I think the "array subscript" warnings doesn't point to a real problem, but we should ideally fix the code so that the compiler also is convinced it is fine. However the "implicit cast" is new and a real issue that can result in wrong error messages at least.

mcuee, 2022-03-28:
The warnings for umockdev is known under Ubuntu 20.04.

```
umockdev.c:37:32: warning: unknown option after ‘#pragma GCC diagnostic’ kind [-Wpragmas]
   37 | #pragma GCC diagnostic ignored "-Wanalyzer-malloc-leak"
      |                                ^~~~~~~~~~~~~~~~~~~~~~~~
umockdev.c:38:32: warning: unknown option after ‘#pragma GCC diagnostic’ kind [-Wpragmas]
   38 | #pragma GCC diagnostic ignored "-Wanalyzer-file-leak"
      |                                ^~~~~~~~~~~~~~~~~~~~~~

```

hjelmn, 2022-03-28:
We broke fuzzing somehow. Plan to take a look today to see how we can best fix it.

tormodvolden, 2022-03-28:
> We broke fuzzing somehow. Plan to take a look today to see how we can best fix it.

Do you refer to CIFuzz? The issue in [this log](https://oss-fuzz-build-logs.storage.googleapis.com/log-4973d6c1-1eec-42fc-b21c-723e1ba9f09f.txt) is missing pkg-config (pkg.m4 macro) when running bootstrap. I found the log at https://bugs.chromium.org/p/oss-fuzz/issues/list searching for `proj:libusb` but is there a "button" or dashboard for this?

agoodney, 2022-03-28:
### MacOS 10.11.6

- [x] build (Xcode project)
- [x] build (command line, Xcode provided compiler clang-800.0.42.1)
- [ ] run

hjelmn, 2022-03-28:
Yes. CIFuzz.

mcuee, 2022-03-29:
Using pyusb to test cystream FW bulk_loop.
```
#!/usr/bin/python

### quick mod from https://github.com/danielkucera/cyusb3014-breakout/blob/master/software/loopback-test.py

import usb.core
import usb.util
import time
import _thread

# find our device
dev = usb.core.find(idVendor=0x04b4, idProduct=0x1003)

# was it found?
if dev is None:
    raise ValueError('Device not found')

# set the active configuration. With no arguments, the first
# configuration will be the active one
dev.set_configuration()

# get an endpoint instance
cfg = dev.get_active_configuration()

dev.set_interface_altsetting(interface = 0, alternate_setting = 2)
intf = cfg[(0,2)]

print(intf)

ep = usb.util.find_descriptor(
    intf,
    custom_match = \
    lambda e: \
        e.bEndpointAddress == 0x06)
        
ep2 = usb.util.find_descriptor(
    intf,
    custom_match = \
    lambda e: \
        e.bEndpointAddress == 0x82)

assert ep is not None
assert ep2 is not None

print ('ep:',ep)
print ('ep2:',ep2)

chunkr = 2**16
chunkw = 2**16

print ("Starting CYUSB FX2 CyStream python bulk loopback test")

def read_loop():
    start = time.time()
    trans = 0
    while True:
        data = dev.read(0x82, chunkr, 1000)
        trans = trans + len(data)
        if time.time() > start + 1:
            bps = trans/(time.time() - start)
            print ("Transfered %d kB/s" % (bps/(1024)) )
            trans = 0
            start = time.time()

def write_loop():
    dat = "A"*chunkw
    while True:
        try:
            ep.write(dat)
        except:
            pass

_thread.start_new_thread( write_loop, () )

read_loop()
```

Windows result using libusbk driver
```
(py39x64venv) PS C:\work\libusb\python_usb> python .\cyusb_fx2_bulk_loopback-test_python3.py
    INTERFACE 0, 2: Vendor Specific ========================
     bLength            :    0x9 (9 bytes)
     bDescriptorType    :    0x4 Interface
     bInterfaceNumber   :    0x0
     bAlternateSetting  :    0x2
     bNumEndpoints      :    0x2
     bInterfaceClass    :   0xff Vendor Specific
     bInterfaceSubClass :    0x0
     bInterfaceProtocol :    0x0
     iInterface         :    0x0
      ENDPOINT 0x82: Bulk IN ===============================
       bLength          :    0x7 (7 bytes)
       bDescriptorType  :    0x5 Endpoint
       bEndpointAddress :   0x82 IN
       bmAttributes     :    0x2 Bulk
       wMaxPacketSize   :  0x200 (512 bytes)
       bInterval        :    0x0
      ENDPOINT 0x6: Bulk OUT ===============================
       bLength          :    0x7 (7 bytes)
       bDescriptorType  :    0x5 Endpoint
       bEndpointAddress :    0x6 OUT
       bmAttributes     :    0x2 Bulk
       wMaxPacketSize   :  0x200 (512 bytes)
       bInterval        :    0x0
ep:       ENDPOINT 0x6: Bulk OUT ===============================
       bLength          :    0x7 (7 bytes)
       bDescriptorType  :    0x5 Endpoint
       bEndpointAddress :    0x6 OUT
       bmAttributes     :    0x2 Bulk
       wMaxPacketSize   :  0x200 (512 bytes)
       bInterval        :    0x0
ep2:       ENDPOINT 0x82: Bulk IN ===============================
       bLength          :    0x7 (7 bytes)
       bDescriptorType  :    0x5 Endpoint
       bEndpointAddress :   0x82 IN
       bmAttributes     :    0x2 Bulk
       wMaxPacketSize   :  0x200 (512 bytes)
       bInterval        :    0x0
Starting CYUSB3014 python loopback test
Transfered 30424 kB/s
Transfered 29803 kB/s
Transfered 29833 kB/s
Transfered 29889 kB/s
Transfered 29114 kB/s
Transfered 29627 kB/s
Transfered 29584 kB/s
Transfered 29424 kB/s
Transfered 29338 kB/s
...
```

mcuee, 2022-03-29:
Using pyusb to test cystream FW isco_loopback:
```
#!/usr/bin/python

### quick mod from https://github.com/danielkucera/cyusb3014-breakout/blob/master/software/loopback-test.py

import usb.core
import usb.util
import time
import _thread

# find our device
dev = usb.core.find(idVendor=0x04b4, idProduct=0x1003)

# was it found?
if dev is None:
    raise ValueError('Device not found')

# set the active configuration. With no arguments, the first
# configuration will be the active one
dev.set_configuration()

# get an endpoint instance
cfg = dev.get_active_configuration()

dev.set_interface_altsetting(interface = 0, alternate_setting = 6)
intf = cfg[(0,6)]

print(intf)

ep = usb.util.find_descriptor(
    intf,
    custom_match = \
    lambda e: \
        e.bEndpointAddress == 0x06)
        
ep2 = usb.util.find_descriptor(
    intf,
    custom_match = \
    lambda e: \
        e.bEndpointAddress == 0x82)

assert ep is not None
assert ep2 is not None

print ('ep:',ep)
print ('ep2:',ep2)

chunkr = 2**16
chunkw = 2**16

print ("Starting CYUSB FX2LP CyStream python isoc loopback test")

def read_loop():
    start = time.time()
    trans = 0
    while True:
        data = dev.read(0x82, chunkr, 1000)
        trans = trans + len(data)
        if time.time() > start + 1:
            bps = trans/(time.time() - start)
            print ("Transfered %d kB/s" % (bps/(1024)) )
            trans = 0
            start = time.time()

def write_loop():
    dat = "A"*chunkw
    while True:
        try:
            ep.write(dat)
        except:
            pass

_thread.start_new_thread( write_loop, () )

read_loop()
```

Windows 11 test result using libusbk driver (WinUSB does not work well for isoc transfer now).
```
#!/usr/bin/python

### quick mod from https://github.com/danielkucera/cyusb3014-breakout/blob/master/software/loopback-test.py

import usb.core
import usb.util
import time
import _thread

# find our device
dev = usb.core.find(idVendor=0x04b4, idProduct=0x1003)

# was it found?
if dev is None:
    raise ValueError('Device not found')

# set the active configuration. With no arguments, the first
# configuration will be the active one
dev.set_configuration()

# get an endpoint instance
cfg = dev.get_active_configuration()

dev.set_interface_altsetting(interface = 0, alternate_setting = 6)
intf = cfg[(0,6)]

print(intf)

ep = usb.util.find_descriptor(
    intf,
    custom_match = \
    lambda e: \
        e.bEndpointAddress == 0x06)
        
ep2 = usb.util.find_descriptor(
    intf,
    custom_match = \
    lambda e: \
        e.bEndpointAddress == 0x82)

assert ep is not None
assert ep2 is not None

print ('ep:',ep)
print ('ep2:',ep2)

chunkr = 2**16
chunkw = 2**16

print ("Starting CYUSB FX2LP CyStream python isoc loopback test")

def read_loop():
    start = time.time()
    trans = 0
    while True:
        data = dev.read(0x82, chunkr, 1000)
        trans = trans + len(data)
        if time.time() > start + 1:
            bps = trans/(time.time() - start)
            print ("Transfered %d kB/s" % (bps/(1024)) )
            trans = 0
            start = time.time()

def write_loop():
    dat = "A"*chunkw
    while True:
        try:
            ep.write(dat)
        except:
            pass

_thread.start_new_thread( write_loop, () )

read_loop()
```

mcuee, 2022-03-29:
> usbutils 007 is from 2013... Didn't we get latest [usbutils](https://formulae.brew.sh/formula/usbutils) properly into homebrew recently? Or are you somehow stuck with mikhailai's version?

Oops I was using the older version from mikhailai. Once I remore the old brew tap and install the latest usbutils, it is now okay. Sorry for the false alarming.

mcuee, 2022-03-29:
Further tested fxload example under my Mac Mini M1 and it works fine as well.
```
mcuee@mcuees-Mac-mini libusb-1.0.26-rc1_arm64 % ./examples/fxload -t fx2lp -i ./CYStream.hex           
found device 'Cypress EZ-USB FX2LP (68013A/68014A/68015A/68016A)' [04b4:8613] (2,10)
microcontroller type: fx2lp
./CYStream.hex: type Intel HEX
stop CPU
... WROTE: 2749 bytes, 40 segments, avg 68
reset CPU
```

mcuee, 2022-03-29:
Cyusb 3 isochronous transfer works fine under macOS as well using the Cypress USB SDK macOS version.

```
mcuee@mcuees-Mac-mini examples % ./download_fx3 -t RAM -i ../fx3_images/cyfxisosrcsink.img
FX3 firmware programming to RAM completed
mcuee@mcuees-Mac-mini examples % ./cyisoread_test 
Usage of binary ./isoread_test <num packets> 
 Ex: ./isoread_test 64 
mcuee@mcuees-Mac-mini examples % ./cyisoread_test 4096

Packet size is 1024 

***************************
 Total Packets Received: 4096 
 Pkts Succeeded    : 4096 
 Pkts Failed       : 0 
****************************
mcuee@mcuees-Mac-mini examples % ./cyisowrite_test 4096

Packet size is 1024 

***************************
 Total Packets sent: 4096 
 Pkts Succeeded    : 4096 
 Pkts Failed       : 0 
****************************```

mcuee, 2022-03-29:
isoc under Linux is also good with Cyusb FX3 SDK 1.0.5 under Ubuntu Linux 20.04 (x86_64) with libusb-1.0.26rc1.
[https://i.imgur.com/O3oJWZ9.png](url)

mcuee, 2022-03-29:
@tormodvolden 

WinUSB driver will have an error.
```
Starting CYUSB FX2LP CyStream python isoc loopback test
Traceback (most recent call last):
  File "C:\work\libusb\python_usb\cyusb_fx2_isoc_loopback-test_python3.py", line 74, in <module>
    read_loop()
  File "C:\work\libusb\python_usb\cyusb_fx2_isoc_loopback-test_python3.py", line 56, in read_loop
    data = dev.read(0x82, chunkr, 1000)
  File "C:\work\python\py39x64venv\lib\site-packages\usb\core.py", line 1029, in read
    ret = fn(
  File "C:\work\python\py39x64venv\lib\site-packages\usb\backend\libusb1.py", line 846, in bulk_read
    return self.__read(self.lib.libusb_bulk_transfer,
  File "C:\work\python\py39x64venv\lib\site-packages\usb\backend\libusb1.py", line 954, in __read
    _check(retval)
  File "C:\work\python\py39x64venv\lib\site-packages\usb\backend\libusb1.py", line 604, in _check
    raise USBError(_strerror(ret), ret, _libusb_errno[ret])
usb.core.USBError: [Errno 5] Input/Output Error
```

<details>
<summary> click for the full debug log </summary>

```
[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.000015] [00001070] libusb: debug [libusb_init] libusb v1.0.26.11717-rc1
[ 0.000024] [00001070] libusb: debug [usbi_add_event_source] add HANDLE 000000000000010C events 0
[ 0.000031] [00001070] libusb: debug [usbi_io_init] using timer for timeouts
[ 0.000033] [00001070] libusb: debug [usbi_add_event_source] add HANDLE 0000000000000118 events 0
[ 0.000061] [00001070] libusb: debug [get_windows_version] Windows 11 64-bit
[ 0.000064] [00001070] libusb: debug [htab_create] using 1021 entries hash table
[ 0.000786] [00001070] libusb: info [winusbx_init] WinUSB DLL available (with isoch support)
[ 0.000991] [00001070] libusb: debug [winusbx_init] libusbK DLL found, version: 3.1.0.0
[ 0.002808] [00001070] libusb: info [windows_init] UsbDk backend is not available
[ 0.002864] [00001070] libusb: warning [libusb_init] installing new context as implicit default
[ 0.002943] [00001070] libusb: debug [libusb_get_device_list]  
[ 0.003692] [00000448] libusb: debug [windows_iocp_thread] I/O completion thread started
[ 0.009349] [00001070] libusb: debug [winusb_get_device_list] allocating new device for session [5]
[ 0.009394] [00001070] libusb: debug [winusb_get_device_list] allocating new device for session [6]
[ 0.009749] [00001070] libusb: debug [get_api_type] driver(s): WUDFRd
[ 0.009775] [00001070] libusb: debug [get_api_type] lower filter driver(s): WinUsb
[ 0.009795] [00001070] libusb: debug [get_api_type] matched lower filter driver name against WinUSB
[ 0.009796] [00001070] libusb: debug [winusb_get_device_list] allocating new device for session [1F]
[ 0.009913] [00001070] libusb: debug [get_api_type] driver(s): BTHUSB
[ 0.009936] [00001070] libusb: debug [get_api_type] lower filter driver(s): ibtusb
[ 0.009940] [00001070] libusb: debug [winusb_get_device_list] allocating new device for session [34]
[ 0.010027] [00001070] libusb: debug [get_api_type] driver(s): WinUSB
[ 0.010046] [00001070] libusb: debug [get_api_type] matched driver name against WinUSB
[ 0.010048] [00001070] libusb: debug [winusb_get_device_list] allocating new device for session [37]
[ 0.010200] [00001070] libusb: debug [get_api_type] driver(s): usbccgp
[ 0.010217] [00001070] libusb: debug [get_api_type] matched driver name against Composite API
[ 0.010220] [00001070] libusb: debug [winusb_get_device_list] allocating new device for session [63]
[ 0.010297] [00001070] libusb: debug [get_api_type] driver(s): usbccgp
[ 0.010315] [00001070] libusb: debug [get_api_type] matched driver name against Composite API
[ 0.010317] [00001070] libusb: debug [winusb_get_device_list] allocating new device for session [64]
[ 0.010498] [00001070] libusb: debug [enumerate_hcd_root_hub] assigning HCD 'PCI\VEN_8086&DEV_A0ED&SUBSYS_15451025&REV_20\3&11583659&0&A0' bus number 1
[ 0.010526] [00001070] libusb: debug [enumerate_hcd_root_hub] assigning HCD 'PCI\VEN_8086&DEV_9A13&SUBSYS_72708086&REV_01\3&11583659&0&68' bus number 2
[ 0.011073] [00001070] libusb: debug [winusb_get_device_list] found existing device for session [34]
[ 0.011130] [00001070] libusb: debug [init_device] found 1 configurations (current config: 1)
[ 0.011142] [00001070] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 200 bytes)
[ 0.011153] [00001070] libusb: debug [init_device] (bus: 1, addr: 4, depth: 1, port: 10): 'USB\VID_8087&PID_0026\5&586B51A&0&10'
[ 0.011200] [00001070] libusb: debug [winusb_get_device_list] extra GUID: {16126A05-3179-4589-9DB8-952AFFD297D5}
[ 0.011207] [00001070] libusb: debug [winusb_get_device_list] found existing device for session [63]
[ 0.011239] [00001070] libusb: debug [init_device] found 1 configurations (current config: 1)
[ 0.011253] [00001070] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 512 bytes)
[ 0.011260] [00001070] libusb: debug [init_device] (bus: 1, addr: 3, depth: 1, port: 5): 'USB\VID_04F2&PID_B6DD\0001'
[ 0.011322] [00001070] libusb: debug [winusb_get_device_list] extra GUID: {20348E47-A897-468E-9D0E-56307802FC2A}
[ 0.011341] [00001070] libusb: debug [winusb_get_device_list] found existing device for session [1F]
[... 469 lines cut ...]
libusb: debug [libusb_open] open 1.31
libusb: debug [libusb_open] open 1.31
libusb: debug [libusb_open] open 1.31
[... cut ...]
libusb: debug [libusb_open] open 1.31
libusb: debug [libusb_open] open 1.31
libusb: debug [libusb_open] open 1.31
libusb: debug [libusb_open] open 1.31
libusb: debug [libusb_open] open 1.31
libusb: debug [libusb_open] open 1.31
libusb: debug [libusb_open] open 1.31
[ 0.022225] [00001070] libusb: debug [usbi_remove_event_source] remove HANDLE 0000000000000118
[ 0.022235] [00001070] libusb: debug [usbi_remove_event_source] remove HANDLE 000000000000010C
libusb: debug [libusb_open] open 1.31

```

</details>

mcuee, 2022-03-29:
Somehow even for libusbk driver, it shows "w

<details>
<summary> click for full debug log </details>

```
[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.000014] [00001c10] libusb: debug [libusb_init] libusb v1.0.26.11717-rc1
[ 0.000023] [00001c10] libusb: debug [usbi_add_event_source] add HANDLE 0000000000000114 events 0
[ 0.000048] [00001c10] libusb: debug [usbi_io_init] using timer for timeouts
[ 0.000050] [00001c10] libusb: debug [usbi_add_event_source] add HANDLE 0000000000000120 events 0
[ 0.000063] [00001c10] libusb: debug [get_windows_version] Windows 11 64-bit
[ 0.000066] [00001c10] libusb: debug [htab_create] using 1021 entries hash table
[ 0.000896] [00001c10] libusb: info [winusbx_init] WinUSB DLL available (with isoch support)
[ 0.001102] [00001c10] libusb: debug [winusbx_init] libusbK DLL found, version: 3.1.0.0
[ 0.002772] [00001c10] libusb: info [windows_init] UsbDk backend is not available
[ 0.002830] [00001c10] libusb: warning [libusb_init] installing new context as implicit default
[ 0.002916] [00001c10] libusb: debug [libusb_get_device_list]  
[ 0.003381] [00001374] libusb: debug [windows_iocp_thread] I/O completion thread started
[ 0.012161] [00001c10] libusb: debug [winusb_get_device_list] allocating new device for session [5]
[ 0.012213] [00001c10] libusb: debug [winusb_get_device_list] allocating new device for session [6]
[ 0.012712] [00001c10] libusb: debug [get_api_type] driver(s): WUDFRd
[ 0.012748] [00001c10] libusb: debug [get_api_type] lower filter driver(s): WinUsb
[ 0.012753] [00001c10] libusb: debug [get_api_type] matched lower filter driver name against WinUSB
[ 0.012756] [00001c10] libusb: debug [winusb_get_device_list] allocating new device for session [1F]
[ 0.012880] [00001c10] libusb: debug [get_api_type] driver(s): libusbK
[ 0.012908] [00001c10] libusb: debug [get_api_type] matched driver name against libusbK
[ 0.012910] [00001c10] libusb: debug [winusb_get_device_list] allocating new device for session [25]
[ 0.013068] [00001c10] libusb: debug [get_api_type] driver(s): BTHUSB
[ 0.013108] [00001c10] libusb: debug [get_api_type] lower filter driver(s): ibtusb
[ 0.013113] [00001c10] libusb: debug [winusb_get_device_list] allocating new device for session [35]
[ 0.013329] [00001c10] libusb: debug [get_api_type] driver(s): usbccgp
[ 0.013352] [00001c10] libusb: debug [get_api_type] matched driver name against Composite API
[ 0.013354] [00001c10] libusb: debug [winusb_get_device_list] allocating new device for session [63]
[ 0.013456] [00001c10] libusb: debug [get_api_type] driver(s): usbccgp
[ 0.013481] [00001c10] libusb: debug [get_api_type] matched driver name against Composite API
[ 0.013484] [00001c10] libusb: debug [winusb_get_device_list] allocating new device for session [64]
[ 0.013702] [00001c10] libusb: debug [enumerate_hcd_root_hub] assigning HCD 'PCI\VEN_8086&DEV_A0ED&SUBSYS_15451025&REV_20\3&11583659&0&A0' bus number 1
[ 0.013738] [00001c10] libusb: debug [enumerate_hcd_root_hub] assigning HCD 'PCI\VEN_8086&DEV_9A13&SUBSYS_72708086&REV_01\3&11583659&0&68' bus number 2
[ 0.014437] [00001c10] libusb: debug [winusb_get_device_list] found existing device for session [35]
[ 0.014502] [00001c10] libusb: debug [init_device] found 1 configurations (current config: 1)
[ 0.014515] [00001c10] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 200 bytes)
[ 0.014525] [00001c10] libusb: debug [init_device] (bus: 1, addr: 4, depth: 1, port: 10): 'USB\VID_8087&PID_0026\5&586B51A&0&10'
[ 0.014575] [00001c10] libusb: debug [winusb_get_device_list] extra GUID: {16126A05-3179-4589-9DB8-952AFFD297D5}
[ 0.014580] [00001c10] libusb: debug [winusb_get_device_list] found existing device for session [63]
[ 0.014617] [00001c10] libusb: debug [init_device] found 1 configurations (current config: 1)
[ 0.014629] [00001c10] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 512 bytes)
[ 0.014635] [00001c10] libusb: debug [init_device] (bus: 1, addr: 3, depth: 1, port: 5): 'USB\VID_04F2&PID_B6DD\0001'
[ 0.014696] [00001c10] libusb: debug [winusb_get_device_list] extra GUID: {20348E47-A897-468E-9D0E-56307802FC2A}
[ 0.014701] [00001c10] libusb: debug [winusb_get_device_list] found existing device for session [1F]
[ 0.014739] [00001c10] libusb: debug [init_device] found 1 configurations (current config: 1)
[ 0.014750] [00001c10] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 46 bytes)
[ 0.014756] [00001c10] libusb: debug [init_device] (bus: 1, addr: 5, depth: 1, port: 7): 'USB\VID_1C7A&PID_0575\077E2F9A'
[ 0.014859] [00001c10] libusb: debug [winusb_get_device_list] extra GUID: {A366578D-D29C-7D3F-2246-16A1FA2E478B}
[ 0.014866] [00001c10] libusb: debug [winusb_get_device_list] found existing device for session [25]
[ 0.014914] [00001c10] libusb: debug [init_device] found 1 configurations (current config: 1)
[ 0.014922] [00001c10] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 135 bytes)
[ 0.014928] [00001c10] libusb: debug [init_device] (bus: 1, addr: 32, depth: 1, port: 1): 'USB\VID_04B4&PID_1003\5&586B51A&0&1'
[ 0.015152] [00001c10] libusb: debug [init_root_hub] root hub 'USB\ROOT_HUB30\4&12C539F&0&0' reports 16 ports
[... cut ...]
[ 0.321195] [00001b28] libusb: debug [windows_handle_transfer_completion] handling transfer 000001F7BFF26A20 completion with errcode 0, length 49152
[ 0.321198] [00001b28] libusb: debug [arm_timer_for_next_timeout] no timeouts, disarming timer
[ 0.321204] [00001b28] libusb: debug [usbi_handle_transfer_completion] transfer 000001F7BFF26A20 has callback 00007FF8BF5098A0
[ 0.321207] [00001b28] libusb: debug [sync_transfer_cb] actual_length=49152
[ 0.321219] [00001b28] libusb: debug [libusb_free_transfer] transfer 000001F7BFF26A20
[ 0.321268] [00001c10] libusb: debug [libusb_free_transfer] transfer 000001F7BF70C940
...
```

</details>

mcuee, 2022-03-29:
Same thing reported here. It shows "winusbx_submit_bulk_transfer" in the debug log.
* https://github.com/libusb/libusb/issues/759

tormodvolden, 2022-03-29:
Interesting. Let's open a separate ticket for this to not spam the RC testing. We don't expect WinUSB isoch to work correctly.

tormodvolden, 2022-03-29:
I managed to cross-build latest libfreenect (with latest freeglut) for Windows, using MinGW on Ubuntu - this was a feat in itself. It actually works, isochronous streaming over WinUSB from an Xbox 360 Kinect camera.

mcuee, 2022-03-30:
> Interesting. Let's open a separate ticket for this (or carry on in 759) to not spam the RC testing. We don't expect WinUSB isoch to work correctly. It is indeed using winusbx_submit_bulk_transfer() because we see `writing 65536 bytes`. Do you have some isoch test in C, without pyusb?

That is the problem. All my known-good FW are tested with libusbk Kbench or Cypress FX3 USB SDK host tools, they are not using libusb under Windows. That is why I was using pyusb -- but pyusb may not be correct. I will see if I can try other methods. But I thiink we are good to go for libusb-1.0.26 with regard to isocu transfer support.

mcuee, 2022-04-03:
One more test with HID device under Windows.
FW for PIC1887J50 and host code here.
https://github.com/mcuee/picusb/tree/master/lvr_genhid_mod

It seems to work fine. The host code sends one extra byte to see if the device will respond or not.

```
$ ./libusb1_lvrhid8.exe
Successfully find the LVR Generic HID device
Successfully set usb configuration 1
Successfully claimed interface
Testing control transfer using loop back test of feature report
03, 04; 21, 21; 22, 22; 23, 00;
Testing control transfer using loop back test of input/output report
02, 01; 31, 31; 32, 32; 33, 00;
Testing interrupt transfer using loop back test of input/output report
02, 01; 41, 41; 42, 42; 43, 00;
```

mcuee, 2022-04-03:
> Edit: Confirmed that also isochronous OUT works with a custom device on WinUSB, although reported packet transferred lengths are zero. The data is transferred in both directions, but the log shows:
> `error [windows_handle_transfer_completion] detected I/O error 87: [87] The parameter is incorrect.`
> both for OUT and IN transfers. Then in both cases the transfer is submitted again, now with success and the packet actual_length correct for IN but zero for OUT. I wonder if it is just missing actual_length assignment for WinUSB/OUT in winusbx_copy_transfer_data() that is the reason for the zero lengths reported, see #1105. I cannot rule out that the custom firmware is causing the error 87, but on Linux everything seems fine.

Your test results seem to be in line with some of my test results, like the transfer size report as 0 and Error 87.
https://github.com/libusb/libusb/issues/1104

martinling, 2022-04-09:
As requested by @tormodvolden I've run some tests with HackRF.

We're pretty sensitive to any transfer throughput issues since we stream at 40MB/s with little latency margin, and we do it with bulk transfers.

I tested on Linux, and on Windows with the WinUSB backend. I ran `hackrf_transfer` in transmit and receive mode, `hackrf_sweep`, and a test program we've been using called `hackrf_loop` which repeatedly starts & stops streaming.

Everything seems to work fine on both Linux and Windows with 1.0.26-rc1 and I don't see any changes to our streaming performance there - Windows has latency problems at full sample rate, but that's unchanged across libusb releases.

I had previously been running 1.0.24, so I also tested 1.0.25 to see if we had been affected by bugs in that release.

On 1.0.25, `hackrf_transfer` on Windows segfaults just before normal exit, after printing:
```
libusb: debug [libusb_free_transfer] transfer 000001e6f888a5b8
```
That problem is fixed by 1.0.26-rc1.

tormodvolden, 2022-04-09:
@martinling Thanks a lot for your testing and success report!

closed 2022-04-10 by tormodvolden