libusb/libusb #1103: Some functions fail on Windows 7

a923169932, 2022-03-30:
In Windows 7，libusb_open success but libusb_get_active_config_descriptor fail.

[image]

[image]

a923169932, 2022-03-30:
In Windows 10 is normal.

mcuee, 2022-03-30:
Please provide the details of the device, the driver used and the debug log (by setting LIBUSB_DEBUG=4).

mcuee, 2022-03-30:
BTW: Try to use text and not picture. Thanks.

a923169932, 2022-03-30:
libusb: debug [windows_get_device_list] setting HID interface for [2E8]:
libusb: debug [set_hid_interface] interface[0] = \\.\HID#VID_25A7&PID_5121#6&2A48F251&0&0000#{4D1E55B2-F16F-11CF-88CB-001111000030}
libusb: debug [get_api_type] driver(s): libusb0
libusb: debug [get_api_type] matched driver name against libusb0
libusb: debug [libusb_get_device_descriptor]
libusb: debug [libusb_get_device_descriptor]
libusb: debug [libusb_get_device_descriptor]
libusb: debug [libusb_open] open 3.6
libusb: debug [libusb_get_configuration]
libusb: debug [libusb_get_configuration] active config 4
config->iConfiguration[5]
config->bNumInterfaces[1]
config->bConfigurationValue[1]
libusb_interface_descriptor [0/1] bInterfaceClass[0x06] bInterfaceSubClass[0x01] bInterfaceProtocol[0x01]
[09:58:55[2] Could not find a suitable USB interface for device 3-6
libusb: debug [libusb_close]

The above is the failed log,
The configuration returned by libusb_get_configuration has only 1 interface,
but it is normal to use libusb-win32，
drive is libusb-win32,  In Windows 10 is normal.
Thanks.

a923169932, 2022-03-30:
//use libusb-win32 API log
usb_config->iConfiguration[8]
usb_config->bNumInterfaces[3]
usb_config->bConfigurationValue[4]
usb_interface_descriptor [0/3] bInterfaceClass[0x06] bInterfaceSubClass[0x01] bInterfaceProtocol[0x01]
usb_interface_descriptor [1/3] bInterfaceClass[0xff] bInterfaceSubClass[0xfe] bInterfaceProtocol[0x02]
usbdev->interface[1]
usbdev->ep_out[04]
usbdev->ep_in[85]
usb_claim_interface [1] success

//use libusb1.0.20 API log
libusb_get_active_config_descriptor start
libusb: debug [libusb_get_configuration] active config 4
config->iConfiguration[5]
config->bNumInterfaces[1]
config->bConfigurationValue[1]
libusb_interface_descriptor [0/1] bInterfaceClass[0x06] bInterfaceSubClass[0x01] bInterfaceProtocol[0x01]
[09:58:55[2] Could not find a suitable USB interface for device 3-6

a923169932, 2022-03-30:
I think libusb_get_active_config_descriptor active config is 4,
but returned config 1 configuration

a923169932, 2022-03-30:
Because libusb1.0 does not support setting configuration, I use libusb-win32 to set configuration, does this matter?

//code
`
	libusb_set_debug(NULL, 255);
	if((res = libusb_get_configuration(handle, &current_config)) != 0) {
		usbmuxd_log(LL_WARNING, "Could not get configuration for device %d-%d: %s", bus, address, libusb_error_name(res));
		libusb_close(handle);
		return -1;
	}
	printf("libusb_get_configuration current_config[%d]\n", current_config);
`

//Windows 10 success log
libusb: debug [libusb_get_config_descriptor] index 0
[0]config[015CEBF0]
[0]config->iConfiguration[5]
[0]config->bNumInterfaces[1]
[0]config->bConfigurationValue[1]
[0]libusb_interface_descriptor [1/1] bInterfaceClass[0x06] bInterfaceSubClass[0x01] bInterfaceProtocol[0x01]
libusb: debug [libusb_get_config_descriptor] index 1
libusb: debug [parse_endpoint] skipping descriptor 25
[1]config[015CEE90]
[1]config->iConfiguration[6]
[1]config->bNumInterfaces[3]
[1]config->bConfigurationValue[2]
[1]libusb_interface_descriptor [1/3] bInterfaceClass[0x01] bInterfaceSubClass[0x01] bInterfaceProtocol[0x00]
[1]libusb_interface_descriptor [2/3] bInterfaceClass[0x01] bInterfaceSubClass[0x02] bInterfaceProtocol[0x00]
[1]libusb_interface_descriptor [3/3] bInterfaceClass[0x03] bInterfaceSubClass[0x00] bInterfaceProtocol[0x00]
libusb: debug [libusb_get_config_descriptor] index 2
[2]config[015CEF10]
[2]config->iConfiguration[7]
[2]config->bNumInterfaces[2]
[2]config->bConfigurationValue[3]
[2]libusb_interface_descriptor [1/2] bInterfaceClass[0x06] bInterfaceSubClass[0x01] bInterfaceProtocol[0x01]
[2]libusb_interface_descriptor [2/2] bInterfaceClass[0xff] bInterfaceSubClass[0xfe] bInterfaceProtocol[0x02]
libusb: debug [libusb_get_config_descriptor] index 3
[3]config[015CF0B0]
[3]config->iConfiguration[8]
[3]config->bNumInterfaces[3]
[3]config->bConfigurationValue[4]
[3]libusb_interface_descriptor [1/3] bInterfaceClass[0x06] bInterfaceSubClass[0x01] bInterfaceProtocol[0x01]
[3]libusb_interface_descriptor [2/3] bInterfaceClass[0xff] bInterfaceSubClass[0xfe] bInterfaceProtocol[0x02]
[3]libusb_interface_descriptor [3/3] bInterfaceClass[0xff] bInterfaceSubClass[0xfd] bInterfaceProtocol[0x01]

//Windows 7 fail log 
libusb: debug [libusb_get_configuration]
libusb: debug [libusb_get_configuration] active config 4
libusb_get_configuration current_config[4]
libusb_get_config_descriptor start
libusb: debug [libusb_get_config_descriptor] index 0
[0]config[0089F730]
[0]config->iConfiguration[5]
[0]config->bNumInterfaces[1]
[0]config->bConfigurationValue[1]
[0]libusb_interface_descriptor [1/1] bInterfaceClass[0x06] bInterfaceSubClass[0x01] bInterfaceProtocol[0x01]
libusb: debug [libusb_get_config_descriptor] index 1
[1]config[0089F6D0]
[1]config->iConfiguration[5]
[1]config->bNumInterfaces[1]
[1]config->bConfigurationValue[1]
[1]libusb_interface_descriptor [1/1] bInterfaceClass[0x06] bInterfaceSubClass[0x01] bInterfaceProtocol[0x01]
libusb: debug [libusb_get_config_descriptor] index 2
[2]config[0089F710]
[2]config->iConfiguration[5]
[2]config->bNumInterfaces[1]
[2]config->bConfigurationValue[1]
[2]libusb_interface_descriptor [1/1] bInterfaceClass[0x06] bInterfaceSubClass[0x01] bInterfaceProtocol[0x01]
libusb: debug [libusb_get_config_descriptor] index 3
[3]config[0089F790]
[3]config->iConfiguration[5]
[3]config->bNumInterfaces[1]
[3]config->bConfigurationValue[1]
[3]libusb_interface_descriptor [1/1] bInterfaceClass[0x06] bInterfaceSubClass[0x01] bInterfaceProtocol[0x01]

mcuee, 2022-03-30:
What is the version of libusb are you using? Please try to use libusb-1.0.25 release and 1.0.26rc1 and test again.

mcuee, 2022-03-30:
What is this device? Please post the USB descriptors using usbview.
https://ftdichip.com/utilities/#microsoft-usbview

a923169932, 2022-03-30:
My libusb0.sys version is 1.2.6.0
I try use libusb1.0.26rc1, the problem still exists
My device is iPhone, VID_05AC PID_12A8.

Using the libusb_set_configuration function also succeeds on Windows 10, fails on Windows 7.
this is libusb1.0.26rc1 the log:
[11:40:13[3] Using libusb 1.0.26
[11:40:13[1] libusb does not support hotplug events
[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.060903] [000013c0] libusb: debug [libusb_get_configuration]
[ 0.060951] [000013c0] libusb: debug [libusb_get_configuration] active config 1
libusb_get_configuration current_config[1]
[ 0.061037] [000013c0] libusb: debug [libusb_set_configuration] configuration 4
[ 0.061075] [000013c0] libusb: debug [libusb_submit_transfer] transfer 00488388
[ 0.061113] [000013c0] libusb: debug [add_to_flying_list] arm timer for timeout in 1000ms (first in line)
[ 0.061189] [000013c0] libusb: debug [libusb_claim_interface] interface 0
[ 0.061261] [000013c0] libusb: debug [winusbx_claim_interface] claimed interface 0
[ 0.061301] [000013c0] libusb: debug [windows_assign_endpoints] (re)assigned endpoint 02 to interface 0
[ 0.061359] [000013c0] libusb: debug [windows_assign_endpoints] (re)assigned endpoint 81 to interface 0
[ 0.061417] [000013c0] libusb: debug [windows_assign_endpoints] (re)assigned endpoint 83 to interface 0
[ 0.061494] [000013c0] libusb: debug [auto_claim] auto-claimed interface 0 for control request
[ 0.061508] [00000d44] libusb: debug [windows_iocp_thread] ignoring overlapped 0035F9B0 for handle 00487798 (device 3.7)
[ 0.061606] [00000d44] libusb: debug [windows_iocp_thread] ignoring overlapped 0035F9B0 for handle 00487798 (device 3.7)
[ 0.061668] [00000d44] libusb: debug [windows_iocp_thread] ignoring overlapped 0035F9B0 for handle 00487798 (device 3.7)
[ 0.061733] [00000d44] libusb: debug [windows_iocp_thread] ignoring overlapped 0035F9B0 for handle 00487798 (device 3.7)
[ 0.061533] [000013c0] libusb: debug [winusbx_submit_control_transfer] will use interface 0
[ 0.061800] [00000d44] libusb: debug [windows_iocp_thread] ignoring overlapped 0035FA30 for handle 00487798 (device 3.7)
[ 0.061870] [000013c0] libusb: warning [winusbx_submit_control_transfer] ControlTransfer failed: [87] 参数错误。
[ 0.062754] [000013c0] libusb: debug [libusb_release_interface] interface 0
[ 0.062801] [000013c0] libusb: debug [auto_release] auto-released interface 0
[ 0.062843] [000013c0] libusb: debug [arm_timer_for_next_timeout] no timeouts, disarming timer
[ 0.062890] [000013c0] libusb: debug [libusb_free_transfer] transfer 00488388
[11:40:13[2] Could not set configuration 4 for device 3-7: LIBUSB_ERROR_IO

a923169932, 2022-03-30:
I noticed that the version of winusb.sys in Windows 7 is 6.1.7601.17514, and the version of winusb.sys in Windows 10 is 10.0.19041.1.

tormodvolden, 2022-03-30:
Is this with libusb 1.0.26-rc1 ?

a923169932, 2022-03-30:
> 这是 libusb 1.0.26-rc1 吗？

yes, this is libusb 1.0.26-rc1

a923169932, 2022-03-30:
In Windows 7, 1.0.26rc1 function libusb_get_string_descriptor_ascii fail, return -5, 1.0.20 not this problem.
But these functions are normal in Windows 10

a923169932, 2022-03-30:
**This is my log using "testlibusb.exe -v", the bConfigurationValue of config1-4 is always 1**

Dev (bus 3, device 21): 05AC - 12A8 speed: 480M
  Product:                   iPhone
  Serial Number:             c80a85d400342a3d730ac3dfe2daa05b5ce7d145
  Configuration:
    wTotalLength:            39
    bNumInterfaces:          1
    bConfigurationValue:     1
    iConfiguration:          5
    bmAttributes:            c0h
    MaxPower:                250
    Interface:
      bInterfaceNumber:      0
      bAlternateSetting:     0
      bNumEndpoints:         3
      bInterfaceClass:       6
      bInterfaceSubClass:    1
      bInterfaceProtocol:    1
      iInterface:            19
      Endpoint:
        bEndpointAddress:    02h
        bmAttributes:        02h
        wMaxPacketSize:      512
        bInterval:           0
        bRefresh:            0
        bSynchAddress:       0
      Endpoint:
        bEndpointAddress:    81h
        bmAttributes:        02h
        wMaxPacketSize:      512
        bInterval:           0
        bRefresh:            0
        bSynchAddress:       0
      Endpoint:
        bEndpointAddress:    83h
        bmAttributes:        03h
        wMaxPacketSize:      64
        bInterval:           10
        bRefresh:            0
        bSynchAddress:       0
  Configuration:
    wTotalLength:            39
    bNumInterfaces:          1
    bConfigurationValue:     1
    iConfiguration:          5
    bmAttributes:            c0h
    MaxPower:                250
    Interface:
      bInterfaceNumber:      0
      bAlternateSetting:     0
      bNumEndpoints:         3
      bInterfaceClass:       6
      bInterfaceSubClass:    1
      bInterfaceProtocol:    1
      iInterface:            19
      Endpoint:
        bEndpointAddress:    02h
        bmAttributes:        02h
        wMaxPacketSize:      512
        bInterval:           0
        bRefresh:            0
        bSynchAddress:       0
      Endpoint:
        bEndpointAddress:    81h
        bmAttributes:        02h
        wMaxPacketSize:      512
        bInterval:           0
        bRefresh:            0
        bSynchAddress:       0
      Endpoint:
        bEndpointAddress:    83h
        bmAttributes:        03h
        wMaxPacketSize:      64
        bInterval:           10
        bRefresh:            0
        bSynchAddress:       0
  Configuration:
    wTotalLength:            39
    bNumInterfaces:          1
    bConfigurationValue:     1
    iConfiguration:          5
    bmAttributes:            c0h
    MaxPower:                250
    Interface:
      bInterfaceNumber:      0
      bAlternateSetting:     0
      bNumEndpoints:         3
      bInterfaceClass:       6
      bInterfaceSubClass:    1
      bInterfaceProtocol:    1
      iInterface:            19
      Endpoint:
        bEndpointAddress:    02h
        bmAttributes:        02h
        wMaxPacketSize:      512
        bInterval:           0
        bRefresh:            0
        bSynchAddress:       0
      Endpoint:
        bEndpointAddress:    81h
        bmAttributes:        02h
        wMaxPacketSize:      512
        bInterval:           0
        bRefresh:            0
        bSynchAddress:       0
      Endpoint:
        bEndpointAddress:    83h
        bmAttributes:        03h
        wMaxPacketSize:      64
        bInterval:           10
        bRefresh:            0
        bSynchAddress:       0
  Configuration:
    wTotalLength:            39
    bNumInterfaces:          1
    bConfigurationValue:     1
    iConfiguration:          5
    bmAttributes:            c0h
    MaxPower:                250
    Interface:
      bInterfaceNumber:      0
      bAlternateSetting:     0
      bNumEndpoints:         3
      bInterfaceClass:       6
      bInterfaceSubClass:    1
      bInterfaceProtocol:    1
      iInterface:            19
      Endpoint:
        bEndpointAddress:    02h
        bmAttributes:        02h
        wMaxPacketSize:      512
        bInterval:           0
        bRefresh:            0
        bSynchAddress:       0
      Endpoint:
        bEndpointAddress:    81h
        bmAttributes:        02h
        wMaxPacketSize:      512
        bInterval:           0
        bRefresh:            0
        bSynchAddress:       0
      Endpoint:
        bEndpointAddress:    83h
        bmAttributes:        03h
        wMaxPacketSize:      64
        bInterval:           10
        bRefresh:            0
        bSynchAddress:       0

tormodvolden, 2022-03-30:
Please attach the complete debug log.

a923169932, 2022-03-30:
> [ 0.404826] [00000e70] libusb: debug [libusb_get_device_descriptor]
> Dev (bus 3, device 21): 05AC - 12A8 speed: 480M
> [ 0.408947] [00000e70] libusb: debug [libusb_open] open 3.21
> [ 0.411082] [00000e70] libusb: debug [libusb_submit_transfer] transfer 00824A78
> [ 0.413093] [00000e70] libusb: debug [add_to_flying_list] arm timer for timeout in 1000ms (first in line)
> [ 0.418233] [00000e70] libusb: debug [libusb_claim_interface] interface 0
> [ 0.422419] [00000e70] libusb: debug [winusbx_claim_interface] claimed interface 0
> [ 0.422586] [00000e70] libusb: warning [windows_assign_endpoints] could not read config descriptor: error -5
> [ 0.424391] [00000e70] libusb: debug [libusb_claim_interface] interface 1
> [ 0.426531] [00000e70] libusb: debug [winusbx_claim_interface] claimed interface 1
> [ 0.428634] [00000e70] libusb: warning [windows_assign_endpoints] could not read config descriptor: error -5
> [ 0.430683] [00000e70] libusb: debug [libusb_claim_interface] interface 2
> [ 0.424422] [00000894] libusb: debug [windows_iocp_thread] ignoring overlapped 0037F494 for handle 0082A608 (device 3.21)
> [ 0.435737] [00000e70] libusb: debug [winusbx_claim_interface] claimed interface 2
> [ 0.439854] [00000e70] libusb: warning [windows_assign_endpoints] could not read config descriptor: error -5
> [ 0.441987] [00000e70] libusb: debug [libusb_claim_interface] interface 3
> [ 0.444090] [00000e70] libusb: debug [libusb_claim_interface] interface 4
> [ 0.446120] [00000e70] libusb: debug [libusb_claim_interface] interface 5
> [ 0.452278] [00000e70] libusb: debug [libusb_claim_interface] interface 6
> [ 0.453231] [00000e70] libusb: debug [libusb_claim_interface] interface 7
> [ 0.455341] [00000e70] libusb: debug [libusb_claim_interface] interface 8
> [ 0.442026] [00000894] libusb: debug [windows_iocp_thread] ignoring overlapped 0037F494 for handle 0082A608 (device 3.21)
> [ 0.457409] [00000e70] libusb: debug [libusb_claim_interface] interface 9
> [ 0.461624] [00000e70] libusb: debug [libusb_claim_interface] interface 10
> [ 0.463691] [00000e70] libusb: debug [libusb_claim_interface] interface 11
> [ 0.459574] [00000894] libusb: debug [windows_iocp_thread] ignoring overlapped 0037F494 for handle 0082A608 (device 3.21)
> [ 0.470766] [00000894] libusb: debug [windows_iocp_thread] ignoring overlapped 0037F494 for handle 0082A608 (device 3.21)
> [ 0.472877] [00000894] libusb: debug [windows_iocp_thread] ignoring overlapped 0037F494 for handle 0082A608 (device 3.21)
> [ 0.468661] [00000e70] libusb: debug [libusb_claim_interface] interface 12
> [ 0.475001] [00000894] libusb: debug [windows_iocp_thread] ignoring overlapped 0037F494 for handle 0082A608 (device 3.21)
> [ 0.475066] [00000e70] libusb: debug [libusb_claim_interface] interface 13
> [ 0.481206] [00000e70] libusb: debug [libusb_claim_interface] interface 14
> [ 0.487093] [00000e70] libusb: debug [libusb_claim_interface] interface 15
> [ 0.488322] [00000e70] libusb: debug [libusb_claim_interface] interface 16
> [ 0.477134] [00000894] libusb: debug [windows_iocp_thread] ignoring overlapped 0037F494 for handle 0082A608 (device 3.21)
> [ 0.490443] [00000e70] libusb: debug [libusb_claim_interface] interface 17
> [ 0.494644] [00000e70] libusb: debug [libusb_claim_interface] interface 18
> [ 0.496706] [00000e70] libusb: debug [libusb_claim_interface] interface 19
> [ 0.501697] [00000e70] libusb: debug [libusb_claim_interface] interface 20
> [ 0.503739] [00000e70] libusb: debug [libusb_claim_interface] interface 21
> [ 0.505860] [00000e70] libusb: debug [libusb_claim_interface] interface 22
> [ 0.507930] [00000e70] libusb: debug [libusb_claim_interface] interface 23
> [ 0.510036] [00000e70] libusb: debug [libusb_claim_interface] interface 24
> [ 0.512005] [00000e70] libusb: debug [libusb_claim_interface] interface 25
> [ 0.514069] [00000e70] libusb: debug [libusb_claim_interface] interface 26
> [ 0.521242] [00000e70] libusb: debug [libusb_claim_interface] interface 27
> [ 0.521378] [00000e70] libusb: debug [libusb_claim_interface] interface 28
> [ 0.523534] [00000e70] libusb: debug [libusb_claim_interface] interface 29
> [ 0.525594] [00000e70] libusb: debug [libusb_claim_interface] interface 30
> [ 0.527743] [00000e70] libusb: debug [libusb_claim_interface] interface 31
> [ 0.529777] [00000e70] libusb: error [auto_claim] could not auto-claim any interface
> [ 0.534747] [00000e70] libusb: debug [arm_timer_for_next_timeout] no timeouts, disarming timer
> [ 0.536811] [00000e70] libusb: debug [libusb_free_transfer] transfer 00824A78
> [ 0.538910] [00000e70] libusb: debug [libusb_submit_transfer] transfer 00824A78
> [ 0.540958] [00000e70] libusb: debug [add_to_flying_list] arm timer for timeout in 1000ms (first in line)
> [ 0.543054] [00000e70] libusb: debug [winusbx_submit_control_transfer] will use interface 0
> [ 0.545358] [00000e70] libusb: debug [windows_force_sync_completion] transfer 00824A78, length 4
> [ 0.545357] [00000894] libusb: debug [windows_iocp_thread] ignoring overlapped 0037F5B8 for handle 0082A608 (device 3.21)
> [ 0.553953] [00000894] libusb: debug [windows_iocp_thread] transfer 00824A78 completed, length 4
> [ 0.547184] [00000e70] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
> [ 0.556475] [00000e70] libusb: debug [handle_events] event sources modified, reallocating event data
> [ 0.558515] [00000e70] libusb: debug [usbi_wait_for_events] WaitForMultipleObjects() for 2 HANDLEs with timeout in 60000ms
[... cut ...]
>         bSynchAddress:       0
>       Endpoint:
>         bEndpointAddress:    81h
>         bmAttributes:        02h
>         wMaxPacketSize:      512
>         bInterval:           0
>         bRefresh:            0
>         bSynchAddress:       0
>       Endpoint:
>         bEndpointAddress:    83h
>         bmAttributes:        03h
>         wMaxPacketSize:      64
>         bInterval:           10
>         bRefresh:            0
>         bSynchAddress:       0
> [ 1.072075] [00000e70] libusb: debug [libusb_close]

a923169932, 2022-03-30:
> Please attach the complete debug log.

### this debug log

> [ 0.404826] [00000e70] libusb: debug [libusb_get_device_descriptor]
Dev (bus 3, device 21): 05AC - 12A8 speed: 480M
[ 0.408947] [00000e70] libusb: debug [libusb_open] open 3.21
[ 0.411082] [00000e70] libusb: debug [libusb_submit_transfer] transfer 00824A78
[ 0.413093] [00000e70] libusb: debug [add_to_flying_list] arm timer for timeout in 1000ms (first in line)
[ 0.418233] [00000e70] libusb: debug [libusb_claim_interface] interface 0
[ 0.422419] [00000e70] libusb: debug [winusbx_claim_interface] claimed interface 0
[ 0.422586] [00000e70] libusb: warning [windows_assign_endpoints] could not read config descriptor: error -5
[ 0.424391] [00000e70] libusb: debug [libusb_claim_interface] interface 1
[ 0.426531] [00000e70] libusb: debug [winusbx_claim_interface] claimed interface 1
[ 0.428634] [00000e70] libusb: warning [windows_assign_endpoints] could not read config descriptor: error -5
[ 0.430683] [00000e70] libusb: debug [libusb_claim_interface] interface 2
[ 0.424422] [00000894] libusb: debug [windows_iocp_thread] ignoring overlapped 0037F494 for handle 0082A608 (device 3.21)
[ 0.435737] [00000e70] libusb: debug [winusbx_claim_interface] claimed interface 2
[ 0.439854] [00000e70] libusb: warning [windows_assign_endpoints] could not read config descriptor: error -5
[ 0.441987] [00000e70] libusb: debug [libusb_claim_interface] interface 3
[ 0.444090] [00000e70] libusb: debug [libusb_claim_interface] interface 4
[ 0.446120] [00000e70] libusb: debug [libusb_claim_interface] interface 5
[ 0.452278] [00000e70] libusb: debug [libusb_claim_interface] interface 6
[ 0.453231] [00000e70] libusb: debug [libusb_claim_interface] interface 7
[ 0.455341] [00000e70] libusb: debug [libusb_claim_interface] interface 8
[ 0.442026] [00000894] libusb: debug [windows_iocp_thread] ignoring overlapped 0037F494 for handle 0082A608 (device 3.21)
[ 0.457409] [00000e70] libusb: debug [libusb_claim_interface] interface 9
[ 0.461624] [00000e70] libusb: debug [libusb_claim_interface] interface 10
[ 0.463691] [00000e70] libusb: debug [libusb_claim_interface] interface 11
[ 0.459574] [00000894] libusb: debug [windows_iocp_thread] ignoring overlapped 0037F494 for handle 0082A608 (device 3.21)
[ 0.470766] [00000894] libusb: debug [windows_iocp_thread] ignoring overlapped 0037F494 for handle 0082A608 (device 3.21)
[ 0.472877] [00000894] libusb: debug [windows_iocp_thread] ignoring overlapped 0037F494 for handle 0082A608 (device 3.21)
[ 0.468661] [00000e70] libusb: debug [libusb_claim_interface] interface 12
[ 0.475001] [00000894] libusb: debug [windows_iocp_thread] ignoring overlapped 0037F494 for handle 0082A608 (device 3.21)
[ 0.475066] [00000e70] libusb: debug [libusb_claim_interface] interface 13
[ 0.481206] [00000e70] libusb: debug [libusb_claim_interface] interface 14
[ 0.487093] [00000e70] libusb: debug [libusb_claim_interface] interface 15
[ 0.488322] [00000e70] libusb: debug [libusb_claim_interface] interface 16
[ 0.477134] [00000894] libusb: debug [windows_iocp_thread] ignoring overlapped 0037F494 for handle 0082A608 (device 3.21)
[ 0.490443] [00000e70] libusb: debug [libusb_claim_interface] interface 17
[ 0.494644] [00000e70] libusb: debug [libusb_claim_interface] interface 18
[ 0.496706] [00000e70] libusb: debug [libusb_claim_interface] interface 19
[ 0.501697] [00000e70] libusb: debug [libusb_claim_interface] interface 20
[ 0.503739] [00000e70] libusb: debug [libusb_claim_interface] interface 21
[ 0.505860] [00000e70] libusb: debug [libusb_claim_interface] interface 22
[ 0.507930] [00000e70] libusb: debug [libusb_claim_interface] interface 23
[ 0.510036] [00000e70] libusb: debug [libusb_claim_interface] interface 24
[ 0.512005] [00000e70] libusb: debug [libusb_claim_interface] interface 25
[ 0.514069] [00000e70] libusb: debug [libusb_claim_interface] interface 26
[ 0.521242] [00000e70] libusb: debug [libusb_claim_interface] interface 27
[ 0.521378] [00000e70] libusb: debug [libusb_claim_interface] interface 28
[ 0.523534] [00000e70] libusb: debug [libusb_claim_interface] interface 29
[ 0.525594] [00000e70] libusb: debug [libusb_claim_interface] interface 30
[ 0.527743] [00000e70] libusb: debug [libusb_claim_interface] interface 31
[ 0.529777] [00000e70] libusb: error [auto_claim] could not auto-claim any interface
[ 0.534747] [00000e70] libusb: debug [arm_timer_for_next_timeout] no timeouts, disarming timer
[ 0.536811] [00000e70] libusb: debug [libusb_free_transfer] transfer 00824A78
[ 0.538910] [00000e70] libusb: debug [libusb_submit_transfer] transfer 00824A78
[ 0.540958] [00000e70] libusb: debug [add_to_flying_list] arm timer for timeout in 1000ms (first in line)
[ 0.543054] [00000e70] libusb: debug [winusbx_submit_control_transfer] will use interface 0
[ 0.545358] [00000e70] libusb: debug [windows_force_sync_completion] transfer 00824A78, length 4
[ 0.545357] [00000894] libusb: debug [windows_iocp_thread] ignoring overlapped 0037F5B8 for handle 0082A608 (device 3.21)
[ 0.553953] [00000894] libusb: debug [windows_iocp_thread] transfer 00824A78 completed, length 4
[ 0.547184] [00000e70] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[ 0.556475] [00000e70] libusb: debug [handle_events] event sources modified, reallocating event data
[ 0.558515] [00000e70] libusb: debug [usbi_wait_for_events] WaitForMultipleObjects() for 2 HANDLEs with timeout in 60000ms
[... cut ...]
        bRefresh:            0
        bSynchAddress:       0
      Endpoint:
        bEndpointAddress:    81h
        bmAttributes:        02h
        wMaxPacketSize:      512
        bInterval:           0
        bRefresh:            0
        bSynchAddress:       0
      Endpoint:
        bEndpointAddress:    83h
        bmAttributes:        03h
        wMaxPacketSize:      64
        bInterval:           10
        bRefresh:            0
        bSynchAddress:       0
[ 1.072075] [00000e70] libusb: debug [libusb_close]

tormodvolden, 2022-03-30:
It is stil not complete. When you run the command on the command line in CMD.EXE for example:
```
set LIBUSB_DEBUG=4
testlibusb.exe -v > debug_log.txt 2>&1
```
and then attach the resulting debug_log.txt here.

a923169932, 2022-03-30:
> It is stil not complete. When you run the command on the command line in CMD.EXE for example:
> 
> ```
> set LIBUSB_DEBUG=4
> testlibusb.exe -v > debug_log.txt 2>&1
> ```
> 
> and then attach the resulting debug_log.txt here.

Thank you, this log

[debug_log.txt](https://github.com/libusb/libusb/files/8379231/debug_log.txt)

a923169932, 2022-03-30:
> 

I get it, thank.
In Windows 7, cache_config_descriptors-->DeviceIoControl always returns the first configuration when getting the configuration descriptor. Is there a way to solve this?

mcuee, 2022-03-30:
> In Windows 7, 1.0.26rc1 function libusb_get_string_descriptor_ascii fail, return -5, 1.0.20 not this problem. But these functions are normal in Windows 10.
> 
> It seems that in Windows 7 and the driver is libusb0.sys, most functions are not available.

Are you saying 1.0.20 works for you under Windows 7 whereas  1.0.25/1.0.26rc1 nor working for you when using libusb0.sys as the device driver?

Can you try 1.0.23 as well to see it works under Windows 7? Just want to narrow down the changes which affect libusb0.sys under Windows 7.

mcuee, 2022-03-30:
> Because libusb1.0 does not support setting configuration, I use libusb-win32 to set configuration, does this matter?
> 

Hmm, by right libusb-1.0 is able to support setting of configurations when the driver libusb0.sys is used. Have you tried that?

tormodvolden, 2022-03-30:
It is anyhow weird, cache_config_descriptors() hasn't changed much since 2010 so I don't understand why it would work in libusb 1.0.20 and not in current code. Can you please attach the complete debug log with 1.0.20? And if possible intermediate versions?

tormodvolden, 2022-03-30:
You also mentioned libusb_get_string_descriptor_ascii() failing. Do you have a debug log for this?

a923169932, 2022-03-30:
> Are you saying 1.0.20 works for you under Windows 7 whereas 1.0.25/1.0.26rc1 nor working for you when using libusb0.sys as the device driver?
>
> Can you try 1.0.23 as well to see it works under [...]

1.0.20
libusb_open : success
libusb_get_string_descriptor_ascii : success
libusb_get_config_descriptor : Returns a successful but error configuration(always returns the 1st configuration)

1.0.26rc1
libusb_open : success
libusb_get_string_descriptor_ascii : fail return -5
libusb_get_config_descriptor : Returns a successful but error configuration(always returns the 1st configuration)

a923169932, 2022-03-30:
> > Because libusb1.0 does not support setting configuration, I use libusb-win32 to set configuration, does this matter?
> 
> Hmm, by right libusb-1.0 is able to support setting of configurations when the driver libusb0.sys is used. Have you tried that?

Use libusb_set_configuration and libusb_get_configuration it is normal.

a923169932, 2022-03-30:
> not in current code. Can you please attach the complete debug log with 1.0.20? And if possible intermediate versions?

> 

cache_config_descriptors is wrong in both 1.0.20 and 1.0.26rc1.

a923169932, 2022-03-30:
> It is anyhow weird, cache_config_descriptors() hasn't changed much since 2010 so I don't understand why it would work in libusb 1.0.20 and not in current code. Can you please attach the complete debug log with 1.0.20? And if possible intermediate versions?

cache_config_descriptors is wrong in both 1.0.20 and 1.0.26rc1.

mcuee, 2022-03-30:
It might be USB 3.0 controller vendor driver issues under your Windows 7 machine. Unfortuntaely I do not have Windows 7 machine to test. Windows 8 onwards the SuperSpeed USB controller drivers are much better. This may help to explan the issue.

If you can try to test with a USB 2.0 port to see if that makes a differences.

What is your USB 3.0 controller vendor driver used?

a923169932, 2022-03-30:
> You also mentioned libusb_get_string_descriptor_ascii() failing. Do you have a debug log for this?

libusb_get_string_descriptor_ascii works fine in 1.0.20, returns -5 in 1.0.26rc1.

tormodvolden, 2022-03-30:
Thanks. Now I understand better. So cache_config_descriptors() and libusb_get_config_descriptor() were maybe broken for multi-config devices since the beginning. Only libusb_get_string_descriptor_ascii() is a regression since 1.0.20(?).

a923169932, 2022-03-30:
> You also mentioned libusb_get_string_descriptor_ascii() failing. Do you have a debug log for this?

> Thanks. Now I understand better. So cache_config_descriptors() and libusb_get_config_descriptor() were maybe broken for multi-config devices since the beginning. Only libusb_get_string_descriptor_ascii() is a regression since 1.0.20(?).

libusb_get_string_descriptor_ascii works fine in 1.0.20, returns -5 in 1.0.26rc1.
this 1.0.26rc1 fail log
[libusb_get_string_descriptor_ascii___debug_log.txt](https://github.com/libusb/libusb/files/8379782/libusb_get_string_descriptor_ascii___debug_log.txt)

tormodvolden, 2022-03-30:
> It might be USB 3.0 controller vendor driver issues under your Windows 7 machine. 
Indeed, it is probably not the fault of libusb. I misunderstood originally and thought it was working on Windows 7 with 1.0.20.

mcuee, 2022-03-30:
Another thing Multi-configuration USB devices are not well supported under Windows in reality. libusbk/WinUSB do not even support changing from the current configuration o a different configuration. Not so relevant for this issue though. Just a comment.

tormodvolden, 2022-03-30:
The issue with libusb_get_string_descriptor_ascii() is the one dealt with in #1018. If you can apply and test with the patch there it would be good.

tormodvolden, 2022-03-30:
Actually I am not sure if 1018 helps in this case (since the important part of it was already merged in 1016) but testing is still welcome.

tormodvolden, 2022-03-30:
The use of IOCTL_USB_GET_DESCRIPTOR_FROM_NODE_CONNECTION in cache_config_descriptors() probably cannot work for multi-configuration devices because it seems from https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/usbioctl/ns-usbioctl-_usb_descriptor_request#remarks that it returns the _current configuration_.

a923169932, 2022-03-30:
> It might be USB 3.0 controller vendor driver issues under your Windows 7 machine. Unfortuntaely I do not have Windows 7 machine to test. Windows 8 onwards the SuperSpeed USB controller drivers are much better. This may help to explan the issue.
> 
> If you can try to test with a USB 2.0 port to see if that makes a differences.
> 
> What is your USB 3.0 controller vendor driver used?
> 
> If you have other Windws 7 machine, please try as well. Thanks.

> The use of IOCTL_USB_GET_DESCRIPTOR_FROM_NODE_CONNECTION in cache_config_descriptors() probably cannot work for multi-configuration devices because it seems from [...]

I replaced a Windows 7 device to test, cache_config_descriptors became normal.
 but libusb_open failed, returning -3 LIBUSB_ERROR_ACCESS

[ 0.085461] [00000ad0] libusb: debug [init_device] found 4 configurations (current config: 1)
[ 0.090111] [00000ad0] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 39 bytes)
[ 0.095477] [00000ad0] libusb: debug [cache_config_descriptors] cached config descriptor 1 (bConfigurationValue=2, 149 bytes)
[ 0.098999] [00000ad0] libusb: debug [cache_config_descriptors] cached config descriptor 2 (bConfigurationValue=3, 62 bytes)
[ 0.104382] [00000ad0] libusb: debug [cache_config_descriptors] cached config descriptor 3 (bConfigurationValue=4, 117 bytes)

[libusb_open_debug_log.txt](https://github.com/libusb/libusb/files/8380118/libusb_open_debug_log.txt)

a923169932, 2022-03-30:
> The use of IOCTL_USB_GET_DESCRIPTOR_FROM_NODE_CONNECTION in cache_config_descriptors() probably cannot work for multi-configuration devices because it seems from [...]

Thanks, but in Windows 10 is normal.

a923169932, 2022-03-30:
> Another thing Multi-configuration USB devices are not well supported under Windows in reality. libusbk/WinUSB do not even support changing from the current configuration o a different configuration. Not so relevant for this issue though. Just a comment.

Thanks, But using libusb-win32's API, all functions work fine.

a923169932, 2022-03-30:
> > I replaced a Windows 7 device to test, cache_config_descriptors became normal.
> 
> Ah, so it works in this case. What did you exactly change, you ran it on another computer, also with Windows 7?
> 
> In the good case it displays `(current config: 1)` and in the bad case `(current config: 4)` but I don't know why.

only replaced a Windows 7 computer. The code is not modified.
(current config: 4) and (current config: 1) both have problems, config 1 -> 4 because i used libusb_set_configuration.

a923169932, 2022-03-30:
this is success log:
```
[ 0.054013] [00000c28] libusb: debug [get_api_type] driver(s): libusb0
[ 0.054261] [00000c28] libusb: debug [get_api_type] matched driver name against libusb0
[ 0.054395] [00000c28] libusb: debug [winusb_get_device_list] allocating new device for session [1B8C]
**[ 0.054813] [00000c28] libusb: debug [get_api_type] driver(s): usbccgp**
[ 0.055043] [00000c28] libusb: debug [get_api_type] matched driver name against Composite API
[ 0.055149] [00000c28] libusb: debug [winusb_get_device_list] allocating new device for session [1BF0]
[ 0.056608] [00000c28] libusb: debug [enumerate_hcd_root_hub] assigning HCD 'PCI\VEN_15AD&DEV_0770&SUBSYS_077015AD&REV_00\4&3
[ 0.057373] [00000c28] libusb: debug [enumerate_hcd_root_hub] assigning HCD 'PCI\VEN_15AD&DEV_0774&SUBSYS_197615AD&REV_00\4&3
[ 0.061499] [00000c28] libusb: debug [init_root_hub] root hub 'USB\ROOT_HUB\5&3BB57B&0' reports 2 ports
[ 0.065860] [00000c28] libusb: debug [init_device] (bus: 2, addr: 0, depth: 0, port: 0): 'USB\ROOT_HUB\5&3BB57B&0'
[ 0.073859] [00000c28] libusb: debug [init_root_hub] root hub 'USB\ROOT_HUB20\5&299E1C9F&0' reports 6 ports
[ 0.074770] [00000c28] libusb: debug [init_device] (bus: 1, addr: 0, depth: 0, port: 0): 'USB\ROOT_HUB20\5&299E1C9F&0'
[ 0.076068] [00000c28] libusb: debug [winusb_get_device_list] extra GUID: {F5CB67FB-48B1-4FC2-9F07-DC61B124F8AB}
[ 0.077165] [00000c28] libusb: debug [winusb_get_device_list] found existing device for session [1B8C]
[ 0.077905] [00000c28] libusb: debug [init_device] found 4 configurations (current config: 1)
[ 0.943426] [00000c28] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 39 bytes)
[ 0.947024] [00000c28] libusb: debug [cache_config_descriptors] cached config descriptor 1 (bConfigurationValue=2, 149 bytes)
[ 0.950498] [00000c28] libusb: debug [cache_config_descriptors] cached config descriptor 2 (bConfigurationValue=3, 62 bytes)
[ 0.954278] [00000c28] libusb: debug [cache_config_descriptors] cached config descriptor 3 (bConfigurationValue=4, 117 bytes)
```

this is error log,
```
[ 0.057139] [000016f0] libusb: debug [get_api_type] driver(s): libusb0
[ 0.057295] [000016f0] libusb: debug [get_api_type] matched driver name against libusb0
[ 0.057364] [000016f0] libusb: debug [winusb_get_device_list] allocating new device for session [1014]
**[ 0.057692] [000016f0] libusb: debug [get_api_type] driver(s): HidUsb**
[ 0.057826] [000016f0] libusb: debug [get_api_type] matched driver name against HID API
[ 0.057888] [000016f0] libusb: debug [winusb_get_device_list] allocating new device for session [146C]
[ 0.058114] [000016f0] libusb: debug [enumerate_hcd_root_hub] assigning HCD 'PCI\VEN_8086&DEV_8C26&SUBSYS_85341043&REV_05\3&
[ 0.058258] [000016f0] libusb: debug [enumerate_hcd_root_hub] assigning HCD 'PCI\VEN_8086&DEV_8C2D&SUBSYS_85341043&REV_05\3&
[ 0.058431] [000016f0] libusb: debug [enumerate_hcd_root_hub] assigning HCD 'PCI\VEN_8086&DEV_8C31&SUBSYS_85341043&REV_05\3&
[ 0.059641] [000016f0] libusb: debug [init_root_hub] root hub 'USB\ROOT_HUB20\4&5854A60&0' reports 2 ports
[ 0.059748] [000016f0] libusb: debug [init_device] (bus: 2, addr: 0, depth: 0, port: 0): 'USB\ROOT_HUB20\4&5854A60&0'
[ 0.060199] [000016f0] libusb: debug [init_root_hub] root hub 'USB\ROOT_HUB20\4&78CD1A&0' reports 2 ports
[ 0.060304] [000016f0] libusb: debug [init_device] (bus: 1, addr: 0, depth: 0, port: 0): 'USB\ROOT_HUB20\4&78CD1A&0'
[ 0.060542] [000016f0] libusb: debug [winusb_get_device_list] extra GUID: {74F1AD1F-4131-483A-996B-019D0BF57221}
[ 0.060662] [000016f0] libusb: debug [winusb_get_device_list] found existing device for session [1014]
[ 0.060824] [000016f0] libusb: debug [init_device] found 4 configurations (current config: 1)
[ 0.062880] [000016f0] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 39 bytes)
[ 0.062986] [000016f0] libusb: debug [cache_config_descriptors] cached config descriptor 1 (bConfigurationValue=1, 39 bytes)
[ 0.063080] [000016f0] libusb: debug [cache_config_descriptors] cached config descriptor 2 (bConfigurationValue=1, 39 bytes)
[ 0.063155] [000016f0] libusb: debug [cache_config_descriptors] cached config descriptor 3 (bConfigurationValue=1, 39 bytes)
```

Is this a problem with "usbccgp" and "HidUsb"?

a923169932, 2022-03-31:
I tried changing the driver to WinUSB or libusbk , cache_config_descriptors also failed, so it's not just a problem with libusb0.

WinUSB fail log
```
[ 0.000095] [00000558] libusb: debug [libusb_init] created default context
[ 0.000137] [00000558] libusb: debug [libusb_init] libusb v1.0.26.11717-rc1
[ 0.000153] [00000558] libusb: debug [usbi_add_event_source] add HANDLE 00000034 events 0
[ 0.000169] [00000558] libusb: debug [usbi_io_init] using timer for timeouts
[ 0.000178] [00000558] libusb: debug [usbi_add_event_source] add HANDLE 00000038 events 0
[ 0.000189] [00000558] libusb: debug [get_windows_version] Windows 7 SP1 64-bit
[ 0.000198] [00000558] libusb: debug [htab_create] using 1021 entries hash table
[ 0.004497] [00000558] libusb: info [winusbx_init] WinUSB DLL available (without isoch support)
[ 0.008041] [00000558] libusb: debug [winusbx_init] libusbK DLL found, version: 3.1.0.0
[ 0.010098] [00000558] libusb: info [windows_init] UsbDk backend is not available
[ 0.010225] [00000558] libusb: debug [libusb_get_device_list]  
[ 0.010340] [00001424] libusb: debug [windows_iocp_thread] I/O completion thread started
[ 0.057978] [00000558] libusb: debug [winusb_get_device_list] found new PnP enumerator string 'IUSB3'
[ 0.058016] [00000558] libusb: debug [winusb_get_device_list] allocating new device for session [17D4]
[ 0.058077] [00000558] libusb: debug [winusb_get_device_list] allocating new device for session [1814]
[ 0.058137] [00000558] libusb: debug [winusb_get_device_list] allocating new device for session [1850]
[ 0.058194] [00000558] libusb: debug [winusb_get_device_list] allocating new device for session [1888]
[ 0.058250] [00000558] libusb: debug [winusb_get_device_list] allocating new device for session [18D8]
[ 0.058547] [00000558] libusb: debug [get_api_type] driver(s): libusbK
[ 0.058621] [00000558] libusb: debug [get_api_type] matched driver name against libusbK
[ 0.058632] [00000558] libusb: debug [winusb_get_device_list] allocating new device for session [1014]
[ 0.058852] [00000558] libusb: debug [get_api_type] driver(s): HidUsb
[ 0.058924] [00000558] libusb: debug [get_api_type] matched driver name against HID API
[ 0.058934] [00000558] libusb: debug [winusb_get_device_list] allocating new device for session [146C]
[ 0.059077] [00000558] libusb: debug [enumerate_hcd_root_hub] assigning HCD 'PCI\VEN_8086&DEV_8C26&SUBSYS_85341043&REV_05\3&11583659&0&E8' bus number 1
[ 0.059152] [00000558] libusb: debug [enumerate_hcd_root_hub] assigning HCD 'PCI\VEN_8086&DEV_8C2D&SUBSYS_85341043&REV_05\3&11583659&0&D0' bus number 2
[ 0.059210] [00000558] libusb: debug [enumerate_hcd_root_hub] assigning HCD 'PCI\VEN_8086&DEV_8C31&SUBSYS_85341043&REV_05\3&11583659&0&A0' bus number 3
[ 0.060209] [00000558] libusb: debug [init_root_hub] root hub 'USB\ROOT_HUB20\4&5854A60&0' reports 2 ports
[ 0.060232] [00000558] libusb: debug [init_device] (bus: 2, addr: 0, depth: 0, port: 0): 'USB\ROOT_HUB20\4&5854A60&0'
[ 0.060577] [00000558] libusb: debug [init_root_hub] root hub 'USB\ROOT_HUB20\4&78CD1A&0' reports 2 ports
[ 0.060597] [00000558] libusb: debug [init_device] (bus: 1, addr: 0, depth: 0, port: 0): 'USB\ROOT_HUB20\4&78CD1A&0'
[ 0.060741] [00000558] libusb: debug [winusb_get_device_list] extra GUID: {6767891E-9B98-431B-AFE8-495501FB5231}
[ 0.060779] [00000558] libusb: debug [winusb_get_device_list] found existing device for session [1014]
[ 0.060871] [00000558] libusb: debug [init_device] found 4 configurations (current config: 1)
[ 0.060889] [00000558] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 39 bytes)
[ 0.060903] [00000558] libusb: debug [cache_config_descriptors] cached config descriptor 1 (bConfigurationValue=1, 39 bytes)
[ 0.060915] [00000558] libusb: debug [cache_config_descriptors] cached config descriptor 2 (bConfigurationValue=1, 39 bytes)
[ 0.060928] [00000558] libusb: debug [cache_config_descriptors] cached config descriptor 3 (bConfigurationValue=1, 39 bytes)
```

libusbK fail log
```
[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.000111] [0000135c] libusb: debug [libusb_init] created default context
[ 0.000161] [0000135c] libusb: debug [libusb_init] libusb v1.0.26.11717-rc1
[ 0.000181] [0000135c] libusb: debug [usbi_add_event_source] add HANDLE 00000034 events 0
[ 0.000200] [0000135c] libusb: debug [usbi_io_init] using timer for timeouts
[ 0.000211] [0000135c] libusb: debug [usbi_add_event_source] add HANDLE 00000038 events 0
[ 0.000224] [0000135c] libusb: debug [get_windows_version] Windows 7 SP1 64-bit
[ 0.000236] [0000135c] libusb: debug [htab_create] using 1021 entries hash table
[ 0.004682] [0000135c] libusb: info [winusbx_init] WinUSB DLL available (without isoch support)
[ 0.004942] [0000135c] libusb: debug [winusbx_init] libusbK DLL found, version: 3.1.0.0
[ 0.006997] [0000135c] libusb: info [windows_init] UsbDk backend is not available
[ 0.007089] [0000135c] libusb: debug [libusb_get_device_list]  
[ 0.007206] [00000d64] libusb: debug [windows_iocp_thread] I/O completion thread started
[ 0.057178] [0000135c] libusb: debug [winusb_get_device_list] found new PnP enumerator string 'IUSB3'
[ 0.057215] [0000135c] libusb: debug [winusb_get_device_list] allocating new device for session [17D4]
[ 0.057295] [0000135c] libusb: debug [winusb_get_device_list] allocating new device for session [1814]
[... cut ...]
[ 0.222554] [0000135c] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 39 bytes)
[ 0.222609] [0000135c] libusb: debug [cache_config_descriptors] cached config descriptor 1 (bConfigurationValue=1, 39 bytes)
[ 0.222659] [0000135c] libusb: debug [cache_config_descriptors] cached config descriptor 2 (bConfigurationValue=1, 39 bytes)
[ 0.222707] [0000135c] libusb: debug [cache_config_descriptors] cached config descriptor 3 (bConfigurationValue=1, 39 bytes)
```

tormodvolden, 2022-03-31:
It is unclear to me in these logs which devices the messages belongs to. I understand it that way that libusb gets the device config descriptors from the hub driver (which already has read them from the devices when they were discovered). It is possible that the hub driver deals with multiple configuration differently if the device is composite or not. Can you provide a testlibusb -v dump for both cases? Maybe also check the version of Usbhub.sys on each machine.

In general Windows doesn't support multiple configurations well. See e.g. http://www.tracesystemsinc.com/USB_Tutorials_web/USB/A1_Overview/Books/A3_USB_Facts_of_Life/slide09.htm

This diagram helps to understand the driver topology:
https://docs.microsoft.com/en-us/windows-hardware/drivers/usbcon/usb-3-0-driver-stack-architecture
Note libusb would be located in the "Applications" box on top. See also https://github.com/libusb/libusb/wiki/FAQ#libusb-win32-libusbk-and-libusb-project about the possible library and driver topologies for libusb.

tormodvolden, 2022-03-31:
> So I think it has something to do with "usbccgp" and "HidUsb", usbccgp is a composite device and can have multiple configurations. Is HidUsb a single configuration?
> 
> What is the difference between the Composite API and the HID API, and does the HID API fail?

Each device on the USB bus gets assigned to a driver depending on the class and other characteristics. In the case of a composite device the usbccgp driver can be assigned, and it splits the interfaces between "virtual" devices which again will be assigned to different normal drivers, like HID.

mcuee, 2022-03-31:
> On the problem device, I tried changing the driver to Winusb or libusbk , cache_config_descriptors also failed, so it's not just a problem with libusb0.

When you use libusb0.sys or libusbk.sys driver, libusbk.dll has to be used. When you use WinUSB, there are two different scenarios -- with libusbk.dll or without libusbk.dll.

Please try one more case -- use WinUSB driver, but remove libusbk.dll.
 
Please also make sure your libusbk.dll is of the latest version (libusbk 3.1.0.0 version)
https://github.com/mcuee/libusbk

a923169932, 2022-03-31:
> When you use libusb0.sys or libusbk.sys driver, libusbk.dll has to be used. When you use WinUSB, there are two different scenarios -- with libusbk.dll or without libusbk.dll.
>
> Please try one more [...]

Ok, this is me using WinUSB and deleting libusbk.dll, still fail.
```
C:\Users\waip002\Desktop\xusb>testlibusb.exe -v
[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.000180] [00000fc8] libusb: debug [libusb_init] created default context
[ 0.000266] [00000fc8] libusb: debug [libusb_init] libusb v1.0.26.11717-rc1
[ 0.000345] [00000fc8] libusb: debug [usbi_add_event_source] add HANDLE 00000034 events 0
[ 0.000439] [00000fc8] libusb: debug [usbi_io_init] using timer for timeouts
[ 0.000518] [00000fc8] libusb: debug [usbi_add_event_source] add HANDLE 00000038 events 0
[ 0.000590] [00000fc8] libusb: debug [get_windows_version] Windows 7 SP1 64-bit
[ 0.000664] [00000fc8] libusb: debug [htab_create] using 1021 entries hash table
[ 0.004479] [00000fc8] libusb: info [winusbx_init] WinUSB DLL available (without isoch support)
[ 0.004599] [00000fc8] libusb: info [winusbx_init] libusbK DLL is not available
[ 0.005716] [00000fc8] libusb: info [windows_init] UsbDk backend is not available
[ 0.005984] [00000fc8] libusb: debug [libusb_get_device_list]
[ 0.006111] [00000d24] libusb: debug [windows_iocp_thread] I/O completion thread started
[ 0.058760] [00000fc8] libusb: debug [winusb_get_device_list] found new PnP enumerator string 'IUSB3'
[ 0.058895] [00000fc8] libusb: debug [winusb_get_device_list] allocating new device for session [17D4]
[ 0.059095] [00000fc8] libusb: debug [winusb_get_device_list] allocating new device for session [1814]
[ 0.059269] [00000fc8] libusb: debug [winusb_get_device_list] allocating new device for session [1850]
[ 0.059489] [00000fc8] libusb: debug [winusb_get_device_list] allocating new device for session [1888]
[ 0.059636] [00000fc8] libusb: debug [winusb_get_device_list] allocating new device for session [18D8]
[ 0.059974] [00000fc8] libusb: debug [get_api_type] driver(s): WinUSB
[ 0.060097] [00000fc8] libusb: debug [get_api_type] matched driver name against WinUSB
[ 0.060159] [00000fc8] libusb: debug [winusb_get_device_list] allocating new device for session [800]
[ 0.060550] [00000fc8] libusb: debug [get_api_type] driver(s): HidUsb
[ 0.060663] [00000fc8] libusb: debug [get_api_type] matched driver name against HID API
[ 0.060708] [00000fc8] libusb: debug [winusb_get_device_list] allocating new device for session [146C]
[ 0.060920] [00000fc8] libusb: debug [enumerate_hcd_root_hub] assigning HCD 'PCI\VEN_8086&DEV_8C26&SUBSYS_85341043&REV_05\3&11583659&0&E8' bus number 1
[ 0.061077] [00000fc8] libusb: debug [enumerate_hcd_root_hub] assigning HCD 'PCI\VEN_8086&DEV_8C2D&SUBSYS_85341043&REV_05\3&11583659&0&D0' bus number 2
[ 0.061223] [00000fc8] libusb: debug [enumerate_hcd_root_hub] assigning HCD 'PCI\VEN_8086&DEV_8C31&SUBSYS_85341043&REV_05\3&11583659&0&A0' bus number 3
[ 0.129180] [00000fc8] libusb: debug [init_root_hub] root hub 'USB\ROOT_HUB20\4&5854A60&0' reports 2 ports
[ 0.129364] [00000fc8] libusb: debug [init_device] (bus: 2, addr: 0, depth: 0, port: 0): 'USB\ROOT_HUB20\4&5854A60&0'
[ 0.222279] [00000fc8] libusb: debug [init_root_hub] root hub 'USB\ROOT_HUB20\4&78CD1A&0' reports 2 ports
[ 0.222667] [00000fc8] libusb: debug [init_device] (bus: 1, addr: 0, depth: 0, port: 0): 'USB\ROOT_HUB20\4&78CD1A&0'
[ 0.223682] [00000fc8] libusb: debug [winusb_get_device_list] extra GUID: {05202847-AF09-4EBF-B0BA-E3720DBEF2D0}
[ 0.224180] [00000fc8] libusb: debug [winusb_get_device_list] found existing device for session [800]
[ 0.224948] [00000fc8] libusb: debug [init_device] found 4 configurations (current config: 1)
[ 0.225262] [00000fc8] libusb: debug [cache_config_descriptors] cached config descriptor 0 (bConfigurationValue=1, 39 bytes)
[ 0.225541] [00000fc8] libusb: debug [cache_config_descriptors] cached config descriptor 1 (bConfigurationValue=1, 39 bytes)
[ 0.225908] [00000fc8] libusb: debug [cache_config_descriptors] cached config descriptor 2 (bConfigurationValue=1, 39 bytes)
[ 0.226222] [00000fc8] libusb: debug [cache_config_descriptors] cached config descriptor 3 (bConfigurationValue=1, 39 bytes)
```

a923169932, 2022-03-31:
Latest information, I installed another device with the same Windows 7 image and the problem can be reproduced. So I think it has something to do with a certain SYS or DLL version.

tormodvolden, 2022-03-31:
If this is USB3 (I didn't see that mentioned before) you must check usbhub3.sys, ucx01000.sys, and usbxhci.sys as well.

mcuee, 2022-04-11:
@a923169932 Please provide the necessary info. Thanks. I will probably keep this issue open for another month.

mcuee, 2022-06-18:
Close this since the OP does not come back. I can only assume it is because of USB 3 driver problems under Windows 7.

closed 2022-06-18 by mcuee